edison-fw/meta-intel-edison #8: Add xfstk recipe

htot, 2017-12-02:
This should build xfstk native for easy flashall --recovery

Exoyds, 2020-07-12:
I am not sure how of help it can be, but Aurelien Ballier ( https://github.com/AurelienBallier/xfstk ) got it working under Qt5 and a newer libboost. It did compile successfully under Ubuntu 20.04 when I tested. Although I did have some issues with the connection stability, I can't say for sure that it was definitely the xfstk that was at fault there.

htot, 2020-07-12:
What would be useful to have is a Yocto recipe that builds the CLI version using recent libs.

Exoyds, 2020-09-09:
Sorry for my late reply!

Do you mean a recipe as in a bitbake one? If so I can have a look if I can piece on together.

I was also thinking I could put xfstk in the Makefile in https://github.com/edison-fw/meta-intel-edison/blob/master/utils/Makefile.mk or would that be a bad place?

htot, 2020-09-10:
A bitbake recipe would be nice as that would go into Yocto and be independent of edison.

Exoyds, 2020-09-11:
I have created a draft pull request for this, but as I am completely new to bitbake and how it should be laid out I would appreciate any pointers as to where it all should go!

Also I'm not sure how to best resolve the dependencies of xfstk - should all of them be built into the image?

htot, 2020-09-11:
That's a good question, I'm no expert either. I'm trying to think of a comparable example recipe...

Exoyds, 2020-09-12:
Thank you very much for your review! I also couldn't find any comparable recipes in edison, at least none that used cmake.

htot, 2020-10-01:
This is resolved by merge commit e8af20375de779cf5aa1ba983c22f355855a98e5

closed 2020-10-01 by htot