edison-fw/meta-intel-edison #13: Bluetooth support

floion, 2017-12-23:
Hi,
have you had luck with getting bluetooth to work on kernel 4.x?
Is the btbcm module working or do we need to fall back to backporting the bcm_bt_lpm?

Thank you

htot, 2017-12-23:
Last time I checked it just works, see #9 under https://github.com/htot/meta-intel-edison/wiki/Current-state-of-Edison-Morty.

floion, 2017-12-23:
So before running btattach you get the device listed when you run "rfkill list"?

htot, 2017-12-23:
Hmm, something is wrong now that worked before. When I do:
```
root@edison:~# /usr/bin/btattach  -B /dev/ttyS0 -P bcm -S 3000000
Attaching Primary controller to /dev/ttyS0
Switched line discipline from 0 to 15
Failed to set protocol: Protocol not supported
No controller attached
```
Looks like the bt device does not get power. Maybe @andy-shev knows what is wrong (this is on linux 4.14 bluez 5.43)

htot, 2017-12-23:
(@floion no, before, when it  worked, first you run btattach then rfkill)

htot, 2017-12-27:
If I remember correctly it worked with linux 4.12, possibly with 4.13. You can easily go back to one of these by changing this file: [poky-edison.conf](https://github.com/htot/meta-intel-edison/blob/pyro64/meta-intel-edison-distro/conf/distro/poky-edison.conf). Let me know if that helps.

andy-shev, 2017-12-28:
Can you revert *commit 276c87054751 ("x86/platform/intel-mid: Make 'bt_sfi_data' const")* and check if it works again?

htot, 2017-12-28:
Fixed with c788c20abbb3c8f9e30265e206780befcf9915a5. Thanks for reporting.

closed 2017-12-28 by htot

floion, 2018-01-03:
Thanks @htot @andy-shev