edison-fw/meta-intel-edison #19: [pyro64-acpi] flashall.sh successfully working on mini breakout board

vrubiolo, 2018-04-30:
Hi there,

Just a note to report that I have been successfully able to use flashall.sh on my mini-breakout board to flash the pyro64-acpi produced kernel and rootfs. I had previously updated u-boot as per the wiki but I am not sure this is necessary as flashall seems to be taking care of that too.
This is a great milestone! Thanks much for the hard work, I could not wait to share that:)
Now onto moving to rocko64 ..

Attached is my [boot log](https://github.com/htot/meta-intel-edison/files/1959712/new_kernel_boot_success.txt), I can provide more information if you need me to.

htot, 2018-04-30:
Good news. I guess eth over usb won't be working yet as we force it to host mode. Maybe in 4.17?

vrubiolo, 2018-05-10:
And I confirm I could also boot the image using the rocko64-acpi branch. [Here](https://github.com/htot/meta-intel-edison/files/1990095/new_kernel_boot_success-kernel_v4.16.txt) is the associated boot log. Great job!

closed 2018-05-10 by vrubiolo