edison-fw/meta-intel-edison #20: [Packages] Would need to have python virtualenv available

vrubiolo, 2018-05-10:
[apologies if this is not the right channel to ask, please tell me where we can best discuss those questions]

Hi Ferry,

I am using the rocko64-acpi branch to package an installation of Octoprint as per the official site instructions.
Those reference the virtualenv binary but I don't see it available in the list of 
[bitbake recipes](https://github.com/htot/meta-intel-edison/files/1990628/bitbake_recipe_list.txt) that I have

It looks to be available in the [meta-openstack](http://layers.openembedded.org/layerindex/recipe/24482/) layer (from the Yocto layer index). Is it a matter of just adding it in (I am a beginner w/ Yocto) when I configure my project? If not, would it be appropriate to include in the list of recipes available (you already provide a good dev setup btw, thanks!)

Another question: what is the difference between a package like 'python-setuptools' and 'nativesdk-python-setuptools'? Is it that the former really runs on the target (via the sysroot) whereas the latter is just picked up from the host?

Thanks much for your attention,
Vincent

htot, 2018-05-10:
Hello Vincent, I'm just a beginner too. But I would do the same, lookup the open embedded layer index, find that meta-openstack has the recipy and meta-openstack is in git://git.yoctoproject.org/meta-cloud-services. Then modify setup.sh to clone that and add it to the layers file. Or I would probably start by cloning manually and manually changing the layers.conf.

About your 2nd question, I'm not sure. But I now many packages have a native variant. Afaik these are built to run on the host, i.e. not picked up from the host. And there are nativesdk packages. Bitbake can build an sdk (installer), I guess the nativesdk packages go in there.

BTW I do use the sdk, but I don't install it, just create a few links to make it work. I need to document that I think.

vrubiolo, 2018-05-10:
Hi Ferry,

Thanks for the quick reply and the good suggestions, I will try to add the layer from Openembedded and report back then.

htot, 2018-11-01:
Did you succeed in making it work?

vrubiolo, 2018-12-03:
Sorry, I was not actively monitoring the project here as I have put that aside for a while. My Printrbot 3D printer got its CNC control board fried and I am busy installing a new one (Printrboard G2). I will get back to the Edison once I have that sorted out.
Thanks again for providing and supporting all those bits!

vrubiolo, 2018-12-03:
Closing until I get to work on this, this is not really an issue.

closed 2018-12-03 by vrubiolo