edison-fw/meta-intel-edison #25: Some recipes refer to non-existing bcm4334x module and old FW

alext-mkrs, 2018-06-17:
When checking the `hostapd` stuff for #21 I've noticed the driver currently used for the wifi NIC has changed - the original Edison image used some semi-custom (? - it looked like the part is based on the public chip, but it either was customized somehow or used a customized FW, so e.g. the docs are not available, some on Intel forums tried) `bcm4334x` one + some again semi-custom FW generated/packaged in dedicated recipes. Same went for the Bluetooth part of it (it's a combo adapter).

Right now with the 4.16 vanilla kernel the wifi seems to work, but the driver used is `brcmfmac` and therefore some of the recipes no longer do the right thing + there are some errors logged in the boot log. I haven't yet done any serious reconnaissance, so let's see if a single issue would make sense. Let's document such things here and fix them all at once or one-by-one as appropriate.

[ ] meta-intel-edison/meta-intel-edison-distro/recipes-support/ap-mode-toggle
[ ] `brcmfmac mmc2:0001:1: Direct firmware load for brcm/brcmfmac43340-sdio.clm_blob failed with error -2` in the boot log

htot, 2018-06-17:
I believe bcm4334x is the bluetooth and brcmfmac is the wifi. The bcm4334x loads the firmware itself, in the past that was done from user space (with the rfkill thingy I removed not so long ago). In the beginning I just renamed the firmware from 01org, but later I found a newer in the android x86 repo's. 
I started hostapd last week (for the first time, don't know exactly how it works) and saw an AP created from my phone. Could not connect though, I think the config might have been wrong.

htot, 2018-06-17:
https://github.com/htot/meta-intel-edison/tree/master/meta-intel-edison-bsp/recipes-kernel/bcm43340

htot, 2018-06-17:
BTW the acpi version powers and loads the bluetooth firmware automatically. On the non-acpi version you need to enable bluetooth_attach.service

alext-mkrs, 2018-06-18:
> I believe bcm4334x is the bluetooth and brcmfmac is the wifi.

I think that was not the case in the official image - there are three recipes total, two for FW (BT and WiFi) + one for the module (named bcm4334x). Here: http://git.yoctoproject.org/cgit/cgit.cgi/meta-intel-edison/tree/meta-intel-edison-bsp/recipes-kernel. brcmfmac therefore should be a vanilla kernel's thing.

Got it with the FW loading - that would explain why it works even with the error present. Finally that ap-mode-toggle is used in that oobe script + postinstall script IIRC and is run. Instead of fixing we may very well just remove it, yes :)

htot, 2018-06-18:
Ah yes, the bcm43341 is in vanilla (maybe in module btbcm?).
```
root@edison:~# journalctl -b | grep -i bcm
Jun 15 19:20:08 edison kernel: Bluetooth: hci0: BCM: chip id 82
Jun 15 19:20:08 edison kernel: Bluetooth: hci0: BCM: features 0x2f
Jun 15 19:20:08 edison kernel: Bluetooth: hci0: BCM43341B0
Jun 15 19:20:08 edison kernel: Bluetooth: hci0: BCM43341B0 (002.001.014) build 0000
Jun 16 17:55:10 edison kernel: Bluetooth: hci0: BCM (002.001.014) build 0176
```

lybtongji, 2018-06-22:
I have a question: What's the difference between `meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/linux-yocto_*.bb` and `out/linux64/poky/meta-yocto-bsp/recipes-kernel/linux/linux-yocto_*.bbappend`?
I found the file downloaded in `bbcache/downloads/linux-*.tar.xz` was only given in `out/linux64/poky/meta-yocto-bsp/recipes-kernel/linux/linux-yocto_*.bbappend`.

alext-mkrs, 2018-06-22:
Bbappend files are used to override parts of respective bb files. You can read more about that in Yocto project reference guide.

htot, 2018-06-22:
The end result is, we are not using the yocto provided kernels, but instead have our own recipe that builds andy-shev's kernel which is basically vanilla + a few patches for Edison.

htot, 2018-07-15:
Ok, I remember: currently bt and wifi are in vanilla, we now have 1 recipe remaining that pulls from 3 sources:
1 - wifi firmware from kernel.org
2- bt firmware android-x86
3 - configuration files from github/01org
This is correct, but could be cleaner.

Looking at ap-mode-toggle, I think it is usefull, but it needs brctl (from bridge-utils). Will resolve that as a separate issue.

We could close this one?

htot, 2019-04-14:
@alext-mkrs Do you have a suggestion what would be needed to close this?

alext-mkrs, 2019-04-25:
I think if that fw loading is not a problem as we discussed and if ap-mode-toggle is no longer used by oobe script (ir is it? I recall you wanted to rework it), we can simply close it.

htot, 2019-04-25:
`ap-mode-toggle` has been reworked and fixed. 
`oobe` will be reworked, will not depend on `ap-mode-toggle` and will basically be changed into a web interface into `connman`, based on `python` and `bottle`.

Closing.

closed 2019-04-25 by htot