edison-fw/meta-intel-edison #28: On first build kernel modules can not be located for installation into initramfs

htot, 2018-07-01:
On rocko on the first build after `make setup` the build fails with:
```
ERROR: core-image-minimal-initramfs-1.0-r0 do_rootfs: Unable to install packages. Command '/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/build/tmp/work/edison-poky-linux/core-image-minimal-initramfs/1.0-r0/recipe-sysroot-native/usr/bin/apt-get  install --force-yes --allow-unauthenticated acpi-tables base-passwd busybox initramfs-live-boot kernel-module-mmc-block kernel-module-sdhci-acpi kernel-module-sdhci-pci run-postinsts udev' returned 100:
Reading package lists...
Building dependency tree...
E: Unable to locate package kernel-module-mmc-block
E: Unable to locate package kernel-module-sdhci-acpi
E: Unable to locate package kernel-module-sdhci-pci

ERROR: core-image-minimal-initramfs-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/build/tmp/work/edison-poky-linux/core-image-minimal-initramfs/1.0-r0/temp/log.do_rootfs.10056
ERROR: Task (/home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/poky/meta/recipes-core/images/core-image-minimal-initramfs.bb:do_rootfs) failed with exit code '1'
```
It appears no kernel modules were built. do_compile_kernelmodules was executed, but it's log file says: NOTE: no modules to compile, which seems to originate from grepping .config for "^CONFIG_MODULES=y$" in do_compile_kernelmodules(). Checking .config I find # CONFIG_MODULES is not set which we do have that set if defconfig. Something seems to go wrong with run.do_kernel_metadata on the first run.

run.do_kernel_metadata generates dir's .meta and .kernel-meta under kernel-source. Here the correct config is generated from .defconfig + .cfg. But it seems this is not used by run.do_configure

alext-mkrs, 2018-07-01:
See also some discussion in #27

alext-mkrs, 2018-07-08:
Looks like that fix was not enough - in #21 we saw that happening again and after additional checking I see it also needs flex. I actually knew that it does (it was a change introduced in 4.16 build process - something that was pre-built before they're now building with bison and flex), but in my tests only the former was enough. Looks like the tests I used didn't show everything, so I'm now (1) adding flex piece and (3) will triple-check with everything built from scratch.

commit aaf1e39a (2018-07-09): kernel: additional fix for kernel build failing the first time

The fix introduced in 535f3a33e6c2b2114719315a7e3ca324a33caa7e
turned out to be incomplete - flex is missing as well.

Fixes #28.
closed 2018-07-09 by alext-mkrs