edison-fw/meta-intel-edison #33: ap-mode-toggle error message in dmesg

alext-mkrs, 2018-07-15:
There's a suspicious message in the `dmesg` noticed by @htot. Needs to be investigated and probably fixed.

```
ap-mode-toggle[2438]: Usage: /usr/bin/ap-mode-toggle start|stop|toggle
ap-mode-toggle[2438]:   start  - Start AP mode
ap-mode-toggle[2438]:   stop   - AP mode is stopped and we return to station mode
ap-mode-toggle[2438]:   toggle - Start or stop AP mode depending on previous mode.
ap-mode-toggle[2438]: This script is only usable in Edison where the Broadcom
ap-mode-toggle[2438]: wifi driver firmware needs to be told which mode to operate.
```

htot, 2018-07-15:
Yes, I see. The script loads a kernel module with a param. And it uses connman to tether. So this is either not necessary anymore or needs to be done different (hostapd, networkd). We need to find the users of this script, maybe we can drop it entirely.

sjpeng, 2019-01-11:
@htot if possible using ap mode in Edison? I got "Failed to initialize interface" message when I using the "hostapd -c /etc/hostapd/hostapd.conf" command.

htot, 2019-01-11:
@sjpeng I have an updated ap-mode-toggle in my branch (https://github.com/htot/meta-intel-edison/commits/sumo64), and plan to push `thud` based branch tonight. If you want this sumo64, you need to be quick as I will be deleting this branch when I push thud.

The fixed version relies on `connman` tethering working and removing `hostapd` completely. If you are impatient, you can already turn on tethering from the console using `connman tether wifi ... `

htot, 2019-04-14:
@alext-mkrs For me this one can be closed. The script has been fixed, and connman provides AP mode nicely.

alext-mkrs, 2019-04-25:
+1 from me for closing

closed 2019-04-25 by htot