edison-fw/meta-intel-edison #34: [discussion] Do we really need meta-intel-iot-middleware?

alext-mkrs, 2018-07-15:
We've touched upon this in another thread and @htot, you've removed the old `node`, `mraa` and `upm` recipes. However let me take this further and ask why would we need this whole layer at all.

I personally see no point of having it - its main piece was `xdk-daemon` and wyliodrin libs/server. The former is no longer supported for a long time (and the cloud it connected to is gone as well) and the latter is most probably EOLed too at the Wyliodrin side, as Edison is EOLed (I'm not 100% on that, that's a hypothesis, but in any case I'm not sure if people used it at all).

Everything else there are just dependencies, which don't look too generally useful to justify preservation. Maybe just the `paho-mqtt` library and `hid-api`, but recipe for the former is utterly outdated (2014 vs 2017 in `meta-oe`) and the latter it not developed since 2016. So my proposal would be to drop the whole layer with *optionally* (if anyone sees a value) adding newer `paho-mqtt` recipe from `meta-oe`.

@htot (or anyone else reading this), what do you think?

htot, 2018-07-15:
I agree. I like to delete as much as possible, so if available from upstream keep (`paho-mqtt` replaced by `paho-mqtt-c`), if shutdown cloud service dependent remove (`parse-embedded` should be removed as well).

Not sure about `xdk-daemon`, I haven't used that. Is it not available any more?
And `Wyliodrin` looked cool, sort of graphical programming like `LabVIEW`, for children? Didn't find the time to try it.

alext-mkrs, 2018-07-15:
Yeah, `xdk-daemon` was used as a board-side agent for [Intel XDK IoT Edition](https://software.intel.com/en-us/getting-started-with-xdk-and-iot), which was discontinued. And it's not the cloud, sorry - I mixed it up with `iotkit-agent`, that one was used for some cloud for storing IoT device data (the cloud was closed as well, even before Edison got EOLed).

As for the Wyliodrin - in general it could be fun, but I wonder how many people would really use it, given that Edisons are no loger sold and are not really the easiest target (and never were intended to be) for beginner developers. Plus, I'm not sure what's the support status from the Wyliodrin side, as that should be in place too. So in the spirit of getting rid of unused cruft, I'd remove it as a whole and add if anyone asks for it - the git history is always there should we need to dig it up.

htot, 2018-07-15:
I see. I thought the XDK was something you would need to develop nodejs applications. But according to this https://software.intel.com/en-us/xdk/articles/alternatives-to-using-the-intel-xdk-to-develop-nodejs-iot-applications you don't it at all. Then, drop them both.

I would be nice if someone got https://software.intel.com/en-us/iot/arduino-create to work at some point, given we have all the ingredients (MRAA, UPM and MQTT).

alext-mkrs, 2018-07-16:
Ok, I'll submit a PR for dropping that one.

Yeah, Arduino Create that could be an interesting one, though I'm not sure it's possible to add new board support without Arduino Create side-support, but that's just at the first glance.

htot, 2018-07-17:
And I see now it's a cloud based IDE so adding boards may be impossible.

commit 28c1c68c (2018-07-24): meta-intel-iot-middleware: remove this layer as obsolete

As discussed in #34, we no longer need this layer.

Closes #34.
closed 2018-07-24 by alext-mkrs