edison-fw/meta-intel-edison #36: Network configuration

htot, 2018-07-17:
I got Edison to operate as a wifi bridge using systemd-networkd, hostapd. 
The DHCP server is in my external router. The bridge receives network settings from the router, device eth0 and wlan0 don't and need none. Laptop connecting to the AP receives network settings from the DHCP server in the router. Very cool. Laptop, Edison, router etc. are all on the same subnet.

This configuration will not work when there is no eth0 present or when it's not connected as there is no DHCP server active then. The laptop will see the AP, send the passphrase but not receive a network address. Not good for initial configuration.

Also, without eth0, initial configuration would probably want to select wlan0 to connect as a client to another AP which will then provide the DHCP server.

Ways out:
1 detect DHCP server present or not. symlink appropriate network file and daemon-reload, restart networkd,
2 create a wireless router, with a DHCP server on AP wlan0. wifi connected devices will be on a subnet

I all cases have configuration script switch from AP to wifi client, which would mean symlinking appropriate file, stopping hostapd, maybe starting wpa_supplicant,  daemon-reload, restarting networkd.

Or we accept that networkd is not very good in on the fly network topology changes and use connman, which has a tether command, that I did not yet get to work.

My gut says stick with networkd and create a router (2).

alext-mkrs, 2018-07-18:
So let me try to undersand this better. Are you talking about the initial Edison configuration after it's unboxed, or about a generic ethernet to wifi "bridging" where Edison would serve as a wifi Access Point, while being connected to a main, wired network?

htot, 2018-07-18:
Both. The 2nd (bridged AP) works, but won't work without upstream network. So for initial configuration is not useful. But a routed AP with its own DHCP server on the wifi subnet would work for both cases.

alext-mkrs, 2018-07-19:
Ok, so I'd vote for KISS - just have two scripts, which will set it up in either mode. And by default the freshly installed image would start in isolated AP mode, with DHCP server, just as official image did, to make it easy to set the board up.

htot, 2018-07-20:
If I can get the routed AP to work, then I can fix the ap-mode-toggle and that should be it. It is probably simple, just not too many examples floating around.

htot, 2018-11-24:
I got this to work using `systemd-networkd` and `systemd-resolved`, but it appears to be much simpler using `connmanctl`. Even simpler than with the factory image as it appears `connmanctl` has matured and can handle tethering, no need to configure `hostapd`.

alext-mkrs, 2018-12-08:
Then it's probably worth it to switch to usign that one. Sorry I haven't been participating actively for q while now - life has changed a bit and I simply have no time to devote to this project besides just reading GH updates (and replying where it makes sense) for now. I haven'tgiven it up though, so I'm still around and if it changes again, who knows, I may get to contribute actively again, I certainly would like to :)

htot, 2018-12-08:
No worries. Nevertheless your comments and reviews are appreciated whenever you have a bit of time.

htot, 2018-12-22:
@alext-mkrs I'm working on a big pull with 4.20 kernel. This enables USB OTG, which can provide wired eth over usb. And with connman tethering working so well (creates a routed AP) we can drop hostapd. For this I am completely rewriting oobe. I am dropping edison-config-server (nodejs based) and instead moving that functionality into configure_edison (python with bottle). I hope you'll find a bit of time to play with this :-)

htot, 2019-02-10:
With PR #58 tethering can be done from connmanctl (tether wifi on ssid passphrase), no need for hostapd. Also connecting USB OTG works. Oobe rewrite is not yet completed, but no need to keep this open as that is a separate issue #32.

closed 2019-02-10 by htot