edison-fw/meta-intel-edison #37: Current master doesn't boot?

alext-mkrs, 2018-07-22:
@htot, let me check something with you - looks like current master built from scratch (no saved state whatsoever, manually removed all the dirs) doesn't boot for me. The kernel boots up until the below and then just stalls, then in a short while the board gets reset by the watchdog timer, and the process repeats endlessly.

Have you seen anything like that? Could you please try building the image from scratch and checking if it boots for you?

```
<...>
[    2.216898] usb usb2: Manufacturer: Linux 4.16.0-edison-no-acpi-standard xhci-hcd
[    2.224414] usb usb2: SerialNumber: xhci-hcd.2.auto
[    2.230064] hub 2-0:1.0: USB hub found
[    2.233897] hub 2-0:1.0: 1 port detected
[    2.238788] usbcore: registered new interface driver usblp
[    2.244474] usbcore: registered new interface driver usb-storage
[    2.250681] usbcore: registered new interface driver pl2303
[    2.256361] usbserial: USB Serial support registered for pl2303
```

<details><summary>Full log inside</summary>
</p>
```
U-Boot 2018.05 (Jun 17 2018 - 12:44:14 +0000)

CPU: x86_64, vendor Intel, device 406a8h
DRAM:  980.6 MiB
MMC:   mmc@ff3fc000: 0, mmc@ff3fa000: 1
Loading Environment from MMC... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Hit any key to stop autoboot:  0
=>
=> run edsboot
10749632 bytes read in 473 ms (21.7 MiB/s)
Valid Boot Flag
Setup Size = 0x00003e00
Magic signature found
Using boot protocol version 2.0d
Linux kernel version 4.16.0-edison-no-acpi-standard (oe-user@oe-host) #1 SMP Sat Jul 21 19:26:21 UTC 2018
Building boot_params at 0x00090000
Loading bzImage at address 100000 (10733760 bytes)
Magic signature found
Kernel command line: "tty1 console=ttyS2,115200n8 root=/dev/mmcblk1 rootfstype=ext4 systemd.unit=multi-user.target hardware_id=00"
Magic signature found

Starting kernel ...

[    0.000000] Linux version 4.16.0-edison-no-acpi-standard (oe-user@oe-host) (gcc version 7.3.0 (GCC)) #1 SMP Sat Jul 21 19:26:21 UTC 2018
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000097fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000003ffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000004000000-0x0000000005ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000006000000-0x000000003f4fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000003f500000-0x000000003fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec04000-0x00000000fec07fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] Notice: NX (Execute Disable) protection cannot be enabled: non-PAE kernel!
[    0.000000] random: fast init done
[    0.000000] SMBIOS 2.6 present.
[    0.000000] DMI: Intel Corporation Merrifield/BODEGA BAY, BIOS 542 2015.01.21:18.19.48
[    0.000000] e820: last_pfn = 0x3f500 max_arch_pfn = 0x100000
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI BIOS Error (bug): A valid RSDP was not found (20180105/tbxfroot-244)
[    0.000000] 129MB HIGHMEM available.
[    0.000000] 883MB LOWMEM available.
[    0.000000]   mapped low ram: 0 - 373fe000
[... 224 lines cut ...]
[    2.129000] xhci-hcd xhci-hcd.2.auto: irq 34, io mem 0xf9100000
[    2.135382] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    2.142234] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.149501] usb usb1: Product: xHCI Host Controller
[    2.154417] usb usb1: Manufacturer: Linux 4.16.0-edison-no-acpi-standard xhci-hcd
[    2.161932] usb usb1: SerialNumber: xhci-hcd.2.auto
[    2.167571] hub 1-0:1.0: USB hub found
[    2.171430] hub 1-0:1.0: 1 port detected
[    2.175937] xhci-hcd xhci-hcd.2.auto: xHCI Host Controller
[    2.181729] xhci-hcd xhci-hcd.2.auto: new USB bus registered, assigned bus number 2
[    2.189547] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[    2.197888] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003
[    2.204728] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.211985] usb usb2: Product: xHCI Host Controller
[    2.216898] usb usb2: Manufacturer: Linux 4.16.0-edison-no-acpi-standard xhci-hcd
[    2.224414] usb usb2: SerialNumber: xhci-hcd.2.auto
[    2.230064] hub 2-0:1.0: USB hub found
[    2.233897] hub 2-0:1.0: 1 port detected
[    2.238788] usbcore: registered new interface driver usblp
[    2.244474] usbcore: registered new interface driver usb-storage
[    2.250681] usbcore: registered new interface driver pl2303
[    2.256361] usbserial: USB Serial support registered for pl2303
```
</p>
</details>

alext-mkrs, 2018-07-22:
Hmm, this is really interesting - bumping up all the log levels and whatnot doesn't provide any  additional information around the point of hang. @andy-shev, have you seen anything like that (this is a non-ACPI kernel, 4.16)?

alext-mkrs, 2018-07-22:
Also tried without your HSU-related kernel patches (just in case, I know you've tested them, but I have no other ideas at the moment) - they are not the reason, the behavior doesn't change.

htot, 2018-07-22:
Could it be you have acpi enabled U-Boot. Master is no-acpi so will not boot with acpi=off on the command line. Can you check?

htot, 2018-07-23:
Yes, I see SFI. This is the U-Boot without ACPI. I haven't tried that for a while, could it be related? I don't think so.
Maybe systemd not starting anything?
Only other difference with me: I'm booting from USB and have combined USB hub / eth port.

alext-mkrs, 2018-07-23:
Thanks for suggestions, gentlemen. Yes, this is a non-ACPI U-Boot with non-ACPI kernel and I've compared the boot logs with 3.10.98 (the latest one I have booting at hand), it does an SDHC init around that place - and I haven't seen any messages about that in my current (failing kernel) log.

@htot, could you please post (attach) a log from your boot? I'll check the difference to see if there's anything interesting there. ACPI or non-ACPI is irrelevant, that's something I can account for when comparing.

htot, 2018-07-23:
Same kernel/U-Boot but with `acpi=off` 
[64b-no-acpi-boot.log](https://github.com/edison-fw/meta-intel-edison/files/2220825/64b-no-acpi-boot.log)

htot, 2018-07-24:
We might just be missing a module in the initramfs. I'll copy my rootfs to a sdhc this evening and try to boot it.

alext-mkrs, 2018-07-24:
Thanks for logs and other information - I'm a bit swamped right now to look deeper into this, but this will be useful when I get to that.

htot, 2018-07-24:
Alright, I just booted from sd (same 64b kernel, acpi enabled). My kernel command line is slightly different than yours, but should not matter. So, it boots the kernel, runs the init script, with systemd 237, waits for rootfs (with the annoying messages I kind seem to shut up), then switches root and continues booting.
```
Kernel command line: "**debugshell=10 quiet** tty1 console=ttyS2,115200n8 root=/dev/mmcblk1 rootfstype=ext4 systemd.unit=multi-user.target hardware_id=00"
Magic signature found

Starting kernel ...

starting version 237
Kernel with acpi enabled detected
Loading acpi tables
**Waiting for root device /dev/mmcblk1**
  10[    2.656202] EXT4-fs (mmcblk1): couldn't mount as ext3 due to feature incompatibilities
[    2.665541] EXT4-fs (mmcblk0p5): couldn't mount as ext3 due to feature incompatibilities
  9[    2.686285] EXT4-fs (mmcblk1): couldn't mount as ext2 due to feature incompatibilities
[    2.695597] EXT4-fs (mmcblk0p5): couldn't mount as ext2 due to feature incompatibilities
[    2.749394] EXT4-fs (mmcblk0p8): couldn't mount as ext3 due to feature incompatibilities
[    2.762104] EXT4-fs (mmcblk0p10): couldn't mount as ext3 due to feature incompatibilities
[    2.762599] EXT4-fs (mmcblk0p8): couldn't mount as ext2 due to feature incompatibilities
[    2.782466] EXT4-fs (mmcblk0p10): couldn't mount as ext2 due to feature incompatibilities
**Found device '/run/media/mmcblk1'**
```

My rootfs I make with:
```
sudo umount /dev/sdc
sudo sh -c 'cat edison-image-edison.ext4 >/dev/sdc' && sync /* this appears to be faster than dd */
sudo e2fsck -f -y -v -C 0 /dev/sdc && sudo resize2fs -p /dev/sdc && sync

```
My [core-image-minimal-initramfs-edison.manifest](https://github.com/edison-fw/meta-intel-edison/files/2225016/core-image-minimal-initramfs-edison.manifest.log).

I can't reproduce with acpi enabled image. Will try by building sumo64-acpi with acpi disabled (shorter build time from where I am now)

alext-mkrs, 2018-07-26:
Thanks. Another difference is that master is 32bit, not 64.

htot, 2018-07-26:
Eh, yes correct. But that would be very surprising if related. Key thing in my logs is the following line:
```
Jul 14 19:22:39 edison kernel:  mmcblk0: p1 p2 p3 p4 p5 p6 p7 p8 p9 p10
Jul 14 19:22:39 edison kernel: mmcblk1: mmc1:e624 SU02G 1.84 GiB 
Jul 14 19:22:39 edison kernel: EXT4-fs (mmcblk1): mounted filesystem with ordered data mode. Opts: (null)
```
which shows the init script waiting for the devices to appear then mounting it, which I don't see in your log.

BTW how did you insert your log with the cool unfold button?

PS I will be on a holiday from Sat 28 until Thu 7 Aug so won't be able to reply in that period.

alext-mkrs, 2018-07-29:
No worries and thanks for the information, have a nice holiday :)

As for the hiding trick - GH's Markdown allows for some HTML tags, including https://html.com/tags/details/, that's what I used. You can see the exact usage by clicking "edit" on my post.

lybtongji, 2018-08-04:
I can not boot too.

It stuck in a bootloop.

<details>
<summary>Boot Info</summary>

******************************
PSH KERNEL VERSION: b0182b2b
                WR: 20104000
******************************

SCU IPC: 0x800000d0  0xfffce92c

PSH miaHOB version: TNG.B0.VVBD.0000000c

microkernel built 11:24:08 Feb  5 2015

******* PSH loader *******
PCM page cache size = 192 KB
Cache Constraint = 0 Pages
Arming IPC driver ..
Adding page store pool ..
PagestoreAddr(IMR Start Address) = 0x04899000
pageStoreSize(IMR Size)          = 0x00080000

*** Ready to receive application ***

U-Boot 2018.05 (Jul 29 2018 - 13:14:33 +0000)

CPU: x86_64, vendor Intel, device 406a8h
DRAM:  980.6 MiB
MMC:   mmc@ff3fc000: 0, mmc@ff3fa000: 1
Loading Environment from MMC... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Hit any key to stop autoboot:  0
Target:blank
Partitioning already done...
Flashing already done...
** Unable to read file vmlinuz **
Error: Invalid Boot Flag (found 0xffef, expected 0xaa55)
## Kernel loading failed ...
zboot - Boot bzImage

Usage:
zboot [addr] [size] [initrd addr] [initrd size]
      addr -        The optional starting address of the bzimage.
                    If not set it defaults to the environment
                    variable "fileaddr".
      size -        The optional size of the bzimage. Defaults to
                    zero.
      initrd addr - The address of the initrd image to use, if any.
      initrd size - The size of the initrd image to use, if any.

Unknown boot mode: boot
Saving Environment to MMC... Writing to redundant MMC(0)... OK
Resetting to default boot mode and reboot...
resetting ...

******************************
PSH KERNEL VERSION: b0182b2b
                WR: 20104000
******************************

SCU IPC: 0x800000d0  0xfffce92c

PSH miaHOB version: TNG.B0.VVBD.0000000c

microkernel built 11:24:08 Feb  5 2015

******* PSH loader *******
PCM page cache size = 192 KB
Cache Constraint = 0 Pages
Arming IPC driver ..
Adding page store pool ..
PagestoreAddr(IMR Start Address) = 0x04899000
pageStoreSize(IMR Size)          = 0x00080000

*** Ready to receive application ***

U-Boot 2018.05 (Jul 29 2018 - 13:14:33 +0000)

CPU: x86_64, vendor Intel, device 406a8h
DRAM:  980.6 MiB
MMC:   mmc@ff3fc000: 0, mmc@ff3fa000: 1
Loading Environment from MMC... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Hit any key to stop autoboot:  0
Target:blank
Partitioning already done...
Flashing already done...
** Unable to read file vmlinuz **
Error: Invalid Boot Flag (found 0xffef, expected 0xaa55)
## Kernel loading failed ...
zboot - Boot bzImage

Usage:
zboot [addr] [size] [initrd addr] [initrd size]
      addr -        The optional starting address of the bzimage.
                    If not set it defaults to the environment
                    variable "fileaddr".
      size -        The optional size of the bzimage. Defaults to
                    zero.
      initrd addr - The address of the initrd image to use, if any.
      initrd size - The size of the initrd image to use, if any.

Unknown boot mode: boot
Saving Environment to MMC... Writing to MMC(0)... OK
Resetting to default boot mode and reboot...
resetting ...

......

</details>

htot, 2018-08-06:
@lybtongji that looks like a different issue. Maybe you press <esc> to interrupt boot and capture `printenv`?

lybtongji, 2018-08-06:
@htot `printenv` output:

```
audio_codec_name=audio_codec="dummy"
boot_edsboot=zboot 0x100000
boot_target_cmd=run do_flash_os;run do_probe_dfu;run do_compute_target;run mmc-bootargs;run load_kernel;zboot ${loadaddr}
bootargs_console=console=ttyS2 earlyprintk=ttyS2,keep
bootargs_debug=loglevel=4 acpi=off
bootargs_edsboot=tty1 console=ttyS2,115200n8 root=/dev/mmcblk1 rootfstype=ext4 systemd.unit=multi-user.target hardware_id=00
bootargs_ethconfig=rndis
bootargs_target=multi-user
bootargs_usbboot=debugshell=10 tty1 console=ttyS2,115200n8 root=/dev/sda rootfstype=ext4 systemd.unit=multi-user.target hardware_id=00
bootcmd=echo "Target:${target_name}"; run do_partition; run do_handle_bootargs_mode;
bootdelay=1
dfu_alt_info_ram=kernel ram ${loadaddr} 0x800000
dfu_alt_info_reset=reset ram 0x0 0x0
dfu_to_sec=3
do_audio_support=setenv audio_support platform_mrfld_audio.${audio_codec_name}
do_boot=run boot_target_cmd;
do_bootargs_rootfs=setenv bootargs_rootfs root=/dev/mmcblk0p8 rootfstype=ext4
do_compute_target=if itest.b ${first_install_retry} -gt ${first_install_max_retries} || itest.b ${ota_update_retry} -gt ${ota_update_max_retries}; then echo "Switch to Rescue target"; setenv bootargs_target rescue; saveenv; fi
do_dfu_alt_info_ifwi=setenv dfu_alt_info "ifwi${hardware_id} raw 0 8192 mmcpart 1;ifwib${hardware_id} raw 0 8192 mmcpart 2"
do_dfu_alt_info_mmc=setenv dfu_alt_info "ifwi${hardware_id} raw 0 8192 mmcpart 1;ifwib${hardware_id} raw 0 8192 mmcpart 2;u-boot0 part 0 1;u-boot-env0 part 0 2;u-boot1 part 0 3;u-boot-env1 part 0 4;boot part 0 7;rootfs part 0 8;update part 0 9;home part 0 10;vmlinuz fat 0 7;initrd fat 0 7"
do_dnx=setenv dfu_alt_info ${dfu_alt_info_ram};dfu 0 ram 0 ram;run bootcmd
do_fallback=echo "Unknown boot mode: $bootargs_mode"; env delete -f bootargs_mode; saveenv; echo "Resetting to default boot mode and reboot..."; reset;
do_flash=run do_force_flash_os;
do_flash_ifwi=run do_dfu_alt_info_ifwi ; dfu 0 mmc 0
do_flash_os=if itest.b ${do_flash_os_done} -eq 1 ; then echo "Flashing already done..." ; else run do_force_flash_os; fi
do_flash_os_done=1
do_flashall=run do_partition;run do_flash_ifwi;run do_flash_os
do_force_flash_os=run do_dfu_alt_info_mmc ; sleep 1 ; setenv do_flash_os_done 1 ; saveenv ; dfu 0 mmc 0
do_force_partition=echo "Partitioning using GPT"; gpt write mmc 0 ${partitions} ; mmc rescan; setenv do_partition_done 1 ; saveenv
do_handle_bootargs_mode=run do_preprocess_bootargs_mode; if itest.s $bootargs_mode == "ota" ; then run do_ota; fi; if itest.s $bootargs_mode == "boot" ; then run do_boot; fi; if itest.s $bootargs_mode == "flash"; then run do_flash; fi; run do_fallback; exit;
do_load_ota_scr=if fatload mmc 0:9 $ota_script_addr ota_update.scr ; then setenv ota_status 0 ; else setenv ota_status 1 ; fi
do_ota=run do_ota_init ; run do_load_ota_scr ; run do_source_ota_scr ; run do_ota_clean
do_ota_clean=saveenv ; reset
do_ota_init=setenv ota_status 1 ; env delete -f bootargs_mode
do_partition=if itest.b ${do_partition_done} -eq 1; then echo "Partitioning already done..."; else run do_force_partition ; fi
do_partition_done=1
do_preprocess_bootargs_mode=if env exists bootargs_mode ; then ; else setenv bootargs_mode "boot" ;fi;
do_probe_dfu=run do_dfu_alt_info_mmc ; dfu 0 mmc 0 ${dfu_to_sec}
do_source_ota_scr=if test $ota_status -eq 0 ; then if source $ota_script_addr ; then setenv ota_status 0 ; else setenv ota_status 2 ; fi ; fi
edsboot=setenv bootargs ${acpi} ${bootargs_edsboot}; run load_edsboot; run boot_edsboot
fdtcontroladdr=3d4541b0
first_install_max_retries=3
first_install_retry=0
hardware_id=00
init_dfu=run do_dfu_alt_info_mmc ; saveenv
load_edsboot=load mmc 0:9 0x100000 bzImage-initramfs
load_kernel=fatload mmc 0:7 ${loadaddr} vmlinuz
load_usbboot=load mmc 0:9 0x100000 bzImage-initramfs-usb
loadaddr=0x100000
mmc-bootargs=run do_bootargs_rootfs; run do_audio_support; setenv bootargs ${bootargs_rootfs} ${bootargs_console} ${bootargs_debug} g_multi.ethernet_config=${bootargs_ethconfig} systemd.unit=${bootargs_target}.target hardware_id=${hardware_id} g_multi.iSerialNumber=${serial#} g_multi.dev_addr=${usb0addr} ${audio_support}
ota_done=0
ota_script_addr=0x100000
ota_update_max_retries=3
ota_update_retry=0
partitions=uuid_disk=${uuid_disk};name=u-boot0,start=1MiB,size=2MiB,uuid=${uuid_uboot0};name=u-boot-env0,size=1MiB,uuid=${uuid_uboot_env0};name=u-boot1,size=2MiB,uuid=${uuid_uboot1};name=u-boot-env1,size=1MiB,uuid=${uuid_uboot_env1};name=factory,size=1MiB,uuid=${uuid_factory};name=panic,size=24MiB,uuid=${uuid_panic};name=boot,size=32MiB,uuid=${uuid_boot};name=rootfs,size=1536MiB,uuid=${uuid_rootfs};name=update,size=768MiB,uuid=${uuid_update};name=home,size=-,uuid=${uuid_home};
serial#=4aa08c597b139dd507f4946de1312c8a
target_name=blank
usb0addr=02:00:86:31:2c:8a
usbboot=setenv bootargs ${acpi} ${bootargs_usbboot}; run load_usbboot; run boot_edsboot
uuid_boot=db88503d-34a5-3e41-836d-c757cb682814
uuid_disk=21200400-0804-0146-9dcc-a8c51255994f
uuid_factory=333a128e-d3e3-b94d-92f4-d3ebd9b3224f
uuid_home=f13a0978-b1b5-1a4e-8821-39438e24b627
uuid_panic=f20aa902-1c5d-294a-9177-97a513e3cae4
uuid_rootfs=012b3303-34ac-284d-99b4-34e03a2335f4
uuid_uboot0=d117f98e-6f2c-d04b-a5b2-331a19f91cb2
uuid_uboot1=8a4bb8b4-e304-ae48-8536-aff5c9c495b1
uuid_uboot_env0=25718777-d0ad-7443-9e60-02cb591c9737
uuid_uboot_env1=08992135-13c6-084b-9322-3391ff571e19
uuid_update=faec2ecf-8544-e241-b19d-757e796da607

Environment size: 5520/65531 bytes

```

htot, 2018-08-06:
@lybtongji it looks like you see 2 problems.
1 - `load_kernel=fatload mmc 0:7 ${loadaddr} vmlinuz` seems to be failing. I can't remember why here `fatload` is used while we use `load` to read from sdhc or usb. I does work for me (as I use this as a fallback after I try to boot a broken kernel from usb). Maybe I modified this in my env manually and forgot to update [edison.env](https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-bsp/recipes-bsp/u-boot/files/edison.env)? I need to check this.
2 - after fatload fails (and then zboot), it reboots to 'boot' mode, but 'boot' seems to be not recognized in `do_handle_bootargs_mode`, so it falls back to `do_fallback`.  This should not be happening, maybe something changed in the processing of `if itest.s $bootargs_mode == "boot"`?

lybtongji, 2018-08-07:
@htot I compared the images of `rocko32` and `sumo32`. And found the difference in `boot` partition.

`rocko32`:
<pre>
=> ls mmc 0:7
    69632   ldlinux.sys
   186500   libcom32.c32
    24148   libutil.c32
      592   syslinux.cfg
    27104   vesamenu.c32
 10666112   <b>vmlinuz</b>
   122308   ldlinux.c32

7 file(s), 0 dir(s)
</pre>

---

`sumo32`:
<pre>
=> ls mmc 0:7
  7414592   <b>bzImage</b>
    69632   ldlinux.sys
   186500   libcom32.c32
    24148   libutil.c32
      592   syslinux.cfg
    27104   vesamenu.c32
   122308   ldlinux.c32

7 file(s), 0 dir(s)
</pre>

`vmlinuz` is not found in `sumo32`, but `bzImage` found.

alext-mkrs, 2018-08-08:
@lybtongji, you seem to have an interesting mix of what @htot has on the wiki as the "on the side" boot setup - with those `*edsboot` variables and loading from SD card, with the default setup. How do you install the image, specifically? You basically need to implement proper U-Boot config for that, which you seem to have in `edsboot` vars. As for the `rocko` image and `vmlinuz` - that's actually strange, IIRC `rocko` produced `bzImage` for me all the time. Are you sure you don't have some leftovers from older versions?

And yes - that's a completely different issue that the one I had.

lybtongji, 2018-08-08:
@alext-mkrs The commands which I build the image.

``` sh
mkdir os
cd os
git clone https://github.com/htot/meta-intel-edison.git
cd meta-intel-edison
git checkout rocko32
cd ..
ln -s meta-intel-edison/utils/Makefile.mk Makefile
```

``` sh
make clean && rm -rf bbcache/sstate-cache/
```

``` sh
make setup
make edison-image
```

Then flash the image to the eMMC by running `flashall.bat` without SD card.

``` bat
flashall.bat
```

I don't why it changes.

htot, 2018-08-09:
In any case it is not intended. Using flashall or Flash Tool Lite the image should just boot as is. Let me check what changed.

BTW I read people have great troubles connecting from Win10 with flashall. What is your experience here?

htot, 2018-08-09:
@lybtongji I can confirm your problem. As @alext-mkrs mentioned it is not related to his. Can you please create a new issue for this so we can track it?

How I confirmed:

When you run `make edison-image` the final step is creating the directory `out/current/build/toFlash/`. The file that contains the boot partition that is installed by flashall is called `edison-image-edison.hddimg`

In linux you can mount this file to view it's contents:
```
sudo losetup --partscan --find --show edison-image-edison.hddimg // in my case returns /dev/loop4
sudo mount /dev/loop4 /mnt
ls -l /mnt
-rw-r--r-- 1 ferry ferry 8725264 jul 14 23:04 bzImage
-r--r--r-- 1 ferry ferry  122308 jul 14 23:04 ldlinux.c32
-r--r--r-- 1 ferry ferry   69632 jul 14 23:04 ldlinux.sys
-rw-r--r-- 1 ferry ferry  186500 jul 14 23:04 libcom32.c32
-rw-r--r-- 1 ferry ferry   24148 jul 14 23:04 libutil.c32
-rw-r--r-- 1 ferry ferry     592 jul 14 23:04 syslinux.cfg
-rw-r--r-- 1 ferry ferry   27104 jul 14 23:04 vesamenu.c32
```
However, my `/dev/mmcblk0p7`
```
root@edison:~# ls -l /mnt
-r-xr-xr-x 1 root root   122308 Dec 27  2017 ldlinux.c32
-r-xr-xr-x 1 root root    69632 Dec 27  2017 ldlinux.sys
-rwxr-xr-x 1 root root   186500 Dec 27  2017 libcom32.c32
-rwxr-xr-x 1 root root    24148 Dec 27  2017 libutil.c32
-rwxr-xr-x 1 root root      592 Dec 27  2017 syslinux.cfg
-rwxr-xr-x 1 root root    27104 Dec 27  2017 vesamenu.c32
-rwxr-xr-x 1 root root 10174352 Dec 27  2017 vmlinuz

```

htot, 2018-08-12:
@alext-mkrs taking another look at your log, it does look you are booting the non-initramfs kernel or otherwise the sdhc modules are not being loaded. Can you please try:
```
setenv acpi 'debugshell=10'
run edsboot
```
If you are booting initramfs kernel but the sdhc subsystem is not working this will fallback to a busybox shell after 10 sec so we can investigate further.

alext-mkrs, 2018-08-12:
> you are booting the non-initramfs kernel

Nope, checked that

> or otherwise the sdhc modules are not being loaded

That is what seems to be happening - I just haven't yet got any time to look deeper into that since my previous updates here. The modules are in the initramfs though. I'll check this out and post the results.

htot, 2018-08-12:
Yeah, your bzImage is 10MB (mine 12MB, but 64bits), would be 6MB or so without initramfs.

Are you using the same hardware (Edison Arduino board)?

alext-mkrs, 2018-08-12:
> Are you using the same hardware (Edison Arduino board)?

Yes

Checked with `debugshell` parameter - no change, it just gets [seemingly] stuck after that `pl2303` driver load and then reset kicks in. No shell appears.

htot, 2018-08-12:
Hmm, new territory. Do you have any i2c device connected (I have nothing on the Ardiuno connector)? What you could try is flash the rootfs on a usb stick and use `run usbboot`. That eliminates the need for emmc/sdhc support.
Another is: enable acpi, flash U-Boot (with acpi patches) and run the new kernel with edsboot. This disables the fix for i2c-6, uses built-in i2c driver. In case it's i2c subsystem hanging.

alext-mkrs, 2018-08-12:
I have no i2c devices connected and honestly I don't think this can be related - RTC driver doesn't touch i2c. As for the USB boot - yeah, that's an option to get it booting, but my goal here is to fix this original problem.

After glancing over the `rtc_cmos` code this starts to vaguely look like something related to ACPi not being fully off in our presumably "no-ACPI" `master`, but this is currently more of a gut feeling than something I can prove :) Will look further into that.

htot, 2018-08-12:
Yes, but still doesn't explain why I can't reproduce. And I've tested all 4 combinations of u-boot/kernel acpi/no-acpi.
And you are right: the no-acpi kernel actually has acpi - just not working. So, when booted from no-acpi U-Boot the interpreter is switched off and nothing bad happens. When booted from acpi enabled U-Boot it will crash unless you have acpi=off on the command line. This is also true for the factory kernel.
The acpi kernel actually has patches for 'hw reduced acpi', which is what the acpi enabled U-Boot provides. But when booted from no-acpi U-Boot, again the interpreter is switched off. And surprisingly afaict everything still works.

alext-mkrs, 2018-08-12:
That's why this is a gut feeling and more investigation is needed :)

Though you've mentioned you rebuilt it 64bit-acpi with ACPI disabled - have you dropped all the caches and intermediate state before rebuilding? I've noticed (when experimenting with 64 vs 32 bits earlier - though that may be more invasive than acpi/noacpi) that it may not be properly refreshing all the pieces unless I drop everything manually. Anyway, I think I'll just either do further debug of `rtc_cmos` or probably first get to 64bit/ACPI just as you to see if that would establish a common reference point - because if not, then we have something else different in our builds.

alext-mkrs, 2018-08-19:
Okay, so I've just built and flashed the `sumo64-acpi` branch (including the U-Boot). Runs fine with `acpi=on` in kernel cmdline, stalls as originally reported with `acpi=off` (the only change). So this must have to do something with our kernel probably not being fully ACPI-free even when we disable it from the command line. Interesting. Will keep looking.

htot, 2018-08-19:
I did `make clean` and deleted bbcache/sstate-cache/, so no 100% fresh (this way I don't re-download everything), but close.

htot, 2018-08-19:
So now you have U-Boot and kernel with acpi. Turning acpi=off it should still boot (and does for me). Coincidentally I just tried to document this stuff here: https://htot.github.io/meta-intel-edison/1.3-ACPI-or-not.html

lybtongji, 2018-08-19:
@htot I flashed `sumo64-acpi` and found `blink-led` not working. Is it due to `ACPI`?

htot, 2018-08-19:
Yes, because of `acpi` pin names/numbers have changed. This affects `blink-led` and `mraa` and probably some other code too. I found the best way to access pins is from the libgpiod provided tools `gpiodetect gpiofind gpioget gpioinfo gpiomon gpioset`. See also [Edison wiki](https://edison.internet-share.com/wiki/ACPI).

For now I think we need the code first to detect if we are using `acpi` (`/sys/kernel/config/acpi/table/arduino/` exists?).

I didn't find time to fix `blink-led`, but if you want to take this, please do.

htot, 2018-08-20:
@alext-mkrs At least the acpi kernel has no i2c patches that could be causing this trouble. 

I remember @andy-shev mentioned that running acpi kernel from non-acpi U-Boot is not exactly the same as running acpi kernel from acpi U-Boot with `acpi=off`. But then, still we don't know why that works for me. Or why your non-acpi U-Boot/kernel crashes.

Looking at your first log (non-acpi U-Boot/kernel) I see:
`platform rtc_cmos: registered platform RTC device (no PNP device found)`

and in your log with debug code:
```
bus: 'pnp': add driver rtc_cmos
bus: 'platform': add driver rtc_cmos
```
As if the rtc_cmos driver gets probed 2x.
What additional logging did you turn on? I'll try the same for comparison.

htot, 2018-08-20:
@alext-mkrs You can download my kernel here: https://drive.google.com/file/d/1B78zm0dYqVPQjft2vyp-QaehWRjzF5-P/view?usp=sharing

This is 64b + acpi. Only I added bridge support and ftrace (is my debug kernel). It should be at least enough to get you into a debugshell. But if you still have a hang at least we know it's not related to the build system.

alext-mkrs, 2018-08-22:
Ok, thanks gentlemen - I'll check it out hopefully onthe upcoming weekend and let you know.

htot, 2018-08-26:
In the meanwhile I have kernel 4.18 + U-Boot v2018.09-rc2 (both acpi enabled) booting. It shows some errors in the log that seem to do no harm. After testing this week I will push to my repo probably next weekend.

alext-mkrs, 2018-08-26:
Now *that's* weird. I've just started looking into this further - by just booting the board as-is, left untouched (with `acpi=off`) from the last week's experiment. And all of a sudden it booted fine. By now I've done about a dozen boots with both `acpi=on` and `off` and it boots just fine in both. So now I don't know what was that when it didn't boot the last week and `on` vs. `off` made a difference.

alext-mkrs, 2018-08-26:
So this rules out the SD card (and it anyway worked fine in Windows where I write it and with official 3.10 kernel) - plus it actually never got to even try to mount it, that rtc driver stall happened much earlier than that.

Kernel version probably as well (unless the 32-bit has a bug 64 version doesn't, which is not impossible, but less likely).

Memory corruption looks like an interesting option, @andy-shev, how did you detect that in your case (or where I can read about that)?

htot, 2018-08-26:
Or maybe it's an initialization thing. We have seen that pressing the reset button does not always reset I2C and also SD card correct. I have even seen that the factory 3.10 had that issue. I have now made it my habit to plug/unplug power when resetting, and then issues with i2c not detecting the mux'es or sd card not appearing. Otoh iirc when booting from sd card, if it doesn't appear I would expect initramfs falling back to the debugshell.

htot, 2018-08-26:
BTW with linux 4.18 and U-Boot v2018.09-rc2 (both acpi enabled) I do get an error - maybe not coincidentally in rtc_cmos:
```
root@edison:~# journalctl -b | grep rtc
Aug 14 19:08:12 edison kernel: platform rtc_cmos: registered platform RTC device (no PNP device found)
Aug 14 19:08:12 edison kernel: genirq: Flags mismatch irq 8. 00000080 (rtc0) vs. 00004000 (0000:00:14.0)
Aug 14 19:08:12 edison kernel:  ? rtc_wake_on+0x20/0x20
Aug 14 19:08:12 edison kernel:  ? rtc_dev_init+0x2b/0x2b
Aug 14 19:08:12 edison kernel:  ? rtc_dev_init+0x2b/0x2b
Aug 14 19:08:12 edison kernel:  ? rtc_dev_init+0x2b/0x2b
Aug 14 19:08:12 edison kernel: rtc_cmos: probe of rtc_cmos failed with error -16
```
0000:00:14.0 is the PMU, with driver intel_mid_pwr. The error goes away when I boot with `acpi=off`, the other way around from what you saw.

alext-mkrs, 2018-08-26:
Yeah, unclean reset (there's AFAIK some hw- or board-level bug/peculiarity, which doesn't reset it properly using reset button) was something I suspected, but I (1) don't use the button and (2) I tried various combinations today (warm/cold resets) and it didn't show up. Oh well. Anyway, I decided to give it one more try by debugging the driver itself. If that doesn't fly, my plan is to get back to our past discussion and check what do we lose by going 64 bit instead of 32 and propose to make 64 bit the default and just focus on ironing out any problems like that UART DMA you've faced. Just to have fewer configs (ideally one acpi-64, if only Andy didn't say ACPI is totally not prime time yet ;) to take care of.

alext-mkrs, 2018-08-26:
...and yes, that error you observe is interesting - there seems to be *something* about RTC settings/driver we have, in ACPI vs. non-ACPI environment.

htot, 2018-08-26:
When I asked Andy he said "I know about IRQ flags mismatch, but this doesn't prevent boot anyhow.
One should create an AML excerpt for RTC eventually (and perhaps HPET) and it might require some refactoring in the kernel.". But I don't remember seeing this error before.

The UART problems on 64b seem to be caused by an enormous interrupt latency that isn't present on 32b. I haven't been able to find the root of that. 

Other then that I haven't seen any other 64b related problems, no particular advantages either, see https://htot.github.io/meta-intel-edison/1.4-i686-or-x86_64.html (this is from my master/docs generated automatically by gh, solves #12 once we find it is complete enough).

alext-mkrs, 2018-08-26:
Got it for IRQs. Interesting page for 32 vs 64 bit. BT while Edison's version of it was never advertised as 64bit (don't know why), the core/SoC used there was also used in Merrifield phone platform - and there it was officially 64bit from the very beginning (https://ark.intel.com/products/codename/41559/Merrifield?q=merrifield).

Ok, I've found it. Your note about the reset (even though I tested it today to no avail) got me thinking further and I actually just fallen prey to my specific workflow I used. In short - everything works ok, it's just that reset (cold vs. warm) that makes all the difference. I used the following flow:
1. Boot into official image
1. Mount the ex-"update" partition, copy the kernel from my build host
1. Boot into U-Boot, start DFU mode (`run do_force_flash_os`), flash the U-Boot via `dfu-util`
1. Run `reset` from U-Boot, when it resets `run edsboot` to try to boot into the new kernel. It would hang on that RTC driver

And now if I do a poweroff instead of `reset` in U-Boot, it would boot normally. The reason is apparently that booting into 3.10-based official image takes up *something* that isn't cleaned up by a simple warm reset. That's why it didn't come up in my testing earlier today - because I was only getting to the new kernel and U-Boot, never "old kernel->new kernel" - and did come up last week (because I was following the full flow described above).

Okay, mistery solved and luckily nothing to be done here, just remember to do cold resets **all the time** :)

Now back to 32 vs 64 and non-ACPI vs ACPI. I've noticed on your "ACPI or non-ACPI" page that

> Work on providing non-acpi U-Boot patches will end with the release of U-Boot v2018.09

is this still the plan (@andy-shev - probably that's a question to you)? If so I think non-ACPI just makes no sense, I'd rather then join (for real this time) the enablement activity for ACPI, to avoid spending my [rather limited, alas] time for non-ACPI version. There are I guess anyway not that many "production" users of our image right now and it would be more beneficial to get ACPI enabled sooner than drag the non-ACPI around in "somewhat working"/"it boots so probably works" state.

And one more note about 32 vs 64 bits - those instruction-related drawbacks look significant enough, however I wonder if we can actually help it by using a different - targeted at Silvermont, unlike the current one - Yocto comiler optimizations. We currently use `core2-*`, while according to the class comments, the one for Silvermont would be `corei7-*` ([link](https://lists.yoctoproject.org/pipermail/poky/2014-January/009540.html)). Actually that was exactly the thing I was trying to test when I got into this boot stall problem. Let me proceed with this task then - if you have any specific benchmarks you want me to try, let me know.

All in all, I'm now strongly inclined to ditch 32bit-non-ACPI and focus on fully enabling and stabilizing 64bit-ACPI, unless there's any showstopper there (interrupt delay might be one, but I hope this is just something that needs more detailed investigation/fix).

htot, 2018-08-26:
With your reset problems you may have actually found a new issue. I believe (but can not prove) the one I had was related to i2c messages being interrupted mid message and the i2c muxes only get reset on cold reset. My suspicion was linux does not properly perform the sw reset sequence on power up.

Actually I had corei7- in the past. It didn't seem to bring any advantages. And when I found the hsu issues thought it was related, changed to core2, didn't see any change and just left it.

Thing is 32b non-acpi is working quite well, mraa works, blink-led, hsu, and I believe but did not test, i2c-6. Only thing is, the pin muxes are fixed. Would be a waste to ditch.

Maybe 64b will never work well, and then we should create 32b acpi branch. But acpi is where we should put our efforts.

The stuff in there that works now, is working well enough. There is just some things missing, like acpi code that tell the SCU to change the mode of i2c-6 pins. And stuff that Andy works on ([pmic](https://edison.internet-share.com/wiki/PMIC) [acpi](https://edison.internet-share.com/wiki/ACPI)). For me, I really would need to have Andy tell me how to do that, step-by-step. And I believe he got the USB gadget stuff working, but haven't tried all that yet.

alext-mkrs, 2018-08-26:
Okay, makes sense. Let's then leave the master as-is for now and focus on ACPI.

Now, @andy-shev - do you use 32 or 64 bit for developing that (AFAIU - 64, but let me confirm)? And the next question would be - which part of what @htot described above (or described on those Edison-wiki pages) I can help you *to test*? I figure I'd start with just helping out with testing while I'm getting up to speed with ACPI in kernel in general (and Edison in particular).

andy-shev, 2018-08-27:
So, to the discontinuation:
- the work on U-Boot w/o ACPI support will be stopped after September release, the new _edison-acpi_ branch there would be WIP one
- the (non-ACPI) kernel de facto already is not changing for a long time and while the branch would be rebased from time to time, the WIP one will be called _eds-acpi_ on GH (now it's on BB with some noisy patches)

staroselskii, 2018-08-30:
Hello, guys!

I'm the guy Andy told you about. Here's the [PR](https://github.com/andy-shev/u-boot/pull/12) with I2C#6 support. It's not ready yet but it's already working. You're welcome to take a look and comment/test it out.

andy-shev, 2018-08-31:
@htot 
It would be nice if you can test it after I merge the mentioned PR. I'm expecting @staroselskii will address my comments today and tonight we will have this in my edison-acpi branch of U-Boot.

htot, 2018-08-31:
I've have been following your discussion closely :-) Do we need patches to the kernel as well or is this independent?

htot, 2018-09-01:
With no device connected i2cdetect -r -y 6 prints very slowly, same as before (-r -y 1 prints fast). It that normal? Will try to  connect a device tomorrow.

staroselskii, 2018-09-02:
@htot 

No, it's not normal. On my setups the former prints fast as well. I tested it on two boards. One of them had an external sensor connected and the other did not. Both gave good results.

Want to know what's wrong before sending upstream. Didn't expect it not to work.

Could you please share a log from U-Boot booting? I'd be very much interested if you could add "#define DEBUG" on the top of pinmux.c. In would give more context. I've been mostly using only two Edisons for the tests. There's a slight possibility something would be working differently on others due to some different settings pins are configured in the firmware. But I seriously doubt it. 

Where are you testing? I've been using rocko64-acpi as a base with kernel checked out at eds-4.16-unified. Built for 64-bit. I boot off an eMMC (i.e. I don't use run edsboot but rather use ACPI for default boot as well).

htot, 2018-09-02:
I am using ~ the same, rocko64-acpi, but with eds-4.18-unified (from my repo) booting off USB. I'll try making a log.

staroselskii, 2018-09-02:
@htot 

If you have the time to test, I'd really appreciate that.

Here's what my U-Boot prints out

```
U-Boot 2018.09-rc2 (Sep 02 2018 - 14:53:01 +0000)

CPU: x86_64, vendor Intel, device 406a8h
DRAM:  980.6 MiB
scu: v: 0x203511 p: 0xff0c1d28 bits: 1, mask: 7 bufcfg: 0xff0c1d28
scu: v: 0x203591 p: 0xff0c1d2c bits: 1, mask: 7 bufcfg: 0xff0c1d2c
MMC:   mmc@ff3fc000: 0, mmc@ff3fa000: 1
Loading Environment from MMC... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
```

htot, 2018-09-02:
I wanted to check these things today but I have a fever and feel sick as a dog.

I remember in the first line (2018.09-rc2...) I have the SHA of the top commit and followed by 'dirty'.

staroselskii, 2018-09-02:
Sorry to hear that. Get well! This thing waits for sure. I'll try to reproduce these on other Edisons. I might have a couple more available for testing.

staroselskii, 2018-09-03:
I tested this patch series with 5 more Edisons and 2 different boards connected to them. All tests actually gave me what I wanted.

staroselskii, 2018-09-03:
I contacted Andy on the issue and said that he got some timeouts but it was better than before.

I just tested it on Intel Breakout board for Edison and it just works as well.

Andy also recommended to share these:
- output of `acpidump -o tables.dat`
- dmesg, when kernel command line has `ignore_loglevel` in it
- output of `grep -H 15 /sys/bus/acpi/devices/*/status`
- /sys/kernel/debug/gpio
- /sys/kernel/debug/pinctrl/.../pins
- output of `i2cdetect -y -r 6`

You can find them all [here](https://gist.github.com/staroselskii/097808e05fd609dbafd4fe5ebd618708)

[20 more replies]

closed 2018-09-13 by alext-mkrs