edison-fw/meta-intel-edison #40: How to add TUN/TAP kernel driver?

lybtongji, 2018-08-06:
I have tried to add `IMAGE_INSTALL_append += " kernel-module-tun"` to `out/linux64/build/conf/local.conf`. But got error:

```
ERROR: edison-image-1.0-r0 do_rootfs: Unable to install packages. Command '/home/recover/os/out/linux64/build/tmp/work/edison-poky-linux/edison-image/1.0-r0/recipe-sysroot-native/usr/bin/apt-get  install --force-yes --allow-unauthenticated alsa-utils-alsactl alsa-utils-alsamixer alsa-utils-amixer alsa-utils-aplay ap-mode-toggle apt asciidoc bash-completion battery-voltage bcm43340-fw bind-utils blink-led bluez5-dev bluez5-noinst-tools bluez5-obex bluez5-testtools bridge-utils libcares2 libcares-dev cleanjournal connman connman-client connman-tools cppzmq-dev crashlog curl curl-dev dnsmasq dosfstools dpkg e2fsprogs-badblocks e2fsprogs-e2fsck e2fsprogs-mke2fs e2fsprogs-tune2fs ethtool file git git-perltools gstreamer1.0 gstreamer1.0-meta-audio gstreamer1.0-meta-base hostapd htop i2c-tools iotop iperf3 iptables iw kernel-module-tun kernel-modules ldd less libcom-err2 libe2p2 libev4 libev-dev libext2fs2 libgomp1 libgpiod libnss-mdns-dev libpam libpulse0 libpulse-simple0 libpulsecommon libpulsecore libsodium-dev libss2 libstdc++6 lrzsz lshw lsof mbedtls mbedtls-dev mcu-fw-bin mcu-fw-load mosquitto-clients mosquitto-dev libmraa-dev libmraa-doc net-tools nodejs-dev nodejs-npm ntp ofono oobe openssh-sftp-server openvpn opkg packagegroup-core-boot packagegroup-core-buildessential packagegroup-core-ssh-openssh packagegroup-core-tools-debug parted pciutils perf perl perl-module-extutils-cbuilder perl-module-extutils-install perl-module-extutils-makemaker perl-module-extutils-manifest perl-module-extutils-parsexs perl-modules post-install powertop pulseaudio-misc pulseaudio-server pwr-button-handler python python-argparse python-audio python-compile python-ctypes python-distutils python-email python-html python-image python-importlib python-json python-misc python-netserver python-numbers python-paho-mqtt-dev python-pip python-pkgutil python-pydoc python-setuptools python-unittest python-xmlrpc python3-pip python3-setuptools resize-rootfs rng-tools run-postinsts screen sketch-check sst-fw-bin sudo swig systemd-analyze tcpdump tzdata u-boot-fw-utils upm-dev usbutils vim watchdog-sample wget wireless-tools wpa-supplicant xmlto zeromq-dev' returned 100:
Reading package lists...
Building dependency tree...
Package kernel-module-tun is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'kernel-module-tun' has no installation candidate

ERROR: edison-image-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/recover/os/out/linux64/build/tmp/work/edison-poky-linux/edison-image/1.0-r0/temp/log.do_rootfs.46661
ERROR: Task (/home/recover/os/meta-intel-edison/meta-intel-edison-distro/recipes-core/images/edison-image.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4623 tasks of which 4562 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/recover/os/meta-intel-edison/meta-intel-edison-distro/recipes-core/images/edison-image.bb:do_rootfs
Summary: There were 2 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
Makefile:53: recipe for target 'edison-image' failed
make: *** [edison-image] Error 1

```

htot, 2018-08-06:
I haven't tried, but I am guessing you need to modify the kernel config for the module to be built. I am still on a holiday but will put instructions how to do it here when I return.

lybtongji, 2018-08-06:
@htot Thanks for reply. Have a nice holiday!

lybtongji, 2018-08-06:
OK. I have made it.

First, I use `bitbake virtual/kernel -c menuconfig` to generate `.config`.
```
Device drivers > Network device support > Universal TUN/TAP device driver support.
```
And I found `CONFIG_TUN=m` appeared in `~/os/out/linux64/build/tmp/work/edison-poky-linux/linux-yocto/4.16.0-r0/linux-edison-standard-build/.config`

Then I copied the file to `recipes-kernel`  directory.
``` sh
cp ~/os/out/linux64/build/tmp/work/edison-poky-linux/linux-yocto/4.16.0-r0/linux-edison-standard-build/.config ~/os/meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/files/defconfig
```
And edited the file `~/os/meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/linux-yocto_4.16.0.bb`.
``` sh
vim ~/os/meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/linux-yocto_4.16.0.bb
```
Add `file://defconfig` to `SRC_URI`.

At last, I rebuilt the image and found the `tun.ko` was appeared in `out/linux64/build/tmp/work/edison-poky-linux/linux-yocto/4.16.0-r0/linux-edison-standard-build/drivers/net/`.

Is there a better way?

alext-mkrs, 2018-08-08:
So replacing defconfig is one way, but it's quite invasive.

Much more modular and focused one would be to use the so called kernel config fragments (a Yocto feature - see developer's manual) - see e.g. all those `*.cfg` files in https://github.com/edison-fw/meta-intel-edison/tree/master/meta-intel-edison-bsp/recipes-kernel/linux/files

htot, 2018-08-09:
What I do to create kernel fragments:
```
bitbake linux-yocto -c kernel_configme -f
bitbake linux-yocto -c menuconfig   // configure the kernel
bitbake linux-yocto -c diffconfig   // create kernel fragment after configuring, manually copy to recipe files
bitbake linux-yocto -c cleansstate  // clean up

```

lybtongji, 2018-08-17:
solved

closed 2018-08-17 by lybtongji