edison-fw/meta-intel-edison #41: the master branch cannot boot

lybtongji, 2018-08-10:
After building `master` branch and flash the image, it stuck in a bootloop.

Here is the boot information:
<details>
<summary>Boot Info</summary>
<pre>
******************************
PSH KERNEL VERSION: b0182b2b
                WR: 20104000
******************************

SCU IPC: 0x800000d0  0xfffce92c

PSH miaHOB version: TNG.B0.VVBD.0000000c

microkernel built 11:24:08 Feb  5 2015

******* PSH loader *******
PCM page cache size = 192 KB
Cache Constraint = 0 Pages
Arming IPC driver ..
Adding page store pool ..
PagestoreAddr(IMR Start Address) = 0x04899000
pageStoreSize(IMR Size)          = 0x00080000

*** Ready to receive application ***

U-Boot 2018.05 (Aug 10 2018 - 03:32:08 +0000)

CPU: x86_64, vendor Intel, device 406a8h
DRAM:  980.6 MiB
MMC:   mmc@ff3fc000: 0, mmc@ff3fa000: 1
Loading Environment from MMC... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Hit any key to stop autoboot:  0
Target:blank
Partitioning already done...
Flashing already done...
** Unable to read file vmlinuz **
Error: Invalid Boot Flag (found 0xffef, expected 0xaa55)
## Kernel loading failed ...
zboot - Boot bzImage

Usage:
zboot [addr] [size] [initrd addr] [initrd size]
      addr -        The optional starting address of the bzimage.
                    If not set it defaults to the environment
                    variable "fileaddr".
      size -        The optional size of the bzimage. Defaults to
                    zero.
      initrd addr - The address of the initrd image to use, if any.
      initrd size - The size of the initrd image to use, if any.

Unknown boot mode: boot
Saving Environment to MMC... Writing to MMC(0)... OK
Resetting to default boot mode and reboot...
resetting ...

******************************
PSH KERNEL VERSION: b0182b2b
                WR: 20104000
******************************

SCU IPC: 0x800000d0  0xfffce92c

PSH miaHOB version: TNG.B0.VVBD.0000000c

[... 686 lines cut ...]
[    3.408527] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    3.417000] CPU: 1 PID: 1 Comm: swapper/0 Not tainted 4.16.0-edison-no-acpi-standard #1
[    3.425192] Hardware name: Intel Corporation Merrifield/BODEGA BAY, BIOS 542 2015.01.21:18.19.48
[    3.434181] Call Trace:
[    3.436708]  dump_stack+0x47/0x66
[    3.440116]  panic+0x8a/0x1cb
[    3.443167]  mount_block_root+0x12b/0x1d0
[    3.447285]  mount_root+0xf6/0x10f
[    3.450778]  prepare_namespace+0x116/0x147
[    3.454977]  kernel_init_freeable+0x18e/0x1a0
[    3.459446]  ? rest_init+0x90/0x90
[    3.462937]  kernel_init+0x8/0xf0
[    3.466336]  ret_from_fork+0x2e/0x38
[    3.470042] Kernel Offset: 0x10c00000 from 0xc1000000 (relocation range: 0xc0000000-0xf7bfdfff)
[    3.478953] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
</pre>
</details>

And get the error:

```
[    3.391957] VFS: Cannot open root device "mmcblk0p8" or unknown-block(0,0): error -6
[    3.399932] Please append a correct "root=" boot option; here are the available partitions:
[    3.408527] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
```

htot, 2018-08-11:
I can confirm all of the above. There appear to be multiple problems that we did not have with Rocko:
1) With Sumo Yocto changed to incorporating `bzImage` into `edison-image-edison.hddimg` instead of `vmlinuz`. This is not a big issue as we only need to change the file name in `meta-intel-edison-bsp/recipes-bsp/u-boot/files/edison.env`.
2) It looks like we need to put the baud rate in the `console` (and `earlyprintk`) option to see any error messages with linux 4.16. Without this fix, when there are no errors, linux will still boot but very quietly.
3) With `bzImage` the kernel crashes as `mmcblk0p8` does not appear. Adding `rootwait` to the kernel arg it waits forever. A usb disk does appear if present, sdhc not. As the mmc driver is built as module we need the initramfs to boot from emmc same as for sdhc. However Yocto builds `bzImage-edison.bin` into `edison-image-edison.hddimg` instead of `bzImage-initramfs-edison.bin`. @alext-mkrs do you know how to change which kernel image to build into the hddimage?

I could test the above because my mmcblk0p9 still contains bootable kernels so I can still boot from USB. mmcblk0p7 is a 9MB fs in a 33MB partition, too small to hold the initramfs kernel (12MB). I created a new fat32 partition in p7, copied `bzImage-initramfs-edison.bin` there and renamed to `bzImage`. Now it boots normally.

lybtongji, 2018-08-11:
I used a temporary solution just now.
After built the image.
I ran the command `bitbake -c cleanall edison-image` to cleanup the hddimg files.
Then added `bzImage-initramfs-edison.bin` to `hddimg` folder manually.
``` sh
mkdir -p ~/os/out/linux64/build/tmp/work/edison-poky-linux/edison-image/1.0-r0/edison-image-1.0/hddimg
cp -v ~/os/out/linux64/build/tmp/work/edison-poky-linux/linux-yocto/4.16.0-r0/deploy-linux-yocto/bzImage-initramfs-edison.bin ~/os/out/linux64/build/tmp/work/edison-poky-linux/edison-image/1.0-r0/edison-image-1.0/hddimg/vmlinuz
```
Rebuilt the image:
``` sh
bitbake edison-image
```
Got the hddimg file:
```
~/os/out/linux64/build/tmp/deploy/images/edison/edison-image-edison.hddimg
```
Then replaced the old hddimg file by the new one and flashed it to the edison.
Now, I can boot into the system.

htot, 2018-08-11:
Good one.

I hope @alext-mkrs will know how to fix this permanently. In the past he found [VM_DEFAULT_KERNEL = "bzImage-initramfs-edison.bin"](https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-distro/recipes-core/images/edison-image.bb#L26) which I think did the trick.

alext-mkrs, 2018-08-12:
Not OTMH, but I'll take a look, there should be something.

htot, 2018-08-12:
I have the bad feeling they want us to use WIC, but have no clue how that works.

alext-mkrs, 2018-08-12:
BTW, @htot, do you remember/know why don't we currently build the mmc drive into the kernel? I sort of assumed that's because of that i2c-6 problem, but re-reading the commit message for that one now, I don't think that's actually related. Built-in module would solve that problem. Initramfs was introduced only for ACPI table loading AFAIU, so I don't see why we couldn't just have the most essential drivers built-in. That will help to avoid the fix I'm about to share.

alext-mkrs, 2018-08-12:
@lybtongji, please give it a try with the above two patches. One ensures proper initramfs kernel is added to the hddimg and another one corrects the U-Boot env file. The second one you can actually do manually using `setenv`, this is just for fix completeness.

lybtongji, 2018-08-12:
@alext-mkrs Not working...
The boot message still shows:
```
[    3.386858] VFS: Cannot open root device "mmcblk0p8" or unknown-block(0,0): error -6
[    3.394831] Please append a correct "root=" boot option; here are the available partitions:
[    3.403430] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
```
`bzImage` was replaced by `bzImage-initramfs-edison.bin`:
``` sh
recover@ilab:~/os$  ll ~/os/out/linux64/build/tmp/work/edison-poky-linux/linux-yocto/4.16.0-r0/deploy-linux-yocto/
total 18384
drwxrwxr-x  2 recover recover     4096 8月  13 00:50 ./
drwxrwxr-x 15 recover recover     4096 8月  13 01:08 ../
lrwxrwxrwx  2 recover recover       44 8月  13 00:50 bzImage -> bzImage--4.16.0-r0-edison-20180812160734.bin
-rw-r--r--  2 recover recover  7445984 8月  13 00:50 bzImage--4.16.0-r0-edison-20180812160734.bin
lrwxrwxrwx  2 recover recover       44 8月  13 00:50 bzImage-edison.bin -> bzImage--4.16.0-r0-edison-20180812160734.bin
-rw-r--r--  2 recover recover 10791680 8月  13 00:50 bzImage-initramfs-4.16.0-r0-edison-20180812160734.bin
lrwxrwxrwx  2 recover recover       53 8月  13 00:50 bzImage-initramfs-edison.bin -> bzImage-initramfs-4.16.0-r0-edison-20180812160734.bin
-rw-rw-r--  2 recover recover   575511 8月  13 00:50 modules--4.16.0-r0-edison-20180812160734.tgz
lrwxrwxrwx  2 recover recover       44 8月  13 00:50 modules-edison.tgz -> modules--4.16.0-r0-edison-20180812160734.tgz
recover@ilab:~/os$  ll ~/os/out/linux64/build/tmp/work/edison-poky-linux/edison-image/1.0-r0/edison-image-1.0/hddimg/
total 10856
drwxr-xr-x 2 recover recover     4096 8月  13 01:08 ./
drwxr-xr-x 3 recover recover     4096 8月  13 01:08 ../
-rw-r--r-- 1 recover recover 10791680 8月  13 01:08 bzImage
-r--r--r-- 1 recover recover    68599 8月  13 01:08 ldlinux.sys
-r--r--r-- 1 recover recover   186500 8月  13 01:08 libcom32.c32
-r--r--r-- 1 recover recover    24148 8月  13 01:08 libutil.c32
-rw-r--r-- 1 recover recover      592 8月  13 01:08 syslinux.cfg
-rw-r--r-- 1 recover recover    27104 8月  13 01:08 vesamenu.c32
```
But still not in hddimg file:
``` sh
recover@ilab:~/os$  sudo mount ~/os/out/linux64/build/tmp/deploy/images/edison/edison-image-edison.hddimg /mnt/
[sudo] password for recover:
recover@ilab:~/os$  ll /mnt/
total 7718
drwxr-xr-x  2 root root   16384 1月   1  1970 ./
drwxr-xr-x 27 root root    4096 8月   8 06:47 ../
-rwxr-xr-x  1 root root 7445984 8月  13 01:08 bzImage*
-r-xr-xr-x  1 root root  122308 8月  13 01:08 ldlinux.c32*
-r-xr-xr-x  1 root root   69632 8月  13 01:08 ldlinux.sys*
-rwxr-xr-x  1 root root  186500 8月  13 01:08 libcom32.c32*
-rwxr-xr-x  1 root root   24148 8月  13 01:08 libutil.c32*
-rwxr-xr-x  1 root root     592 8月  13 01:08 syslinux.cfg*
-rwxr-xr-x  1 root root   27104 8月  13 01:08 vesamenu.c32*
```

alext-mkrs, 2018-08-12:
@htot, understood, thanks, will look deeper into that then when I have time.

@lybtongji, thanks for testing. Looks like the approach I took is doing the file substitution too late, at the end of the day. And the variable they now use in live-vm-common looks too generic to mess up with directly. I'll see what else could be done here.

alext-mkrs, 2018-08-12:
@htot, thanks, that now makes more sense to me. I wonder though if this has changed since then, I remember there were some kernel patches mentioning SDHC/MMC power or something along these lines...

@lybtongji, here comes the second try - please see the same branch (I've force-pushed the updated two commits). As they have no hooks whatsoever in that new live-vm-common implementation, the only [more or less meaningful] way I see right now is the per-recipe variable override. That means the kernel file name will need to change as well, to the one that's generated. I've checked it in a very limited way (just by building and *not* from scratch), but at least it now builds the hddimg with initramfs kernel.

lybtongji, 2018-08-12:
@alext-mkrs How can I find the commit of yours?
I used the command `git cherry-pick -n <SHA>` to apply your commits.
But `git log alext-mkrs/master --not origin/master` shows nothing.
I can not find the commits in `https://github.com/alext-mkrs/meta-intel-edison/commits/master`  either.
[image]

htot, 2018-08-14:
@lybtongji wrong branch. You need @alext-mkrs fix-hddimg-gen branch.

@alext-mkrs I just built and flashed your fix-hddimg-gen rebased onto top of sumo64-acpi (after `make clean`, `make setup`. I used linux version of Flash Tool Lite to flash. It is working just fine, you are a wizard.

2 things could use improvement:
1) bootargs_console could use 115200n8. As is, it all boots but very quiet (nothing) until root: prompt.
2) bootargs_debug turn off acpi, even when we are building acpi U-Boot and kernel.

I propose you test this with no-acpi/32b see if it solves your boot problems.

lybtongji, 2018-08-15:
@htot Thanks. I find it now.

alext-mkrs, 2018-08-15:
@htot, thanks for checking this out. I haven't modified the rest of the standard U-Boot env pieces for now, just for @lybtongji to check the kernel itself. If that works I think we'll need a separate issue to do an overhaul of those. I'm currently using the "dual-booting" scheme on mine (4.16 kernel on the SD card while the official image is still on eMMC), to have a reference, so I haven't bumped into this just yet.

htot, 2018-08-15:
@alext-mkrs I'm not complaining :-) It's more of a note to self that we add 2 more commits to these before pulling.

htot, 2018-08-16:
@lybtongji I created PR https://github.com/edison-fw/meta-intel-edison/pull/42 (from here https://github.com/htot/meta-intel-edison/tree/sumo32 ) that is equal to current master + fix-hddimg-gen + fixes the baud rate. If you have already tested, can you let us know the results?
I tested the sumo64-acpi version myself and that is working fine.

lybtongji, 2018-08-16:
@htot It's fine. I have no problem with booting now. :smile:

PR #42 (2018-08-16): Sumo32
This pull in in the hddimg generation changes + sets baud rate

Closes #41
commit 290cdb85 (2018-08-16): edison-image: correct live image generation config

Openembedded-core commit

5493bb5ba4b4520f944d38b214a3d53a5282e0be
live-vm-common.bbclass: Don't use vmlinuz or VM_DEFAULT_KERNEL

broke our mechanism for including initramfs kernel into live image.

As that image (hddimg) is used for flashing the Edison's MMC boot
partition, that in turn breaks the board boot as it's not able to
find the rootfs - it's on MMC and we build MMC/sdhc as modules right now.

Fixes #41.
commit 8effdf60 (2018-08-16): edison.env: correct kernel file name

In sumo the live image generation class no longer uses a hardcoded "vmlinuz"
name for the kernel file. As we use "bzImage" in Edison, update accordingly.

I've left the "initrd" piece in the dfu_alt_info_mmc just in case we ever
want to use it.

Fixes #41.
commit 2b103700 (2018-08-16): edison.env: correct kernel file name

In sumo the live image generation class no longer uses a hardcoded "vmlinuz"
name for the kernel file. As we use "bzImage" in Edison, update accordingly.

I've left the "initrd" piece in the dfu_alt_info_mmc just in case we ever
want to use it.

Fixes #41.
closed 2018-08-16 by htot