edison-fw/meta-intel-edison #44: On sumo64-acpi blink-led is not working

htot, 2018-08-20:
@lybtongji wrote:  
> I flashed sumo64-acpi and found blink-led not working. Is it due to ACPI?

@htot wrote:
>Yes, because of acpi pin names/numbers have changed. This affects blink-led and mraa and probably some other code too. I found the best way to access pins is from the libgpiod provided tools gpiodetect gpiofind gpioget gpioinfo gpiomon gpioset. See also Edison wiki.
For now I think we need the code first to detect if we are using acpi (Edit: /sys/kernel/config/acpi/table/arduino/ /sys/firmware/acpi exists).
I didn't find time to fix blink-led, but if you want to take this, please do.

htot, 2018-08-20:
@lybtongji I created a new issue for this.

lybtongji, 2018-08-20:
Ah. It just a question, not very necessary. But I'm glad to hear that you can help with it.

In addition. Can I make the LED (`DS2`) to be an activity light of `CPU` or `eMMC`?

which I found in kernel configurations:

```
...
CONFIG_LEDS_TRIGGER_DISK
CONFIG_LEDS_TRIGGER_ACTIVITY
...
```

htot, 2018-08-20:
The blink-led program is probably relatively easy to fix. Also you should be able to turn LED on and off from command line, You need to figure out the name of the pin, try `gpioinfo`.
To turn on: gpioset `gpiofind pin_name`=1

I haven't studied `CONFIG_LEDS_TRIGGER_DISK` and `CONFIG_LEDS_TRIGGER_ACTIVITY`, so I don't have the answer to that.

htot, 2018-10-14:
Fixed with PR #48

closed 2018-10-14 by htot