edison-fw/meta-intel-edison #49: Master build failure

7Grok, 2018-10-13:
OS: Ubuntu 14.0.0
Procedure:
- Downloaded the last intel source
- Put this meta-intel-edison in the correct place

`make setup`
`make debian_image`

Fails building node with linking errors around libssl. Can figure out what the problem is. Do you have any advice?

Looks like Node v8.9.4 and libssl 1.0.2j

alext-mkrs, 2018-10-13:
I wouldn't honestly run this on 14.00 (did you mean 14.04 BTW?) these days, I'm not sure `sumo` supports it - 16.04 or 17.10 were tested to work ok.

7Grok, 2018-10-13:
Yea sorry, I mean 14.0.4. OK let my try on 17.10

--
André

htot, 2018-10-14:
I don't know what `make debian_image` does. Should be `make image`. 

Our WIP documentation can be found here: [Intel Edison Image Builder](https://htot.github.io/meta-intel-edison/).

htot, 2018-10-17:
I just built sumo64-acpi from scratch, this is just a few commits ahead of sumo32. Master is identical to sumo32. There were not problems fetching.
The build `make image` does stop at a certain point (missing modules to the initramfs). But starting `make image` again it builds until completion.

@7Grok are you still experiencing this problem?

7Grok, 2018-10-21:
Oh yea sorry, I mean the source from here. I was able to get a clean build on a 16.04 box, but now dfu-util fails with `Device has DFU interface, but has no DFU functional descriptor` ... Not sure what that means. I tried different cables on different USB ports

htot, 2018-10-21:
Maybe try Flashtool Lite?

htot, 2018-11-01:
Have you been able to flash it?

closed 2019-02-10 by htot