edison-fw/meta-intel-edison #56: sumo branch build failure

sjpeng, 2019-01-09:
Build Configuration:
BB_VERSION           = "1.38.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "i686-poky-linux"
MACHINE              = "edison"
DISTRO               = "poky-edison"
DISTRO_VERSION       = "2.5.2"
TUNE_FEATURES        = "m32 corei7"
TARGET_FPU           = ""
meta
meta-poky
meta-yocto-bsp       = "sumo:623b77885051174d0e05198843e739110977bd18"
meta-oe
meta-python
meta-networking      = "sumo:8760facba1bceb299b3613b8955621ddaa3d4c3f"
meta-intel           = "sumo:90af97d23fb2a56187c2fe2a3f4f4190d7cc2605"
meta-intel-edison-bsp
meta-intel-edison-distro = "master:d0de39d612c6cb032f19acd3b89f2f4264c444b9"
meta-acpi            = "eds:477a31d2bad06de9c240753462695c46569cf162"

ERROR: Multiple versions of openssl are due to be built (/home/cloud/edison/out/linux64/poky/meta/recipes-connectivity/openssl/openssl_1.1.0i.bb /home/cloud/edison/out/linux64/poky/meta/recipes-connectivity/openssl/openssl_1.0.2p.bb). Only one version of a given PN should be built in any given build. You likely need to set PREFERRED_VERSION_openssl to select the correct version or don't depend on multiple versions.

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
Makefile:53: recipe for target 'edison-image' failed
make: *** [edison-image] Error 1

htot, 2019-01-09:
@sjpeng that is surprising. I never saw this one.And we actually do set PREFERRED_VERSION_openssl here: https://github.com/edison-fw/meta-intel-edison/blob/sumo32/meta-intel-edison-distro/conf/distro/poky-edison.conf.

Which branch did you checkout? (there is master, sumo32 and a sumo64-acpi)

sjpeng, 2019-01-09:
thank you for your fast reply, I'm using sumo32 branch.

sjpeng, 2019-01-10:
@htot I'm reference from here:
https://github.com/edison-fw/meta-intel-edison/blob/sumo32/README

In phase 2 (Get this layer), I have make sure checkout at sumo32 branch

htot, 2019-01-10:
@sjpeng I can reproduce if I do new checkout of poky. Strange. Maybe due to change in poky/sumo.

Can you try this: in meta-intel-edison/meta-intel-edison-distro/conf/distro/poky-edison.conf remove the line: `PREFERRED_VERSION_openssl ?= "1.1.0%"`

sjpeng, 2019-01-10:
@htot Yes, It's work when mark the line: **PREFERRED_VERSION_openssl ?= "1.1.0%"**

htot, 2019-01-11:
@sjpeng are you able to successfully  build the image? I would like to close this issue but I am experiencing build errors myself in:
- bison-3.0.4
- cross-localedef-native_2.27
- coreutils-8.29

It looks like sumo has been updated creating problems with gnulib and not all fixes have been up-streamed yet.

```
virtual:native:/home/ferry/Develop/tmp/out/linux64/poky/meta/recipes-devtools/bison/bison_3.0.4.bb:do_compile

| ../bison-3.0.4/lib/fseterr.c: In function 'fseterr':
| ../bison-3.0.4/lib/fseterr.c:77:3: error: #error "Please port gnulib fseterr.c to your platform! Look at the definitions of ferror and clearerr on your system, then report this to bug-gnulib."
| #error "Please port gnulib fseterr.c to your platform! Look at the definitions of ferror and clearerr on your system, then report this to bug-gnulib."
| ^~~~~

/home/ferry/Develop/tmp/out/linux64/poky/meta/recipes-core/glibc/cross-localedef-native_2.27.bb:do_compile

| /home/ferry/Develop/tmp/out/linux64/build/tmp/hosttools/ld: argp-fmtstream.c:(.text+0x525): undefined reference to `_IO_fwide'
| /home/ferry/Develop/tmp/out/linux64/build/tmp/hosttools/ld: argp-help.o: in function `argp_failure':
| argp-help.c:(.text+0x20b3): undefined reference to `_IO_fwide'
| collect2: error: ld returned 1 exit status

virtual:native:/home/ferry/Develop/tmp/out/linux64/poky/meta/recipes-core/coreutils/coreutils_8.29.bb:do_compile

| ../coreutils-8.29/lib/freadseek.c: In function 'freadptrinc':
| ../coreutils-8.29/lib/freadseek.c:68:3: error: #error "Please port gnulib freadseek.c to your platform! Look at the definition of getc, getc_unlocked on your system, then report this to bug-gnulib."
| #error "Please port gnulib freadseek.c to your platform! Look at the definition of getc, getc_unlocked on your system, then report this to bug-gnulib."
| ^~~~~
```

sjpeng, 2019-01-11:
@htot sure, I have successfully build the image

htot, 2019-01-11:
@sjpeng I am guessing your host machine <> Ubuntu 18.10 and has another glibc then v2.28?

htot, 2019-01-11:
@sjpeng I am on 18.10. It appears the build errors are cause by 18.10 using glibc v2.28. I have now created patches that solve the above build errors for 18.10. It would be very helpful if you could test to see if it still builds on your machine which presumably has a older version of glibc. 
I can imagine you don't want to wait hours for this to build from scratch. The quickest test would be:
-  pull from https://github.com/htot/meta-intel-edison/tree/sumo32
- `make update` (ignore the white space errors)
- your favorite build method, either `make image`, or in the build dir `bitbake -k edison-image`
- just check if it builds

If you could do this that would be great.

sjpeng, 2019-01-14:
Sorry, I am late in my rely. I'm build host machine is Ubuntu 16.04.

htot, 2019-01-14:
@sjpeng it would be great to know if it build the current sumo32

sjpeng, 2019-01-18:
@htot unfortunately build failed, If unmark `PREFERRED_VERSION_openssl ?= "1.1.0%"`

htot, 2019-01-18:
With so my latest patches I mean. I force pushed so you might need to reset --hard against remote.

sjpeng, 2019-01-18:
yes, I have do that.

[https://imgur.com/a/YpoPI0w](https://imgur.com/a/YpoPI0w)

htot, 2019-01-19:
That would be 4b7e12986f31cf28336407eb251c228f4ad8f353

htot, 2019-01-19:
Never mind, I just tested this on a xenial (16.04) container, I it builds just fine.

htot, 2019-01-27:
Fix by PR #57

PR #57 (2019-01-27): Revert me: patches to make build on Ubuntu Cosmic 18.10
These patches are needed to build on Ubuntu Cosmic due to glibc 2.28. I tested this on Cosmic and on Xenial to verify backwards compatibility. It has been independantly tested by @sjpeng. I expect at some point these patches will be backported to Sumo at which point this will break and need to be reverted.

This closes #56
closed 2019-01-27 by htot