edison-fw/meta-intel-edison #60: Power button not working

htot, 2019-05-11:
Edison has a power button handler, but it currently doesn't work. Reason is systemd is handling the events, so pressing the power button shuts down the Edison.

To disable that we need to change /etc/systemd/logind.conf:
`HandlePowerKey=ignore`

With this set, power button handler detects:

1. Edison PWR button was pressed more than 2s
2. Edison PWR button was pressed more than 2s and released
3. Edison PWR button was pressed 3 times

Further:
- press 10s and release reboots Edison

I have patch in my queue that turns on Bluetooth for pairing on 3 presses, without further intervention. When this works we can enable phone pairing with tethering, and initial setup (oobe) done from phone browser. A single press could enable WiFi AP to do the same.

But I need to know best way to get login.conf modified. It could be done from post-install but that seems ugly. 

@alext-mkrs Do you have better suggestion?

htot, 2019-05-17:
@alext-mkrs I found systemd provides for conf directories that can be filled by packages to override default behavior. https://www.freedesktop.org/software/systemd/man/logind.conf.html

So instead of modifying `/etc/systemd/logind.conf` we just seem to need a snipped in `/usr/lib/systemd/logind.conf.d/pwr-button-handler.conf` containing:
```
HandlePowerKey=ignore
```
And this can be provided by the power-button-handler recipe.

Agreed?

alext-mkrs, 2019-07-20:
@htot, yes, I think that's the way to go.

Sorry, literally zero time for the last couple of months for anything Edison-related...

htot, 2019-07-20:
Then this patch resolves the issue https://github.com/htot/meta-intel-edison/commit/631d6e182a030d454af98beeb2cf0e97c9a2da6e

closed 2020-01-03 by htot