edison-fw/meta-intel-edison #63: SoC_DT0 and SoC_DT1 (intel temperature sensors) not available

razvan-becheriu, 2019-07-02:
Hi,

The intel temperature drivers are not available in thud version.
I have tried to enable the drivers in the kernel with no success.
Can you point out what are the kernel configuration parameters I need to enable/disable so that I can access the intel temperature sensors trough /sys/class/thermal/ interface?

Thank you,
Razvan

htot, 2019-07-02:
Good question. I wasn't aware that the original kernel had access to thermal sensors.
AFAIK we have access to the PMIC which has some temperature sensors attached: https://edison.internet-share.com/wiki/PMIC

Otherwise, we are using the kernel from Andy Shevchenko and it might be better to post this question there: https://github.com/andy-shev/linux

razvan-becheriu, 2019-07-03:
https://github.com/andy-shev/linux/issues/26

andy-shev:
In the original BSP the drivers/thermal/intel_mrfl_thermal.c provides such functionality. As far as I know there is no similar functionality provided in vanilla kernel. Note, the vanilla should support rather ACPI approach to use DTS.

razvan-becheriu, 2019-07-03:
I have found that using hwmon you can actually get the temperature:
root@edison:~# cat /sys/class/hwmon/hwmon0/temp2_label 
Core 0
root@edison:~# cat /sys/class/hwmon/hwmon0/temp2_input 
55000
root@edison:~# cat /sys/class/hwmon/hwmon0/temp3_label 
Core 1
root@edison:~# cat /sys/class/hwmon/hwmon0/temp3_input 
56000

this works for me.
you can close this issue if it is of no interest for you either.

andy-shev, 2019-11-05:
@htot, I guess you may close this for now. I doubt we will see the request to have a comprehensive thermal solution for this any time in the future.

htot, 2019-11-05:
I see. This same hwmon is also enabled on ubuntu. I'll consider to make it optionally enabled.

closed 2019-11-05 by htot