edison-fw/meta-intel-edison #68: boot environment not updated on ota

razvan-becheriu, 2019-09-18:
the flashall script does flash the u-boot-envs/*.bin files but the boot environment does not define such action (do_ota, etc...), so doing ota to a different version which might have a different boot environment might not work, as the old environment is preserved

razvan-becheriu, 2019-09-18:
I believe this can be achieved by either modifying ./utils/flash/ota_update.cmd which is the upgrade script run on ota, or by modifying meta-intel-edison-bsp/recipes-bsp/u-boot/files/edison.env, or both.

htot, 2019-09-18:
I think Intel gave up on ota at a certain point when the partition to hold the ota became to small. Or when the root partition became to small so they had to sacrifice the ota partition. Correct me if I'm wrong.
I'm not even sure how the ota is supposed to work. IIUC the binary is booted instead of the kernel. I don't think we have the sources of the binary. My long term plan was to create a 2nd kernel and use that to install the rootfs ota.
But if you have a better solution I would be happy to take your patches.

razvan-becheriu, 2019-09-19:
Hi,

Here is a patch.
You can test it. Please let me know if you find any issue with it.

[ota_upgrade.patch.txt](https://github.com/edison-fw/meta-intel-edison/files/3631340/ota_upgrade.patch.txt)

It does write the u-boot-env0 and u-boot-env1 partitions just as flashall would do, and then loads the content using 'env import' and saves it as current env using 'saveenv'.

razvan-becheriu, 2019-09-19:
To test this, I use the old (official) intel edison image and do ota to thud.

razvan-becheriu, 2019-09-20:
It seems that the ota is broken in thud version.
the ota_update.cmd is not supported by the u-boot.
I'll provide a patch to fix this as well.

htot, 2019-09-20:
I see. I wanted to ask you that.

I think I removed ota when adding reboot usb and reboot sdhc.

I think we could use a doc page explaining how the ota mechanism works.

razvan-becheriu, 2019-09-20:
no. it is not related.
the new uboot has 'part' command with different sintax.

old sintax:
part find mmc 0 label:${u_part_lbl} u_part_num
part info mmc 0:${u_part_num} u_part_start u_part_sz u_part_blksz

new sintax:
part start mmc 0 ${u_part_num} u_part_start
part size mmc 0 ${u_part_num} u_part_sz

where ${u_part_num} is ${u_part_lbl}
not sure if the conversion function is still needed.

htot, 2019-09-20:
That would be very nice.

razvan-becheriu, 2019-09-20:
I believe thad doing downgrade won’t work as the script on the old image has to be modified also to support the new syntax. so doing an OTA to an older version will use the new ‘part’ syntax, but the commands from the old script within the image with old ‘part’ syntax...so the best I can do is add support for OTA from old to new and from new to new versions.

razvan-becheriu, 2019-09-20:
another way to do this is to actually modify the ‘part’ command to support both ‘syntax’ so we can leave the scripts as they are and have full compatibility between versions

htot, 2019-09-20:
From old to new, and new to new sounds good enough to me. To go back to factory state, people can better flashall (which doesn't work well on win10 I hear) or Flash Tool Lite.

Does OTA work even I you prepare that partition for usb/sdhc booting? https://edison-fw.github.io/meta-intel-edison/2.2-Building-and-installing-the-kernel.html#installing-the-kernel

razvan-becheriu, 2019-09-21:
Hi,

I have a working ota script.
The script allows old (pocky) version to be upgraded to thud.
As I have described, it is not compatible with pocky when it comes to downgrading because:
1. the pocky version does not have the update script with compatibility of 'part' command running in u-boot on the thud board
2. the pocky version does not have the update script which updates the environment from u-boot-envs/*.bin files

I am not really happy with the fix, as it hardcodes the partition numbers and block size (512 bytes/block).
However, there are many things hardcoded in the boot environment, like: 'root=/dev/mmcblk0p8'.
Also the partitioning is hardcoded, but I believe we can work with this approach for now.

A better solution would be to modify the 'part' command to retrieve the same way it did on pocky, but there will still a need to fix item 2.
I am not sure (sincerely I don't think so) if we can change the thud image to boot with the environment on pocky, which concludes that a change on the pocky image (upgrade script and maybe even more files) to make it work with OTA from thud.

[patch.txt](https://github.com/edison-fw/meta-intel-edison/files/3638798/patch.txt)

I'll investigate item 1 for a better solution. Until then, you can experiment with this.

razvan-becheriu, 2019-09-21:
I have modified the part command to support 'block' and 'id' which return the partition block size and partition id respectively.

I have also modified the dfu to support either mmc or raw type (both act like default raw type).

add files:
meta-intel-edison-bsp/recipes-bsp/u-boot/files/dfu-add-mmc-support.diff
meta-intel-edison-bsp/recipes-bsp/u-boot/u-boot-fw-utils_2019.04.bbappend
meta-intel-edison-bsp/recipes-bsp/u-boot/u-boot_2019.04.bbappend

I guess we should fine a better name for the files.

Now we can modify the utils/flash/ota_update.cmd to use runtime (not hard coded) values for partition id and block size (update ota_find_partition_alternative and ota_get_partition_attributes_alternative functions).

[u-boot-fw-utils_2019.04.bbappend.txt](https://github.com/edison-fw/meta-intel-edison/files/3638919/u-boot-fw-utils_2019.04.bbappend.txt)
[u-boot_2019.04.bbappend.txt](https://github.com/edison-fw/meta-intel-edison/files/3638920/u-boot_2019.04.bbappend.txt)
[dfu-add-mmc-support.diff.txt](https://github.com/edison-fw/meta-intel-edison/files/3638921/dfu-add-mmc-support.diff.txt)

I'll update the utils/flash/ota_update.cmd and come back with a patch.

htot, 2019-09-21:
You patch u-boot directly (no need bbappend). They are ours meta-intel-edison/meta-intel-edison-bsp/recipes-bsp/u-boot/

razvan-becheriu, 2019-09-21:
ok.

here is the patch for the OTA upgrade script (without any harcoded value):

[ota_update.cmd.diff.txt](https://github.com/edison-fw/meta-intel-edison/files/3639001/ota_update.cmd.diff.txt)

enjoy!

razvan-becheriu, 2019-09-21:
you can put these changes in their respective repo.
I only have a fork from meta-intel-edison, os this is the reason I put everything relative to it.

razvan-becheriu, 2019-09-21:
Ups.. I did a mistake..
reloading the patch

[dfu-add-mmc-support.diff.txt](https://github.com/edison-fw/meta-intel-edison/files/3639036/dfu-add-mmc-support.diff.txt)

razvan-becheriu, 2019-09-21:
I have tested on my side and everything seems fine. Please let me know what you think.

razvan-becheriu, 2019-09-22:
If I come to think about it, we can leave drivers/dfu/dfu_mmc.c unmodified, as we will not try to resolve issue 2.
So you can just delete the modifications related to that file in dfu-add-mmc-support.diff.txt

htot, 2019-09-22:
> you can put these changes in their respective repo.
> I only have a fork from meta-intel-edison, os this is the reason I put everything relative to it.

I see. You are right, adding the patch to the u-boot recipe is more future proof then adding to u-boot fork. Unless you get @andy-shev to take your patch (is our upstream).

If you rebase to current meta-intel-edison/master (or warrior) and commit your changes then push your local branch, you can send me a PR. That would prevent me from making mistakes and also takes care of the attribution of your work. Would do you think?

razvan-becheriu, 2019-09-23:
Ok. I'll try to do that but I am busy with something else right now. Will try to do this at the end of the week.

Another thing, it seems that the modification in drivers/dfu/dfu_mmc.c are necessary.
If you flash a board with the new thud image and try to do flashall.sh using an pocky image, it won't work unless you also add the modification in drivers/dfu/dfu_mmc.c to the thud image.

The flashall will overwrite the old dfu, but it uses the currently installed dfu on the board (the thud version), so if it does not support mmc (which is equivalent to raw), it won't work.

can you please also specify the upstream git repository address. Thank you.

andy-shev, 2019-09-23:
@htot, my branch is merely an intermittent place for patches to go to U-Boot upstream. So, if the patch could fit pure upstream U-Boot, I can take it. Otherwise I wouldn't touch U-Boot repository for that.

htot, 2019-09-23:
@andy-shev understood, we will apply patches through u-boot recipe.

razvan-becheriu, 2019-09-30:
Hi,

I havecreated 2 PR:

u-boot: Add part command support for block size and partition number:
https://github.com/edison-fw/u-boot/pull/1

meta-intel-edison: Fixed OTA upgrade script to update environment and to
use alternative syntax to support upgrade from both pocky or thud:
https://github.com/edison-fw/meta-intel-edison/pull/70

Regards,
Razvan

razvan-becheriu, 2019-09-30:
I believe that you have to update some hashes/commit ids if you are going
to use the new HEAD (after merge) on https://github.com/edison-fw/u-boot

If you prefer to use (just as I did) bbappend, then I'll create another PR
for meta-intel-edison.

I can also try to apply the patch to https://github.com/andy-shev/u-boot

Please let me know.

Razvan

razvan-becheriu, 2019-09-30:
Moved https://github.com/edison-fw/meta-intel-edison/pull/70 to
https://github.com/edison-fw/meta-intel-edison/pull/71

fixed typo in branch name

htot, 2019-09-30:
I'll try it, but need a bit of time. Hold on :-)

htot, 2020-01-03:
I believe it is fixed in warrior in my repo https://github.com/htot/meta-intel-edison/tree/warrior. Eventually this will land in edison-fw/master. Closing for now.

closed 2020-01-03 by htot