edison-fw/meta-intel-edison #69: enabling CONFIG_WATCHDOG_NOWAYOUT=y in kernel does not work

razvan-becheriu, 2019-09-22:
I have enabled CONFIG_WATCHDOG_NOWAYOUT=y in kernel config, but when I start the watchdog service and stop it, I'd expect a reboot after 60 sec.
However, the watchdog timer is stopped, just like I would have not enabled CONFIG_WATCHDOG_NOWAYOUT at all.

htot, 2019-09-22:
Maybe that is something @andy-shev can answer?

htot, 2019-09-22:
Eh, how do you enable it? Did you make a kernel fragment?

razvan-becheriu, 2019-09-22:
as described in intel edison configure kernel documentation:
https://software.intel.com/en-us/node/593593

razvan-becheriu, 2019-09-22:
If I just kill the service process, the board reboots, but stopping the service should not stop the watchdog.

htot, 2019-09-22:
> as described in intel edison configure kernel documentation:
> https://software.intel.com/en-us/node/593593

We have made some changes to more closely follow the Yocto way. It is documented here: https://htot.github.io/meta-intel-edison/5.1-Bitbake-tricks#configuring-the-kernel-and-grab-the-kernel-fragment

htot, 2019-09-22:
Afaiu the service interacts with the systemd sw watchdog. CONFIG_WATCHDOG_NOWAYOUT is a hw watchdog controlled by the kernel? I don't know how these interact.

razvan-becheriu, 2019-09-22:
hm..I am certain that it works the way I build it.
I have enabled other flags, like:
CONFIG_RFKILL_GPIO=y
CONFIG_SENSORS_CORETEMP=y
CONFIG_INTEL_MFLD_THERMAL=y
CONFIG_PANIC_TIMEOUT=10
CONFIG_IKCONFIG=y
CONFIG_IKCONFIG_PROC=y
CONFIG_WATCHDOG_NOWAYOUT=y
CONFIG_WATCHDOG_SYSFS=y

the only one not working is the CONFIG_WATCHDOG_NOWAYOUT.

htot, 2019-09-22:
Ok.

But the intel instructions shouldn't be working afaik. The defconfig from the kernel is used on each build (`bitbake linux-yocto -c cleansstate` deletes the whole linux directory) if all is well.

So either you need to switch to your own kernel fork and put all your patches on top. Or tell the recipe to apply the fragments (is like a patch for .config) on each checkout of the kernel.

I did notice unpredictable behavior in the past when the checked out kernel is locally modified and gave up on working that way.

razvan-becheriu, 2019-09-22:
Ok. But as I have said, I can retrieve the kernel config at runtime doing zcat /proc/config.gz and I can see the flag enabled. I am only curious about the intel watchdog. As you have said, it seems to be implemented in software (kernel doing panic) instead on hw.
Either way, the flag should work.
Maybe you can also give some insight related to watchdog-sample service, which I think is implemented in hw.

Thanks

htot, 2019-09-22:
Ok, so the flag is set.The watchtdog sample code refers to systemd sw watchdog: https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-distro/recipes-support/watchdog-sample/watchdog-sample/watchdog-sample.c.

That's all I know, sorry.

andy-shev, 2019-09-24:
I see. Driver doesn't support it. I will send soon a patch, please test.

razvan-becheriu, 2019-09-25:
Hi,
I want to test the patch, but I don't know how.

I have added a file
meta-intel-edison-bsp/recipes-kernel/linux/linux-yocto_5.0.0.bbappend

and added the patch th the files directory but the patch can not be applied

I have also modified the already existing file:
meta-intel-edison-bsp/recipes-kernel/linux/files/0001-serial-8250_dma-use-linear-buffer-for-transmit.patch

but it seems that it is unable to apply the patch.
What is the procedure (the easiest, I don't want to create another recipe or alternative platform).

I only have a fork for meta-intel-edison

thank you.

razvan-becheriu, 2019-09-25:
nevermind. I was able to apply the patch by modifying:
meta-intel-edison-bsp/recipes-kernel/linux/linux-yocto_5.0.0.bb

adding another line:
SRC_URI_append = " file://0001-no_way_out.patch"

testing now

razvan-becheriu, 2019-09-25:
it works!

thanks!

razvan-becheriu, 2019-09-25:
unrelated question:

does Intel Edison board has a HW watchdog as well?

htot, 2019-09-25:
> nevermind. I was able to apply the patch by modifying:
> meta-intel-edison-bsp/recipes-kernel/linux/linux-yocto_5.0.0.bb
> 
> adding another line:
> SRC_URI_append = " file://0001-no_way_out.patch"
> 
> testing now

This is exactly how I do it.

htot, 2019-09-25:
> it works!
> 
> thanks!

You mean the patch works as you expect? If so, you might want to add you test-by: tag?

BTW this is also where you would add a kernel fragment (see the other *.cfg files in the files directory) to enable CONFIG_WATCHDOG_NOWAYOUT.

If you send a PR for the linux-yocto recipe I'll take it.

razvan-becheriu, 2019-09-26:
What do you mean add my test by tag?

I am not sure if people would like this enabled by default.
Usually, on reboot, systemd waits for 1:30/1:35 (90s/95s) for services to
exit. This would cause a watchdog reset on some reboots.
Also, if this is disabled in the kernel, the watchdog service is just
stopped on systemd reboot sequence and the watchdog counter is disabled so
a forced reboot caused by the watchdog is never executed.
If you want to enable this by default, we should also extend the watchdog
timer to at lease 120s to allow a gratuitous shutdown.

htot, 2019-09-26:
Like this: https://marc.info/?l=linux-kernel&m=156942376014758&w=2

I think Andy's patch should always go in the kernel. But I see what you mean, the systemd watchdog service should be shutdown last, after all others.

andy-shev, 2019-09-27:
> What do you mean add my test by tag?

To tell maintainers that you have tested it and it works.
See about tags in this documentation: https://www.kernel.org/doc/html/latest/process/submitting-patches.html#using-reported-by-tested-by-reviewed-by-suggested-by-and-fixes

Note, it has nothing to do with default settings.

razvan-becheriu, 2019-09-30:
How about this:

```
Tested-by: Razvan Becheriu <razvan.becheriu@gmail.com>
The patch works.
By configuring the CONFIG_WATCHDOG_NOWAYOUT in the kernel and stopping the
watchdog.service, the board reboots.
Previously, the watchdog.service was able to disable the watchdog counter,
and no reboot was performed.
```

Thank you,
Razvan

andy-shev, 2019-09-30:
> How about this: ``` Tested-by: Razvan Becheriu <razvan.becheriu@gmail.com> The patch works. By configuring the CONFIG_WATCHDOG_NOWAYOUT in the kernel and stopping the watchdog.service, the board [...]

Perfect!

andy-shev, 2019-10-01:
> ``` Tested-by: Razvan Becheriu <razvan.becheriu@gmail.com> The patch works. By configuring the CONFIG_WATCHDOG_NOWAYOUT in the kernel and stopping the watchdog.service, the board reboots. Previously, the watchdog.service was able to disable the watchdog counter, and no reboot was performed. ```

I didn't see any email from you. Hope you will send it some time soon. The text above is good, but has to be send as a reply to my patch by email!

razvan-becheriu, 2019-10-01:
I have just replied to your patch email.

Hope it is OK now.

Thank you again.
Razvan

razvan-becheriu, 2019-10-01:
I did send an email, but I get:

Message may contain a virus
Your message wasn't delivered to linux-watchdog@vger.kernel.orgbecause it may contain a virus. Learn how to check for and prevent viruses.
The response from the remote server was:
550 5.7.1 Content-Policy reject msg: The message contains HTML subpart, therefore we consider it SPAM or Outlook Virus. TEXT/PLAIN is accepted.! BF:<U 0.499989>; S1726642AbfJATJI

andy-shev, 2019-10-01:
> I did send an email, but I get: The message contains HTML subpart, therefore we consider it SPAM or Outlook Virus. TEXT/PLAIN is accepted.!

I left summary above. I hope you know how to send in plain text.

htot, 2019-10-27:
I took the kernel patch and added it into the linux-yocto recipe. We are now on v5.3 and may expect this patch to go into the kernel with v5.5. CONFIG_WATCHDOG_NOWAYOUT is not enabled by default. To enable it create a kernel fragment like this [Configuring the kernel and grab the kernel fragment](https://htot.github.io/meta-intel-edison/5.1-Bitbake-tricks#configuring-the-kernel-and-grab-the-kernel-fragment) and add the fragment to the kernel recipe.

@razvan-becheriu I force pushed this into [htoth/meta-intel-edison](https://github.com/htot/meta-intel-edison) master and warrior. Eventually this will go into edison-fw. It builds but I would appreciate if you test it..

andy-shev, 2019-10-28:
Can somebody ping maintainers about it? @razvan-becheriu, we may have even chance to get it into v5.4 if you have a good justification.

razvan-becheriu, 2019-10-28:
Hi,

No need to hurry. I was interested in thud version. If this is going into
warrior version, then it is the same for me if it is 5.4 or 5.5.
I'll use custom 'patch-file' in my thud version.

Regards,
Razvan

htot, 2019-10-28:
You can cherry-pick this patch on top of thud ec3327e294cb6e0012faffb8e73a8a63a885c674

andy-shev, 2019-11-29:
JFYI: patch on its way to Torvalds' tree.

htot, 2020-01-03:
I believe this is fixed, so closing.

closed 2020-01-03 by htot