edison-fw/meta-intel-edison #72: bitbake -k edison-image error

shawnhsoia97, 2019-10-16:
Hi,i trying to build image,but  i have encountered some problems.it shows [error: Nothing PROVIDES 'edison-image' ] when i bitbake -k edison-image.how can i solve it?thank

htot, 2019-10-16:
Please setup according to https://edison-fw.github.io/meta-intel-edison/1.2-Setting-up.html#setting-up
and build https://edison-fw.github.io/meta-intel-edison/2.1-Building-and-installing-the-rootfs.html#building-the-rootfs

Thanks

shawnhsoia97, 2019-10-18:
Hi @htot ,thank you for telling me how to solve it,but i have a new problem .when i finished make image,there was an error with 「fetcher failure for URL:'http://download.xdk.intel.com/iot/xdk-daemon-0.0.37.tar.bz2', Unable to fetch URL from any soure」.how can solve it?thank

htot, 2019-10-18:
Strange. Which version are you building? Afaik we don't build that on master/thud due to https://github.com/edison-fw/meta-intel-edison/blob/7499c1d689aec15c1a6ed65980b9594a443e8016/meta-intel-edison-distro/recipes-core/images/edison-image.bb#L69

shawnhsoia97, 2019-10-19:
my building version is dizzy-tr/dizzy-latest
OS: Ubuntu 14.04

htot, 2019-10-19:
I haven't built that in a long time. Since then Intel has stopped supporting xdk, so I think it will be best to comment out the line in edison-image.bb that tries to install it.

shawnhsoia97, 2019-10-19:
OK,I get it ,thank you

htot, 2019-10-19:
Once you get it building I would gladly take your PR to make this branch up to date again. :-)

htot, 2019-10-19:
(note that the most up to date branch is thud), with development work going on on warrior here: https://github.com/htot/meta-intel-edison/tree/warrior).

shawnhsoia97, 2019-10-23:
hi,@htot
i tried your method and tried other versions
but still there is an error
https://user-images.githubusercontent.com/53549747/67348558-db429d00-f577-11e9-9c08-fbca42d40bee.JPG
version:master OS:Ubuntu 18.04

[diaay-latest](https://user-images.githubusercontent.com/53549747/67348654-252b8300-f578-11e9-8e44-7a13466e14aa.JPG)
version:dizzy-latest OS:Ubuntu 16.04

 thank you for helping me

shawnhsoia97, 2019-10-23:
when i building thud,that shows 
[image]
am i remove ftp.sudo.ws?

htot, 2019-10-24:
Thud: it looks like https://www.sudo.ws/download.html changed certificate and directory structure. So that breaks not only thud, but also warrior and zeus. You can fix it by going into out/linux64/poky/meta/recipes-extended/sudo/sudo_1.8.23.bb and fix the path into https://www.sudo.ws/dist/sudo-${PV}.tar.gz or wget ftp://ftp.sudo.ws/pub/sudo/sudo-${PV}.tar.gz.

This an issue in openembedded-core.

htot, 2019-10-26:
I added a patch 084f87da657485f8efea8a95d211b1b518f9cc17 on top of thud that fixes building sudo. Apparently once the sources are in your bbcache it doesn't matter when upstream becomes unreachable. And I hadn't deleted my bbcache in a long while. 

I tested this patch by rebuilding from scratch and now will test on the new branch thud32-v2.0 which is v2.0 (master) + backported patch.

htot, 2019-10-27:
Also I will not send in a patch to the ML, master already has a fix.

@shawnhsoia97 Would you care to test?

Also I have the patch in have [htoth/meta-intel-edison](https://github.com/htot/meta-intel-edison) master and warrior.

shawnhsoia97, 2019-10-28:
Thank ,I successfully built thud ,but i found that expat-2.2.6.tar.bz2 will also make mistake.so i changed the expat_2.2.6.bb to solve it.

Also i will test that ^_^

htot, 2019-10-28:
Strange, I built from scratch and did not get that error. How did you solve it?

ML is mailing list.

You can search for packages here: https://layers.openembedded.org/layerindex/branch/thud/recipes/?q=v4l, add the layer where you find the package (if needed), then add the package to the image here https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-distro/recipes-core/images/edison-image.bb

shawnhsoia97, 2019-10-29:
i fixed it by going into out/linux64/poky/meta/recipes-extended/expat/expat-2.2.6.bb and fix the path into https://sourceforge.net/projects/expat/files/expat/2.2.6/ expat-${PV}.tz.bz2

htot, 2019-10-29:
I wonder why you see so many errors that I don't. After changing branch, did you do make clean && make setup?

shawnhsoia97, 2019-10-30:
yes,i am re-establishing a new directory and git clone git://git.yoctoproject.org/meta-intel-edison and make setup,but i removed directory and rebuild master.then this time,no errors were successfully completed,and so i think it should be a mistake when make setup,but i didn't find it.

also have you used Edison's Wifi?i am successful in ping 168.95.1.1,but i ping www.google.com.tw that shows bad address 'www.google.com.tw',i am sure i have connected to the internet,do you know why?

also do you konw why apt-get install something(unzip...) that show unable to locate package,is it because i can't solve domain name problem?

htot, 2019-10-30:
Sometimes, switching to a new branch, going 32b <-> 64b, or changing kernel / U-Boot version doesn't cleanup well and build errors happen.

By default there are no deb repo's configured. It's not that you can just configure debian or ubuntu repo's, That won't work. You can build you own repo, configure edison to use that. See https://edison-fw.github.io/meta-intel-edison/5.0-Creating-a-deb-repository.html

closed 2019-11-05 by htot