edison-fw/meta-intel-edison #73: i2c not work in master

xlla, 2019-10-29:
Hi,
I am very excited to found this wonderful project! It make my Edison board reborn.
I have successful build branch sumo but and branch master.
but I can't get my i2c device work with it. I have read each section of manual very careful, and can't find a clue.
I am use Edison Arduino board, and I had followed guide [I2C](https://htot.github.io/meta-intel-edison/4.4-hsu-i2c-spi.html),  it was 4 multiplexers in bus-1, address 20-23, but can't find any device in bus-6,
my device is ssd-1306 oled, it works on arduino uno board at address 0x3C,
I have found some error message in log:

Using boot protocol version 2.0d
Linux kernel version 5.2.0-edison-acpi-standard (oe-user@oe-host) #1 SMP Mon Oct 28 07:13:03 UTC 2019
Building boot_params at 0x00090000
Loading bzImage at address 100000 (11919392 bytes)
Magic signature found
Kernel command line: "root=/dev/mmcblk0p8 rootfstype=ext4 console=ttyS2,115200n8 earlyprintk=ttyS2,115200n8,keep loglevel=4 g_multi.ethernet_config=cdc systemd.unit=multi-user.target hardware_id=00 g_multi.iSerialNumber=8452e52a5b73f5f38c917327f40a577c g_multi.dev_addr=02:00:86:0a:57:7c platform_mrfld_audio.audio_codec=dummy"
Magic signature found

Starting kernel ...

starting version 239
Kernel with acpi enabled detected
Loading acpi tables
[    1.985141] pca953x i2c-INT3491:00: failed writing register
gpioset: at least one GPIO line offset to value mapping must be specified
gpioset: at least one GPIO line offset to value mapping must be specified
Waiting for root device /dev/mmcblk0p8
Found device '/run/media/mmcblk0p8'

htot, 2019-10-29:
Sure you have sumo? According to last line (thud).

The error lines are prior to mounting rootfs. They originate from initramfs loading acpi tables, I believe this https://github.com/edison-fw/meta-acpi/blob/eds-5.0.0/recipes-bsp/acpi-tables/files/edison/acpi-tables-load. It seems gpiofind is not working. Maybe `sleep 0.1` is just a little too fast?

What do you get when you `gpiofind TRI_STATE_ALL` from the command line?

xlla, 2019-10-30:
it's a long story, I have built sumo first, and then I have found i2c can't work, so I think it maybe caused by ACPI, then I  have created another env and switch to master branch and built thud.

it return nothing
root@edison:/media/sdcard/git/luma.examples# gpiofind TRI_STATE_ALL
root@edison:/media/sdcard/git/luma.examples# gpiofind TRI_STATE_ALL

I found that built in led in pin 13 don't work, only when I do this
rmmod ledtrig_heartbeat
rmmod leds_gpio

led start blink with some warning:
root@edison:/media/sdcard/git/luma.examples# blink-led 
can't find line TRI_STATE_ALL, type=gpiod.LINE_REQ_DIR_OUT
can't find line TRI_STATE_ALL, type=gpiod.LINE_REQ_DIR_OUT
led owner name: gpiochip0, name 'gpiochip0:40 /unnamed/' 
^CSignal intercepted: de-initing LED GPIOs
clean and quit.

I have connect second spi oled, it does not work too.
root@edison:/media/sdcard/git/luma.examples# ls /dev/spi*
/dev/spidev5.1
python3 examples/demo.py -i spi --spi-device=1 --spi-port=5 --gpio-data-command 41 --gpio-reset 49
Version: luma.oled 3.3.0 (luma.core 1.12.0)
Display: ssd1306
Interface: spi
Dimensions: 128 x 64
demo.py: error: GPIO access not available

sck -> pin 13,gpio 40
mosi -> pin11, gpio 43
D/C -> pin 10, gpio 41
CS -> pin 9 , 
Reset -> pin 8, gpio 49

xlla, 2019-10-30:
this is full gpio info
root@edison:/media/sdcard/git/luma.examples# gpiofind TRI_STATE_ALL
root@edison:/media/sdcard/git/luma.examples# gpioinfo              
gpiochip0 - 192 lines:
	line   0:      unnamed       unused   input  active-high 
	line   1:      unnamed       unused   input  active-high 
	line   2:      unnamed       unused  output  active-high 
	line   3:      unnamed       unused   input  active-high 
	line   4:      unnamed       unused  output  active-high 
	line   5:      unnamed       unused  output  active-high 
	line   6:      unnamed       unused  output  active-high 
	line   7:      unnamed       unused   input  active-high 
	line   8:      unnamed       unused   input  active-high 
	line   9:      unnamed       unused  output  active-high 
	line  10:      unnamed       unused  output  active-high 
	line  11:      unnamed       unused  output  active-high 
	line  12:      unnamed       unused  output  active-high 
	line  13:      unnamed       unused  output  active-high 
	line  14:      unnamed       unused   input  active-high 
	line  15:      unnamed       unused   input  active-high 
	line  16:      unnamed       unused   input  active-high 
	line  17:      unnamed       unused   input  active-high 
	line  18:      unnamed       unused   input  active-high 
	line  19:      unnamed       unused   input  active-high 
	line  20:      unnamed       unused   input  active-high 
	line  21:      unnamed       unused   input  active-high 
	line  22:      unnamed       unused   input  active-high 
	line  23:      unnamed       unused   input  active-high 
	line  24:      unnamed       unused   input  active-high 
	line  25:      unnamed       unused   input  active-high 
	line  26:      unnamed       unused   input  active-high 
	line  27:      unnamed       unused   input  active-high 
	line  28:      unnamed       unused   input  active-high 
	line  29:      unnamed       unused   input  active-high 
	line  30:      unnamed       unused   input  active-high 
	line  31:      unnamed       unused   input  active-high 
	line  32:      unnamed       unused   input  active-high 
	line  33:      unnamed       unused   input  active-high 
	line  34:      unnamed       unused   input  active-high 
	line  35:      unnamed       unused   input  active-high 
	line  36:      unnamed       unused   input  active-high 
	line  37:      unnamed       unused   input  active-high 
	line  38:      unnamed       unused   input  active-high 
	line  39:      unnamed       unused   input  active-high 
	line  40:      unnamed       unused  output  active-high 
	line  41:      unnamed       unused   input  active-high 
	line  42:      unnamed       unused   input  active-high 
	line  43:      unnamed       unused   input  active-high 
	line  44:      unnamed       unused   input  active-high 
	line  45:      unnamed       unused   input  active-high 
	line  46:      unnamed       unused   input  active-high 
	line  47:      unnamed       unused   input  active-high 
	line  48:      unnamed       unused   input  active-high 
	line  49:      unnamed       unused   input  active-high 
	line  50:      unnamed       unused  output  active-high 
	line  51:      unnamed       unused   input  active-high 
	line  52:      unnamed       unused   input  active-high 
	line  53:      unnamed       unused   input  active-high 
	line  54:      unnamed       unused  output  active-high 
	line  55:      unnamed       unused   input  active-high 
	line  56:      unnamed       unused   input  active-high 
	line  57:      unnamed       unused   input  active-high 
	line  58:      unnamed       unused   input  active-high 
	line  59:      unnamed       unused   input  active-high 
	line  60:      unnamed       unused   input  active-high 
	line  61:      unnamed       unused   input  active-high 
	line  62:      unnamed       unused   input  active-high 
	line  63:      unnamed       unused   input  active-high 
	line  64:      unnamed       unused   input  active-high 
	line  65:      unnamed       unused   input  active-high 
[... 157 lines cut ...]
	line  13:  "U16_IO1.5"       unused   input  active-high 
	line  14:  "U16_IO1.6"       unused   input  active-high 
	line  15:  "U16_IO1.7"       unused   input  active-high 
gpiochip3 - 16 lines:
	line   0:  "MUX33_DIR" "uart1-rx-oe" output active-high [used]
	line   1:  "MUX31_DIR" "uart1-tx-oe" output active-high [used]
	line   2:  "MUX29_DIR"       unused   input  active-high 
	line   3:  "MUX27_DIR"       unused   input  active-high 
	line   4:  "MUX24_DIR"       unused   input  active-high 
	line   5:  "MUX21_DIR"       unused   input  active-high 
	line   6:  "MUX19_DIR"       unused   input  active-high 
	line   7:  "MUX32_DIR"       unused   input  active-high 
	line   8:  "MUX30_DIR"       unused   input  active-high 
	line   9:  "MUX28_DIR"       unused   input  active-high 
	line  10:  "MUX26_DIR"       unused   input  active-high 
	line  11:  "MUX23_DIR"       unused   input  active-high 
	line  12:  "MUX20_DIR"       unused   input  active-high 
	line  13:  "MUX18_DIR"       unused  output  active-high 
	line  14:  "MUX22_SEL"       unused   input  active-high 
	line  15:  "MUX25_SEL"       unused   input  active-high 
gpiochip4 - 4 lines:
	line   0:      unnamed       unused   input  active-high 
	line   1:      unnamed       unused   input  active-high 
	line   2:      unnamed       unused   input  active-high 
	line   3:      unnamed       unused   input  active-high

htot, 2019-10-30:
> it's a long story, I have built sumo first, and then I have found i2c can't work, so I think it maybe caused by ACPI, then I have created another env and switch to master branch and built thud.

env has nothing to do with this. There is a no-acpi branch (sumo32) that makes mraa/upm working by adding a kernel patch. So i2c works.
https://edison-fw.github.io/meta-intel-edison/0.6-News.html#whats-new-for-this-release

In the acpi enabled version (sumo64-acpi, thud, thud32-v2.0), i2c works too. But not through mraa, mraa needs to be patched.
Here we select which acpi tables to load (leds!) and what stuff to enable (i2c!):
https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-bsp/conf/machine/edison.conf

> it return nothing
> root@edison:/media/sdcard/git/luma.examples# gpiofind TRI_STATE_ALL
> root@edison:/media/sdcard/git/luma.examples# gpiofind TRI_STATE_ALL

That is not normal. I am on warrior now (there is only an acpi enabled version now).
```
root@edison:~# gpiofind TRI_STATE_ALL
gpiochip1 14
```

> I found that built in led in pin 13 don't work, only when I do this
> rmmod ledtrig_heartbeat
> rmmod leds_gpio

With leds.asl loaded the kernel takes the led and runs the heartbeat trigger (this beats faster with higher cpu load). With the kernel taking control of the led, you can not access it from user space, so blink-led service fails.

If you don't want leds controlled from the kernel remove it from edison.conf.

> led start blink with some warning:
> root@edison:/media/sdcard/git/luma.examples# blink-led
> can't find line TRI_STATE_ALL, type=gpiod.LINE_REQ_DIR_OUT
> can't find line TRI_STATE_ALL, type=gpiod.LINE_REQ_DIR_OUT

Same problem, TRI_STATE_ALL doesn't exist.

> led owner name: gpiochip0, name 'gpiochip0:40 /unnamed/'
> ^CSignal intercepted: de-initing LED GPIOs
> clean and quit.
>
> I have connect second spi oled, it does not work too. [...]

I don't know this demo. Maybe something else is wrong. But note, in `ACPI_FEATURES_edison ?= "uart_2w i2c"` spi is not enabled. And you can't combine spi and leds.

> Version: luma.oled 3.3.0 (luma.core 1.12.0)
> Display: ssd1306
> Interface: spi
> Dimensions: 128 x 64
> demo.py: error: GPIO access not available
> 
> sck -> pin 13,gpio 40
> mosi -> pin11, gpio 43
> D/C -> pin 10, gpio 41
> CS -> pin 9 ,
> Reset -> pin 8, gpio 49

htot, 2019-10-30:
My gpio listing (note: 5 devices against your 4):
<details><summary>gpio listing</summary>
          
```
gpiochip0 - 192 lines:
        line   0:      unnamed       unused   input  active-high 
        line   1:      unnamed       unused   input  active-high 
        line   2:      unnamed       unused  output  active-high 
        line   3:      unnamed       unused   input  active-high 
        line   4:      unnamed       unused  output  active-high 
        line   5:      unnamed       unused  output  active-high 
        line   6:      unnamed       unused  output  active-high 
        line   7:      unnamed       unused   input  active-high 
        line   8:      unnamed       unused   input  active-high 
        line   9:      unnamed       unused  output  active-high 
        line  10:      unnamed       unused  output  active-high 
        line  11:      unnamed       unused  output  active-high 
        line  12:      unnamed       unused  output  active-high 
        line  13:      unnamed       unused  output  active-high 
        line  14:      unnamed       unused   input  active-high 
        line  15:      unnamed  "interrupt"   input  active-high [used]
        line  16:      unnamed       unused   input  active-high 
        line  17:      unnamed       unused   input  active-high 
        line  18:      unnamed       unused   input  active-high 
        line  19:      unnamed       unused   input  active-high 
        line  20:      unnamed       unused   input  active-high 
        line  21:      unnamed       unused   input  active-high 
        line  22:      unnamed       unused   input  active-high 
        line  23:      unnamed       unused   input  active-high 
        line  24:      unnamed       unused   input  active-high 
        line  25:      unnamed       unused   input  active-high 
        line  26:      unnamed       unused   input  active-high 
        line  27:      unnamed       unused   input  active-high 
        line  28:      unnamed       unused   input  active-high 
        line  29:      unnamed       unused   input  active-high 
        line  30:      unnamed       unused   input  active-high 
        line  31:      unnamed       unused   input  active-high 
        line  32:      unnamed       unused   input  active-high 
        line  33:      unnamed       unused   input  active-high 
        line  34:      unnamed       unused   input  active-high 
        line  35:      unnamed       unused   input  active-high 
        line  36:      unnamed       unused   input  active-high 
        line  37:      unnamed       unused   input  active-high 
        line  38:      unnamed       unused   input  active-high 
        line  39:      unnamed       unused   input  active-high 
        line  40:      unnamed          "?"  output  active-high [used]
        line  41:      unnamed       unused   input  active-high 
        line  42:      unnamed       unused   input  active-high 
        line  43:      unnamed       unused   input  active-high 
        line  44:      unnamed       unused   input  active-high 
        line  45:      unnamed       unused   input  active-high 
        line  46:      unnamed       unused   input  active-high 
        line  47:      unnamed       unused   input  active-high 
        line  48:      unnamed       unused   input  active-high 
        line  49:      unnamed       unused   input  active-high 
        line  50:      unnamed       unused  output  active-high 
        line  51:      unnamed       unused   input  active-high 
        line  52:      unnamed       unused   input  active-high 
        line  53:      unnamed       unused   input  active-high 
        line  54:      unnamed       unused  output  active-high 
        line  55:      unnamed       unused   input  active-high 
        line  56:      unnamed       unused   input  active-high 
        line  57:      unnamed       unused   input  active-high 
        line  58:      unnamed       unused   input  active-high 
        line  59:      unnamed       unused   input  active-high 
        line  60:      unnamed       unused   input  active-high 
        line  61:      unnamed       unused   input  active-high 
        line  62:      unnamed       unused   input  active-high 
        line  63:      unnamed       unused   input  active-high 
        line  64:      unnamed       unused   input  active-high 
[... 179 lines cut ...]
        line   0:  "MUX33_DIR" "uart1-rx-oe" output active-high [used]
        line   1:  "MUX31_DIR" "uart1-tx-oe" output active-high [used]
        line   2:  "MUX29_DIR"       unused   input  active-high 
        line   3:  "MUX27_DIR"       unused   input  active-high 
        line   4:  "MUX24_DIR"       unused   input  active-high 
        line   5:  "MUX21_DIR"       unused   input  active-high 
        line   6:  "MUX19_DIR"       unused   input  active-high 
        line   7:  "MUX32_DIR"       unused   input  active-high 
        line   8:  "MUX30_DIR"       unused   input  active-high 
        line   9:  "MUX28_DIR"       unused   input  active-high 
        line  10:  "MUX26_DIR"       unused   input  active-high 
        line  11:  "MUX23_DIR"       unused   input  active-high 
        line  12:  "MUX20_DIR"       unused   input  active-high 
        line  13:  "MUX18_DIR"       unused  output  active-high 
        line  14:  "MUX22_SEL"       unused   input  active-high 
        line  15:  "MUX25_SEL"       unused   input  active-high 
gpiochip5 - 4 lines:
        line   0:      unnamed       unused   input  active-high 
        line   1:      unnamed       unused   input  active-high 
        line   2:      unnamed       unused   input  active-high 
        line   3:      unnamed       unused   input  active-high 
```
</details>

Do you have the same if you power down the edison (pull the power cord) instead of pushing reset (cold v.s warm boot).

xlla, 2019-11-01:
what I said 'env' is meaning another environment, a cloned vmware os, I just kept bbcache/downloads folder to reduce fetch time.

I want to try new kernel and new packages, so I use master branch

xlla@u18-1:~/develop/git/meta-intel-edison$ git status
On branch master
Your branch is up to date with 'origin/master'.

and I want to learn how to patch mraa too.

since kernel controled led, and led is conflict with spi, I commented ledtrig-heartbeat

#KERNEL_MODULE_AUTOLOAD += " ledtrig-heartbeat"
KERNEL_MODULE_AUTOLOAD += "fbtft_device"
#module_conf_fbtft_device = "options fbtft_device name=er_oled028 busnum=5 debug=7
# FIXME: file parameter should be based on partition UUID (from U-Boot) or fixed
# with label (label seems to work, but driver is probed too early)
#module_conf_g_multi = "options g_multi file=/dev/mmcblk0p9 stall=0 idVendor=0x8087 idProduct=0x0A9E iProduct=Edison iManufacturer=Intel"
#KERNEL_MODULE_PROBECONF += "g_multi"

#ACPI_TABLES ?= "arduino.asl spidev.asl leds.asl"
ACPI_TABLES ?= "arduino.asl spidev.asl"

ACPI_FEATURES_edison ?= "uart_2w i2c spi"

after build new image and flash to edison,  and poweroff edison, unplug two usb cable, then plug in.
TRI_STATE_ALL is appear

root@edison:~# gpiofind TRI_STATE_ALL
gpiochip1 14

gpioinfo shows 5 devices now!

but the i2c oled ssd1306 device still not appear at 0x3C
root@edison:~# i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: UU UU UU UU -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
root@edison:~# i2cdetect -y -r 6
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

I have connect oled sda/scl -> A4/A5, and yet ->sda/scl , either connection failure.

htot, 2019-11-01:
> what I said 'env' is meaning another environment, a cloned vmware os, I just kept bbcache/downloads folder to reduce fetch time.

OK

> I want to try new kernel and new packages, so I use master branch

If you want the latest, I am working on warrior here: https://github.com/htot/meta-intel-edison

> xlla@u18-1:~/develop/git/meta-intel-edison$ git status
> On branch master
> Your branch is up to date with 'origin/master'.
> 
> and I want to learn how to patch mraa too.

That will be most welcome.

> since kernel controled led, and led is conflict with spi, I commented ledtrig-heartbeat

Actually it is the acpi table `leds.asl` that defines the control (hogs the gpio).
You can find the acpi tables here: https://github.com/edison-fw/meta-acpi/tree/eds-5.0.0/recipes-bsp/acpi-tables/samples/edison.

> ` #KERNEL_MODULE_AUTOLOAD += " ledtrig-heartbeat"`

This module only tells the kernel how to blink the led device. We have other `ledtrig` as well.

> ` KERNEL_MODULE_AUTOLOAD += "fbtft_device"`

We do have display acpi tables, but I don't know how to use/create them. @andy-shev knows of course.

>` #module_conf_fbtft_device = "options fbtft_device name=er_oled028 busnum=5 debug=7`

<snip>

> `#ACPI_TABLES ?= "arduino.asl spidev.asl leds.asl"`
>` ACPI_TABLES ?= "arduino.asl spidev.asl"`

This will load these 2 tables and the LED pin will be available.

> `ACPI_FEATURES_edison ?= "uart_2w i2c spi"`

This sets conditional for building the tables. There will be a uart, i2c and spi.
 
> after build new image and flash to edison, and poweroff edison, unplug two usb cable, then plug in.
> TRI_STATE_ALL is appear

So maybe the problem was caused by a locked up i2c device.

> root@edison:~# gpiofind TRI_STATE_ALL
> gpiochip1 14
> 
> gpioinfo shows 5 devices now!
> 
> but the i2c oled ssd1306 device still not appear at 0x3C

```
i2cdetect -y -r 6
 0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

> I have connect oled sda/scl -> A4/A5, and yet ->sda/scl , either connection failure.

There is a jumper to select 3V / 5V operation. What does your device need?
Also: TRI_STATE_ALL must be high. You could try this before i2cdetect:
```
gpioset `gpiofind TRI_STATE_ALL`=0
gpioset `gpiofind TRI_STATE_ALL`=1
```

xlla, 2019-11-02:
Aha! that's the key!
root@edison:~# modprobe -i i2c-dev
root@edison:~# i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: UU UU UU UU -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
root@edison:~# gpioset `gpiofind TRI_STATE_ALL`=0
root@edison:~# gpioset `gpiofind TRI_STATE_ALL`=1
root@edison:~# i2cdetect -y -r 6
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
root@edison:~# 

my device can tolerate 3v-5v, so I just left jumper to default location, I think it's 5v.

htot, 2019-11-02:
It may be that documentation can be improved. We take patches for that too. You can find mark down here: https://github.com/edison-fw/meta-intel-edison/tree/master/docs/_docs/Edison. Github generates https://edison-fw.github.io/meta-intel-edison/ from that.

xlla, 2019-11-02:
it work!  amazing! thank you very much!
but I am still a little bit confused by what goes on.
according original Intel edison arduino board manual,  before we can use i2c , we should pull down gpio214, then select pin-mux, set pin direction, set pin pull-up, and then pull up gpio214.

now I just simple pull down gpio214 and pull up it, everything is done. 
if I missed anything or kernel prepared everything in between 214 down and up according ACPI table?

htot, 2019-11-02:
It's done in the acpi tables. Don't thank me, thank @andy-shev who did the kernel work and the acpi tables.

xlla, 2019-11-03:
yes we can!
many thanks for @andy-shev 's hard work, and thanks for your patience!

one more thing, 
I saw there already is a call in
recipes-bsp/acpi-tables/files/edison/acpi-tables-load

but when I first login, it was still remain low
root@edison:~# gpioget `gpiofind TRI_STATE_ALL`
0

closed 2019-11-03 by xlla

andy-shev, 2019-11-05:
Unfortunaly this discrete pin control on Edison/Arduino has an (unknown) flaw when I2C GPIO expanders, connected to V_SHLD_IO, are simple disappeared. This requires the cold power off / on cycle. The desing of TRI_STATE_ALL is another flaw on this board. So, we have no mechanisms to set that pin reliable. (In correct design the TRI_STATE_ALL must be a pin of the GPIO which always present in the system, like the main one on SoC).

andy-shev, 2019-11-05:
@htot, may we put the above to some wiki if it's not there? Can you recommend the page where that information suits better to be on?

htot, 2019-11-05:
We can put it in the wiki https://github.com/edison-fw/edison-wiki may be create a new page i2c.md?

Or we can put it in the documentation https://github.com/edison-fw/meta-intel-edison/blob/master/docs/_docs/Edison/Current%20State/4.4-hsu-i2c-spi.md

andy-shev, 2019-11-05:
@htot, just sent a PR, please consider to merge. Thanks!

htot, 2019-11-05:
Thanks, merged that. [Edison Wiki](https://edison-fw.github.io/edison-wiki/) should be more or less good to go now and a nice addition to the [Intel Edison Image Builder](https://edison-fw.github.io/meta-intel-edison/) site. Closing this issue.