edison-fw/meta-intel-edison #75: how to use MCU SDK in thud branch or most recent branch

xlla, 2019-11-03:
in original firmware, we use tty channels and sysfs nodes to coummunicate with the MCU.
in thud branch, the /dev/ttymcuX is missing,  /sys/devices/platform/intel_mcu does not exist, 
so how can develop a MCU application and upload/debug it ?

htot, 2019-11-03:
According to @andy-shev 

> The MCU is not supported right now until forward ported

xlla, 2019-11-04:
alright.

andy-shev, 2019-11-05:
And to the record I'm not willing to enable it at all. It has a lot of problems, in particular with pin control since the original (Linux kernel) solution is not compatibnle with upstream.

xlla, 2019-11-22:
what a pity that a 100Mhz quark MCU was abandoned.

andy-shev, 2019-11-22:
> what a pity that a 100Mhz quark MCU was abandoned.

Agree, and unfortunately I can't do anything about it. I can, though not recommend (waste of time for cancelled products), to complain to Intel thru the official channels.

htot, 2019-11-22:
The idea was brilliant. Would have been great to run Zephyr on the MCU. Maybe it will return in a future product.

xlla, 2019-11-23:
> waste of time for cancelled products

Reasonably~but why we devoted ourself with enthusiasm to maintain this cancelled product(intel edison), and add many new features to it, such as new kernel, patched u-boot, ACPI, 64bit, etc.

andy-shev, 2019-11-23:
> > waste of time for cancelled products
> 
> Reasonably~but why we devoted ourself with enthusiasm to maintain this cancelled product(intel edison), and add many new features to it, such as new kernel, patched u-boot, ACPI, 64bit, etc.

First of all, **just for fun** and there also few more useful (for me personally as an Intel employee) giveaways.

xlla, 2019-11-23:
me too, just for fun , haha~

xlla, 2020-03-09:
@andy-shev as I was stuck in acpi,u-boot and kernel, I would like to port the old implement to current , could you mind to help me as a guides.

andy-shev, 2020-03-09:
@xlla, simplest way is to take PSH "driver" as is from stock image (sources) and compile it in the new kernel environment. It should work like the original one. Somewhere I even had a patch, I can share (tell me your email for that). The pin muxing issue is there, though. That's why enabling pin muxing has much more priority over PSH, besides the fact the PSH is not convenient to use due to absence of Zephyr kernel for it.

xlla, 2020-03-17:
@andy-shev sorry for the late reply, I have sent a email to you.
I have final rescued my development environment and migrate to a ssd, hope it will speed up build process.
I have prepare a workspace for kernel module development,  plan to migrate module intel_mcu.

andy-shev, 2020-03-17:
I have sent you the patch (it's basically compound of what we used to have in stock kernel. Feel free to clean up and send a PR. Maybe @htot will include it in the kernel. (Note, it's not upstreamable material from my perspective, though you always may try to go directly to upstream with it)

xlla, 2020-03-19:
I have compiled it into kernel, flash to ota partition and boot from sdcard.
it works but very lag.
<pre>

Kernel command line: "tty1=console=ttyS2,115200n8 root=/dev/mmcblk1p1 rootfstype=ext4 systemd.unit=multi-user.target hardware_id=00"
Magic signature found

Starting kernel ...

Application available at (physical) address 0x04819000
    VRL mapped to 0xff217000
    App size = 11508 bytes

    App Authentication feature is disabled!
    Resetting IPC

*** Ready to receive application *** 

Poky (Yocto Project Reference Distro) 2.7.3 edison ttyS2

</pre>

cpu load up to 43% and keep on.
<pre>
Mem: 161068K used, 816884K free, 576K shrd, 8356K buff, 104880K cached
CPU:   4% usr  54% sys   0% nic   0% idle  41% io   0% irq   0% sirq
Load average: 17.99 17.61 12.77 12/108 750
Mem: 161068K used, 816884K free, 576K shrd, 8356K buff, 104880K cached
CPU:   4% usr  52% sys   0% nic   0% idle  43% io   0% irq   0% sirq
Load average: 17.99 17.62 12.80 12/108 751
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
   89     2 root     RW       0   0%  43% [irq/24-intel_ps]
  751   661 root     R     3764   0%   4% top
  656   655 root     S    26952   3%   0% (sd-pam)
    1     0 root     D    26156   3%   0% {systemd} /sbin/init tty1=console=ttyS2,115200n8 root=/dev/mmcblk1p1 rootfstype=ext4 systemd.unit=m
  498     1 root     S    22248   2%   0% /usr/sbin/rngd -f -r /dev/hwrng
</pre>

dmesg flood with
<pre>

[  509.134958] intel_psh_ipc 0000:00:16.0: Ignore message from channel 6
[  509.135010] intel_psh_ipc 0000:00:16.0: Ignore message from channel 6
[  509.135136] intel_psh_ipc 0000:00:16.0: Ignore message from channel 6
[  509.135206] intel_psh_ipc 0000:00:16.0: Ignore message from channel 6
[  509.135344] intel_psh_ipc 0000:00:16.0: Ignore message from channel 6
[  509.135473] intel_psh_ipc 0000:00:16.0: Ignore message from channel 6
[  509.135542] intel_psh_ipc 0000:00:16.0: Ignore message from channel 6
[  509.135593] intel_psh_ipc 0000:00:16.0: Ignore message from channel 6
[  509.135709] intel_psh_ipc 0000:00:16.0: Ignore message from channel 6
[  509.135763] intel_psh_ipc 0000:00:16.0: Ignore message from channel 6
</pre>

intel_mcu node exist
<pre>
root@edison:~# ls /sys/devices/platform/intel_mcu -al
total 0
drwxr-xr-x  4 root root    0 Mar 18 17:38 .
drwxr-xr-x 17 root root    0 Mar 18 17:38 ..
--w-------  1 root root 4096 Mar 18 17:44 control
lrwxrwxrwx  1 root root    0 Mar 18 17:46 driver -> ../../../bus/platform/drivers/intel_mcu
-rw-r--r--  1 root root 4096 Mar 18 17:46 driver_override
-r--------  1 root root 4096 Mar 18 17:42 fw_version
-rw-------  1 root root 4096 Mar 18 17:43 log_level
-r--r--r--  1 root root 4096 Mar 18 17:46 modalias
drwxr-xr-x  2 root root    0 Mar 18 17:46 power
lrwxrwxrwx  1 root root    0 Mar 18 17:38 subsystem -> ../../../bus/platform
drwxr-xr-x  5 root root    0 Mar 18 17:38 tty
-rw-r--r--  1 root root 4096 Mar 18 17:38 uevent
</pre>

if I missing something or default mcu app try to send something that new kernel can't understand?

xlla, 2020-03-19:
I upload a simple [mcu app](https://software.intel.com/en-us/articles/creating-applications-with-mcu-sdk-for-intel-edison-board#Getting_debug_messages_from_the_MCU) just read from tty and print debug info, to avoid involve any mux/pin.

I can't get any message from ttymcu0

<pre>
root@edison:/mnt/sdcard/test# echo debug > /sys/devices/platform/intel_mcu/log_level 
-sh: echo: write error: Operation not permitted

root@edison:/mnt/sdcard/test# cat /sys/devices/platform/intel_mcu/log_level
cat: /sys/devices/platform/intel_mcu/log_level: Operation not permitted

root@edison:~# cat /dev/ttymcu0
^C

root@edison:~# echo "start" > /dev/ttymcu0 
root@edison:~# cat /dev/ttymcu0 
^C

root@edison:/mnt/sdcard/test# cat /proc/interrupts | grep intel_psh_ipc
 24:   33181819          0   IO-APIC  47-fasteoi   intel_psh_ipc

</pre>

xlla, 2020-03-21:
after clear interrupt flags, no more complain messages, cpu load drop to 0, 
<pre>
root@edison:~# cat /proc/interrupts | grep intel_psh_ipc
 24:          1          0   IO-APIC  47-fasteoi   intel_psh_ipc

root@edison:~#top
CPU:   0% usr   0% sys   0% nic  99% idle   0% io   0% irq   0% sirq

</pre> 

but still can't access log_level
<pre>
root@edison:~# cat /sys/devices/platform/intel_mcu/log_level
cat: /sys/devices/platform/intel_mcu/log_level: Operation not permitted

</pre>

xlla, 2020-03-22:
I don't know how to debug kernel module.
after add many trace log, I finally found intel_mcu was load before intel_psh_ipc, so something not prepared correctly.
how to force load module_platform_driver A after module_pci_driver B?
should I move intel_mcu files from drivers/hwmon into drivers/platform/x86 , and change the appearance order in Makefile ?

xlla, 2020-03-22:
Bingo! 
after adjust those modules load senquence, I got it.
It is very exciting even just seen a few logs from MCU. @htot 

[image]
[image]

<pre>
root@edison:~# cat /sys/devices/platform/intel_mcu/log_level
debug

</pre>

Many thanks to @andy-shev !

htot, 2020-03-22:
Don't forget to send a PR when you get stuff to work!

xlla, 2020-03-22:
look carefully at those logs and compare with source code, the sequence was  RANDOM!

xlla, 2020-03-22:
@htot I will. 
it now on bare status, I need to check every point with edison-mcusdk.
first of all, I want to config usb0 to 192.168.2.15 automatically, since each mcusdk upload an app, it will reboot edison, then I lost usb ethernet.
I have created a service to do that, but failure on reboot.
<pre>
root@edison:~# systemctl status network-gadget-init.service
● network-gadget-init.service - USB Ethernet gadget
   Loaded: loaded (/lib/systemd/system/network-gadget-init.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Mon 2020-03-23 00:17:16 CST; 41s ago
  Process: 573 ExecStart=/bin/sh -c  ifconfig usb0 192.168.2.15 netmask 255.255.255.0 (code=exited, status=255/EXCEPTION)
 Main PID: 573 (code=exited, status=255/EXCEPTION)

Mar 23 00:17:15 edison systemd[1]: Starting USB Ethernet gadget...
Mar 23 00:17:15 edison sh[573]: SIOCSIFADDR: No such device
Mar 23 00:17:15 edison sh[573]: usb0: ERROR while getting interface flags: No such device
Mar 23 00:17:15 edison sh[573]: SIOCSIFNETMASK: No such device
Mar 23 00:17:16 edison systemd[1]: network-gadget-init.service: Succeeded.
Mar 23 00:17:16 edison systemd[1]: Started USB Ethernet gadget.

</pre> 

should I wait some service then launch my service?
I don't know which service will prepare usg gadget.

or add this in to service unit file?
<pre>
ExecStartPre=-/bin/sh -c "echo 'connect' > /sys/class/udc/dwc3.0.auto/soft_connect"
</pre>

another question, when mcusdk upload an app, it will reboot edison; it seems edison must be reboot when mcu app upload, but I notice these is a service will upload mcu app, why that upload do not need a reboot?
<pre>
root@edison:~# systemctl status mcu_fw_loader.service
● mcu_fw_loader.service - Daemon to load edison mcu app binary
   Loaded: loaded (/lib/systemd/system/mcu_fw_loader.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Mon 2020-03-23 00:17:16 CST; 16min ago
  Process: 582 ExecStart=/etc/intel_mcu/mcu_fw_loader.sh (code=exited, status=0/SUCCESS)
 Main PID: 582 (code=exited, status=0/SUCCESS)

Mar 23 00:17:15 edison systemd[1]: Started Daemon to load edison mcu app binary.
Mar 23 00:17:16 edison systemd[1]: mcu_fw_loader.service: Succeeded.
</pre>

If I first power on edison, this service will upload mcu app, then edison need reboot immediately, and the app will upload again, and need another reboot.
should I disable this service , manual upload mcu app , do a reboot to do the test?

htot, 2020-03-22:
Network: https://edison-fw.github.io/meta-intel-edison/4.2-networking.html#gadget
The network should already be up, no service needed. The connection is made from the host side.

MCU problems: I don't now, never tried, not even with the factory firmware.

xlla, 2020-03-23:
I have execute those commands in that article, and I found every time I reboot the Edison, only need manual call 
<pre>
ifconfig usb0 192.168.2.15 netmask 255.255.255.0
</pre>
then usb ethernet would work.

Since I changed gadget protocol to RNDIS, on mac host, network was recognized automatically.
[image]

> The network should already be up, no service needed

it was up when I login in,  but not got an IP like this.
<pre>
root@edison:~# ifconfig usb
usb0      Link encap:Ethernet  HWaddr aa:bb:cc:dd:ee:f0  
          inet6 addr: fe80::a8bb:ccff:fedd:eef0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:20285 errors:0 dropped:0 overruns:0 frame:0
          TX packets:38 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:828488 (809.0 KiB)  TX bytes:6518 (6.3 KiB)

</pre>

would you tell me which service load /usr/bin/conf-gadget.sh, so I can let my service wait after it.

xlla, 2020-03-23:
or another service had execute those command auto
<pre>
connmanctl
enable gadget
services
connect gadget_aabbccddeef1_usb
</pre>

each time I power on edison or reboot it, wifi and usb0 was got connected.
<pre>
root@edison:~# connmanctl              
connmanctl> services
*AO wifissid                wifi_fcc2de3408cc_6970617135_managed_psk
*   Wired                gadget_aabbccddeef0_usb
connmanctl> 

</pre>

wifi was fine, since dhcp on router, but usb0 need manual assign IP.

htot, 2020-03-23:
> but usb0 need manual assign IP.

No that is wrong. usb gets ip from host. But you need to tell host to have dhcp server (shared ip connection). Reed instructions again.

Also usb should have power enabled automatically.

xlla, 2020-03-24:
to emulate the original Edison's behavior, I found it was called by udev rules, so I invoke my service from udev rules too, it works.

<pre>
[  OK  ] Started Hostname Service.

Poky (Yocto Project Reference Distro) 2.7.3 edison ttyS2

edison login: root
Last login: Wed Mar 25 00:13:23 CST 2020 on ttyS2
root@edison:~# ifconfig usb0
usb0      Link encap:Ethernet  HWaddr aa:bb:cc:dd:ee:f0  
          inet addr:192.168.2.15  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::a8bb:ccff:fedd:eef0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:59 errors:0 dropped:0 overruns:0 frame:0
          TX packets:43 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:19119 (18.6 KiB)  TX bytes:9168 (8.9 KiB)

root@edison:~# systemctl status  sys-subsystem-net-devices-usb0.device
● sys-subsystem-net-devices-usb0.device - Merrifield USB Device Controller (OTG)
   Loaded: loaded
   Active: active (plugged) since Wed 2020-03-25 00:16:19 CST; 38s ago
   Device: /sys/devices/pci0000:00/0000:00:11.0/dwc3.0.auto/gadget/net/usb0
root@edison:~# systemctl status usb-gadget@usb0.service               
● usb-gadget@usb0.service - USB Gadget ifconfig
   Loaded: loaded (/lib/systemd/system/usb-gadget@.service; static; vendor preset: enabled)
   Active: active (exited) since Wed 2020-03-25 00:16:20 CST; 1min 1s ago
  Process: 613 ExecStart=/sbin/ifconfig usb0 192.168.2.15 netmask 255.255.255.0 (code=exited, status=0/SUCCESS)
 Main PID: 613 (code=exited, status=0/SUCCESS)

Mar 25 00:16:19 edison systemd[1]: Starting USB Gadget ifconfig...
Mar 25 00:16:20 edison systemd[1]: Started USB Gadget ifconfig.
</pre>

htot, 2020-03-24:
Sure, there may be multiple ways to do the same thing.

xlla, 2020-03-25:
in my case, after reboot, no need do anything,  it is Wired without "A" prefix.
my udev rules will config ip automatic, so does not need host run dhcp service.
as there is showed "Connected = False", I can do connection test successful.
<pre>
root@edison:~# connmanctl
connmanctl> services
*AO ssid                wifi_xxxxx_managed_psk
*   Wired                gadget_aabbccddeef0_usb
    phone                 bluetooth_xxxx

connmanctl> technologies
/net/connman/technology/gadget
  Name = Gadget
  Type = gadget
  Powered = True
  Connected = False
  Tethering = False

root@edison:~# iperf3 --server
-----------------------------------------------------------
Server listening on 5201
</pre>
on host
<pre>
==> Pouring iperf3-3.7_1.high_sierra.bottle.1.tar.gz
🍺  /usr/local/Cellar/iperf3/3.7_1: 11 files, 380.4KB
rmbp:workspace-mcu xlla$ iperf3 --client 192.168.2.15
Connecting to host 192.168.2.15, port 5201
[  5] local 192.168.2.20 port 58492 connected to 192.168.2.15 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  26.9 MBytes   226 Mbits/sec                  
[  5]   1.00-2.00   sec  27.0 MBytes   227 Mbits/sec                  
[  5]   2.00-3.00   sec  26.8 MBytes   225 Mbits/sec                  

</pre>

htot, 2020-03-25:
That's right.

In fact you are 'manually assigning an address' but through a script. That means your host doesn't know the address edison or edison.local unless you also add it to /etc/hosts, right?

xlla, 2020-03-26:
yes! that's the original Edison board's behavior, at that time, every [tutorial](https://software.intel.com/en-us/articles/intel-edison-board-user-guide) and manual just use that special IP for edison usb ethernet. intel xdk/ mcusdk/iot all are try to connect edison use that IP.
I am lazy to change those app.

htot, 2020-03-26:
I see that's fine.

Can you create a PR for changing to rndis?

htot, 2020-03-30:
I use `qgit` to visually select files to create a commit.

But I think Andy pointed out you can undo 1 commit with `git reset HEAD~1`, the files will be uncommited but not erased.

Each commit (git commit) will be done on the files that you added (git add).

https://www.atlassian.com/git/tutorials/saving-changes/git-commit

We really don't want to have a single commit with everything in it. It's impossible to understand / debug / rollback.

andy-shev, 2020-05-19:
The ProGit Book is very good and free: https://git-scm.com/book/en/v2

htot, 2021-03-15:
@xlla 
> In my adventure, I'v change a lot of files to get things run, I don't know how to pick them out individually right now, may I create a PR with all of them, or create a clean PR when I have learned how to do that in the future.
> 
> I will push it [here](https://github.com/xlla/meta-intel-edison/tree/warrior-32-sketch-v2.1) temporary in period.

Are you still working on this?
There is quite some useful stuff in your branch like enable rndis. Did you get the mcu to work in the end?

mwallnoefer, 2025-07-08:
@xlla ?

But now that we have the PREEMPT_RT kernels in place we do not really need it anymore, correct?

htot, 2025-07-08:
I think with PREEMPT_RT we have about 50 us latency, maybe up to 100 us. And we have the HSU working up to 3.5Mb/s. The HCU afaik only goes (with ViperOS) to 115.2kb/s, but maybe the latency is less. If so you might use that for really low latency applications.

Or with battery powered apps sleep the uP and wake up by the MCU to save power.

Actually right now I am working on an interim scarthgap release with the MCU working. Finalizing the docs now.

beckmx, 2026-09-23:
Hello there, i wanted to share with you guys, i managed to run and deploy code for the MCU quark, i am reusing this old hardware to build some general robots that move garbage can around the house.

https://github.com/beckmx/edison-quark-compiler

Greetings!

htot, 2026-09-23:
Cool. 

Q: You copy your program directly into the MCU memory space?
Does that work better than loading “/lib/firmware/intel_mcu.bin”?

beckmx, 2026-09-28:
> Cool.
> 
> Q: You copy your program directly into the MCU memory space? Does that work better than loading “/lib/firmware/intel_mcu.bin”?

Yes, as far as I understood, intel originally also loaded the firmware for the quark "on the fly" directly on assignated RAM space, so i could say (not an expert yet) it's compatible, i just place my firmware anywhere and load it when the edison boots. I do noticed that there are some wierd pins for instance i initiate the I2C device (an mpu5060) first on the linux side, close the connection and start on the MCU and it continues to work and in my tests with a robot that moves, is very very precise, it runs like a charm, so my wheels are more "real time" than in linux, rotations integrate with .5°