edison-fw/meta-intel-edison #77: how to install driver

shawnhsoia97, 2019-11-11:
Hi i want to install sierra driver,but that show "not found kernel modules" when i make install($(MAKE) -C $(/lib/modules/$(shell uname -r)/build) M=$(PWD) modules).So i think it should be no  kernel header,then i add" IMAGE_INSTALL_append = linux-libc-header" to edison-image.bb,but i have encountered an error that show "unable to locate package linux-libc-header",can you help me?thank you.
(thud branch)

htot, 2019-11-11:
I don't know Sierra. But this is not really the Yocto way to add a module to the kernel.

I am working on the documentation here (warrior but it is valid for thud too in this case): https://htot.github.io/meta-intel-edison/5.1-Bitbake-tricks#configuring-the-kernel-and-grab-the-kernel-fragment

The idea is to configure the kernel, then create a fragment, which is like a diff of the configuration changes you make. Then add the fragment to the recipe https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-bsp/recipes-kernel/linux/linux-yocto_5.2.0.bb like ftdi_sio.cfg.

shawnhsoia97, 2019-11-13:
i want to  use usb serial,because ttyUSB* not found,so i trying to create a usb_serialKH.cfg,content as follows 
CONFIG_USB_SERIAL=y
CONFIG_USB_SUPPORT=y
CONFIG_USB=y
CONFIG_TTY=y
CONFIG_HAS_IOMEM=y
CONFIG_USB_SERIAL_CONSOLE=y
CONFIG_USB_SERIAL_GENERIC=y
CONFIG_USB_SERIAL_SIMPLE=y
CONFIG_USB_SERIAL_SIERRAWIRELESS=m
CONFIG_USB_SIERRA_NET=y
CONFIG_PHY_CADENCE_SIERRA=y
then add file://usb_serialKH.cfg to linux-yocto_5.2.0.bb ,finally i executed make edison-image
but i still failed after the test,do you know where i am doing wrong?

sideralis, 2019-11-13:
I had the same problem as you. In my case the device I was connecting to Edison board was using a CH340 USB to serial chip.
In general, it is either a Prolific chip or a CH340 chip.
By default CH340 is not supported in Edison kernel. To activate it, I added the file `ch341.cfg` in `.../meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/files`
I modified `linux-yocto_5.0.0.bb` in `.../meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux` 
by adding
`	file://ch341.cfg \`
to 
`SRC_URI`

And my file `ch341.cfg` contains:
`CONFIG_USB_SERIAL_CH341=m`

You recompile and reflash.

If you device is using prolific, you have to find the right config.

shawnhsoia97, 2019-11-19:
Sorry,i still don't understand how to use kernel headers.
i adding config_ikheaders=m,insmod kheaders.ko,but these don't work.
i can't find Modules in the /lib/modules/$(shell uname -r)/build.
Do you know how to install driver?please help me,thank you
(thud branch)

htot, 2019-11-19:
What do you mean by using kernel headers? With Yocto you don't build drivers on the Edison, but on the host. Yocto builds the kernel and modules already, so headers and stuff are already in place, nothing extra to install. You are probably googling around how to build modules on the edison itself.

I think you want to configure the kernel to build the driver you need. Did you follow my instructions on how to configure the kernel?

ochrin, 2019-11-19:
As htot, I don' t understand exactly what you want to do. But anyway:
If you add a file _ikheaders.cfg_ in _meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/files/ikheaders.cfg_ which contains:
`CONFIG_IKHEADERS=m`
And add this line (2nd line) in _meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/linux-yocto_5.2.0.bb_:
```
        file://bpf.cfg \
	file://ikheaders.cfg \
        "
```
Then build with make edison-image and flash, then you will find:
_kheaders.ko_ in _/lib/modules/5.2.0-edison-acpi-standard/kernel/kernel_ on edison board.

ochrin, 2019-11-19:
I don't know exactly which SIERRA USB driver you want to install, but looking at kernel config, this may help:
Do the same as my previous comment but replace 
`CONFIG_IKHEADERS=m`
by 
`CONFIG_USB_SERIAL_SIERRAWIRELESS=m`
which is for "USB Sierra Wireless Driver"
There is also a:
`CONFIG_USB_SIERRA_NET` available for "USB-to-WWAN Driver for Sierra Wireless modems". Don't know it it makes sense for you.

htot, 2019-11-19:
> Sorry ,i did not explain clearly.
> i try build external module,because i try to follow your instructions on configure the kernel,but it can't achieve the function I want(i want to let Edison connect to the internet with sierra modules),so i need install external driver and external module(GobiNet)

What do you mean by "external module"? If you mean you need to build a kernel module, that is exactly what my instructions result to. The kernel will be built with the configuration file, the module will be built and will be included on the image. There is no need to add kernel-devsrc.

However, if you mean out-of-tree module, then I don't know how to do that exactly.

> sierra.cfg contains:
> 
> ```
> CONFIG_USB_SERIAL_QUALCOMM=y
> CONFIG_USB_SERIAL_SIERRAWIRELESS=m
> CONFIG_USB_SIERRA_NET=y
> CONFIG_PHY_CADENCE_SIERRA=m
> CONFIG_USB_HSO=y
> CONFIG_USB_NET_QMI_WWAN =y
> CONFIG_N_GSM=y
> CONFIG_USB_NET_HUAWEI_CDC_NCM=y
> CONFIG_IKHEADERS=m
> ```
> 
> However,i have a "build" folder in the `/lib/modules/5.2.0-edison-acpi-standard/` by add `IMAGE_INSTALL =kernel-devsrc`, so i solved the problem of not having an kernel header.

I have under this path `out/linux64/build/tmp/work/edison-poky-linux/linux-yocto/`:
[image]

As you can see there are plenty kernel modules built including ftdi_sio, all are installed onto the image due to [this line](https://github.com/edison-fw/meta-intel-edison/blob/7499c1d689aec15c1a6ed65980b9594a443e8016/meta-intel-edison-distro/recipes-core/images/edison-image-minimal.bb#L41) , not necessarily automatically loaded. For autoload you may need [this line](https://github.com/edison-fw/meta-intel-edison/blob/7499c1d689aec15c1a6ed65980b9594a443e8016/meta-intel-edison-bsp/conf/machine/edison.conf#L26).

shawnhsoia97, 2019-11-20:
i mean is out-of-tree module.

i follow your method,but i didn't achieve the results i wanted,so i need to use out-of-tree module.

thank you for help me.
Have a good day.

shawnhsoia97, 2019-11-25:
Hi @htot
i have a question ,i don't know what i did.My edison is boot floder can't enter,that show `-sh: cd: /boot/: No such device`,also there is nothing in the folder.
```
root@edison:~# ls / -l
total 53
drwxr-xr-x   2 root root  4096 Nov 25  2019 bin
drwxr-xr-x   2 root root     0 Nov  1 02:05 boot
drwxr-xr-x  16 root root  3500 Nov  1 02:05 dev
drwxr-xr-x  38 root root  4096 Nov  1 02:05 etc
drwxr-xr-x   3 root root  1024 Jun 19  2015 factory
drwxr-xr-x   6 root root  4096 Aug  1  2018 home
drwxr-xr-x   9 root root  4096 Nov 25  2019 lib
drwx------   2 root root 16384 Nov 25  2019 lost+found
drwxr-xr-x   2 root root  4096 Nov  1  2019 media
drwxr-xr-x   2 root root  4096 Nov  1  2019 mnt
dr-xr-xr-x 113 root root     0 Jan  1  2000 proc
drwxr-xr-x  15 root root   400 Nov  1 02:05 run
drwxr-xr-x   2 root root  4096 Nov 25  2019 sbin
dr-xr-xr-x  12 root root     0 Jan  1  2000 sys
drwxrwxrwt   9 root root   220 Nov  1 02:05 tmp
drwxr-xr-x  11 root root  4096 Nov 25  2019 usr
drwxr-xr-x  10 root root  4096 Nov  1 02:05 var
```

htot, 2019-11-25:
It should automount when you enter (this when I boot using usbboot so boot is on p9):
```
root@edison:/# cat /etc/fstab | grep boot
/dev/mmcblk0p9     /boot       auto    noauto,x-systemd.automount,nosuid,nodev,noatime,discard     1   1
```
After automount:
```
root@edison:~# mount | grep boot
systemd-1 on /boot type autofs (rw,relatime,fd=42,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
/dev/mmcblk0p9 on /boot type vfat (rw,nosuid,nodev,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro,discard,x-systemd.automount)
```

shawnhsoia97, 2019-11-26:
My Edison also shows automount:
```
root@edison:~# cat /etc/fstab | grep boot
/dev/mmcblk0p9     /boot       auto    noauto,x-systemd.automount,nosuid,nodev,noatime,discard     1   1
```
but that didn't automount:
```
root@edison:~# mount | grep boot
systemd-1 on /boot type autofs (rw,relatime,fd=28,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
root@edison:~#

```
I don't know why it has to be like this.

Also do you know how to configure external RTC on boot, I configure `CONFIG_RTC_DRV_RV3028 =m`,but i still need `echo rv3028 0x52 > /sys/class/i2c-adapter/i2c-1/new_device`,i want to run it automatically at boot.

Also i'm using QMI interface with a LE910 modem. This creates WWAN0 interface on the Edison.but my "wwan0" interface became "wwp0s20u7c2i12":
`qmi_wwan 1-1.3:1.2 wwp0s20u7c2i12: renamed from wwan0`
This didn't happen on Edison.
Can I reset to default name wwan0?

shawnhsoia97, 2019-11-28:
Hi @htot ,i want to use sysconfig and init.d  to set ip,but i not found that.
How can i find that?

htot, 2019-11-28:
1) we are using systemd to start up services
2) we use connmanctl to setup the network

Alternatively you could drop a .network file and use systemd-networkd to configure the network. Possibly i.c.w. bridgectl if you need a bridge. See [systemd-networkd](https://www.freedesktop.org/software/systemd/man/systemd.network.html) Example 1. Static network configuration. 

I found systemd-networkd does not automatically handle network changes as connman does, but for a static network if might be good enough. Currently Edison can get a network via usb, wifi, bluetooth and ethernet adapters simultaneously. And can be a wifi access point. That's not easy to do with systemd-networkd.

There used to be a usb0.network file like [this](https://github.com/edison-fw/meta-intel-edison/blob/sumo32/meta-intel-edison-distro/recipes-core/systemd/systemd_%25.bbappend).

shawnhsoia97, 2019-11-28:
thank you,let me try

also i still haven't resolved the problem of the boot folder, i don't know why there is no automatic mount.
i can't even mount from /dev/mmcblk0p9.
Do you have any solution? thank

htot, 2019-11-28:
How are you trying to mount exactly?

Normally you receive an ip address from the dhcp server. Have you tried connmanctl? Try 'help'

shawnhsoia97, 2019-11-29:
It should automount when me enter
```
root@edison:~# cat /etc/fstab | grep boot
/dev/mmcblk0p9     /boot       auto    noauto,x-systemd.automount,nosuid,nodev,noatime,discard     1   1
```
but it no
```
root@edison:~# mount | grep boot
systemd-1 on /boot type autofs (rw,relatime,fd=28,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
root@edison:~#
```
so i tried to mount manually
`mount -t noauto,x-systemd.automount,nosuid,nodev,noatime,discard /dev/mmcblk0p9 /boot`
but it obviously failed
```
root@edison:/dev# cat /proc/partitions |grep 179
 179        0    3817472 mmcblk0
 179        1       2048 mmcblk0p1
 179        2       1024 mmcblk0p2
 179        3       2048 mmcblk0p3
 179        4       1024 mmcblk0p4
 179        5       1024 mmcblk0p5
 179        6      24576 mmcblk0p6
 179        7      32768 mmcblk0p7
root@edison:/dev# ls -al /dev/block/179*
lrwxrwxrwx 1 root root 10 Nov  1 02:05 /dev/block/179:0 -> ../mmcblk0
lrwxrwxrwx 1 root root 12 Nov  1 02:05 /dev/block/179:1 -> ../mmcblk0p1
lrwxrwxrwx 1 root root 15 Nov  1 02:05 /dev/block/179:16 -> ../mmcblk0boot1
lrwxrwxrwx 1 root root 12 Nov  1 02:05 /dev/block/179:2 -> ../mmcblk0p2
lrwxrwxrwx 1 root root 12 Nov  1 02:05 /dev/block/179:3 -> ../mmcblk0p3
lrwxrwxrwx 1 root root 12 Nov  1 02:05 /dev/block/179:4 -> ../mmcblk0p4
lrwxrwxrwx 1 root root 12 Nov  1 02:05 /dev/block/179:5 -> ../mmcblk0p5
lrwxrwxrwx 1 root root 12 Nov  1 02:05 /dev/block/179:6 -> ../mmcblk0p6
lrwxrwxrwx 1 root root 12 Nov  1 02:05 /dev/block/179:7 -> ../mmcblk0p7
lrwxrwxrwx 1 root root 15 Nov  1 02:05 /dev/block/179:8 -> ../mmcblk0boot0
```
I don't know how to mount the boot properly.
Do you have any ideas?

shawnhsoia97, 2019-11-29:
Hi @htot 
i try rebuild thud
but whan i building thud,that shows
```
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: linux-yocto-5.2.0-r0 do_fetch: Failed to fetch URL git://github.com/edison-fw/linux.git;protocol=https;branch=eds-acpi-5.2.0, attempting MIRRORS if available
ERROR: linux-yocto-5.2.0-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"; export SSH_AGENT_PID="1519"; export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh"; export PATH="/home/shawn/my_Edison_Workspace/thud/out/linux64/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/home/shawn/my_Edison_Workspace/thud/out/linux64/poky/scripts:/home/shawn/my_Edison_Workspace/thud/out/linux64/build/tmp/work/edison-poky-linux/linux-yocto/5.2.0-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux:/home/shawn/my_Edison_Workspace/thud/out/linux64/build/tmp/work/edison-poky-linux/linux-yocto/5.2.0-r0/recipe-sysroot/usr/bin/crossscripts:/home/shawn/my_Edison_Workspace/thud/out/linux64/build/tmp/work/edison-poky-linux/linux-yocto/5.2.0-r0/recipe-sysroot-native/usr/sbin:/home/shawn/my_Edison_Workspace/thud/out/linux64/build/tmp/work/edison-poky-linux/linux-yocto/5.2.0-r0/recipe-sysroot-native/usr/bin:/home/shawn/my_Edison_Workspace/thud/out/linux64/build/tmp/work/edison-poky-linux/linux-yocto/5.2.0-r0/recipe-sysroot-native/sbin:/home/shawn/my_Edison_Workspace/thud/out/linux64/build/tmp/work/edison-poky-linux/linux-yocto/5.2.0-r0/recipe-sysroot-native/bin:/home/shawn/my_Edison_Workspace/thud/out/linux64/poky/bitbake/bin:/home/shawn/my_Edison_Workspace/thud/out/linux64/build/tmp/hosttools"; export HOME="/home/shawn"; LANG=C git -c core.fsyncobjectfiles=0 clone --bare --mirror https://github.com/edison-fw/linux.git /home/shawn/my_Edison_Workspace/thud/bbcache/downloads/git2/github.com.edison-fw.linux.git --progress failed with exit code 128, output:
Cloning into bare repository '/home/shawn/my_Edison_Workspace/thud/bbcache/downloads/git2/github.com.edison-fw.linux.git'...
remote: Enumerating objects: 7008798, done.        
error: RPC failed; curl 56 GnuTLS recv error (-54): Error in the pull function.
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

ERROR: linux-yocto-5.2.0-r0 do_fetch: Fetcher failure for URL: 'git://github.com/edison-fw/linux.git;protocol=https;branch=eds-acpi-5.2.0'. Unable to fetch URL from any source.
ERROR: linux-yocto-5.2.0-r0 do_fetch: Function failed: base_do_fetch
ERROR: Logfile of failure stored in: /home/shawn/my_Edison_Workspace/thud/out/linux64/build/tmp/work/edison-poky-linux/linux-yocto/5.2.0-r0/temp/log.do_fetch.11265
ERROR: Task (/home/shawn/my_Edison_Workspace/thud/meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/linux-yocto_5.2.0.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 746 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/shawn/my_Edison_Workspace/thud/meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/linux-yocto_5.2.0.bb:do_fetch
Summary: There was 1 WARNING message shown.
Summary: There were 3 ERROR messages shown, returning a non-zero exit code.
Makefile:53: recipe for target 'edison-image' failed
make: *** [edison-image] Error 1
```
how can solve it?

htot, 2019-11-29:
May be one thing at a time:
- to enter the dir `ls /boot`
- to mount `mount /dev/mmcblk0p9 /boot`
(to mount with a type: -t, to mount with options -o)
- if your boot partition is broken, on a unmounted device: `fsck /dev/mmcblk0p9`

shawnhsoia97, 2019-12-02:
Hi @htot 
> May be one thing at a time:
> 
> * to enter the dir `ls /boot`
> * to mount `mount /dev/mmcblk0p9 /boot`
>   (to mount with a type: -t, to mount with options -o)
> * if your boot partition is broken, on a unmounted device: `fsck /dev/mmcblk0p9`

I tried to do it your way,but i still  run into problems:
```
root@edison:/# ls boot/
[ 1272.977797] FAT-fs (mmcblk0p9): error, fat_get_cluster: invalid cluster chain (i_pos 1)
[ 1272.986022] FAT-fs (mmcblk0p9): Filesystem has been set read-only
ls: cannot access 'boot/': No such device
root@edison:/# mount /dev/mmcblk0p9 /boot
[ 1294.417946] FAT-fs (mmcblk0p9): error, fat_get_cluster: invalid cluster chain (i_pos 1)
[ 1294.426178] FAT-fs (mmcblk0p9): Filesystem has been set read-only
mount: /boot: can't read superblock on /dev/mmcblk0p9.
root@edison:/# fsck /dev/mmcblk0p9
fsck from util-linux 2.32.1
CP437: Invalid argument
fsck.fat 4.1 (2017-01-24)
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
1) Remove dirty bit
2) No action
? 2
There are differences between boot sector and its backup.
This is mostly harmless. Differences: (offset:original/backup)
  65:01/00, 440:28/00, 441:d6/00, 442:1f/00, 443:3d/00, 448:01/00, 449:20/00
  , 450:0b/00, 451:03/00, 452:d0/00, 453:ff/00, 455:08/00, 459:f8/00
  , 460:17/00
1) Copy original to backup
2) Copy backup to original
3) No action
? 3
Both FATs appear to be corrupt. Giving up.

```
I don't know what i need to do,can you tell me how to solve it? thank you

htot, 2019-12-02:
No, the file system is damaged (not uncommon on FAT). It needs to be fixed. Remove dirty bit. And possibly copy original to backup.

shawnhsoia97, 2019-12-02:
I tried your method(Remove dirty bit. And possibly copy original to backup.),but still not solve.
```
root@edison:~# fsck /dev/mmcblk0p9
fsck from util-linux 2.32.1
CP437: Invalid argument
fsck.fat 4.1 (2017-01-24)
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
1) Remove dirty bit
2) No action
? 1
There are differences between boot sector and its backup.
This is mostly harmless. Differences: (offset:original/backup)
  440:28/00, 441:d6/00, 442:1f/00, 443:3d/00, 448:01/00, 449:20/00, 450:0b/00
  , 451:03/00, 452:d0/00, 453:ff/00, 455:08/00, 459:f8/00, 460:17/00
1) Copy original to backup
2) Copy backup to original
3) No action
? 1
Both FATs appear to be corrupt. Giving up.
root@edison:~# ls /boot
[  118.619873] FAT-fs (mmcblk0p9): error, fat_get_cluster: invalid cluster chain (i_pos 1)
[  118.628122] FAT-fs (mmcblk0p9): Filesystem has been set read-only
ls: cannot open directory '/boot': No such device

```

htot, 2019-12-02:
Maybe you need to try to flashall your device again, this will rewrite the boot partition. But I actually use [Flash Tool Lite](https://htot.github.io/meta-intel-edison/2.0-Building-and-installing-the-image.html#flash-tool-lite): 

Or you could try manually using this line from flashall.sh:
https://github.com/edison-fw/meta-intel-edison/blob/7499c1d689aec15c1a6ed65980b9594a443e8016/utils/flash/flashall.sh#L192

shawnhsoia97, 2019-12-02:
> Maybe you need to try to flashall your device again, this will rewrite the boot partition. But I actually use [Flash Tool [...]

I have tried flashall many time,but still not solve.
I think it's been a problem since the build image,but i don't know where the problem.

htot, 2019-12-02:
This is maybe what is going on:
- If you are booting from the emmc (i.e what was written by flashall) the boot partition is on /dev/mmcblk0p7

But it seems p7 is not automatically mounted because /dev/mmcblk0p9 tries to be mounted on /boot. We might consider this to be a bug.

If you want to see contents then temporarily mount on /mnt: `mount /dev/mmcblk0p7 /mnt`

- If your are booting from usb or sdhc  the boot partition is on /dev/mmcblk0p9. 

To make this work you need to manually copy the kernel to /boot. But before you can do that you must prepare the partition. This is described [here](https://edison-fw.github.io/meta-intel-edison/2.2-Building-and-installing-the-kernel.html#installing-the-kernel), in short you need to format the partition first.

So: even when booting from usb or sdhc, the kernel must be located on the emmc for U-Boot to find it. We leave the kernel on p7 untouched so you can always boot the edison, even if you installed a broken kernel on p9, or a broken rootfs on the usb/sdhc.

shawnhsoia97, 2019-12-03:
thank you

Also I am now rebuilding a new edison-image(branch master),but it made a mistake at automake-1.14,i don't know how to solve it.
error contents:
```
NOTE: Preparing runqueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: Function failed: do_compile (log file is located at /home/shawn/my_Edison_Workspace/master/out/linux64/build/tmp/work/x86_64-linux/automake-native/1.14.1-r0/temp/log.do_compile.28176)
ERROR: Logfile of failure stored in: /home/shawn/my_Edison_Workspace/master/out/linux64/build/tmp/work/x86_64-linux/automake-native/1.14.1-r0/temp/log.do_compile.28176
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4
| : && /bin/mkdir -p doc && { PATH='/home/shawn/my_Edison_Workspace/master/out/linux64/build/tmp/work/x86_64-linux/automake-native/1.14.1-r0/build/t/wrap':$PATH && export PATH; } && /usr/bin/perl /home/shawn/my_Edison_Workspace/master/out/linux64/build/tmp/work/x86_64-linux/automake-native/1.14.1-r0/automake-1.14.1/doc/help2man --output=doc/automake-1.14.1 automake-1.14
| help2man: can't get `--help' info from automake-1.14
| Try `--no-discard-stderr' if option outputs to stderr
| Makefile:3707: recipe for target 'doc/automake-1.14.1' failed
| make: *** [doc/automake-1.14.1] Error 255
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/shawn/my_Edison_Workspace/master/out/linux64/build/tmp/work/x86_64-linux/automake-native/1.14.1-r0/temp/log.do_compile.28176)
ERROR: Task 1472 (virtual:native:/home/shawn/my_Edison_Workspace/master/out/linux64/poky/meta/recipes-devtools/automake/automake_1.14.1.bb, do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 66 tasks of which 62 didn't need to be rerun and 1 failed.
Waiting for 0 running tasks to finish:

Summary: 1 task failed:
  virtual:native:/home/shawn/my_Edison_Workspace/master/out/linux64/poky/meta/recipes-devtools/automake/automake_1.14.1.bb, do_compile
Summary: There was 1 WARNING message shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
Makefile:49: recipe for target 'edison-image' failed
make: *** [edison-image] Error 1
```

shawnhsoia97, 2019-12-03:
I modify /etc/fstab:`/dev/mmcblk0p7     /boot       auto    noauto,x-systemd.automount,nosuid,nodev,noatime,discard     1`

something appeared on boot:
[image]
so now what part do i want to modify so that i can mount the boot after flash?

htot, 2019-12-03:
https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-distro/recipes-core/base-files/base-files/fstab

htot, 2019-12-03:
You got to get your stuff together man. You are building automake-1.14. If you go [here](https://layers.openembedded.org/layerindex/branch/dizzy/recipes/?q=automake) you can see 1.14 is in Yocto Dizzy. Thud has [this](https://layers.openembedded.org/layerindex/branch/thud/recipes/?q=automake) (1.16.1).

shawnhsoia97, 2019-12-04:
thank you for help me

I have another question,I want to pre-declare the I2C devices which live on this bus. This is done with an array of struct i2c_board_info which is registered by calling i2c_register_board_info().

but i don't know where i want to call i2c_register_board_info(),i found it seems to be in /arch/arm/mach-xxxx/board_xxxx.c,but i still don't know the exact location ,so do you know where i want to edit?

i refer [http://lxr.linux.no/linux+v2.6.34/Documentation/i2c/instantiating-devices](here)

htot, 2019-12-04:
You are looking at a very old kernel documentation. Better use this one: https://github.com/edison-fw/linux/blob/eds-acpi-5.4.0/Documentation/i2c/instantiating-devices.rst

Our preferred method (when the kernel driver handles the i2c device) is to define in an acpi table. Examples are here: https://github.com/westeri/meta-acpi/tree/master/recipes-bsp/acpi-tables/samples/edison

shawnhsoia97, 2019-12-05:
Hi @htot ,I follow your suggestion to use acpi table.I created two files in `my_Edison_Workspace/thud/meta-acpi/recipes-bsp/acpi-tables/samples/edison`

content:
i2c3028.asl
```
/*
 * Intel Edison
 *
 * Copyright (C) 2017, Intel Corporation
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights
 * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in
 * all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
 * THE SOFTWARE.
 */
DefinitionBlock ("i2c3028.aml", "SSDT", 5, "", "i2c", 1)
{
    #include "i2c3028.asli"
}
```
i2c3028.asli
```
/*
 * Intel Edison
 *
 * Provides GPIO expanders and names for all GPIO pins.
 *
 * Copyright (C) 2017, Intel Corporation
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights
 * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in
 * all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
 * THE SOFTWARE.
 *
 * Include this file from an ASL providing SSDT DefinitionBlock.
 */

External (\_SB_.PCI0.I2C1, DeviceObj)

Scope (\_SB.PCI0.I2C1)
    {
        Device (PMON)
        {
            Name (_HID, "PRP0001")
            Name (_DDN, "call i2c register board")
            Name (_CRS, ResourceTemplate () {
                I2cSerialBus (
                    0x52,                   // Bus address
                    ControllerInitiated,    // Don't care
                    400000,                 // Fast mode (400 kHz)
                    AddressingMode7Bit,     // 7-bit addressing
                    "\\_SB.PCI0.I2C1",      // I2C host controller
                    0                       // Must be 0
                )
            })

            Name (_DSD, Package () {
                ToUUID("daffd814-6eba-4d8c-8a91-bc9bbf4aa301"),
                Package () {
                    Package () {"compatible", "rtc,rv3028"},
                }
            })
        }
    }

```
(my i2c bus is 1,and Bus address 0x52)
then `$ make edison-image`

But my test results were unsuccessful,is there step I am missing?

htot, 2019-12-05:
If all is well a i2c3028.aml is built. But it is not automatically built and loaded into the kernel yet. Currently we are loading some tables into the kernel from the initramfs. But while you are debugging, you can build outside Yocto and load from the edison command line. Read more here: https://edison-fw.github.io/meta-intel-edison/1.3-ACPI-or-not.html#acpi-tables on building and in particular on "Run time loading through configfs".

The tables that get loaded automatically are done from here: https://github.com/westeri/meta-acpi/blob/master/recipes-bsp/acpi-tables/files/acpi-tables-load (currently all that get built).
The ones that are built are done from here: https://github.com/westeri/meta-acpi/blob/master/recipes-bsp/acpi-tables/acpi-tables.bb, so controlled from a variable ACPI_TABLES, which is set here: https://github.com/edison-fw/meta-intel-edison/blob/7499c1d689aec15c1a6ed65980b9594a443e8016/meta-intel-edison-bsp/conf/machine/edison.conf#L32

shawnhsoia97, 2019-12-05:
also I want to change root device from mmcblk0p8 to mmcblk0p10(because the mmcblk0p8 memory capacity is too small)

but I only found how to change U-boot rootFS from [https://github.com/edison-fw/meta-intel-edison/blob/b2e865a29109cb7b034c436608945d63e5833cff/meta-intel-edison-bsp/recipes-bsp/u-boot/files/edison.env](here)

So how can I solve it?thank

htot, 2019-12-05:
On my edison:
```
root@edison:/boot# lsblk
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk1      179:0    0  1.9G  0 disk /
mmcblk0      179:8    0  3.7G  0 disk 
|-mmcblk0p1  179:9    0    2M  0 part 
|-mmcblk0p2  179:10   0    1M  0 part 
|-mmcblk0p3  179:11   0    2M  0 part 
|-mmcblk0p4  179:12   0    1M  0 part 
|-mmcblk0p5  179:13   0    1M  0 part /factory
|-mmcblk0p6  179:14   0   24M  0 part 
|-mmcblk0p7  179:15   0   32M  0 part 
|-mmcblk0p8  259:0    0  1.5G  0 part 
|-mmcblk0p9  259:1    0  768M  0 part /boot
`-mmcblk0p10 259:2    0  1.3G  0 part /home
mmcblk0boot0 179:16   0    4M  1 disk 
mmcblk0boot1 179:24   0    4M  1 disk 
```
This was when booting from the sdhc, so root is `mmcblk1`.

As you can see /home is on `mmcblk0p10`.

To boot the sd card or the usb copy the kernel as described here: https://edison-fw.github.io/meta-intel-edison/2.2-Building-and-installing-the-kernel.html#installing-the-kernel
and the rootfs as here: https://edison-fw.github.io/meta-intel-edison/2.1-Building-and-installing-the-rootfs.html#installing-the-rootfs-on-a-sd-card

Then from the U-Boot cli do_edsboot or do_usbboot. Or from a running edison `reboot sdhc` or `reboot usb`.
If you do this you can also manually resize partitions on emmc.

If you still want to go ahead and change rootfs:
```
root@edison:~# fw_printenv | grep bootargs_      
...
bootargs_edsboot=debugshell=10 console=ttyS2,115200n8 earlyprintk=ttyS2,115200n8,keep root=/dev/mmcblk1 rootfstype=ext4 systemd.unit=multi-user.target hardware_id=00
bootargs_usbboot=debugshell=10 tty1 console=ttyS2,115200n8 root=/dev/sda rootfstype=ext4 systemd.unit=multi-user.target hardware_id=00
...
do_bootargs_rootfs=setenv bootargs_rootfs root=/dev/mmcblk0p8 rootfstype=ext4
...
```
It's the `root` keyword that tells the initramfs where to look for the rootfs.

shawnhsoia97, 2019-12-06:
Hi @htot ,I sovled acpi_table as you suggested.Thank you.

Also if I don't want to boot with sdhc or usb,can it be resize partitions on mmcblk0?
```
root@edison:/boot# lsblk
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0      179:0    0  3.7G  0 disk
|-mmcblk0p1  179:1    0    2M  0 part
|-mmcblk0p2  179:2    0    1M  0 part
|-mmcblk0p3  179:3    0    2M  0 part
|-mmcblk0p4  179:4    0    1M  0 part
|-mmcblk0p5  179:5    0    1M  0 part /factory
|-mmcblk0p6  179:6    0   24M  0 part
|-mmcblk0p7  179:7    0   32M  0 part /boot
|-mmcblk0p8  259:0    0  1.5G  0 part /
|-mmcblk0p9  259:1    0  768M  0 part
`-mmcblk0p10 259:2    0  1.3G  0 part /home
mmcblk0boot0 179:8    0    4M  1 disk
mmcblk0boot1 179:16   0    4M  1 disk
root@edison:/boot# df -h
Filesystem       Size  Used Avail Use% Mounted on
none             463M  4.0K  463M   1% /dev
/dev/mmcblk0p8   928M  707M  155M  83% /
tmpfs            470M     0  470M   0% /dev/shm
tmpfs            470M  732K  469M   1% /run
tmpfs            470M     0  470M   0% /sys/fs/cgroup
tmpfs            470M  4.0K  470M   1% /tmp
tmpfs            470M     0  470M   0% /var/volatile
/dev/mmcblk0p5  1003K   19K  913K   3% /factory
/dev/mmcblk0p10  1.3G   72M  1.2G   6% /home
tmpfs             94M     0   94M   0% /run/user/0
/dev/mmcblk0p7    15M   14M  556K  97% /boot
```
Such as changing  the size of mmcblk0p10 to 800M and the size of mmcblk0p08 to 2G.

shawnhsoia97, 2019-12-06:
Also I want to use external rtc1 to set the system time.I have solved the problem of register i2c because your help thank!But I encountered new problems, I still don't  use external rtc1 to set the system time.I found Kernel complains about a non existing rtc device on early boot.
System clock may not set on early boot.

```
root@edison:/sys/class/rtc# dmesg | grep hctosys
[    2.086705] hctosys: unable to open rtc device (rtc1)
root@edison:/sys/class/rtc# dmesg | grep rtc
[    2.045441] rtc_cmos 00:00: registered as rtc0
[    2.045609] rtc_cmos 00:00: no alarms, 242 bytes nvram
[    2.086705] hctosys: unable to open rtc device (rtc1)
[    3.158197] rtc-rv3028 i2c-PRP0001:02: registered as rtc1

```
How can I solve it?Thank!

ochrin, 2019-12-06:
Regarding resize of partition. Here is what you may be able to do.
As far as I know mmcblk0p9 is not used anymore, so you can remove it with:
`parted /dev/mmcblk0 rm 9`
Then you can increase size of mmcblk0p8 by using previous removed space with:
```
growpart /dev/mmcblk0 8
resize2fs /dev/mmcblk0p8
```
I don't know if these utilities are part of Edison. If not you may have to add them.

htot, 2019-12-06:
@shawnhsoia97 Great that you made acpi table work. Maybe you can contribute it to westeri/meta-acpi or else to edison-fw/meta-acpi. Also, we could use some documentation for beginners how to write an acpi-table (under docs/).

I think originally root was just 1GB and the intel firmware provided a U-Boot script that did a one time resize to 1.5G and decreasing p9 (this was the OTA partition which then became too small). But correct me if I'm wrong.

So you can do as @ochrin suggests (or decrease p9 so you can still use it to boot sdhc/usb rootfs) or do it from U-Boot.

htot, 2019-12-06:
@shawnhsoia97 about rtc1, I don't know what is wrong. Possibly something wrong with your acpi table above. Maybe @andy-shev knows?

shawnhsoia97, 2019-12-09:
@ochrin ,thank you.I succeed.but [ database](https://layers.openembedded.org/layerindex/branch/thud/recipes/) not found growpart,so I had to find  an installation file from the internet to install.Is there a way to increase growpart to datebase?

@htot ,of course,my pleasure.But i actually don't know mach about acpi_table,also how to i ask @andy-shev ?

htot, 2019-12-09:
@shawnhsoia97 
> Is there a way to increase growpart to datebase?

Yes you need to write the recipe, publish it somewhere (github) then register your layer with openembedded so they can index it.

shawnhsoia97, 2019-12-10:
@htot ,I get it.Thank you.

Also I have asked apt source list question,you said **By default there are no deb repo's configured. It's not that you can just configure debian or ubuntu repo's, That won't work. You can build you own repo, configure edison to use that.**,so do you mean I can't use  debian repo's,just only use  own repo?
because I tried to  use Debian source list,but that won't work.That show
 ```
Err:8 http://archive.debian.org/debian jessie/main Translation-en
  404  Not Found [IP: 193.62.202.28 80]
```
That is work in other kernel.

htot, 2019-12-10:
> ...so do you mean I can't use debian repo's,just only use own repo?
> because I tried to use Debian source list,but that won't work.That show

Correct. deb is only a package format. There are dependencies between deb depending on lib versions, causing you can not mix packages from stretch and buster, or even between ubuntu and debian. And also not from your own yocto build.

With yocto you build your own distribution, with your own repo with your own deb's.

If you want debian checkout this PR https://github.com/htot/meta-intel-edison/pull/4.
This is about enabling debian stretch build script. Yocto build just a minimal set of kernel, modules, scripts. And that is then incorporated into a debootstrap image. So this is a real Debian image, not especially optimized for Edison.

shawnhsoia97, 2019-12-11:
OK,I get it.thank you.

Also how do I set my root password or add a new user when building an image?

htot, 2019-12-11:
I don't know. You need to check the Yoco manual for that.

lybtongji, 2019-12-11:
> 
> 
> OK,I get it.thank you.
> 
> Also how do I set my root password or add a new user when building an image?

Maybe you can write a script to modify the `$EDISON_ROOT/etc/passwd` and `$EDISON_ROOT/etc/shadow` with `passwd` command

htot, 2019-12-11:
https://www.yoctoproject.org/docs/current/ref-manual/ref-manual.html#ref-classes-extrausers

shawnhsoia97, 2019-12-12:
> https://www.yoctoproject.org/docs/current/ref-manual/ref-manual.html#ref-classes-extrausers

I follow this method to modify  extrausers.bbclass.
content:

but  that not sets the root password and that not adds users .Am I missing any steps besides editing extrausers.bbclass?

Also I want to change hostname,so I follow this [Miscellaneous Changes](https://www.yoctoproject.org/docs/current/ref-manual/ref-manual.html#migration-3.0-miscellaneous-change),that show need edit base-files recipe,but I don't know where to change hostname in base-files recipe,do you have any ideas?

htot, 2019-12-12:
Don't modify extrausers.bbclass. Follow the manual: 6.34. extrausers.bbclass¶

> Here is an example that uses this class in an image recipe:
     inherit extrausers
     EXTRA_USERS_PARAMS = "\
         useradd -p '' tester; \
         groupadd developers; \
         userdel nobody; \
         groupdel -g video; \
         groupmod -g 1020 developers; \
         usermod -s /bin/sh tester; \
         "
       
>Here is an example that adds two users named "tester-jim" and "tester-sue" and assigns passwords:
     inherit extrausers
     EXTRA_USERS_PARAMS = "\
         useradd -P tester01 tester-jim; \
         useradd -P tester01 tester-sue; \
         "
        
> Finally, here is an example that sets the root password to "1876*18":
     inherit extrausers
     EXTRA_USERS_PARAMS = "\
         usermod -P 1876*18 root; \

andy-shev, 2019-12-13:
> Also I want to use external rtc1 to set the system time.I have solved the problem of register i2c because your help thank!But I encountered new problems, I still don't use external rtc1 to set the [...]

It seems you are using `CONFIG_RTC_HCTOSYS_DEVICE=rtc1` when rtc1 is connected to I²C bus and being initialized later (ordering problems). To mitigate this you have to try built-in the driver and supply ACPI table in *initrd*. I'm not sure it will fix it, but that's most what we can do without patching a kernel code.

shawnhsoia97, 2019-12-16:
Hi @andy-shev,I have tried  built-in the driver(such as` CONFIG_I2C_CHARDEV=y`、`CONFIG_RTC_DRV_RV3028=y`),but no effect,and so is there any driver I can try?Also I don't  know how to  supply ACPI table in initrd,do you have any document?

htot, 2019-12-16:
@shawnhsoia97 I load tables late, either from the initramfs or while debugging from the command line. Documentation is here: https://htot.github.io/meta-intel-edison/1.3-ACPI-or-not.html

shawnhsoia97, 2019-12-16:
Hi @htot ,awesome,I get it,thank you
I have a new question,I want to add [softether](https://gitlab.com/meta-layers/meta-resy/tree/master/recipes-connectivity/softether) in edison-image,and so I try to clone that recipe to `/poky/mate/`,but not effect,that show
```
 NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'softether' (but /home/shawn/my_Edison_Workspace/thud/meta-intel-edison/meta-intel-edison-distro/recipes-core/images/edison-image.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'softether' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['softether']
ERROR: Required build target 'edison-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['edison-image', 'softether']
```
And so do you know how to add a recipe that does not belong to OpenEmbedded?

Also Howto generate a Yocto image with .utf8 by default,My current edison-image.bb setup is this:
```
GLIBC_GENERATE_LOCALES = "en_GB.UTF-8 en_US.UTF-8"	    
IMAGE_LINGUAS = "de-de fr-fr en-gb en-us"
```

andy-shev, 2019-12-16:
> Hi @andy-shev,I have tried built-in the driver(such as` CONFIG_I2C_CHARDEV=y`、`CONFIG_RTC_DRV_RV3028=y`),but no effect,and so is there any driver I can try?Also I don't know how to supply ACPI table in initrd,do you have any document?
Read this documentation material
https://www.kernel.org/doc/html/latest/admin-guide/acpi/ssdt-overlays.html

htot, 2019-12-16:
> And so do you know how to add a recipe that does not belong to OpenEmbedded?
 
You either copy the recipe directory into meta-intel-edison/meta-intel-edison-distro, or add the whole layer (as you do) and add the layer location to $yocto_conf_dir/bblayers.conf  (see setup.sh)

> Also Howto generate a Yocto image with .utf8 by default,My current edison-image.bb setup is this:

Don't know. Maybe in the Yocto mega manual?

shawnhsoia97, 2019-12-18:
Hi @andy-shev ,I load ACPI SSDTs from configfs according to that documentation,I found that the `/config/acpi/table/ `  already the SSDT aml code and the aml attribute.
```
root@bovia-deb:/sys/kernel/config/acpi/table# ls
arduino  i2c3028  leds  spidev
root@bovia-deb:/sys/kernel/config/acpi/table# ls i2c3028/
aml  asl_compiler_id  asl_compiler_revision  length  oem_id  oem_revision  oem_table_id  revision  signature
root@bovia-deb:/sys/kernel/config/acpi/table#
```
And so unfortunately this issue is not resolved.
```
root@bovia-deb:/sys/kernel/config/acpi/table# dmesg | grep rtc
[    2.043898] rtc_cmos 00:00: registered as rtc0
[    2.044020] rtc_cmos 00:00: no alarms, 242 bytes nvram
[    2.085949] hctosys: unable to open rtc device (rtc1)
[    3.075209] rtc-rv3028 i2c-PRP0001:02: registered as rtc1

```

shawnhsoia97, 2019-12-18:
Hi @htot ,awesome, ,I succeed ,thank you for help me.

andy-shev, 2019-12-20:
> Hi @andy-shev ,I load ACPI SSDTs from configfs according to that documentation,I found that the `/config/acpi/table/ ` already the SSDT aml code and the aml attribute.

No, configfs is not what I told you about. You need to attach it to initramfs (initrd).

htot, 2019-12-28:
@shawnhsoia97 Maybe we can close this long thread and you can start a new one per particular issue?

That way it will also be easier for other people to find an answer to their problems.

[1 more reply]

closed 2019-12-30 by shawnhsoia97