edison-fw/meta-intel-edison #78: arduino sketch does not work in branch thud

xlla, 2019-11-22:
I am use arduino library, 1.6.7+1.0

after compile and upload sketch, it was running but not work as expect.

should I upgrade sysroot in packages/Intel/hardware/i686/1.6.7+1.0, and patch some file to work with chardev ?

since there are tons of arduino libraries, if we can still support arduino framework, then everything is easy.

htot, 2019-11-22:
I don't know. What sketch is problematic?

xlla, 2019-11-23:
sorry, I am messed up.
It even can't upload sketch into thud version edison, since there is none tty serial port under list. 
[image]
and no sketch folder under root

root@edison:~# ls /sketch
ls: cannot access '/sketch': No such file or directory

compare to firmware 160606, poky 1.7.3 board
[image]
root@localhost:~# ls /sketch/
sketch.elf  sketch.elf.old
root@localhost:~# ps |grep sketch
  207 root      1728 S    /opt/edison/sketch_reset -i 207 -o 215 -s /opt/ediso
  360 root      2660 S    grep sketch

I found that the [arduino framework lib](https://github.com/intel/corelibs-edison) need be patched to use chardev interface to manipulate mux/pullup/gpio,  this is beyond my ability :(

I think I should give up arduino framework and looking for some python/nodejs libs as substitute.

htot, 2019-11-23:
I never sketches. But are you sure the problem is with corelibs-edison?

I think we are probably missing a server on edison, f.i. [clloader](https://github.com/edison-fw/meta-intel-edison/blob/084f87da657485f8efea8a95d211b1b518f9cc17/meta-intel-edison-distro/recipes-core/images/edison-image.bb#L61).

I removed this at a certain point as it didn't want to build, but it may be easier to fix than it seems. There are other servers removed too, compare thud to original.

htot, 2019-11-24:
I just added a [patch](https://github.com/htot/meta-intel-edison/commit/ac602ea00a1a13a97113073e39052b8dbf7d0e02) to my warrior branch that fixes `clloader`. Would that restore the connection to the Edison for you?

xlla, 2020-02-16:
I am glad to heard that the Arduino sketches might be got support.
after migrate to branch warrior,  many hours to fetch sources and compile packages, flash firmware, and now clloader and /sketch was back.

root@edison:~# uname -a
Linux edison 5.4.0-edison-acpi-standard #1 SMP Sat Feb 15 19:22:41 UTC 2020 i686 i686 i386 GNU/Linux
root@edison:~# ls /sketch/
root@edison:~# ps |grep sketch
 2485 root      3768 S    grep sketch
root@edison:~# ls /opt/edison/
clloader         launcher.sh      sketch_reset     sketch_reset.sh
root@edison:~# systemctl status clloader
● clloader.service - Daemon to handle arduino sketches
   Loaded: loaded (/lib/systemd/system/clloader.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-02-15 17:20:12 UTC; 6min ago
 Main PID: 538 (launcher.sh)
   CGroup: /system.slice/clloader.service
           ├─ 538 /bin/sh /opt/edison/launcher.sh
           └─2710 usleep 200000

Feb 15 17:20:12 edison systemd[1]: Started Daemon to handle arduino sketches.

to restore Arduino IDE's function, I will try to change gadget vid/pid to original 8087:0a99, then manage to  patch script to use libgpiod, and build a new sdk for Arduino library.

I will report result late.

htot, 2020-02-16:
@xlla that would be great!

In the meanwhile I am messing with Zeus and starting to use btrfs as a file system so I need a bit of time to stabilize that before I can test your work.

xlla, 2020-02-16:
@htot after changed vid/pid, it still not present in Arduino IDE's port list.

xlla$ lsusb
Bus 020 Device 018: ID 0403:6001 Future Technology Devices International Limited FT232R USB UART  Serial: A903BX28
Bus 020 Device 022: ID 8087:0a9e Intel Corporation USBArmory Gadget  Serial: 0123456789abcdef
xlla$ ls /dev/cu. (press Tab)
cu.usbserial-A903BX28

no tty.usbmodem-xxx appeared.

compare to official firmware, 
xlla$ lsusb
Bus 020 Device 015: ID 0403:6001 Future Technology Devices International Limited FT232R USB UART  Serial: A402EXFU
Bus 020 Device 017: ID 8087:0a9e Intel Corporation Edison  Serial: c2xxxxxxxxxxxxxx
xlla$ ls /dev/cu. (press Tab)
cu.usbserial-A402EXFU
cu.usbmodem14213

I don't know wherever should be modified to let my os recognize the usb gadget as original Edison board's.

current usb Name was "Intel Corporation USBArmory Gadget" instead of "Intel Corporation Edison".

xlla, 2020-02-16:
@andy-shev , thanks for your hard work to bring new version of u-boot to us.
when I look at boot log, I found that kernel boot args was not appeared.

U-Boot 2020.01 (Feb 15 2020 - 18:23:23 +0000)

CPU:   Genuine Intel(R) CPU   4000  @  500MHz
DRAM:  980.6 MiB
WDT:   Started with servicing (60s timeout)
MMC:   mmc@ff3fc000: 0, mmc@ff3fa000: 1
Loading Environment from MMC... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Hit any key to stop autoboot:  0
Target:blank
Partitioning already done...
Flashing already done...
11967552 bytes read in 516 ms (22.1 MiB/s)
Valid Boot Flag
Setup Size = 0x00004000
Magic signature found
Using boot protocol version 2.0d
Linux kernel version 5.4.0-edison-acpi-standard (oe-user@oe-host) #1 SMP Sat Feb 15 19:22:41 UTC 2020
Building b
          Starting version 241-214-gd1cc09a+
Kernel with acpi enabled detected
Loading acpi tables

this is log of u-boot 2019.04

U-Boot 2019.04 (Oct 28 2019 - 07:55:21 +0000)
CPU:   Genuine Intel(R) CPU   4000  @  500MHz
DRAM:  980.6 MiB
MMC:   mmc@ff3fc000: 0, mmc@ff3fa000: 1
Loading Environment from MMC... OK
In:    serial
Out:   serial
Err:   serial
Saving Environment to MMC... Writing to redundant MMC(0)... OK
Saving Environment to MMC... Writing to MMC(0)... OK
Net:   No ethernet found.

Kernel command line: "root=/dev/mmcblk0p8 rootfstype=ext4 console=ttyS2,115200n8 earlyprintk=ttyS2,115200n8,keep loglevel=4 g_multi.ethernet_config=cdc systemd.unit=multi-user.target hardware_id=00 g_multi.iSerialNumber=8xxxxxxxxxxxxxxxxxc g_multi.dev_addr=02:00:86:0a:57:7c platform_mrfld_audio.audio_codec=dummy"

htot, 2020-02-16:
Yeah, the whole gadget subsystem has been revised. The file configuring it now is here:
https://github.com/edison-fw/meta-intel-edison/blob/warrior/meta-intel-edison-bsp/recipes-support/gadget/files/conf-gadget.sh

xlla, 2020-02-16:
I see it, 
in original firmware, 
root@localhost:~# cat /etc/modprobe.d/g_multi.conf 
options g_multi file=/dev/mmcblk0p9 stall=0 idVendor=0x8087 idProduct=0x0A9E iProduct=Edison iManufacturer=Intel

in warrior
root@edison:~# cat /etc/modprobe.d/g_multi.conf
cat: /etc/modprobe.d/g_multi.conf: No such file or directory

andy-shev, 2020-02-16:
> @andy-shev , thanks for your hard work to bring new version of u-boot to us.
> when I look at boot log, I found that kernel boot args was not appeared.
> ...
> Kernel command line: "root=/dev/mmcblk0p8 rootfstype=ext4 console=ttyS2,115200n8 earlyprintk=ttyS2,115200n8,keep loglevel=4 g_multi.ethernet_config=cdc systemd.unit=multi-user.target hardware_id=00 g_multi.iSerialNumber=8xxxxxxxxxxxxxxxxxc g_multi.dev_addr=02:00:86:0a:57:7c platform_mrfld_audio.audio_codec=dummy"

I didn't quite get this. do you mean that in newer U-Boot you can't boot it because there is no kernel command line? Or it simple didn't appear on the console and everything is fine when you do `cat /proc/cmdline` in the shell?

htot, 2020-02-16:
@xlla AFAIK these lines `options g_multi file=/dev/mmcblk0p9 stall=0 idVendor=0x8087 idProduct=0x0A9E iProduct=Edison iManufacturer=Intel` and `g_multi.dev_addr=02:00:86:0a:57:7c` are not used at all any more.

As I said gadget vid/pid is set in [conf-gadget.sh](https://github.com/edison-fw/meta-intel-edison/blob/warrior/meta-intel-edison-bsp/recipes-support/gadget/files/conf-gadget.sh)

xlla, 2020-02-17:
@andy-shev yes, It dit boot, just simple did not appear on console.
  I can see kernel command line when I execute 'cat /proc/cmdline'.  I learned it!

@htot I get confused about gadget settings.
 in conf-gadget.sh,
echo "${DEV_ETH_ADDR}" > functions/eem.usb0/dev_addr
echo "0123456789abcdef" > strings/0x409/serialnumber

in kernel command line, 
g_multi.iSerialNumber=8452e525252525252525252
g_multi.dev_addr=02:00:86:0a:57:7c

htot, 2020-02-17:
```
echo 0x1d6b > idVendor # Linux Foundation
echo 0x0104 > idProduct # Multifunction Composite Gadget
```

xlla, 2020-02-17:
reference to https://software.intel.com/en-us/installing-drivers-for-intel-edison-board-with-windows
we need "Composite Device Class (CDC) for programming the board via the Arduino* IDE"
I think it need acm instead of gser to work with Arduino IDE, but I don't know how to configure it correctly.

andy-shev, 2020-02-17:
> @andy-shev yes, It dit boot, just simple did not appear on console.
> I can see kernel command line when I execute 'cat /proc/cmdline'. I learned it!

It's probably U-Boot configuration or U-Boot itself change... However, I don't see any changes in the code for this.

xlla, 2020-02-17:
If I change gser to acm, then everything is disappear.
mkdir functions/acm.usb0                          
ln -s functions/acm.usb0 configs/c.1/                              

so I restore them to gser.
I can't get my mac osx to recognize this gadget even I am using a known pid/vid, it is only in lsusb list, but there is none  in /dev/.

then I connect it to another windows 10 host, no luck.

finally, I connect it to a Linux host, still no interface under /dev; after execute "echo 0x8087 0x0a99 >/sys/bus/usb-serial/drivers/generic/new_id",   a new interface ttyUSB1 appears, and Arduino IDE recognize it as Intel Edison board.
[image]

xlla, 2020-02-17:
now I can move on and try build arduino sdk, but encounter many errors:

`
xlla@u18-1:~/develop/git/out/linux64/build$ bitbake arduino-toolchain
Loading cache: 100% |#############################################################################################| Time: 0:00:01
Loaded 3283 entries from dependency cache.
Parsing recipes: 100% |###########################################################################################| Time: 0:00:02
Parsing of 2213 .bb files complete (2212 cached, 1 parsed). 3284 targets, 301 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "i686-poky-linux"
MACHINE              = "edison"
DISTRO               = "poky-edison"
DISTRO_VERSION       = "2.7.2"
TUNE_FEATURES        = "m32 corei7"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "warrior:4edf4eb9cc86fbd4b51240fae527c1b97a7ace74"
meta-oe              
meta-python          
meta-networking      = "warrior:a24acf94d48d635eca668ea34598c6e5c857e3f8"
meta-intel           = "warrior:29ee4852a05931dcf856670d9d8a3c3077a40fe8"
meta-intel-edison-bsp 
meta-intel-edison-distro 
meta-intel-arduino   
meta-arduino         = "master:d66c23e7c5c51512e9ad292034c8cdb368a65f2e"
meta-acpi            = "eds-5.0.0:abf2a7c62b372110f3fddf7af26debf753742e40"

Initialising tasks: 100% |########################################################################################| Time: 0:00:04
Sstate summary: Wanted 91 Found 0 Missed 91 Current 843 (0% match, 90% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: nativesdk-util-linux-2.32.1-r0 do_package_qa: QA Issue: /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/bin/swaplabel contained in package nativesdk-util-linux-swaplabel requires libuuid.so.1(UUID_1.0)(64bit), but no providers found in RDEPENDS_nativesdk-util-linux-swaplabel? [file-rdeps]
ERROR: nativesdk-util-linux-2.32.1-r0 do_package_qa: QA Issue: /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/bin/swaplabel contained in package nativesdk-util-linux-swaplabel requires libblkid.so.1(BLKID_2.17)(64bit), but no providers found in RDEPENDS_nativesdk-util-linux-swaplabel? [file-rdeps]
ERROR: nativesdk-util-linux-2.32.1-r0 do_package_qa: QA Issue: /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/bin/swaplabel contained in package nativesdk-util-linux-swaplabel requires libblkid.so.1(BLKID_2.15)(64bit), but no providers found in RDEPENDS_nativesdk-util-linux-swaplabel? [file-rdeps]
ERROR: nativesdk-util-linux-2.32.1-r0 do_package_qa: QA Issue: /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/bin/swaplabel contained in package nativesdk-util-linux-swaplabel requires libblkid.so.1()(64bit), but no providers found in RDEPENDS_nativesdk-util-linux-swaplabel? [file-rdeps]
ERROR: nativesdk-util-linux-2.32.1-r0 do_package_qa: QA Issue: /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/bin/swaplabel contained in package nativesdk-util-linux-swaplabel requires libuuid.so.1()(64bit), but no providers found in RDEPENDS_nativesdk-util-linux-swaplabel? [file-rdeps]
ERROR: nativesdk-util-linux-2.32.1-r0 do_package_qa: QA Issue: /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/bin/sfdisk contained in package nativesdk-util-linux-sfdisk requires libsmartcols.so.1(SMARTCOLS_2.25)(64bit), but no providers found in RDEPENDS_nativesdk-util-linux-sfdisk? [file-rdeps]
ERROR: nativesdk-util-linux-2.32.1-r0 do_package_qa: QA Issue: /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/bin/sfdisk contained in package nativesdk-util-linux-sfdisk requires libfdisk.so.1(FDISK_2.28)(64bit), but no providers found in RDEPENDS_nativesdk-util-linux-sfdisk? [file-rdeps]
ERROR: nativesdk-util-linux-2.32.1-r0 do_package_qa: QA Issue: /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/bin/sfdisk contained in package nativesdk-util-linux-sfdisk requires libfdisk.so.1(FDISK_2.29)(64bit), but no providers found in RDEPENDS_nativesdk-util-linux-sfdisk? [file-rdeps]
ERROR: nativesdk-util-linux-2.32.1-r0 do_package_qa: QA Issue: /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/bin/sfdisk contained in package nativesdk-util-linux-sfdisk requires libfdisk.so.1(FDISK_2.30)(64bit), but no providers found in RDEPENDS_nativesdk-util-linux-sfdisk? [file-rdeps]
ERROR: nativesdk-util-linux-2.32.1-r0 do_package_qa: QA Issue: /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/bin/sfdisk contained in package nativesdk-util-linux-sfdisk requires libfdisk.so.1(FDISK_2.27)(64bit), but no providers found in RDEPENDS_nativesdk-util-linux-sfdisk? [file-rdeps]
ERROR: nativesdk-util-linux-2.32.1-r0 do_package_qa: QA Issue: /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/bin/sfdisk contained in package nativesdk-util-linux-sfdisk requires libfdisk.so.1(FDISK_2.31)(64bit), but no providers found in RDEPENDS_nativesdk-util-linux-sfdisk? [file-rdeps]
ERROR: nativesdk-util-linux-2.32.1-r0 do_package_qa: QA Issue: /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/bin/sfdisk contained in package nativesdk-util-linux-sfdisk requires libfdisk.so.1(FDISK_2.26)(64bit), but no providers found in RDEPENDS_nativesdk-util-linux-sfdisk? [file-rdeps]
[... cut ...]
ERROR: Logfile of failure stored in: /home/xlla/develop/git/out/linux64/build/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-util-linux/2.32.1-r0/temp/log.do_package_qa.2392
ERROR: Task (virtual:nativesdk:/home/xlla/develop/git/out/linux64/poky/meta/recipes-core/util-linux/util-linux_2.32.1.bb:do_package_qa) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2161 tasks of which 2148 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:nativesdk:/home/xlla/develop/git/out/linux64/poky/meta/recipes-core/util-linux/util-linux_2.32.1.bb:do_package_qa
Summary: There were 250 ERROR messages shown, returning a non-zero exit code.

`

xlla, 2020-02-17:
I have manual replace content of poky/meta/recipes-core/util-linux/util-linux_2.32.1.bb with http://git.yoctoproject.org/cgit.cgi/poky/plain/meta/recipes-core/util-linux/util-linux_2.32.1.bb?h=warrior&id=bd374216048581bcb3ea54d69998342525cf2da5, and remove each line which contain "OLDHOST", now the errors was gone.

xlla, 2020-02-17:
Errors back again, :(

ERROR: nativesdk-util-linux-2.32.1-r0 do_package: QA Issue: nativesdk-util-linux: Files/directories were installed but not shipped in any package:
  /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/sbin/partx
  /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/sbin/ldattach
  /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/sbin/readprofile
  /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/sbin/resizepart
  /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/sbin/uuidd
  /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/sbin/delpart
  /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/sbin/rfkill
  /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/sbin/addpart
  /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/sbin/rtcwake
  /opt/poky-edison/2.7.2/sysroots/x86_64-pokysdk-linux/usr/sbin/fdformat
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
nativesdk-util-linux: 10 installed and not shipped files. [installed-vs-shipped]
ERROR: nativesdk-util-linux-2.32.1-r0 do_package: Fatal QA errors found, failing task.
ERROR: nativesdk-util-linux-2.32.1-r0 do_package: 
ERROR: nativesdk-util-linux-2.32.1-r0 do_package: Function failed: do_package
ERROR: Logfile of failure stored in: /home/xlla/develop/git/out/linux64/build/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-util-linux/2.32.1-r0/temp/log.do_package.11706
ERROR: Task (virtual:nativesdk:/home/xlla/develop/git/out/linux64/poky/meta/recipes-core/util-linux/util-linux_2.32.1.bb:do_package) failed with exit code '1'

htot, 2020-02-17:
> > @andy-shev yes, It dit boot, just simple did not appear on console.
> > I can see kernel command line when I execute 'cat /proc/cmdline'. I learned it!
> 
> It's probably U-Boot configuration or U-Boot itself change... However, I don't see any changes in the code for this.

I don't have problems with U-Boot. Log file?

htot, 2020-02-17:
@xlla the problems you see may be related to your host. I am building on:
```
ferry@kalamata:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 19.10
Release:        19.10
Codename:       eoan
ferry@kalamata:~$ uname -a
Linux kalamata 5.3.0-29-generic #31-Ubuntu SMP Fri Jan 17 17:27:26 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
Nevertheless I see if I run `make arduino-toolchain` it complains in the very last step that `tmp/deploy/sdk/poky-edison-glibc-x86_64-arduino-toolchain-corei7-64-edison-toolchain-2.7.
2.tar.xz does not exist!`. Actually `tmp/deploy/sdk` is not existing.

And when I run `make sdk` only:
```
ferry@kalamata:~$ ls Develop/tmp/out/linux64/build/tmp/deploy/sdk
poky-edison-glibc-x86_64-edison-image-corei7-64-edison-toolchain-2.7.2.host.manifest
poky-edison-glibc-x86_64-edison-image-corei7-64-edison-toolchain-2.7.2.sh
poky-edison-glibc-x86_64-edison-image-corei7-64-edison-toolchain-2.7.2.target.manifest
poky-edison-glibc-x86_64-edison-image-corei7-64-edison-toolchain-2.7.2.testdata.json
```
It seems a small detail broke resulting in a missing .xz, shouldn't be to hard to fix.

xlla, 2020-02-18:
> However when installing the board installation fails as the absolute path to the sdk does not exist (/opt/poky-edison/... wrong version).

can you refer this [mode +111](https://askubuntu.com/questions/764715/unable-to-install-intel-i586-library-intel-galileo-gen-2-in-arduino-ide-on-ubu) to fixed board installation problem, then try upload a simple test sketch ?

I have do upload but failed,  just stuck and timeout.

xlla, 2020-02-18:
directly execute "bitbake arduino-toolchain" or execute "make arduino-toolchain" all give out same errors. maybe I should update my build env.

xlla@u18-1:~/develop/git$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic
xlla@u18-1:~/develop/git$ uname -a
Linux u18-1 5.3.0-28-generic #30~18.04.1-Ubuntu SMP Fri Jan 17 06:14:09 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

but "bitbake edison-image -c populate_sdk" was success.

htot, 2020-02-18:
> can you refer this [mode +111](https://askubuntu.com/questions/764715/unable-to-install-intel-i586-library-intel-galileo-gen-2-in-arduino-ide-on-ubu) to fixed board installation problem, then try [...]

i never installed the sdk through the sh script. Instead I manually make a link to the files that i already have (no need to have them 2x on the same machine). I do that as described here: https://edison-fw.github.io/meta-intel-edison/3-Building-the-SDK.html

The problem is that the path contains the yocto version name and the Arduino sdk expects another.

htot, 2020-02-18:
> directly execute "bitbake arduino-toolchain" or execute "make arduino-toolchain" all give out same errors. maybe I should update my build env.

The only error I get is about the missing xz. 
> but "bitbake edison-image -c populate_sdk" was success.

In my case succeeds to build sh installer, but with warnings about uninstallable packages.

xlla, 2020-02-18:
> i never installed the sdk through the sh script. Instead I manually make a link to the files that i already have (no need to have them 2x on the same machine).

We are now try to restore the Arduino sketch compatibilities.
first I found missing folder "/sketch",  you manage to restore "clloader" and fixed it.
then you point out we can manual create a tty [port](https://htot.github.io/meta-intel-edison/4.4-hsu-i2c-spi.html#gadget-device-mode) to restore missing port, it did in host linux.
then you said that you just download Arduino IDE , and encounter some errors when install it.
you think it was caused by missing /opt/poky-edison... and wrong sdk version. 
I encounter it too, after some research, I think it is due to os version upgrade , the parameter format changes, as I mentioned, when I manual change +111 to /111 in ~/.arduino15/packages/Intel/tools/core2-32-poky-linux/1.6.2+1.0/i686/install_script.sh, and manual execute it, everything was ok, even I have not folder "/opt/poky-edison".

(yes, the installed sdk is old, wrong version, and 2x disk space occupy, but we just want to check sketch upload procedure now. It can work well with original firmware edison, so it should upload sketch to our new firmware edison successfully too.)

after that, you can choose "menu->File->Examples->04.Communication->SerialPassthrough", (choose this to avoid GPIO manipulation now)

I am stuck in this point, after compile sketch, timeout when uploading.
I found it invoke "sketchUploader/1.6.2+1.0/clupload/cluploadEdison_linux.sh" to do upload, then invoke "~/.arduino15/packages/Intel/tools/sketchUploader/1.6.2+1.0/x86/bin/lsz".
in lsz's help usage page, it said "Send file(s) with ZMODEM/YMODEM/XMODEM protocol", I don't know if current g_serial port can handle that protocal, since original edison use cdc acm port.

I am now try to compile kernel again to add acm usb gadget support, then configure a acm port and try to upload sketch again.

xlla, 2020-02-18:
after change to acm port, it still stuck in uploading.

[image]

xlla, 2020-02-18:
the good new is the acm port can be recognized by osx and the propertly driver was load.
<pre>
xlla$ ls /dev/cu.usb*
/dev/cu.usbmodem14221		/dev/cu.usbserial-A903BX28

</pre>
[image]

htot, 2020-02-18:
> (yes, the installed sdk is old, wrong version, and 2x disk space occupy, but we just want to check sketch upload procedure now. It can work well with original firmware edison, so it should upload sketch to our new firmware edison successfully too.)

The old sdk uses old compiler and links to wrong libc, mraa libs etc. Even if you can upload the sketch it will not work.

xlla, 2020-02-18:
> The old sdk uses old compiler and links to wrong libc, mraa libs etc. Even if you can upload the sketch it will not work.

If we can upload the sketch then we can manage to switch to right sdk, fixed mraa, chardev  etc, but now we can focus on upload itself only.

xlla, 2020-02-18:
If I stop service clloader, and manual execute some command in terminal, then sketch will upload successful.

<pre>
root@edison:/# systemctl stop clloader.service
root@edison:/# systemctl status clloader.service
● clloader.service - Daemon to handle arduino sketches
   Loaded: loaded (/lib/systemd/system/clloader.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Sat 2020-02-15 17:42:34 UTC; 1s ago
  Process: 543 ExecStart=/opt/edison/launcher.sh (code=killed, signal=TERM)
 Main PID: 543 (code=killed, signal=TERM)

Feb 15 17:20:07 edison systemd[1]: Started Daemon to handle arduino sketches.
Feb 15 17:42:34 edison systemd[1]: Stopping Daemon to handle arduino sketches...
Feb 15 17:42:34 edison systemd[1]: clloader.service: Succeeded.
Feb 15 17:42:34 edison systemd[1]: Stopped Daemon to handle arduino sketches.
root@edison:/# export EDISON_PATH="/opt/edison"
root@edison:/# export CLLOADER="$EDISON_PATH/clloader"
root@edison:/# export CLLOADER_OPTS="--escape --binary --zmodem --disable-timeouts"
root@edison:/# ls /dev/ttyGS0 
/dev/ttyGS0
root@edison:/# $CLLOADER $CLLOADER_OPTS -v < /dev/ttyGS0 > /dev/ttyGS0 
Opened /dev/ttyGS0 as inputOpened /dev/ttyGS0 as output/sketch/sketch.elf file does not exist or invalid permissions
clloader waiting to receive.clloader: remote command execution requested
clloader: mv -f /sketch/sketch.elf /sketch/sketch.elf.old
mv: cannot stat '/sketch/sketch.elf': No such file or directory
signal- child ignore  signal rxed
Zmodem wait finished .. restart to look for sketch
/sketch/sketch.elf file does not exist or invalid permissions
Receiving: SerialPassthrough.ino.elf                                 

Clloader: fclose: 0x1127170
Zmodem wait finished .. restart to look for sketch
/sketch/sketch.elf file does not exist or invalid permissions
clloader waiting to receive.clloader: remote command execution requested
clloader: mv SerialPassthrough.ino.elf /sketch/sketch.elf; chmod +x /sketch/sketch.elf
Zmodem wait finished .. restart to look for sketch
Sketch child process started, pid=07269
Starting /sketch/sketch.elf /dev/pts/0 U??S???????,?
[ 1677.817073] pinctrl-merrifield INTC1002:00: request() failed for pin 111
[ 1677.824000] pinctrl-merrifield INTC1002:00: pin-111 (0000:00:0c.0:27) status -16
[ 1677.832268] pinctrl-merrifield INTC1002:00: request() failed for pin 112
[ 1677.839186] pinctrl-merrifield INTC1002:00: pin-112 (0000:00:0c.0:28) status -16
</pre>
[image]

but with the running clloader service, sketch upload will timeout and failure. 
It seems to me the acm port,  the communication between lsz and clloader was right.
I will modify Arduino IDE to let it use latest sdk and repeat this test, maybe the sketch will successful launch...

htot, 2020-02-18:
Maybe this patch I applied is the problem https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-arduino/recipes-devtools/clloader/files/0001-clloader-silence-when-ttyGS0-not-found.patch 

If so you could revert this 6d7f5f31

xlla, 2020-02-18:
yes, after comment out those check statements, and reboot, upload works.

htot, 2020-02-18:
If you are looking to fix the arduino-toolchain.bb image, it builds the xz (on x86_64) here: 
`tmp/work/corei7-64-poky-linux/arduino-toolchain/1.0-r7/x86_64-deploy-arduino-toolchain-populate-sdk/` (that is `SDKDEPLOYDIR = "${WORKDIR}/${SDKMACHINE}-deploy-${PN}-populate-sdk"`)
but it looks for it here:
`${SDK_DEPLOY}/${TOOLCHAIN_OUTPUTNAME}`

So, I changed that line to:
`source_tarball="${WORKDIR}/${SDKMACHINE}-deploy-${PN}-populate-sdk/${TOOLCHAIN_OUTPUTNAME}.tar.xz"`
and that fixes the error and `arduino-toolchain-x86_64.tar.bz2` is build in `deploy/sdk`.

But there is some other stuff that hardcodes edison as 'i686', and I didn't test the sdk.

Like I said, it needs work. I hope you will send a PR when you get it all to work!

xlla, 2020-02-23:
> AFAIK these lines `options g_multi file=/dev/mmcblk0p9 stall=0 idVendor=0x8087 idProduct=0x0A9E iProduct=Edison iManufacturer=Intel` and `g_multi.dev_addr=02:00:86:0a:57:7c` are not used at all any more.

can we remote those lines from edison.env.

xlla, 2020-02-23:
> Like I said, it needs work. I hope you will send a PR when you get it all to work!

When I have finished, I will.

I am still in confusion now.

1. I can't build arduino-toolchain recipe, see above reply, no progress.
2. Mac can't driver eem device, I am trying patch kernel and switch to rndis to satisfy osx.
3. I can built populate_sdk successful, so I am trying replace Arduino IDE's sdk to latest by myself and fix hardcode reference, then do some basic sketch compile/upload/running test.
4. I think I also need to fix sketch_reset.service to get sketch to launch in edison.
5. I think I should fix files in hardware/i686/corelibs-edison/variants/edison_fab_c/ to work with chardev.

...long road to go.

xlla, 2020-02-23:
how to fix thoses warning,
<pre>
xlla@u18-1:~/develop/git/out/linux64/build$ bitbake edison-image -c populate_sdk
Parsing recipes: 100% |#######################################################################################################################| Time: 0:00:32
Parsing of 2297 .bb files complete (0 cached, 2297 parsed). 3404 targets, 301 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "i686-poky-linux"
MACHINE              = "edison"
DISTRO               = "poky-edison"
DISTRO_VERSION       = "2.7.2"
TUNE_FEATURES        = "m32 corei7"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "warrior:4edf4eb9cc86fbd4b51240fae527c1b97a7ace74"
meta-oe              
meta-python          
meta-networking      = "warrior:a24acf94d48d635eca668ea34598c6e5c857e3f8"
meta-intel           = "warrior:29ee4852a05931dcf856670d9d8a3c3077a40fe8"
meta-intel-edison-bsp 
meta-intel-edison-distro 
meta-intel-arduino   
meta-arduino         = "master:a3e5220a808ba179f63b5f7dd9fe9669e0062797"
meta-acpi            = "eds-5.0.0:ce3e7c4e063ea84f6162aa88d6d3dd034e2a6cfd"
meta-oic             = "master:f3e52b40178d966b260744b0ed91fb5b5e2062bd"
meta-qt5             = "warrior:20331bd6b696793a7aa23a3c035e5ebcf3075aa3"

Initialising tasks: 100% |####################################################################################################################| Time: 0:00:11
Sstate summary: Wanted 223 Found 214 Missed 9 Current 1012 (95% match, 99% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: edison-image-1.0-r0 do_populate_sdk: Unable to install packages. Command '/home/xlla/develop/git/out/linux64/build/tmp/work/edison-poky-linux/edison-image/1.0-r0/recipe-sysroot-native/usr/bin/apt-get  install --force-yes --allow-unauthenticated sqlite3-src libgdbm-dbg sudo-dbg shadow-securetty-dbg screen-dev libgpg-error-dev glibc-src systemd-compat-units-dbg alsa-plugins-dbg e2fsprogs-dev curl-dbg blink-led-dev gadget-dbg rng-tools-dbg dpkg-dev bzip2-src base-passwd-dbg libgnutls-src crashlog-dbg libsoup-2.4-src sudo-src sst-fw-bin-dbg e2fsprogs-dbg libgmp-dbg gettext-dbg bridge-utils-dev flex-dbg libgcc-s-src bcm43340-fw-dbg libgomp-dev libdaemon-dbg libgpg-error-dbg post-install-dev bcm43340-addr-dbg attr-dbg sqlite3-dbg python3-pygobject-dbg python-dbg parted-dbg alsa-state-dbg debianutils-src file-src ofono-dev shared-mime-info-dev sst-fw-bin-dev attr-src libgpg-error-src libnss-mdns-src htop-src db-src libogg-dev gcc-runtime-dbg dpkg-dbg libmpfr-dev openssh-dbg libexpat-src boost-src icu-src libvorbis-dev perl-src gstreamer1.0-src kmod-dbg target-sdk-provides-dummy-dev packagegroup-tzdata-dev libsodium-dbg i2c-tools-dev packagegroup-core-standalone-sdk-target-dev packagegroup-tzdata-dbg usbutils-dev glib-networking-dev target-sdk-provides-dummy-dbg bcm43340-addr-dev python3-setuptools-dbg python3-pygobject-src libxml2-dev glibc-locale-dbg libglib-2.0-dbg parted-dev lsof-dev swig-dbg iperf3-dbg iptables-src libsndfile-dev dbus-src openssh-dev base-passwd-src lshw-dbg iptables-dbg u-boot-fw-utils-dev nano-dev sketch-check-dbg zeromq-src ethtool-dbg shared-mime-info-src os-release-dev sysfsutils-dev gstreamer1.0-plugins-good-dev packagegroup-core-ssh-openssh-dev kbd-dev python-paho-mqtt-dbg libxml2-dbg packagegroup-core-standalone-sdk-target-dbg gstreamer1.0-plugins-base-dev debianutils-dev sudo-dev gstreamer1.0-meta-base-dbg pulseaudio-dev post-install-dbg alsa-utils-src orc-src apt-src pulseaudio-src mosquitto-dbg libcrypt-src libmpfr-dbg libmpc-dev acl-src gobject-introspection-dbg usbutils-src net-tools-dbg elfutils-dev run-timezone-dbg less-src pwr-button-handler-dbg gstreamer1.0-plugins-good-dbg gstreamer1.0-meta-audio-dev packagegroup-core-ssh-openssh-dbg swig-dev libasound-dev python3-src openssl-src libpsl-dbg ethtool-dev shadow-dbg libgnutls-dbg avahi-src bluez5-src run-postinsts-dev gstreamer1.0-plugins-base-dbg run-postinsts-dbg dosfstools-dbg tcpdump-dev m4-dbg pulseaudio-dbg libcrypt-dbg busybox-dev libxml2-src lsof-dbg libmpc-src libgcrypt-dev libunwind-dbg systemd-dbg libffi-src autoconf-dev gstreamer1.0-dev libgpiod-dbg libspeexdsp-dbg nodejs-dbg elfutils-src mosquitto-src perl-dbg volatile-binds-dbg binutils-dev libnsl-src avahi-dbg pwr-button-handler-src libtool-src watchdog-sample-dev gcc-dev connman-dbg nano-src lshw-src wpa-supplicant-dev i2c-tools-src libnsl-dbg libcroco-dbg libvorbis-src libusb-1.0-dbg util-linux-dbg clloader-src kbd-dbg libgcc-s-dbg sysfsutils-src iperf3-src libpam-dbg parted-src perf-dbg mcu-fw-bin-dev acl-dbg autoconf-dbg os-release-dbg shadow-src libpcre-dbg make-dbg powertop-dbg openssh-src iotop-dbg libunwind-dev blink-led-dbg clloader-dev libmpfr-src libgdbm-src libpsl-src automake-dev libsodium-src kbd-src netbase-dbg libgcrypt-dbg libunistring-src wget-dbg python-src binutils-src alsa-utils-dbg binutils-dbg glib-networking-src iotivity-src automake-dbg rng-tools-src alsa-utils-dev nodejs-src nettle-dbg systemd-serialgetty-dbg bash-completion-dbg libtool-dbg elfutils-dbg mcu-fw-load-dbg libexpat-dbg pciutils-src shared-mime-info-dbg packagegroup-core-buildessential-dev tcpdump-dbg boost-dbg battery-voltage-dbg battery-voltage-dev ca-certificates-dbg libusb-1.0-src util-linux-src xz-dbg clloader-dbg htop-dev libz-src libtirpc-sr [... cut ...]
                    Recommends: bash-dev
 pwr-button-handler-dev : Depends: pwr-button-handler (= 0.1-r0) but it is not going to be installed
                          Recommends: bash-dev
 target-sdk-provides-dummy-dev : Depends: target-sdk-provides-dummy (= 1.0-r0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

NOTE: Tasks Summary: Attempted 4152 tasks of which 4127 didn't need to be rerun and all succeeded.

Summary: There was 1 WARNING message shown.
</pre>

xlla, 2020-02-23:
DEFAULTTUNE =   "corei7-32"
Should I change CPP compiler args from "-march=core2 -mtune=core2" to "-march=corei7 -mtune=corei7"

htot, 2020-02-23:
? where is that set? Because you are on warrior, we already have [this](https://github.com/edison-fw/meta-intel-edison/commit/decb4c8f1d025d98e0659f67d908912ddd53c2f3) to change to correct arch.

xlla, 2020-02-24:
where is the create_devtools_package.sh.
not found meta-intel-edison-devtools too.

htot, 2020-02-24:
Good question. If you check out original, you will see it never existed (in the version published on https://git.yoctoproject.org/cgit/cgit.cgi/meta-intel-edison/)

xlla, 2020-02-25:
May I still stick to 32-bit version to avoid performance lose.

xlla, 2020-02-25:
> Good question. If you check out original, you will see it never existed (in the version published on https://git.yoctoproject.org/cgit/cgit.cgi/meta-intel-edison/)

Alright, I am still try to work out osx-version toolchain, so I had study Makefile carefully to see if I missed something. I thought maybe some clues in that script. 

after fixed a lot files, recipe odcctools2-crosssdk and osx-runtime was built.
now stuck on nativesdk-glibc/nativesdk-libc/nativesdk-libc-locale
<pre>
xlla@u18-1:~/develop/git/out/macosx/build$ bitbake arduino-toolchain
Loading cache: 100% |#########################################################################################################################| Time: 0:00:00
Loaded 3286 entries from dependency cache.
Parsing recipes: 100% |#######################################################################################################################| Time: 0:00:04
Parsing of 2216 .bb files complete (2215 cached, 1 parsed). 3287 targets, 310 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'nativesdk-glibc' (but virtual:nativesdk:/home/xlla/develop/git/out/macosx/poky/meta/recipes-core/glibc/glibc-locale_2.29.bb RDEPENDS on or otherwise requires it)
nativesdk-glibc was skipped: PREFERRED_PROVIDER_virtual/nativesdk-libc set to nativesdk-osx-runtime, not nativesdk-glibc
NOTE: Runtime target 'nativesdk-glibc' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['nativesdk-glibc']
NOTE: Runtime target 'nativesdk-bash' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['nativesdk-bash', 'virtual/nativesdk-libc-locale', 'nativesdk-glibc']
NOTE: Runtime target 'nativesdk-bash-completion' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['nativesdk-bash-completion', 'nativesdk-bash', 'virtual/nativesdk-libc-locale', 'nativesdk-glibc']
NOTE: Runtime target 'gdb-cross-canadian-i686' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['gdb-cross-canadian-i686', 'nativesdk-python3', 'nativesdk-util-linux', 'nativesdk-bash-completion', 'nativesdk-bash', 'virtual/nativesdk-libc-locale', 'nativesdk-glibc']
NOTE: Runtime target 'packagegroup-cross-canadian-edison' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-cross-canadian-edison', 'gdb-cross-canadian-i686', 'nativesdk-python3', 'nativesdk-util-linux', 'nativesdk-bash-completion', 'nativesdk-bash', 'virtual/nativesdk-libc-locale', 'nativesdk-glibc']
ERROR: Required build target 'arduino-toolchain' has no buildable providers.
Missing or unbuildable dependency chain was: ['arduino-toolchain', 'packagegroup-cross-canadian-edison', 'gdb-cross-canadian-i686', 'nativesdk-python3', 'nativesdk-util-linux', 'nativesdk-bash-completion', 'nativesdk-bash', 'virtual/nativesdk-libc-locale', 'nativesdk-glibc']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
</pre>

htot, 2020-02-25:
Sorry, can't help you with osx. But maybe there is info on the yocto site on how to build on osx? Maybe there is updated darwin stuff?

xlla, 2020-02-28:
> If you are looking to fix the arduino-toolchain.bb image

after many days research, I had to give up Mac version toolchain, due to the lack of support and ancient source repo.
for linux version, when I fixed arduino-toolchain.bb, it is built!

htot, 2020-02-28:
That's good news. Too bad of the Mac version, I can't help there.

xlla, 2020-02-29:
@htot while play with acpi-tables, I found it is still use configfs in acpi-tables-load.
to make life easier, can we just amend those pin number, then everything is work?
for example, recipes-devtools/clloader/files/sketch_reset.service
replace 
<pre>
/opt/edison/sketch_reset -i 207 -o 215 -s /opt/edison/sketch_reset.sh
</pre>
to 
<pre>
/opt/edison/sketch_reset -i 503 -o 511 -s /opt/edison/sketch_reset.sh
</pre>

htot, 2020-02-29:
Yes, you missed my [reply](https://github.com/edison-fw/meta-intel-edison/issues/74#issuecomment-590984207). You need to change one of you U-Boot variable so that you will see "quiet **skiptables** debugshell=10" n the kernel command line. Really.

And life is already easy :-), I created a test branch [sketch_reset_v2](https://github.com/htot/meta-intel-edison/tree/sketch_reset_v2) that patches [sketch_reset](https://github.com/intel/clloader/blob/edison/galileo-reset/galileo-reset.c) to use libgpiod instead of sysfs. As you can see the code is now greatly simplified.

Using `libgpiod` really is the way forward. I believe development is now continuing to create a daemon that listens to dbus to rootfs will no longer be needed to manipulate gpios.

The trick is to make `mraa` use `libgpiod` as well. The `sketch_reset` code may serve as an example.

xlla, 2020-03-01:
> Yes, you missed my [reply](https://github.com/edison-fw/meta-intel-edison/issues/74#issuecomment-590984207).

I see, I have manage to put this arg in to the kernel command line.
<pre>
fw_setenv bootargs_debug 'quiet skiptables'
</pre>
then it did work.
<pre>
root@edison:~# systemctl status acpi-tables-load
● acpi-tables-load.service - ACPI tables load service
   Loaded: loaded (/lib/systemd/system/acpi-tables-load.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Sat 2020-02-15 17:20:08 UTC; 16s ago
  Process: 551 ExecStart=/usr/bin/acpi-tables-load (code=exited, status=0/SUCCESS)
 Main PID: 551 (code=exited, status=0/SUCCESS)

Feb 15 17:20:07 edison systemd[1]: Started ACPI tables load service.
Feb 15 17:20:08 edison acpi-tables-load[551]: gpioset: at least one GPIO line offset to value mapping must be specified
Feb 15 17:20:08 edison systemd[1]: acpi-tables-load.service: Succeeded.

</pre>

but empty in journalctl.
<pre>
root@edison:~# journalctl -b | grep "Command line"
root@edison:~# 
</pre>
I can confirm it in screen connected to ttyUSB0
<pre>
Linux kernel version 5.4.0-edison-acpi-standard (oe-user@oe-host) #1 SMP Sat Feb 29 18:08:20 UTC 2020
Building boot_params at 0x00090000
Loading bzImage at address 100000 (11950784 bytes)
Magic signature found
Kernel command line: "root=/dev/mmcblk0p8 rootfstype=ext4 console=ttyS2,115200n8 earlyprintk=ttyS2,115200n8,keep quiet skiptables g_multi.ethernet_config=cdc systemd.unit=multi-user.target hardware_id=00 g_multi.iSerialNumber=8452e52a5b73f5f38c917327f40a577c g_multi.dev_addr=02:00:86:0a:57:7c platform_mrfld_audio.audio_codec=dummy"
Magic signature found
</pre>

xlla, 2020-03-01:
> Using `libgpiod` really is the way forward

Yes, libdpiod seems easy to monitor gpio pin edge signal than configfs.
I am confused in sketch_reset code logic, 
first it set -o line 215 to output mode, then pull up follow pull down, it keep 1 on line 215.
then it set -i line 207 to input mode, wait a falling edge from shield_reset button push, then wait a raising edge for shield_reset button release.
question is, line 215 and 205 is connected, when push down shield_reset button, does it short the line 215?

xlla, 2020-03-01:
> I created a test branch [sketch_reset_v2](https://github.com/htot/meta-intel-edison/tree/sketch_reset_v2) that patches [sketch_reset](https://github.com/intel/clloader/blob/edison/galileo-reset/galileo-reset.c) to use libgpiod instead of sysfs.

could you patch this [edison_fab_c](https://github.com/intel/corelibs-edison/blob/master/variants/edison_fab_c/variant.cpp) file to use libgpiod, then arduino sketch will reborn.

I think we can make an intermediate table to translate those configfs line numbers to gpio chip name and line index,  then use chardev functions to implement original invoke.

htot, 2020-03-01:
> then it did work.

Good

> but empty in journalctl.

Just check `journalctl -b` it must be there. 

> `gpioset: at least one GPIO line offset to value mapping must be specified`

That doesn't seem right, I haven't seen that before.

> Yes, libdpiod seems easy to monitor gpio pin edge signal than configfs.

sysfs

> I am confused in sketch_reset code logic,

Yes you are confused, you are looking at the original code, not at the patched code. The patched code will be under (x86_64) `out/linux64/build/tmp/work/corei7-64-poky-linux/clloader/1.0-r0/git/galileo-reset/`

xlla, 2020-03-02:
> As you can see the code is now greatly simplified.

Yes,  the code you rewrite is simple and elegant! It works well.
<pre>
root@edison:~# /opt/edison/sketch_reset -i SHLD_RESET0 -o SHLD_RESET1 -s /opt/edison/sketch_reset.sh  -v
Sketch Reset button pressed:
Sketch Reset button released: Calling /opt/edison/sketch_reset.sh
Sketch Reset button pressed:
Sketch Reset button released: Calling /opt/edison/sketch_reset.sh

</pre>

When I try to upgrate kernel to 5.5.0, recipe perf raise errors:
<pre>
|   CC       /home/xlla/develop/git/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/fs/fs.o
| Traceback (most recent call last):
|   File "/usr/lib/python3.6/site.py", line 570, in <module>
|     main()
|   File "/usr/lib/python3.6/site.py", line 556, in main
|     known_paths = addusersitepackages(known_paths)
|   File "/usr/lib/python3.6/site.py", line 288, in addusersitepackages
|     user_site = getusersitepackages()
|   File "/usr/lib/python3.6/site.py", line 264, in getusersitepackages
|     user_base = getuserbase() # this will also set USER_BASE
|   File "/usr/lib/python3.6/site.py", line 254, in getuserbase
|     USER_BASE = get_config_var('userbase')
|   File "/usr/lib/python3.6/sysconfig.py", line 607, in get_config_var
|     return get_config_vars().get(name)
|   File "/usr/lib/python3.6/sysconfig.py", line 550, in get_config_vars
|     _init_posix(_CONFIG_VARS)
|   File "/usr/lib/python3.6/sysconfig.py", line 421, in _init_posix
|     _temp = __import__(name, globals(), locals(), ['build_time_vars'], 0)
| ModuleNotFoundError: No module named '_sysconfigdata'
| Makefile:185: recipe for target '/home/xlla/develop/git/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/bpf_helper_defs.h' failed

</pre>

htot, 2020-03-02:
Yes, @mwallnoefer reported that [too](https://github.com/edison-fw/meta-intel-edison/issues/91#issuecomment-591540766). I had that too, I think I fixed that by doing -c cleansstate for linux-yocto and perf, but @mwallnoefer reports it didn't work for him.

I did a clean build `make clean && make setup && make image` without errors.

In your case your were building i686, but this branch sketch_reset_v2 is x86_64, so you need to do `make clean` anyway.

xlla, 2020-03-03:
> Just check `journalctl -b` it must be there.

alright, I found it by eyes.
<pre>
root@edison:~# journalctl -b |grep "Kernel command"
Feb 15 17:20:01 edison kernel: Kernel command line: root=/dev/mmcblk0p8 rootfstype=ext4 console=ttyS2,115200n8 earlyprintk=ttyS2,115200n8,keep quiet skiptables g_multi.ethernet_config=cdc systemd.unit=multi-user.target hardware_id=00 g_multi.iSerialNumber=8452e52a5b73f5f38c917327f40a577c g_multi.dev_addr=02:00:86:0a:57:7c platform_mrfld_audio.audio_codec=dummy
root@edison:~# journalctl -b | grep "Command line"
root@edison:~# 

</pre>

xlla, 2020-03-03:
> Yes, @mwallnoefer reported that [too](https://github.com/edison-fw/meta-intel-edison/issues/91#issuecomment-591540766).

I have manual apply patch 3a1a18ba9d28adb5562eabe9ec354f6d93154f5c from git.yoctoproject.org, but still error, I comment out perf and built.

3a1a18ba9d28adb5562eabe9ec354f6d93154f5c
--

xlla, 2020-04-10:
perf was work now.
I wonder how to amend arduino hardware lib for edison.
it does not rely on any external lib,  should I introduce libgpiod to it, or mraa lib.

htot, 2020-04-11:
I don't know the status of mraa. It has moved over to the eclipse foundation and development seems to have slowed down. The most recent work (in forks) seems to be done by you https://github.com/eclipse/mraa/network.
So, if you have lots of code depending on mraa, you might want to fix mraa. Else, libgpiod and libiio might be the way forward.

limaiem, 2020-05-26:
Hi all
I have nearly the same problem as xlla

I tried to follow all the exchanges so I can figure out how to solve the issue.

Here few information about my setup

My host is: 
- Ubuntu 18.04.4 LTS
- Linux io 4.15.0-101-generic # 102-Ubuntu SMP Mon May 11 10:07:26 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

I followed the instructions here (https://edison-fw.github.io/meta-intel-edison/1.2-Setting-up.html ) and I was able to build images and to load them in the Edison board, to boot it, access by screen using **/dev/ttyUSB0** and configure the wifi and so. 

- Poky (Yocto Project Reference Distro) 2.7.3 edison ttyS2
- Linux edison 5.4.0-edison-acpi-standard # 1 SMP Fri May 22 02:04:59 UTC 2020 i686 i686 i386 GNU/Linux

However I was unable to get my host recognize the port **/dev/ttyACM3** to load the arduino codes. Everything worked fine with the original images of Edison board.

I tired to apply the code echo 0x1d6b 0x0104 >/sys/bus/usb-serial/drivers/generic/new_id without any success (https://edison-fw.github.io/meta-intel-edison/4.4-hsu-i2c-spi.html). 
I tried with ID 8087:0a9e instead of 0x1d6b:0x0104 but the same issue was here. 

Correct me if I am wrong but someone managed to fix this and get it work. What is the branch used to make compile the images (master ?). Any patch applied ?

Thank your for the effort to maintain updated images for Edison.

[6 more replies]

closed 2025-04-05 by htot