edison-fw/meta-intel-edison #80: nano text editor

mwallnoefer, 2019-12-03:
I propose to add the _nano_ text editor to the manifest (https://github.com/edison-fw/meta-intel-edison/blob/master/docs/edison-image-edison.manifest) as an alternative to _vi_, for the sake of simplicity of its UI but also its availability in the original Yocto image.

Btw. does the apt package manager work out of the box? I couldn't find any instructions about how to configure repositories for installing packages etc...

htot, 2019-12-03:
I didn't know nano was included, but sure why not.

You can build a package using `bitbake somepackage` and then follow instructions here to create a repo and install packages from the repo onto edison. If you have time and space you can probably `bitbake world` or something similar (I haven't tried).

mwallnoefer, 2019-12-04:
For now I have added this line to _out/current/build/conf/local.conf_:
```
IMAGE_INSTALL_append = " nano"
```
But I do not know how it could be added per default.

htot, 2019-12-04:
I goes directly into https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-distro/recipes-core/images/edison-image.bb

mwallnoefer, 2019-12-05:
Something like this https://github.com/edison-fw/meta-intel-edison/commit/b5a654003a9cb581559980a14844161155accd0c (I guess that _sketch_check_ should belong to Arduino)?

mwallnoefer, 2019-12-15:
@htot Could we merge something like my patch? Should I open an apposite pull request?

htot, 2019-12-15:
Sure. I guess it should go into [warrior](https://github.com/htot/meta-intel-edison/tree/warrior) , did you see my comments at the bottom of b5a6540?

htot, 2020-01-03:
I merged your patches into warrior, thanks. They will eventually reach edison-fw/master. Closing this for now.

closed 2020-01-03 by htot

mwallnoefer, 2021-03-21:
Unfortunately the syntax highlighting support is still not enabled by default, which would be very handy when it comes to programming/debugging.
I remember that I once had posted this to the Yocto package maintainer but apparently he didn't change it...😠

htot, 2021-03-21:
There is a recipe option "tiny", maybe that is related? Other then that the recipe does not enable or disable stuff.

htot, 2021-03-22:
Ok I figured this one out for you. The tiny option would actually disable syntax highlighting. It's on by default. All the syntax files are also packaged and installed to `/usr/share/nano`. The only thing missing is the `nanorc `file in `/etc`. It seems this is not provided by the sources. I copied over the `nanorc` from my Ubuntu and voila, colors.

I see that debian adds `nanorc` in their `nano_5.4-2.debian.tar.xz` sauce. So you could send a patch for the nano recipe to package `nanorc` to the openembedded ML, or we could add modified nano recipe to meta-intel-edison (that will eventually get pickup by [layers.openembedded.org](layers.openembedded.org), or write a bbappend so we always get latest nano with our own sauce.

mwallnoefer, 2021-03-23:
Let's see what the openembedded nano package maintainer responds us.