edison-fw/meta-intel-edison #81: configure_edison

mwallnoefer, 2019-12-03:
The cmdline tool _configure_edison_ doesn't work properly any more. For instance you aren't able to set up a wifi connection using ```configure_edison --wifi``` which had been a neat feature in the past.

Checking the sources under https://github.com/intel/edison-oobe/blob/master/src/configure_edison revealed that there no longer exists an _interfaces_ file (appearently no network service) and also some paths seem have been changed (lines 473 till 476 ```/etc/wpa_supplicant/wpa_supplicant.conf``` -> ```/etc/wpa_supplicant.conf```).

No idea if it makes sense to maintain this tool or better to look for an alternative.

htot, 2019-12-03:
I'm working on this (it is coming from the oobe or out-of-the-box-experience repo https://github.com/intel/edison-oobe).

The CLI is not very useful as you can, once you have a terminal, do everything from connmanctl these days. But there was also the web server version based on nodejs. I have proted all to python3, replaced web server by bottle. This will allow us to make initial connection over wifi AP or bluetooth tether and configure using browser. It still needs a little work though.

mwallnoefer, 2019-12-04:
Ah okay - glad to hear that this still represents a priority for you. I just has this quick look at the Intel OOBE repo and said, "wow - three years without a commit, this cannot work".

And then I thought that also similar embdedded SoCs or SoMs suffer from the same problem (e.g. Raspberry Pis, Beagleboard etc. etc.); so a joint solution for Wifi, network, Bluetooth and hostname configuration should make sense.

htot, 2019-12-04:
I didn't consider that, but I guess it should be possible to make oobe work on other platforms too. But for general purpose use, it will need additional work like https / session / passwords.

mwallnoefer, 2020-03-02:
We may close this in favour of #32 .

closed 2020-03-02 by mwallnoefer