edison-fw/meta-intel-edison #83: ERROR:free(): double free detected in tcache 2

shawnhsoia97, 2020-01-30:
Hi,I build thud on ubuntu 18.04.3.Just at the end when it seems to be creating the image files, I run into this "double free detected" error for both minimal and example image.

```
ERROR: core-image-minimal-initramfs-1.0-r0 do_rootfs: Unable to install packages. Command '/mnt/yocto/out/linux64/build/tmp/work/edison-poky-linux/core-image-minimal-initramfs/1.0-r0/recipe-sysroot-native/usr/bin/opkg --volatile-cache -f /mnt/yocto/out/linux64/build/tmp/work/edison-poky-linux/core-image-minimal-initramfs/1.0-r0/opkg.conf -t /mnt/yocto/out/linux64/build/tmp/work/edison-poky-linux/core-image-minimal-initramfs/1.0-r0/temp/ipktemp/ -o /mnt/yocto/out/linux64/build/tmp/work/edison-poky-linux/core-image-minimal-initramfs/1.0-r0/rootfs  --force_postinstall --prefer-arch-to-version   install acpi-tables base-passwd busybox initramfs-framework-base initramfs-module-setup-live initramfs-module-udev kernel-module-extcon-intel-mrfld kernel-module-intel-soc-pmic-mrfld kernel-module-mmc-block kernel-module-sdhci-acpi kernel-module-sdhci-pci run-postinsts udev' returned -6:
free(): double free detected in tcache 2

```
I dont know why.Haven't seen this error before.Do you have any idea?

htot, 2020-01-30:
I am using dpkg. It seems you are using opkg here?

shawnhsoia97, 2020-01-30:
Yes,I using opkg.

htot, 2020-01-30:
Then I wouldn't know.

shawnhsoia97, 2020-01-30:
Ok,I reverted to using dpkg and it returned to normal.
Thank you.

closed 2020-01-30 by shawnhsoia97

marler8997, 2021-09-13:
I might have run into the same issue but we use opkg over dpkg for other reasons.  These commits look promising:
* https://github.com/ndmsystems/opkg/commit/23ed947e20d43d8ac6bcc620a8633f6ffa20749a
* https://github.com/ndmsystems/opkg/commit/39cdebb5951d7f0dacc3784062825cc05342c7f8

htot, 2021-09-13:
So, it might be automatically resolved once we move to  Yocto Hardknott?