edison-fw/meta-intel-edison #84: Not installing file to /home

shawnhsoia97, 2020-02-10:
Hi ,I encounter some problem .My Edison has nothing in /home after the flashing.
But my yocto rootfs have some thing.
```
ls out/linux64/build/tmp/work/edison-poky-linux/edison-image/1.0-r0/rootfs/home/ -l
total 16
drwxr-xr-x 2 shawn shawn 4096 Feb  7 10:08 bovia
drwxr-sr-x 2 shawn shawn 4096 Jan 30 08:53 root
drwxr-xr-x 2 shawn shawn 4096 Feb  7 10:08 tester
-rw-r--r-- 1 shawn shawn  167 Feb  7 10:01 wpa_supplicant-wlan0.conf
```
Other filles can be installed to root normally,but there is no way to install to /home.

htot, 2020-02-10:
/home is on our rootfs image. But currently we don't build a hoe image. Actually we don't even flash the home partition (checkout flashall.sh).

So it might be that there are files on /home, but then the home partition is mounted on top of it and hiding the files?

shawnhsoia97, 2020-02-11:
Yes,you are right.Files is  appeared after I unmount home partition.
Because my user directory is install on /home,so if home partition hiding the files,user directory can't be found. `No directory, logging in with HOME=/`
Thank you!
Is there a way to clear this partition separately?(Don't build a home image)

htot, 2020-02-11:
You could add things to postinstall and let them be created once run time (after mount).
Also I think Yocto provides a mechanism to add users, but I haven't investigated that yet. That should take care of user directory creation.

I have also been wondering why the user root has its home on /home/root. In case the home partition mount fails this causes problems to log in. This is not what debian does, that has user root home on /root. It can be changed see https://www.yoctoproject.org/docs/latest/mega-manual/mega-manual.html#var-ROOT_HOME

shawnhsoia97, 2020-02-11:
Yes,use to `EXTRA_USERS_PARMS` can create user directory,but this install to /hoom of rootsf,so I change MountPoint of `/dev/disk/by-partlabel/home`.

I think maybe /home/root was created on home partition ,because I can't build it after I remove  /home/root.

htot, 2020-02-11:
I am working to create one btrfs partition to replace home and rootfs, with @, @boot and @home subvolume. But after you do this flashall won't work any more. So, multiple things need changing.

shawnhsoia97, 2020-02-12:
I understand.Thank you!

closed 2020-02-12 by shawnhsoia97