edison-fw/meta-intel-edison #85: failure to build SDK for macosx

xlla, 2020-02-19:
already add meta-darwin in build/conf/bblayers.conf.
set SDKMACHINE = "i386-darwin" in build/conf/local.conf
have created OSX-sdk.zip for osx 10.13
<pre>
xlla$ ditto `xcrun --sdk macosx10.13 --show-sdk-path` .
xlla$ zip -yr OSX-sdk OSX-sdk
</pre>

I have put OSX-sdk.zip in download folder according to Makefile.
<pre>
echo "Please make sure that OSX-sdk.zip is available in your bitbake download directory"
mv ~/develop/OSX-sdk.zip ~/develop/git/bbcache/downloads/
</pre> 

<pre>
xlla@u18-1:~/develop/git/out/macosx$ ls
build  poky
xlla@u18-1:~/develop/git/out/macosx$ source poky/oe-init-build-env build

### Shell environment set up for builds. ###

You can now run 'bitbake <target>'

Common targets are:
    core-image-minimal
    core-image-sato
    meta-toolchain
    meta-ide-support

You can also run generated qemu images with a command like 'runqemu qemux86'
xlla@u18-1:~/develop/git/out/macosx/build$ bitbake odcctools2-crosssdk -c cleansstate
WARNING: Layer meta-darwin should set LAYERSERIES_COMPAT_meta-darwin in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer meta-darwin should set LAYERSERIES_COMPAT_meta-darwin in its conf/layer.conf file to list the core layer names it is compatible with.
Parsing recipes: 100% |###########################################################################################| Time: 0:00:37
Parsing of 2216 .bb files complete (0 cached, 2216 parsed). 3287 targets, 311 skipped, 0 masked, 0 errors.
ERROR: No recipes available for:
  /home/xlla/develop/git/out/macosx/poky/meta-darwin/recipes-core/zlib/zlib_1.2.8.bbappend
  /home/xlla/develop/git/out/macosx/poky/meta-darwin/recipes-devtools/binutils/binutils-cross-canadian_2.24.bbappend
  /home/xlla/develop/git/out/macosx/poky/meta-darwin/recipes-devtools/gcc/gcc-cross-canadian_4.9.bbappend
  /home/xlla/develop/git/out/macosx/poky/meta-darwin/recipes-devtools/gcc/gcc-crosssdk-initial_4.9.bbappend
  /home/xlla/develop/git/out/macosx/poky/meta-darwin/recipes-devtools/gcc/gcc-crosssdk_4.9.bbappend
  /home/xlla/develop/git/out/macosx/poky/meta-darwin/recipes-devtools/gcc/gcc-runtime_4.9.bbappend
  /home/xlla/develop/git/out/macosx/poky/meta-darwin/recipes-devtools/gcc/libgcc_4.9.bbappend

Summary: There were 2 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
</pre>

then I change those 7 xxx.bbapend files to the version same as meta.
<pre>
zlib_1.2.11.bbappend
binutils-cross-canadian_2.32.bbappend
xlla@u18-1:~/develop/git/out/macosx$ ls /home/xlla/develop/git/out/macosx/poky/meta-darwin/recipes-devtools/gcc/
gcc-cross-canadian_8.3.bbappend  gcc-crosssdk_8.3.bbappend  gcc-crosssdk-initial_8.3.bbappend  gcc-runtime_8.3.bbappend  libgcc_8.3.bbappend
</pre>

then new errors come out.
<pre>
xlla@u18-1:~/develop/git/out/macosx/build$ bitbake odcctools2-crosssdk -c cleansstate
WARNING: Layer meta-darwin should set LAYERSERIES_COMPAT_meta-darwin in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer meta-darwin should set LAYERSERIES_COMPAT_meta-darwin in its conf/layer.conf file to list the core layer names it is compatible with.
Loading cache: 100% |#############################################################################################| Time: 0:00:00
Loaded 3286 entries from dependency cache.
ERROR: ExpansionError during parsing /home/xlla/develop/git/out/linux64/poky/meta/recipes-devtools/gcc/gcc-crosssdk_8.3.bb-:--:--
Traceback (most recent call last):
bb.data_smart.ExpansionError: Failure expanding variable EXTRA_OECONF_darwin[:=], expression was ${@oe_filter_out('--with-gnu-ld', '${@oe_filter_out('--with-linker-hash-style=sysv', '    --enable-clocale=generic     --with-gnu-ld     --enable-shared     --enable-languages=c,c++,objc     --enable-threads=posix     --disable-multilib     --enable-default-pie     --enable-c99     --enable-long-long     --enable-symvers=gnu     --enable-libstdcxx-pch     --program-prefix=i386-pokysdk-darwin-     --without-local-prefix         --disable-lto     --disable-libssp     --enable-libitm     --disable-bootstrap     --disable-libmudflap     --with-system-zlib     --with-linker-hash-style=sysv     --enable-linker-build-id     --with-ppl=no     --with-cloog=no     --enable-checking=release     --enable-cheaders=c_global     --without-isl               --with-gxx-include-dir=/not/exist/opt/poky-edison/2.7.2/sysroots/i386-pokysdk-darwin/usr/include/c++/8.3.0     --with-sysroot=/not/exist     --with-build-sysroot=/home/xlla/develop/git/out/macosx/build/tmp/work/x86_64-linux/gcc-crosssdk-i386-pokysdk-darwin/8.3.0-r0/recipe-sysroot                --without-long-double-128 libgcc_cv_powerpc_float128=no       --enable-poison-system-directories     --with-system-zlib  --disable-static ${PACKAGECONFIG_CONFARGS} --disable-nls --with-glibc-version=2.28  --enable-initfini-array', d)} --disable-static ${PACKAGECONFIG_CONFARGS} ${@gettext_oeconf(d)} --with-glibc-version=2.28  --enable-initfini-array', d)} which triggered exception SyntaxError: invalid syntax (Var <EXTRA_OECONF_darwin[:=]>, line 1)

Summary: There were 2 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
</pre>

after read poky/meta-darwin/recipes-devtools/osx-runtime/nativesdk-osx-runtime_10.8.bb
I move OSX-sdk.zip into poky/meta-darwin/recipes-devtools/osx-runtime/files/
<pre>
xlla@u18-1:~/develop/git/out/macosx/build$ mv ~/develop/git/bbcache/downloads/OSX-sdk.zip ../poky/meta-darwin/recipes-devtools/osx-runtime/files/

xlla@u18-1:~/develop/git/out/macosx$ ls -alh poky/meta-darwin/recipes-devtools/osx-runtime/files/
total 79M
drwxrwxr-x 2 xlla xlla 4.0K 2月  19 03:09 .
drwxrwxr-x 3 xlla xlla 4.0K 2月  19 16:11 ..
-rw-rw-r-- 1 xlla xlla   11 2月  15 20:27 LICENSE
-rw-r--r-- 1 xlla xlla  79M 2月  19 02:01 OSX-sdk.zip
-rw-rw-r-- 1 xlla xlla 5.9K 2月  15 20:27 remove_unsupported_attribute_availability_with_message_for_gcc.patch
</pre>

same errors.

htot, 2020-02-19:
I don't have a Mac so can't reproduce. But you can fix the `WARNING: Layer meta-darwin should set LAYERSERIES_COMPAT_meta-darwin...` by adding something similar as here https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-bsp/conf/layer.conf to the meta-darwin/conf/layer.conf.

xlla, 2020-02-19:
In theory, can I build a SDK for Mac  in a linux host? 
Or I must run those build script in Mac host to produce a SDK.

xlla, 2020-02-19:
after update os, rerun make image, the /home is back!

how to simple update kernel inside edison, can I create some kernel.deb , then in edison execute "apt upgrade kernel" ?

each time when I make some tiny change to kernel module, I have to pack whole images and flashall,
maybe emmc would wear out.

htot, 2020-02-19:
setup.sh --list_sdk_hosts=macosx ?

htot, 2021-03-02:
Have you tried this? Either way i think we should mark this as won't fix because it is not supported by upstream Yocto anymore.

closed 2021-11-01 by htot

xlla, 2023-06-10:
@Ecordonnier , I am glad to heard this hibernated layer has been awake again, I will check it in the future.