edison-fw/meta-intel-edison #86: missing /home after change partitions size

xlla, 2020-02-19:
in file  meta-intel-edison/meta-intel-edison-bsp/recipes-bsp/u-boot/files/edison.env
rootfs, size=1536MiB
update, size=768MiB

in file meta-intel-edison/meta-intel-edison-distro/recipes-core/images/edison-image-minimal.bb
IMAGE_ROOTFS_SIZE = "1048576"  

after flash firmware, /home was missing.

<pre>
Poky (Yocto Project Reference Distro) 2.7.2 edison ttyS2

edison login: root
Last login: Sat Feb 15 22:45:35 UTC 2020 on ttyS2
No directory, logging in with HOME=/
root@edison:/# ls
bin  boot  dev	etc  factory  home  lib  lost+found  media  mnt  opt  proc  run  sbin  sketch  sys  tmp  usr  var
root@edison:/# df -h
Filesystem      Size  Used Avail Use% Mounted on
none            474M  4.0K  474M   1% /dev
/dev/mmcblk0p8  928M  602M  260M  70% /
tmpfs           478M     0  478M   0% /dev/shm
tmpfs           478M  604K  478M   1% /run
tmpfs           478M     0  478M   0% /sys/fs/cgroup
tmpfs           478M   52K  478M   1% /tmp
tmpfs           478M     0  478M   0% /var/volatile
/dev/mmcblk1p1   30G   16G   13G  55% /media/sdcard
/dev/mmcblk0p5 1003K   19K  913K   3% /factory
</pre>

why mmcblk0p8 is still size 928M instead of 1536M?
should IMAGE_ROOTFS_SIZE = (rootfs size in edison.env) * 1024 ?
to use OTA, should IMAGE_ROOTFS_SIZE <= (update size in edison.env) ?

htot, 2020-02-19:
Normally, the home partition is not touched during flash, so your files there persist. Looks like something now went wrong during flash destroying the home partition.

Then on first boot partitions are resized (try `partx /dev/mmc0blkp8`) . And then AFAIU the filesystem is resized: `https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-distro/recipes-core/resize-rootfs/resize-rootfs.bb`.

Maybe you need to manually fix?

xlla, 2020-02-19:
I am afraid I will lost data while flash firmware so I keep all my data on sdcard.
In fact I do not need home partition in emmc, so does OTA partition, can I alloc all space to rootfs?

xlla, 2020-02-19:
after update os, rerun make image, the /home is back!

how to simple update kernel inside edison, can I create some kernel.deb , then in edison execute "apt upgrade kernel" ?

each time when I make some tiny change to kernel module, I have to pack whole images and flashall,
maybe emmc would wear out.

htot, 2020-02-19:
I prepared ota partition as simple fat partition and [put kernel there](https://edison-fw.github.io/meta-intel-edison/2.2-Building-and-installing-the-kernel.html#installing-the-kernel), and [rootfs on usb](https://edison-fw.github.io/meta-intel-edison/2.1-Building-and-installing-the-rootfs.html) (fast writing) or sdhc (slower), then boot `run usbboot` or `run edsboot`.

This works best if you have working system on emmc (that you don't touch so it won't break) andyou can use that to scp the kernel to the ota partition.

xlla, 2020-02-20:
> Normally, the home partition is not touched during flash

so the home partition just hold it's original size(1.3GB), and won't change even I have adjust size of rootfs & update in edison.env?

xlla, 2020-02-20:
> I prepared ota partition as simple fat partition and [put kernel there](https://edison-fw.github.io/meta-intel-edison/2.2-Building-and-installing-the-kernel.html#installing-the-kernel)

In sumo branch built, I have put a kernel in there, and forgot to update it when I flash new warrior images.

Now I have put warrior kernel in OTA, and boot successful from sdcard!
<pre>
root@edison:~# df -h
Filesystem       Size  Used Avail Use% Mounted on
none             474M  4.0K  474M   1% /dev
/dev/mmcblk1p1    28G  2.5G   24G  10% /
tmpfs            478M     0  478M   0% /dev/shm
tmpfs            478M  620K  478M   1% /run
tmpfs            478M     0  478M   0% /sys/fs/cgroup
tmpfs            478M  4.0K  478M   1% /tmp
tmpfs            478M     0  478M   0% /var/volatile
/dev/mmcblk0p10  1.3G   15M  1.3G   2% /home
/dev/mmcblk0p5  1003K   19K  913K   3% /factory
tmpfs             96M     0   96M   0% /run/user/0
</pre>

I wonder if I can shrink partition mmcblk0p10 and leave more space for "rootfs / update".

htot, 2020-02-20:
If the end of ota partition overlaps home, home will break. But if you increase rootfs and decrease ota by the same there will be no problem.

For Zeus I am working to replace rootfs / ota/ home by a single btrfs partition, with subvolumes @boot, @, @home, @ota. This way all free space is available to all subvolumes.

xlla, 2020-02-21:
> This way all free space is available to all subvolumes.

It is great!

Eventually, the rootfs size is identical to edison.env definition.
<pre>
root@edison:~# partx /dev/mmcblk0p8 
NR  START     END SECTORS SIZE NAME   UUID
 8 131072 3276799 3145728 1.5G rootfs 04ea6535-c373-4140-ac7a-6f6e99ed7da4
root@edison:~# systemctl start resize-rootfs.service
[24631.865037] resize2fs[886]: resize2fs 1.44.5 (15-Dec-2018)
[24632.392594] resize2fs[886]: Resizing the filesystem on /dev/disk/by-partlabel/rootfs to 393216 (4k) blocks.
[24632.393354] resize2fs[886]: The filesystem on /dev/disk/by-partlabel/rootfs is now 393216 (4k) blocks long.
[24632.489107] systemctl[889]: Removed /etc/systemd/system/default.target.wants/resize-rootfs.service.
root@edison:~# df -h
Filesystem       Size  Used Avail Use% Mounted on
none             474M  4.0K  474M   1% /dev
/dev/mmcblk0p8   1.4G  992M  325M  76% /
tmpfs            478M     0  478M   0% /dev/shm
tmpfs            478M  620K  478M   1% /run
tmpfs            478M     0  478M   0% /sys/fs/cgroup
tmpfs            478M  4.0K  478M   1% /tmp
tmpfs            478M     0  478M   0% /var/volatile
/dev/mmcblk1p1    28G  2.6G   24G  10% /media/sdcard
/dev/mmcblk0p5  1003K   19K  913K   3% /factory
/dev/mmcblk0p10  1.3G   15M  1.3G   2% /home
tmpfs             96M     0   96M   0% /run/user/0

</pre>

closed 2020-02-21 by xlla