edison-fw/meta-intel-edison #88: /boot -> No such device

mwallnoefer, 2020-02-22:
With the latest Yocto (thud?) I get:
```
root@edison:/# cd /boot
-sh: cd: /boot: No such device
```

htot, 2020-02-22:
We are now at warrior.

Where does your fstab point to?

mwallnoefer, 2020-02-22:
Here my whole _fstab_:
```
/dev/mmcblk1    /          auto    nodev,noatime,discard,barrier=1,data=ordered,noauto_da_alloc    1  1
proc            /proc      proc    defaults              0  0
devpts          /dev/pts   devpts  mode=0620,gid=5       0  0
tmpfs           /run               tmpfs      mode=0755,nodev,nosuid,strictatime 0  0
tmpfs           /var/volatile      tmpfs      defaults              0  0

/dev/disk/by-partlabel/factory  /factory        auto    noauto,x-systemd.automount,nosuid,nodev,noatime,discard     1   1
/dev/mmcblk0p9     /boot       auto    noauto,x-systemd.automount,nosuid,nodev,noatime,discard     1   1
/dev/disk/by-partlabel/home     /home       auto    noauto,x-systemd.automount,nosuid,nodev,noatime,discard,barrier=1,data=ordered,noauto_da_alloc 1   1
```

mwallnoefer, 2020-02-22:
With this change everything works:
```
/dev/disk/by-partlabel/boot     /boot       auto    noauto,x-systemd.automount,nosuid,nodev,noatime,discard     1   1
```

mwallnoefer, 2020-02-22:
#92

htot, 2020-02-22:
Ah, I remember. I [reformatted ota](https://edison-fw.github.io/meta-intel-edison/2.2-Building-and-installing-the-kernel.html#installing-the-kernel) so I can easily put kernels there to boot a rootfs on usb or sdhc. It needs formatting otherwise U-Boot can't read the kernel from the partition. Apparently linux also doesn't like it as it was.

closed 2020-02-23 by mwallnoefer