edison-fw/meta-intel-edison #91: sketch_reset.service

mwallnoefer, 2020-02-22:
We should disable the _sketch_reset_ service if there is no GPIO access:
```
root@edison:/# systemctl status sketch-reset.service
Unit sketch-reset.service could not be found.
root@edison:/# systemctl status sketch_reset.service
● sketch_reset.service - Daemon to reset sketches
   Loaded: loaded (/lib/systemd/system/sketch_reset.service; enabled; vendor pr>
   Active: failed (Result: exit-code) since Fri 2020-02-21 06:56:08 UTC; 40min >
 Main PID: 552 (code=exited, status=1/FAILURE)

Feb 21 06:56:08 edison sketch_reset[552]: sh: line 0: echo: write error: Invali>
Feb 21 06:56:08 edison sketch_reset[552]: sh: /sys/class/gpio/gpio215/direction>
Feb 21 06:56:08 edison sketch_reset[552]: sh: /sys/class/gpio/gpio215/value: No>
Feb 21 06:56:08 edison sketch_reset[552]: sh: /sys/class/gpio/gpio215/value: No>
Feb 21 06:56:08 edison sketch_reset[552]: sh: line 0: echo: write error: Invali>
Feb 21 06:56:08 edison sketch_reset[552]: sh: /sys/class/gpio/gpio207/direction>
Feb 21 06:56:08 edison sketch_reset[552]: sh: /sys/class/gpio/gpio207/edge: No >
Feb 21 06:56:08 edison sketch_reset[552]: Failed to open:/sys/class/gpio/gpio20>
Feb 21 06:56:08 edison systemd[1]: sketch_reset.service: Main process exited, c>
Feb 21 06:56:08 edison systemd[1]: sketch_reset.service: Failed with result 'ex>
```

htot, 2020-02-22:
I see, needs to be fixed to use libgpio

mwallnoefer, 2020-02-23:
@htot Have you fixed also this one in your new _zeus_ branch? So I will re-test everything in one shot.

htot, 2020-02-23:
Not yet. I looking if I fix this to use libgpiod.

htot, 2020-02-24:
@mwallnoefer @xlla would you like to have a look at https://github.com/htot/meta-intel-edison/tree/sketch_reset_v1?

I rewrote most of galileo-reset.c to use libgpiod. This uses pin names instead of numbers, so on the command line it also changed. In the make file I had to discard some options to make errors go away, it's crazy. Now there are still many warnings, but only in the `clloader` code which was probably already the case. Ugly.

I tested this by running on edison:
```
root@edison:~# systemctl stop sketch_reset
root@edison:~# /opt/edison/sketch_reset -i SHLD_RESET0 -o SHLD_RESET1 -s /opt/edison/sketch_reset.sh
killall: clloader: no process killed
killall: clloader: no process killed
^C
root@edison:~# /opt/edison/sketch_reset -i SHLD_RESET0 -o SHLD_RESET1 -s /opt/edison/sketch_reset.sh -vv
Toggle reset line
Select event received : from GPIO interrupt pin
Sketch Reset button released: Calling /opt/edison/sketch_reset.sh
killall: clloader: no process killed
killall: clloader: no process killed
Select event received : from GPIO interrupt pin
Sketch Reset button pressed:
Select event received : from GPIO interrupt pin
Sketch Reset button released: Calling /opt/edison/sketch_reset.sh
killall: clloader: no process killed
killall: clloader: no process killed
```
I find at least I got button released and button push mixed up. Line toggle on start I didn't test, may be it's even to fast and needs a timer. For the reset seems to do what it needs - except I'm not sure what it needs :-)

mwallnoefer, 2020-02-25:
Will do. Please don't forget my battery-voltage service suggestions!

mwallnoefer, 2020-02-25:
Error message which I get when calling _bitbake_:
```
Failed to fetch URL file://btrfs.cfg
```

htot, 2020-02-25:
Oops. When I rebased kernel 5.5 recipe patch it must have sneaked in that line. I'll fix that too. In the meanwhile you can remove [that](https://github.com/htot/meta-intel-edison/blob/ae79a4fcda656e4dfbca0ccf3ca9862387fde427/meta-intel-edison-bsp/recipes-kernel/linux/linux-yocto_5.5.0.bb#L17). It is supposed to enable btrfs in the kernel, something you don't need now.

mwallnoefer, 2020-02-25:
Ok, other issue:
```
ERROR: oobe-1.2.1+gitAUTOINC+250de53880-r0 do_package_qa: QA Issue: /usr/bin/configure_edison contained in package oobe requires /usr/bin/python, but no providers found in RDEPENDS_oobe? [file-rdeps]
ERROR: oobe-1.2.1+gitAUTOINC+250de53880-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: oobe-1.2.1+gitAUTOINC+250de53880-r0 do_package_qa: 
ERROR: oobe-1.2.1+gitAUTOINC+250de53880-r0 do_package_qa: Function failed: do_package_qa
```

htot, 2020-02-25:
I see, another moved too far down. I think this will go away if you revert eadb3a8.

mwallnoefer, 2020-02-26:
Okay, but now I still get a build error with `perf`.
[log.do_compile.31056.txt](https://github.com/edison-fw/meta-intel-edison/files/4257177/log.do_compile.31056.txt)

htot, 2020-02-26:
This might be caused by an old perf? I think you are compiling using `make image`? That is good as it cleans up yocto and kernel before bitbaking the reset. But you might want to try to manually clean perf too:
```
cd out/linux64/build
source poky/oe-init-build-env
bitbake -c cleansstate perf
```

mwallnoefer, 2020-02-27:
Did not work.

htot, 2020-02-27:
I found `make setup` does not complete (zeus updated causing patch to fail). Then `meta-acpi` compatibility not yet set for zeus. So, pushed 1 patch on meta-acpi (automatically pulled with `make setup`) and one patch on meta-intel-edison.

So best is if you `make clean && make setup && make image` after pulling meta-intel-edison.

I am now at 50% (perf not yet built), need another hour.

mwallnoefer, 2020-03-02:
Okay will re-try the build!

andy-shev, 2020-03-02:
`perf` tool should come from the kernel sources.

xlla, 2020-03-03:
it works well.
<pre>
root@edison:~# systemctl status sketch_reset.service
● sketch_reset.service - Daemon to reset sketches
   Loaded: loaded (/lib/systemd/system/sketch_reset.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-02-15 17:20:07 UTC; 2h 22min ago
 Main PID: 567 (sketch_reset)
   CGroup: /system.slice/sketch_reset.service
           └─567 /opt/edison/sketch_reset -i SHLD_RESET0 -o SHLD_RESET1 -s /opt/edison/sketch_reset.sh

Feb 15 17:20:07 edison systemd[1]: Started Daemon to reset sketches.
root@edison:~# 

</pre>

htot, 2020-03-03:
@andy-shev yes, there is a recipe for kernel and a recipe for perf. After you built both and then switch kernel and rebuild it seems some artifacts from kernel remain interfere with building perf. Cleaning everything, then building seems to help. It's a Yocto problem i think.

mwallnoefer, 2020-03-05:
Tried another full rebuild, but always ending up in the `perf` issue:
```
make: Leaving directory '/datadisk/edison-fw/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/tools/perf'
WARNING: exit code 1 from a shell command.
```
Referring to your last comment, how may I circumvent this?

htot, 2020-03-05:
That is weird. I meant `bitbake -c cleansstate perf` cleans perf, while `make image` cleans linux and u-boot and then starts build (all in the right directory).

But if you did `make clean && make setup && make image` it doesn't get cleaner. I have done this (64 bit) and then again (32 bit) on another machine with no errors. In all cases the host is Ubuntu 19.10.

mwallnoefer, 2020-03-06:
That's the bug: https://www.mail-archive.com/openembedded-core@lists.openembedded.org/msg133569.html

htot, 2020-03-06:
Yes, but strange that I can build without problem. As I understand the patch makes the native (yocto built) python to be used and not the host python?

mwallnoefer, 2020-03-06:
My host is still on Ubuntu 18.04 LTS, this might make the difference (you're on Ubuntu 19.10).

mwallnoefer, 2020-03-06:
Btw. is this a new service? Does it make sense to have it enabled per default?
```
[FAILED] Failed to start Set timezone by geolocation service.
See 'systemctl status run-timezone.service' for details.
```

htot, 2020-03-06:
Run once should be enough, if it succeeds. I haven't figured out what to do here exactly. I just wanted to have time zone set automatically. Did it fail for you?
```
root@edison:~# systemctl start run-timezone.service
root@edison:~# systemctl status run-timezone.service
● run-timezone.service - Set timezone by geolocation service
   Loaded: loaded (/lib/systemd/system/run-timezone.service; disabled; vendor preset: enabled)
   Active: inactive (dead)

Mar 06 18:18:54 edison systemd[1]: Starting Set timezone by geolocation service...
Mar 06 18:18:54 edison run-timezone[1136641]: PING ipapi.co (104.26.8.44): 56 data bytes
Mar 06 18:18:54 edison run-timezone[1136641]: --- ipapi.co ping statistics ---
Mar 06 18:18:54 edison run-timezone[1136641]: 1 packets transmitted, 1 packets received, 0% packet loss
Mar 06 18:18:54 edison run-timezone[1136641]: round-trip min/avg/max = 16.242/16.242/16.242 ms
Mar 06 18:18:55 edison systemd[1]: run-timezone.service: Succeeded.
Mar 06 18:18:55 edison systemd[1]: Started Set timezone by geolocation service.
root@edison:~# timedatectl 
               Local time: Fri 2020-03-06 18:23:45 CET
           Universal time: Fri 2020-03-06 17:23:45 UTC
                 RTC time: Thu 2000-01-13 18:16:02
                Time zone: Europe/Amsterdam (CET, +0100)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
```

I'll add a patch to fix perf.

mwallnoefer, 2020-03-06:
It fails until you enable connectivity (wifi) with a subsequent reboot respectively service restart operation.

Also here I would not make the service fail (terminate with <>0 exit code), since it is not an exceptional behaviour when the connectivity hasn't been set up yet.

htot, 2020-03-06:
Got it

mwallnoefer, 2020-03-07:
Okay, so back to `sketch_reset.service`, this is what I get on branch _zeus_:
```
root@edison:~# /opt/edison/sketch_reset -i SHLD_RESET0 -o SHLD_RESET1 -s /opt/edison/sketch_reset.sh
Shield reset input GPIO invalid or not specified
```
How do I set up the pin mapping with _libgpiod_?

htot, 2020-03-07:
I force pushed. Are you on 58e37e72?

There is nothing to map, as it is mapped already through acpi tables. You can see that using `gpioinfo`.

mwallnoefer, 2020-03-07:
Yes, I am on that commit. But in `gpioinfo` I do not see these ports mapped.

htot, 2020-03-07:
```
root@edison:~# gpioinfo | grep "SHLD"
        line   7: "SHLD_RESET0" unused input active-high 
        line  15: "SHLD_RESET1" unused output active-high 
```
Afaik these are defined in a aml in U-Boot. What do you have when you just `gpioinfo`?

mwallnoefer, 2020-03-08:
[gpioinfo.log](https://github.com/edison-fw/meta-intel-edison/files/4303030/gpioinfo.log)

htot, 2020-03-08:
Wow, you are missing gpiochip1 (see mine below).
It might be fixed by doing a cold boot (unplug power).

<details>
<summary>gpioinfo.log</summary>

```
root@edison:~# gpioinfo
gpiochip0 - 192 lines:
        line   0:      unnamed       unused   input  active-high 
        line   1:      unnamed       unused   input  active-high 
        line   2:      unnamed       unused  output  active-high 
        line   3:      unnamed       unused   input  active-high 
        line   4:      unnamed       unused  output  active-high 
        line   5:      unnamed       unused  output  active-high 
        line   6:      unnamed       unused  output  active-high 
        line   7:      unnamed       unused   input  active-high 
        line   8:      unnamed       unused   input  active-high 
        line   9:      unnamed       unused  output  active-high 
        line  10:      unnamed       unused  output  active-high 
        line  11:      unnamed       unused  output  active-high 
        line  12:      unnamed       unused  output  active-high 
        line  13:      unnamed       unused  output  active-high 
        line  14:      unnamed       unused   input  active-high 
        line  15:      unnamed  "interrupt"   input  active-high [used]
        line  16:      unnamed       unused   input  active-high 
        line  17:      unnamed       unused   input  active-high 
        line  18:      unnamed       unused   input  active-high 
        line  19:      unnamed       unused   input  active-high 
        line  20:      unnamed       unused   input  active-high 
        line  21:      unnamed       unused   input  active-high 
        line  22:      unnamed       unused   input  active-high 
        line  23:      unnamed       unused   input  active-high 
        line  24:      unnamed       unused   input  active-high 
        line  25:      unnamed       unused   input  active-high 
        line  26:      unnamed       unused   input  active-high 
        line  27:      unnamed       unused   input  active-high 
        line  28:      unnamed       unused   input  active-high 
        line  29:      unnamed       unused   input  active-high 
        line  30:      unnamed       unused   input  active-high 
        line  31:      unnamed       unused   input  active-high 
        line  32:      unnamed       unused   input  active-high 
        line  33:      unnamed       unused   input  active-high 
        line  34:      unnamed       unused   input  active-high 
        line  35:      unnamed       unused   input  active-high 
        line  36:      unnamed       unused   input  active-high 
        line  37:      unnamed       unused   input  active-high 
        line  38:      unnamed       unused   input  active-high 
        line  39:      unnamed       unused   input  active-high 
        line  40:      unnamed  "heartbeat"  output  active-high [used]
        line  41:      unnamed       unused   input  active-high 
        line  42:      unnamed       unused   input  active-high 
        line  43:      unnamed       unused   input  active-high 
        line  44:      unnamed       unused   input  active-high 
        line  45:      unnamed       unused   input  active-high 
        line  46:      unnamed       unused   input  active-high 
        line  47:      unnamed       unused   input  active-high 
        line  48:      unnamed       unused   input  active-high 
        line  49:      unnamed       unused   input  active-high 
        line  50:      unnamed       unused  output  active-high 
        line  51:      unnamed       unused   input  active-high 
        line  52:      unnamed       unused   input  active-high 
        line  53:      unnamed       unused   input  active-high 
        line  54:      unnamed       unused  output  active-high 
        line  55:      unnamed       unused   input  active-high 
        line  56:      unnamed       unused   input  active-high 
        line  57:      unnamed       unused   input  active-high 
        line  58:      unnamed       unused   input  active-high 
        line  59:      unnamed       unused   input  active-high 
        line  60:      unnamed       unused   input  active-high 
[... 181 lines cut ...]
        line  15:  "U16_IO1.7"       unused   input  active-high 
gpiochip4 - 16 lines:
        line   0:  "MUX33_DIR" "uart1-rx-oe" output active-high [used]
        line   1:  "MUX31_DIR" "uart1-tx-oe" output active-high [used]
        line   2:  "MUX29_DIR"       unused   input  active-high 
        line   3:  "MUX27_DIR"       unused   input  active-high 
        line   4:  "MUX24_DIR"       unused   input  active-high 
        line   5:  "MUX21_DIR"       unused   input  active-high 
        line   6:  "MUX19_DIR"       unused   input  active-high 
        line   7:  "MUX32_DIR"       unused   input  active-high 
        line   8:  "MUX30_DIR"       unused   input  active-high 
        line   9:  "MUX28_DIR"       unused   input  active-high 
        line  10:  "MUX26_DIR"       unused   input  active-high 
        line  11:  "MUX23_DIR"       unused   input  active-high 
        line  12:  "MUX20_DIR"       unused   input  active-high 
        line  13:  "MUX18_DIR"       unused  output  active-high 
        line  14:  "MUX22_SEL"       unused   input  active-high 
        line  15:  "MUX25_SEL"       unused   input  active-high 
gpiochip5 - 4 lines:
        line   0:      unnamed       unused   input  active-high 
        line   1:      unnamed       unused   input  active-high 
        line   2:      unnamed       unused   input  active-high 
        line   3:      unnamed       unused   input  active-high
```
</details>

mwallnoefer, 2020-03-10:
That did it, thanks!

closed 2020-03-10 by mwallnoefer