edison-fw/meta-intel-edison #93: can we support I2S

xlla, 2020-02-23:
In edison-arduino-hardware-guide.pdf, it said 

``` Some additional functions are available on certain SoC pins, such as I2S and UART flow control, but they are not currently supported by the Arduino library. However, it may be possible to use these from Linux.

but for now, an [Arudino I2S library](https://www.arduino.cc/en/Reference/I2S) is exist, how to add an i2s feature.

plbossart, 2020-06-02:
IIRC you need to enable the SSP2 pins. In the past I did this with a script, not sure how we would do this now?

htot, 2020-12-23:
Correct me if I'm wrong.

We now have a ACPI_FEATURES i2s in [meta-acpi](https://github.com/edison-fw/meta-acpi/commit/068da094f91d241529f5f9e8e02633c5604ea66e). This can be enabled in the [edison.conf](https://github.com/edison-fw/meta-intel-edison/blob/d06bc2177e3b5bb2fa6c8d4805e7d904f30d6e8e/meta-intel-edison-bsp/conf/machine/edison.conf#L31).

When enabled the acpi tables (notably arduino.asl) will be build with MUX_I2S defined and the I2S will be active with sof codec nocodec enabled in the kernel.

So, in a sense we are supporting i2s already.

Iirc the normal way to proceed would be:

1. enable ACPI_FEATURES i2s
2. add a shield with a codec on the edison-arduino.
3. add the driver to the kernel in a kernel fragment (like [here](https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-bsp/recipes-kernel/linux/files/sof_nocodec.cfg))
4. install firmware for the codec in needed (like [here](https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-bsp/recipes-kernel/sof/sof-fw.bb))

If so can we close this here with reference to https://thesofproject.github.io/latest/index.html?

Or do we need to write a little instruction? @xlla do you have hardware that you can test?

plbossart, 2021-01-11:
> * add the driver to the kernel in a kernel fragment (like [here](https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-bsp/recipes-kernel/linux/files/sof_nocodec.cfg))

yes, but this would need to add the relevant configurations for the machine driver and codec. It's no longer SOF-only.

> * install firmware for the codec if needed (like [here](https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-bsp/recipes-kernel/sof/sof-fw.bb))

You are missing the part where we need an ACPI snippet to describe the codec. Here's what we use for the Up2 board with the initrd override method:
https://github.com/thesofproject/acpi-scripts/blob/master/Up2/PCM512X.asl