edison-fw/meta-intel-edison #94: exception after long time running

xlla, 2020-02-23:
I left edison running there for about 12 hours, then I saw some exception raise.

<pre>
root@edison:~# [ 6029.496802] ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
[ 6029.507425] brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
[ 6029.513721] brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
[ 6029.523112] ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
[ 6029.531197] ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)

</pre>

htot, 2020-02-23:
Never seen that before (mine is running continuously).

xlla, 2020-03-05:
as I am using it, errors appears again.
<pre>
root@edison:/media/sdcard/study/git/luma.oled# python3 -m pip install --upgrade Pillow
[12061.509152] ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
[12061.517562] brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
[12061.523921] brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
[12061.536441] ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
[12061.544571] ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)

Collecting Pillow
  Downloading https://pypi....
</pre>

xlla, 2020-03-05:
in reboot logs
<pre>
[  OK  ] Started Create Volatile Files and Directories.
[    9.759668] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43340-sdio for chip BCM43340/2
[    9.927663] sof-audio-pci 0000:00:0d.0: error: no matching ASoC machine driver found - aborting probe
[    9.937150] sof-audio-pci 0000:00:0d.0: error: failed to get machine info -19
[    9.944510] sof-audio-pci 0000:00:0d.0: error: failed to probe DSP hardware!
[   10.044346] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43340-sdio for chip BCM43340/2
[   10.076572] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   10.149253] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43340/2 wl0: Oct 23 2017 08:41:23 version 6.10.190.70 (r674464) FWID 01-98d71006
[  OK  ] Found device Merrifield Serial IO HSUART Controller.
[
</pre>

htot, 2020-03-05:
Your last log is not related. But I can reproduce this directly after link comes up:
```
Mar 05 20:21:25 edison kernel: IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Mar 05 20:21:25 edison systemd-networkd[576]: wlan0: Gained carrier
Mar 05 20:21:26 edison systemd-networkd[576]: wlan0: Gained IPv6LL
Mar 05 20:21:29 edison kernel: ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
Mar 05 20:21:29 edison kernel: brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
Mar 05 20:21:29 edison kernel: brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
Mar 05 20:21:29 edison kernel: ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
Mar 05 20:21:29 edison kernel: ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)
```

This on:
```
root@edison:~# uname -a
Linux edison 5.5.0-edison-acpi-standard #1 SMP Sun Feb 16 17:32:42 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

Maybe @andy-shev knows.

andy-shev, 2020-03-05:
No idea, I didn't use WiFi for few month. But Broadcom usually responsive in the mailing list(s), so, I would recommend to drop an email to linux-wireless@ mailing list with Cc to the driver maintainer.

htot, 2020-03-05:
@xlla does this happen in linux v5.4.0 too? Or is it new for v5.5.0?

xlla, 2020-03-07:
I think it occur in both.
I have switched to v5.5.0 recently because @andy-shev said that fbtft should work in v5.5 otherwise many patches must be apply.
when I first notice those errors , it must be on v5.4.0.

htot, 2020-03-07:
It doesn't seem to be harmful, I see it only once in journalctl -b (once per boot)

xlla, 2020-03-09:
@htot   I saw some articles said
<pre>
Arduino pin 7 can sometimes negatively impact   the WiFi capability, if using WiFi avoid using this pin
</pre>
 does that D7 interference exist in our new system? it is HW issue or software cause?

htot, 2020-03-09:
Not that I know of. Wifi and BT are internal to the Edison module.

mwallnoefer, 2020-03-28:
These messages appear also when switching from WiFi client to AP mode and vice-versa, but I did not notice any downtime. So I guess they're not causing any harm...

htot, 2021-06-12:
I don't see these messages when using linux 5.10.41, so whatever it was, it seems fixed upstream. Feel free to reopen if needed.

closed 2021-06-12 by htot

leshniak, 2025-03-30:
I'm also experiencing the issue on kernel 5.15 and 6.1. It spams a lot.

```
...
[nie mar 30 06:38:45 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
[nie mar 30 06:38:45 2025] ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)
[nie mar 30 06:40:49 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
[nie mar 30 06:40:49 2025] brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
[nie mar 30 06:40:49 2025] brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
[nie mar 30 06:40:49 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
[nie mar 30 06:40:49 2025] ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)
[nie mar 30 06:45:49 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
[nie mar 30 06:45:49 2025] brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
[nie mar 30 06:45:49 2025] brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
[nie mar 30 06:45:49 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
[nie mar 30 06:45:49 2025] ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)
[nie mar 30 06:46:56 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
[nie mar 30 06:46:56 2025] brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
[nie mar 30 06:46:56 2025] brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
[nie mar 30 06:46:56 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
[nie mar 30 06:46:56 2025] ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)
[nie mar 30 06:47:39 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
[nie mar 30 06:47:39 2025] brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
[nie mar 30 06:47:39 2025] brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
[nie mar 30 06:47:39 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
[nie mar 30 06:47:39 2025] ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)
[nie mar 30 06:49:43 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
[nie mar 30 06:49:43 2025] brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
[nie mar 30 06:49:43 2025] brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
[nie mar 30 06:49:43 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
[nie mar 30 06:49:43 2025] ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)
[nie mar 30 06:51:03 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
[nie mar 30 06:51:03 2025] brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
[nie mar 30 06:51:03 2025] brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
[nie mar 30 06:51:03 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
[nie mar 30 06:51:03 2025] ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)
[nie mar 30 06:54:52 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
[nie mar 30 06:54:52 2025] brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
[nie mar 30 06:54:52 2025] brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
[nie mar 30 06:54:52 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
[nie mar 30 06:54:52 2025] ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)
[nie mar 30 06:57:10 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
[nie mar 30 06:57:10 2025] brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
[nie mar 30 06:57:10 2025] brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
[nie mar 30 06:57:10 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
[nie mar 30 06:57:10 2025] ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)
[nie mar 30 06:58:16 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
[nie mar 30 06:58:16 2025] brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
[nie mar 30 06:58:16 2025] brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
[nie mar 30 06:58:16 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
[nie mar 30 06:58:16 2025] ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)
[nie mar 30 06:59:08 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
[nie mar 30 06:59:08 2025] brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
[nie mar 30 06:59:08 2025] brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
[nie mar 30 06:59:08 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
[nie mar 30 06:59:08 2025] ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)
[nie mar 30 07:07:49 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
[nie mar 30 07:07:49 2025] brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
[nie mar 30 07:07:49 2025] brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
[... cut ...]
[nie mar 30 07:07:49 2025] ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)
[nie mar 30 07:08:24 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
[nie mar 30 07:08:24 2025] brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
[nie mar 30 07:08:24 2025] brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
[nie mar 30 07:08:24 2025] ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
[nie mar 30 07:08:24 2025] ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)
...
```

htot, 2025-03-31:
Could it be related to this:
https://patchwork.kernel.org/project/linux-wireless/patch/20170224163246.7748-1-zajec5@gmail.com/

leshniak, 2025-06-04:
I have still tons of those messages (830 occurrences in 12 hours, it affects the stability at this rate), now on `6.1.55-edison-acpi-standard`. Usually it starts after some time, reboot helps. Are you using systemd-networkd to setup the network? Or something different?

htot, 2025-06-04:
OK I have 2 edison's and use `iperf3 -s` on the one and `iperf3 --bidir -c 192.168.178.59 -t 0` on the other. Here and there retries happens. One has errors in the log:
```
Jun 04 18:44:01 edison kernel: ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
Jun 04 18:44:01 edison kernel: brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
Jun 04 18:44:01 edison kernel: brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
Jun 04 18:44:01 edison kernel: ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
Jun 04 18:44:01 edison kernel: ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)
Jun 04 18:54:12 edison kernel: ieee80211 phy0: brcmf_psm_watchdog_notify: PSM's watchdog has fired!
Jun 04 18:54:12 edison kernel: brcmfmac: brcmf_sdiod_ramrw: membytes transfer failed
Jun 04 18:54:12 edison kernel: brcmfmac: brcmf_sdio_bus_get_memdump: error -84 on reading 2048 membytes at 0x00000000
Jun 04 18:54:12 edison kernel: ieee80211 phy0: brcmf_psm_watchdog_notify: Failed to get memory dump, -84
Jun 04 18:54:12 edison kernel: ieee80211 phy0: brcmf_fweh_event_worker: event handler failed (41)
```
the other nothing (even after reversing roles. Without errors:
```
root@yuna:~# uname -a
Linux yuna 6.12.23-edison-acpi-preempt-rt #1 SMP PREEMPT_RT Thu Apr 10 12:39:41 UTC 2025 x86_64 GNU/Linux
```
with errors (is an older build) 6.12.0-rc5-edison-acpi-preempt-rt. So it could be something got finally fixed, or is just a coincidence. This is using our scarthgap.