edison-fw/meta-intel-edison #95: MRAA usage issues

shawnhsoia97, 2020-03-04:
Hi, I try to control gpio and spi  now,so I follow examples/c/spi.c and gpio.c.
But when i run it shows `MRAA: Invalid Handle.`
dmesg shows:
`edison kernel: export_store: invalid GPIO 214`

Then i found https://github.com/edison-fw/meta-intel-edison/issues/50#issue-369939660
This description:

> This indicates MRAA is using the wrong pin numbering - and in fact it should be using the pin names.
> Doing gpioinfo shows TRI_STATE_ALL is on gpiochip1 line 14.
> You find gpiochip1 under /sys/class/gpio/gpiochip496/device/, so the base address has become 496 and the gpio is 510.
> 
> We need to modify intel_edison_fab_c.c to use either libgpiod or the new gpio chardev interface directly to get mraa working.

But I not use Edison Arduino board,I use to my own circuit board,so I don't have  TRI_STATE_ALL on gpiochip1.

```
root@bovia-deb:/bovia/flashmcu# gpiofind TRI_STATE_ALL
root@bovia-deb:/bovia/flashmcu#
```

```
 root@bovia-deb:/# gpioinfo
gpiochip0 - 192 lines:
        line   0:      unnamed       unused   input  active-high
        line   1:      unnamed       unused   input  active-high
        line   2:      unnamed       unused  output  active-high
        line   3:      unnamed       unused   input  active-high
        line   4:      unnamed       unused  output  active-high
        line   5:      unnamed       unused  output  active-high
        line   6:      unnamed       unused  output  active-high
        line   7:      unnamed       unused   input  active-high
        line   8:      unnamed       unused   input  active-high
        line   9:      unnamed       unused  output  active-high
        line  10:      unnamed       unused  output  active-high
        line  11:      unnamed       unused  output  active-high
        line  12:      unnamed       unused  output  active-high
        line  13:      unnamed       unused  output  active-high
        line  14:      unnamed       unused   input  active-high
        line  15:      unnamed       unused   input  active-high
        line  16:      unnamed       unused   input  active-high
        line  17:      unnamed       unused   input  active-high
        line  18:      unnamed       unused   input  active-high
        line  19:      unnamed       unused   input  active-high
        line  20:      unnamed       unused   input  active-high
        line  21:      unnamed       unused   input  active-high
        line  22:      unnamed       unused   input  active-high
        line  23:      unnamed       unused   input  active-high
        line  24:      unnamed       unused   input  active-high
        line  25:      unnamed       unused   input  active-high
        line  26:      unnamed       unused   input  active-high
        line  27:      unnamed       unused   input  active-high
        line  28:      unnamed       unused   input  active-high
        line  29:      unnamed       unused   input  active-high
        line  30:      unnamed       unused   input  active-high
        line  31:      unnamed       unused   input  active-high
        line  32:      unnamed       unused   input  active-high
        line  33:      unnamed       unused   input  active-high
        line  34:      unnamed       unused   input  active-high
        line  35:      unnamed       unused   input  active-high
        line  36:      unnamed       unused   input  active-high
        line  37:      unnamed       unused   input  active-high
        line  38:      unnamed       unused   input  active-high
        line  39:      unnamed       unused   input  active-high
        line  40:      unnamed  "heartbeat"  output  active-high [used]
        line  41:      unnamed       unused   input  active-high
        line  42:      unnamed       unused   input  active-high
        line  43:      unnamed       unused   input  active-high
        line  44:      unnamed       unused   input  active-high
        line  45:      unnamed       unused   input  active-high
[... 133 lines cut ...]
        line 179:      unnamed       unused  output  active-high
        line 180:      unnamed       unused   input  active-high
        line 181:      unnamed       unused  output  active-high
        line 182:      unnamed       unused  output  active-high
        line 183:      unnamed       unused   input  active-high
        line 184:      unnamed "device-wakeup" output active-high [used]
        line 185:      unnamed "host-wakeup" input active-high [used]
        line 186:      unnamed       unused   input  active-high
        line 187:      unnamed       unused   input  active-high
        line 188:      unnamed       unused   input  active-high
        line 189:      unnamed       unused  output  active-high
        line 190:      unnamed       unused  output  active-high
        line 191:      unnamed       unused   input  active-high
gpiochip1 - 4 lines:
        line   0:      unnamed       unused   input  active-high
        line   1:      unnamed       unused   input  active-high
        line   2:      unnamed       unused   input  active-high
        line   3:      unnamed       unused   input  active-high

```
Do I have to modify my own circuit board?

Besides,I don't know how to modify [intel_edison_fab_c.c](https://github.com/intel-iot-devkit/mraa/blob/master/src/x86/intel_edison_fab_c.c),can you help me? Thanks.

Also do you know why insmod i2c-dev.ko on boot?

htot, 2020-03-04:
> But I not use Edison Arduino board,I use to my own circuit board,so I don't have TRI_STATE_ALL on gpiochip1.
> Do I have to modify my own circuit board?

No.
The description of your board should by in a acpi table. There is one for the Edison-Arduino. For the Edison breakout board none is needed, the tables are in U-Boot already. Maybe you board needs none, maybe it does. Examples [here](https://github.com/edison-fw/meta-acpi/tree/master/recipes-bsp/acpi-tables/samples/edison).

> Besides,I don't know how to modify [intel_edison_fab_c.c](https://github.com/intel-iot-devkit/mraa/blob/master/src/x86/intel_edison_fab_c.c),can you help me? Thanks.

Short answer: with an editor :-) From that create a patch, then let the recipe apply that patch.
Longer answer:  `intel_edison_fab_c.c` defines pins and their properties. Now we have libgpio, so we need to map pin numbers to pin names and manipulate using libgpio. I think the mraa repo already has examples for other boards doing this. 
Also I have a patch [here](https://github.com/htot/meta-intel-edison/commit/5bb27e317e76ad8ec72b0709c93c2e0fc4d68cc0) that ports the sketch-reset recipe to use `libgpiod`. This is an example of how to modify a recipe to apply a patch, and also how to port from using sysfs to using libgpiod.

> Also do you know why insmod i2c-dev.ko on boot?

I don't know, on my machine that doesn't happen:
```
root@edison:~# lsmod | grep i2c
root@edison:~# modprobe -i i2c-dev
root@edison:~# lsmod | grep i2c
i2c_dev                20480  0
```

shawnhsoia97, 2020-03-06:
Sorry,I still don't  understand,I just want to control the input and output of gpio,E.g
[image]
I want to control GP183 output low,so i use to` /usr/share/mraa/examples/c/gpio.c`,
```
/* gpio declaration */
#define GPIO_PIN_1 183 

status = mraa_gpio_write(gpio_1, 0);
        if (status != MRAA_SUCCESS) {
            goto err_exit;
        }
```
But run gpio.o show  `Failed to initialize GPIO 183`,dmesg show  `export_store: invalid GPIO 214`,
GPIO 214 is "TRI_STATE_ALL",but like I said I don't have "TRI_STATE_ALL",because I am not using  Edison Arduino board.[arduino.asli](https://github.com/edison-fw/meta-acpi/blob/master/recipes-bsp/acpi-tables/samples/edison/arduino.asli) should be defined on IO0~IO19 of Edison Arduino board? but I don't have this,I just want to control on gpio pin of  edison.
So do I still need modify  intel_edison_fab_c.c?

>  Also I have a patch here that ports the sketch-reset recipe to use libgpiod. This is an example of how to modify a recipe to apply a patch, and also how to port from using sysfs to using libgpiod.

Do you mean using libgpio not MRAA?then I still need modify  intel_edison_fab_c.c?
Also can I still use mraa spi without modifying anything?I have /dev/spidev5.1.

Also i try use to branch "zeus",but "make setup" is fail. show :
```
error: patch failed: meta/recipes-bsp/u-boot/u-boot.inc:85
error: meta/recipes-bsp/u-boot/u-boot.inc: patch does not apply
Makefile:25: recipe for target 'setup' failed
make: *** [setup] Error 1

```

shawnhsoia97, 2020-03-06:
Also I not found gpiod.h
```
gpio183.c:1:10: fatal error: gpiod.h: No such file or directory
 #include <gpiod.h>
```

htot, 2020-03-06:
> I want to control GP183 output low,so i use to` /usr/share/mraa/examples/c/gpio.c`,
MRAA needs to be fixed, or you need to use libgpiod instead.
> GPIO 214 is "TRI_STATE_ALL",but like I said I don't have "TRI_STATE_ALL",because I am not using Edison Arduino [...]

MRAA is not working and needs to be fixed

> So do I still need modify intel_edison_fab_c.c?

If you want to use MRAA, yes.
> > Also I have a patch here that ports the sketch-reset recipe to use libgpiod. This is an example of how to modify a recipe to apply a patch, and also how to port from using sysfs to using libgpiod.
> 
> Do you mean using libgpio not MRAA?then I still need modify intel_edison_fab_c.c?

You can use libgpiod directly, but if you want to use MRAA (to keep existing code working), one way would be to change intel_edison_fab_c.c to translate pin numbers into the pins chip/line, and use libgpiod inside to access that.

The way to access pins (at the low level) has changed completely, that does not depend on the arduino board. See otherwise the changes in the blink-led code.

> Also can I still use mraa spi without modifying anything?I have /dev/spidev5.1.

That I haven't tried. MRAA serial works, because I got a patch upstreamed.
I2C too, https://htot.github.io/meta-intel-edison/4.5-mraa-upm.html

> Also i try use to branch "zeus",but "make setup" is fail. 

Yes, sorry about that. I am testing the sketch_reset_v2 branch, it builds without errors.

shawnhsoia97, 2020-03-06:
Thank you.So if i want to modify intel_edison_fab_c.c ,how do i know the right pin now?E.g
```
static mraa_intel_edison_pinmodes_t pinmodes[MRAA_INTEL_EDISON_PINCOUNT];
static unsigned int outputen[] = { 248, 249, 250, 251, 252, 253, 254, 255, 256, 257,
                                   258, 259, 260, 261, 232, 233, 234, 235, 236, 237 };
static mraa_gpio_context agpioOutputen[sizeof(outputen) / sizeof(outputen[0])];

static unsigned int pullup_map[] = { 216, 217, 218, 219, 220, 221, 222, 223, 224, 225,
                                     226, 227, 228, 229, 208, 209, 210, 211, 212, 213 };
```
Originally should follow
[image]
How should I find the right pin now?
```
root@edison:/home/testgpio# ls /sys/class/gpio/
export       gpiochip0/   gpiochip508/ unexport
```
I just know GPX number now.follow  Edison Pinout(linux GPIO pin),but i don't know current Linux pin of gpio pin mux and output enable liinux pin and pullup enable linux pin.
So how do find that?Thank

> Yes, sorry about that. I am testing the sketch_reset_v2 branch, it builds without errors.

Many thanks for your hard work,I wait for your good news.^_^

Also I not include <gpiod.h>,I install libgpiod,but that still not work.

htot, 2020-03-06:
You might use the blink-led python as an example. Here we replaced:
```
# Init GPIO mux for LED control
subprocess.call("""
echo 40 >/sys/class/gpio/export
echo 214 >/sys/class/gpio/export
echo 243 >/sys/class/gpio/export
echo 261 >/sys/class/gpio/export
echo high >/sys/class/gpio/gpio214/direction
echo mode0 > /sys/kernel/debug/gpio_debug/gpio40/current_pinmux
echo low >/sys/class/gpio/gpio243/direction
echo high >/sys/class/gpio/gpio261/direction
echo low >/sys/class/gpio/gpio214/direction""", shell=True)
```
by
```
set_line("TRI_STATE_ALL", 0)
set_line("SPI_CLK_SEL", 0)
set_line("MUX18_DIR", 1)
set_line("TRI_STATE_ALL", 0)
led = gpiod.Chip("gpiochip0").get_line(40)
led.request(consumer=led.owner().name(), type=gpiod.LINE_REQ_DIR_OUT)
if led.get_value() == 0:
    initial_led_state = "low"
else:
    initial_led_state = "high"
```
and
```
def write_led(value):
    with open("/sys/class/gpio/gpio40/direction","w") as lf:
        lf.write(value)
```
by
```
def write_led(value):
    if value == "high":
        level = 1
    else:
        level = 0
    led.set_value(level)
```
So the problem is not really with gpiochip0, but with the others as the pins received names from the signal names in the schematic.

shawnhsoia97, 2020-03-09:
Hi @htot ,I test blink-led,but not found gpiod.
```
root@edison:/usr/bin# python blink-led
Traceback (most recent call last):
  File "blink-led", line 35, in <module>
    import gpiod
ImportError: No module named gpiod
```
I have installed libgpiod,but this not work.
How to install gpiod?
I try to pip install gpiod and libgpiod ,but failed.
```
root@edison:/usr/bin# pip3 install gpiod
ERROR: Could not find a version that satisfies the requirement gpiod (from versions: none)
ERROR: No matching distribution found for gpiod
root@edison:/usr/bin# pip3 install libgpiod
ERROR: Could not find a version that satisfies the requirement libgpiod (from versions: none)
ERROR: No matching distribution found for libgpiod

```

htot, 2020-03-09:
It is already installed. Don't use pip.

I don't understand what image you have.

shawnhsoia97, 2020-03-09:
my edison-image.bb
```
shawn@bovia-dev:/mnt/yocto/yocto_warrior$ cat meta-intel-edison/meta-intel-edison-distro/recipes-core/images/edison-image.bb
require edison-image-minimal.bb

DESCRIPTION = "A fully functional image to run EDISON"
KERNEL_IMAGETYPE_pn-edison-image = "bzImage-initramfs-edison.bin"

#edison setup
inherit extrausers
EXTRA_USERS_PARAMS = "\
        useradd -p '' tester; \
        "
GLIBC_GENERATE_LOCALES = "en_GB.UTF-8 en_US.UTF-8"
IMAGE_LINGUAS = "de-de fr-fr en-gb en-us"

#IMAGE_INSTALL_append = " connman"
#IMAGE_INSTALL_append = " connman-client"
#IMAGE_INSTALL_append = " connman-tools"
# IMAGE_INSTALL_append = " connman-init-systemd"
IMAGE_INSTALL_append = " ap-mode-toggle"
IMAGE_INSTALL_append = " ethtool"
IMAGE_INSTALL_append = " pwr-button-handler"
IMAGE_INSTALL_append = " blink-led"
IMAGE_INSTALL_append = " post-install"
IMAGE_INSTALL_append = " resize-rootfs"
IMAGE_INSTALL_append = " bluez5-dev"
IMAGE_INSTALL_append = " bluez5-noinst-tools"
IMAGE_INSTALL_append = " bluez5-testtools"
IMAGE_INSTALL_append = " bluez5-obex"
IMAGE_INSTALL_append = " systemd-analyze"
# Allows to enable open pam feature
IMAGE_INSTALL_append = " libpam"
# Allows to enable OpenMP feature
IMAGE_INSTALL_append = " libgomp"

# Add audio firmware
IMAGE_INSTALL_append = " sst-fw-bin"

# ALSA lib and utilities
#IMAGE_INSTALL_append = " alsa-lib"
IMAGE_INSTALL_append = " alsa-utils-alsamixer alsa-utils-alsactl alsa-utils-aplay alsa-utils-amixer"

# OOBE
IMAGE_INSTALL_append = " run-timezone"

# Python and some basic modules
# removed python-imaging python-pygobject python-dbus
IMAGE_INSTALL_append = " python"
IMAGE_INSTALL_append = " python-distutils python-pkgutil python-audio python-image python-email python-netserver python-xmlrpc python-ctypes python-html python-json python-compile python-misc python-numbers python-unittest python-pydoc"

IMAGE_INSTALL_append = " crashlog"

# Adds various other tools
IMAGE_INSTALL_append = " tcpdump"
IMAGE_INSTALL_append = " net-tools"
IMAGE_INSTALL_append = " iperf3"

# Add pulseaudio
IMAGE_INSTALL_append = " pulseaudio-server libpulsecore libpulsecommon libpulse libpulse-simple pulseaudio-misc"

#Add Gstreamer
IMAGE_INSTALL_append = " gstreamer1.0 gstreamer1.0-meta-base gstreamer1.0-meta-audio"

# Edison Arduino stuff
IMAGE_INSTALL_append = " clloader"
IMAGE_INSTALL_append = " sketch-check"

# Edison Middleware stuff
IMAGE_INSTALL_append = " packagegroup-core-buildessential"
IMAGE_INSTALL_append = " zeromq-dev"
IMAGE_INSTALL_append = " cppzmq-dev"
IMAGE_INSTALL_append = " python-paho-mqtt-dev"
IMAGE_INSTALL_append = " libnss-mdns-dev"
#IMAGE_INSTALL_append = " xdk-daemon"
IMAGE_INSTALL_append = " oobe"

# mosquitto and dependencies
IMAGE_INSTALL_append = " mosquitto-dev"
IMAGE_INSTALL_append = " mosquitto-clients"

# node and sub-components
IMAGE_INSTALL_append = " nodejs-dev"
IMAGE_INSTALL_append = " nodejs-npm"

# MRAA
IMAGE_INSTALL_append = " mraa-dev"
IMAGE_INSTALL_append = " mraa-doc"

# UPM
IMAGE_INSTALL_append = " upm-dev"

# INTEL MCU FW
IMAGE_INSTALL_append = " mcu-fw-load"
IMAGE_INSTALL_append = " mcu-fw-bin"

# nfs
#IMAGE_INSTALL_append = " nfs-utils"

# Add oFono
IMAGE_INSTALL_append = " ofono"

# Add battery level detection
IMAGE_INSTALL_append = " battery-voltage"

# Add linux performance monitoring tool
IMAGE_INSTALL_append = " perf"

# package management - where to find this?
#IMAGE_INSTALL_append = " aptitude"

# Provides strace and gdb
# IMAGE_FEATURES += "tools-debug" fail with thud

# Add the nano text editor
IMAGE_INSTALL_append = " nano"

# SWIG
IMAGE_INSTALL_append = " swig"

# Add monitoring tools
IMAGE_INSTALL_append = " iotop"
IMAGE_INSTALL_append = " powertop"

# Clean corrupted journald entries
IMAGE_INSTALL_append = " cleanjournal"

#Customized
IMAGE_INSTALL_append = " v4l-utils"
IMAGE_INSTALL_append = " gnupg"
IMAGE_INSTALL_append = " minicom"
IMAGE_INSTALL_append = " hostapd"
IMAGE_INSTALL_append = " dnsmasq"
IMAGE_INSTALL_append = " usb-modeswitch"
IMAGE_INSTALL_append = " usb-modeswitch-data"
IMAGE_INSTALL_append = " wvdial"
IMAGE_INSTALL_append = " ppp"
IMAGE_INSTALL_append = " ppp-dialin"
IMAGE_INSTALL_append = " libqmi"
IMAGE_INSTALL_append = " kernel-devsrc"
IMAGE_INSTALL_append = " cmake"
IMAGE_INSTALL_append = " opkg"
IMAGE_INSTALL_append = " util-linux"
IMAGE_INSTALL_append = " gawk"
IMAGE_INSTALL_append = " openssh"
IMAGE_INSTALL_append = " libnetfilter-conntrack"
IMAGE_INSTALL_append = " nginx"
IMAGE_INSTALL_append = " exfat-utils"
IMAGE_INSTALL_append = " fuse-exfat"
IMAGE_INSTALL_append = " fuse"
IMAGE_INSTALL_append = " gunicorn"
IMAGE_INSTALL_append = " python3-pip"
IMAGE_INSTALL_append = " python3-dbus"
IMAGE_INSTALL_append = " python3-flask"
IMAGE_INSTALL_append = " python3-pika"
IMAGE_INSTALL_append = " python3-flask-login"
IMAGE_INSTALL_append = " python3-flask-sqlalchemy"
IMAGE_INSTALL_append = " python3-flask-restful"
IMAGE_INSTALL_append = " python3-gevent"
IMAGE_INSTALL_append = " python3-psutil"
IMAGE_INSTALL_append = " rtc-3028"
IMAGE_INSTALL_append = " language"
IMAGE_INSTALL_append = " openvpn"
IMAGE_INSTALL_append = " openconnect"
IMAGE_INSTALL_append = " zlib"
IMAGE_INSTALL_append = " softether"
IMAGE_INSTALL_append = " libtext-iconv-perl"
IMAGE_INSTALL_append = " jq"
IMAGE_INSTALL_append = " networkmanager"
IMAGE_INSTALL_append = " alsa-lib"
IMAGE_INSTALL_append = " alsa-utils"
IMAGE_INSTALL_append = " pulseaudio"
IMAGE_INSTALL_append = " tzdata"
IMAGE_INSTALL_append = " gdb"
IMAGE_INSTALL_append = " networking"
IMAGE_INSTALL_append = " libgpiod"
IMAGE_INSTALL_append = " python3-spidev"

```

shawnhsoia97, 2020-03-09:
Sorry I used python before,I should use to python3.

but this show the error now.
```
root@edison:/usr/bin# python3 blink-led
Traceback (most recent call last):
  File "blink-led", line 81, in <module>
    set_line("TRI_STATE_ALL", 0)
  File "blink-led", line 66, in set_line
    line.request(consumer=line.owner().name(), type=gpiod.LINE_REQ_DIR_OUT)
AttributeError: 'NoneType' object has no attribute 'request'

```

htot, 2020-03-09:
Strange. I get:
```
root@edison:~# blink-led 
Traceback (most recent call last):
  File "/usr/bin/blink-led", line 86, in <module>
    led.request(consumer=led.owner().name(), type=gpiod.LINE_REQ_DIR_OUT)
OSError: [Errno 16] Device or resource busy
```
and that is because the led gpio _is_ busy. It's used by the kernel:
```
root@edison:~# lsmod | grep led
leds_gpio              16384  0
ledtrig_heartbeat      16384  0
led_class              20480  2 leds_gpio,sdhci
```

I meant to look at the code as an example.

htot, 2020-03-09:
Try this:
```
root@edison:~# ls -l /sys/kernel/config/acpi/table
total 0
drwxr-xr-x 2 root root 0 Jan  1  2000 arduino
drwxr-xr-x 2 root root 0 Jan  1  2000 leds
drwxr-xr-x 2 root root 0 Jan  1  2000 spidev
root@edison:~# rmdir /sys/kernel/config/acpi/table/leds/
# heartbeat stops
root@edison:~# systemctl start blink-led
# blink-led runs
```

shawnhsoia97, 2020-03-10:
I tried,but not work.
```
root@edison:/usr/bin# python3 blink-led
Traceback (most recent call last):
  File "blink-led", line 81, in <module>
    set_line("TRI_STATE_ALL", 0)
  File "blink-led", line 66, in set_line
    line.request(consumer=line.owner().name(), type=gpiod.LINE_REQ_DIR_OUT)
AttributeError: 'NoneType' object has no attribute 'request'
root@edison:/usr/bin# ls -l /sys/kernel/config/acpi/table
total 0
drwxr-xr-x 2 root root 0 Jan  1  2000 i2c3028
drwxr-xr-x 2 root root 0 Jan  1  2000 spidev
root@edison:/usr/bin#  lsmod | grep led
led_class              20480  1 sdhci

```

htot, 2020-03-10:
> I tried,but not work.
>
> ```
> root@edison:/usr/bin# python3 blink-led
> Traceback (most recent call last):
>   File "blink-led", line 81, in <module>
>     set_line("TRI_STATE_ALL", 0)
>   File "blink-led", [...]

Maybe the problem is caused by this table?

Ah, no, you haven't loaded the arduino table, so the TRI_STATE_ALL line is not defined. See:
```
root@edison:~# gpioinfo | grep -i TRI
        line  14: "TRI_STATE_ALL" unused output active-high 
```

> drwxr-xr-x 2 root root 0 Jan  1  2000 spidev
> root@edison:/usr/bin#  lsmod | grep led
> led_class              20480  1 sdhci

shawnhsoia97, 2020-03-11:
Because I use other platform not Arduino.so  I not found "TRI_STATE_ALL",arduino.asl is not work for me.
```
root@edison:/home# gpioinfo | grep -i TRI
root@edison:/home#
```
I think should to create ACPI table for my platform,but I don't know how to do,I not understand arduino.asl,is there any reference?

Do you have reference of gpiod function?Because I not understand how to use `line.request(consumer=line.owner().name(), type=gpiod.LINE_REQ_DIR_OUT)`.Can I use  "TRI_STATE_ALL" directly,not use ACPI table? because that should can use any platform.
[image]

Also I tried blink-led on Arduino,but "TRI_STATE_ALL" is busy.
```
root@edison:~# gpioinfo | grep -i TRI
        line  14: "TRI_STATE_ALL" unused output active-high
root@edison:~# blink-led
Traceback (most recent call last):
  File "/usr/bin/blink-led", line 82, in <module>
    set_line("SPI_CLK_SEL", 0)
  File "/usr/bin/blink-led", line 66, in set_line
    line.request(consumer=line.owner().name(), type=gpiod.LINE_REQ_DIR_OUT)
OSError: [Errno 16] Device or resource busy
root@edison:~#

```

htot, 2020-03-11:
> Because I use other platform not Arduino.so I not found "TRI_STATE_ALL",arduino.asl is not work for me.

If you don't have this pin, you can't toggle it.

> I think should to create ACPI table for my platform,but I don't know how to do,I not understand arduino.asl,is there any reference?

There exists https://www.acpica.org/documentation

> Do you have reference of gpiod function?Because I not understand how to use `line.request(consumer=line.owner().name(), type=gpiod.LINE_REQ_DIR_OUT)`.Can I use "TRI_STATE_ALL" directly,not use ACPI table? because that should can use any platform.
> [image]

It's just a pin. We are getting a handle to the pin, but you don't have the pin.

Do you have a LED on pin 40?
 
> Also I tried blink-led on Arduino,but "TRI_STATE_ALL" is busy.

> root@edison: blink-led
> Traceback (most recent call last):
>   File "/usr/bin/blink-led", line 82, in <module>
>     set_line("SPI_CLK_SEL", 0)
>   File "/usr/bin/blink-led", line 66, in set_line [...]

No, SPI_CLK_SEL is busy. That means it is being used in one of you acpi tables.

shawnhsoia97, 2020-03-11:
I read the arduino.asl
```
External (_SB_.PCI0.I2C1, DeviceObj)

Scope (\_SB.PCI0.I2C1)
{
    #include "gpioexp.asli"

    // GPIO expander (U17)
```
I get it,that means it is "TRI_STATE_ALL" not exists edison,it exists in Arduino,Edison use I2C expand it?

So is spidev.asl exclusive to Arduino?

I can control gpio40 output 0/1.
```
root@edison:~# python3
Python 3.7.6 (default, Mar  4 2020, 03:10:38)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gpiod
>>> led = gpiod.Chip("gpiochip0").get_line(40)
>>> led.request(consumer=led.owner().name(), type=gpiod.LINE_REQ_DIR_OUT)
>>> led.set_value(1) 
>>> led.set_value(0)
>>> led.set_value(1)
```

But  if I want to control Edison pinout 52. ,change GP40 to I2C_2_CLK or GPIO input,how do I?

Also will spidev.asl be affected if I don't use Arduino. Because I use to python control spi,but spi_clk is not work.

htot, 2020-03-11:
> I read the arduino.asl
> 
> ```
> External (_SB_.PCI0.I2C1, DeviceObj)
> 
> Scope (\_SB.PCI0.I2C1)
> {
>     #include "gpioexp.asli"
> 
>     // GPIO expander (U17)
> ```
> 
> I get it,that means it is "TRI_STATE_ALL" not exists edison,it exists in Arduino,Edison use I2C expand it?

Yes, see the schematics.
 
> So is spidev.asl exclusive to Arduino?

This I don't know.

> I can control gpio40 output 0/1.

Because the pin is a GPIO. When you would load leds.aml the pin is taken by the kernel and in use.

> But if I want to control Edison pinout 52. ,change GP40 to I2C_2_CLK or GPIO input,how do I?
> [image]

Unused (by the kernel) gpios can be controlled from user user space same as gpio40.
To change mode gpio->spi must be done from acpi table (or platform driver).
To change gpio from input->output  use gpio.

> Also will spidev.asl be affected if I don't use Arduino? Because I use to python control spi,but spi_clk is not work.

I don't know. You have to compare your schematics with the Edison Arduino?

shawnhsoia97, 2020-03-12:
> Because the pin is a GPIO. When you would load leds.aml the pin is taken by the kernel and in use.

No,I not load leds.aml

> Unused (by the kernel) gpios can be controlled from user user space same as gpio40.
> To change mode gpio->spi must be done from acpi table (or platform driver).
> To change gpio from input->output use gpio.

I have a one question.
```
DefinitionBlock ("spidev.aml", "SSDT", 5, "", "SPIDEV", 1)
{
    External (_SB_.PCI0.SPI5, DeviceObj)

    Scope (\_SB.PCI0.SPI5)
    {
        Device (TP0)
        {
            Name (_HID, "SPT0001")
            Name (_DDN, "SPI test device")
            Name (_CRS, ResourceTemplate () {
                SpiSerialBus (
                    1,                      // Chip select
                    PolarityLow,            // Chip select is active low
                    FourWireMode,           // Full duplex
                    8,                      // Bits per word is 8 (byte)
                    ControllerInitiated,    // Don't care
                    1000000,                // 1 MHz
                    ClockPolarityLow,       // SPI mode 0
                    ClockPhaseFirst,        // SPI mode 0
                    "\\_SB.PCI0.SPI5",      // SPI host controller
                    0                       // Must be 0
                )
            })
        }
    }
}
```
 Why is it SPI5,not SPI2? Because I read schematics with edison and arduino,but that is both only show SPI2,not SPI5.
[image]
[image]

> I don't know. You have to compare your schematics with the Edison Arduino?

Yes,my schematics is not connect MUX  ,that connect the device directly .
But I don't have /dev/spi5.1 now,after modify edison.conf. Strange,It was something before.
I just remove spidev.asl and re-add it.
```

root@edison:/home#  ls -l /sys/kernel/config/acpi/table
total 0
drwxr-xr-x 2 root root 0 Jan  1  2000 i2c3028
drwxr-xr-x 2 root root 0 Jan  1  2000 spidev
root@edison:/home# ls /dev/s
sequencer   sequencer2  serial/     shm/        snapshot    snd/        stderr      stdin       stdout
```

htot, 2020-03-12:
> No,I not load leds.aml

Exactly, therefore you (user space)  control gpio40. When you load leds.aml the kernel controls the gpio and user space can not.

shawnhsoia97, 2020-03-16:
Hi @htot ,I cannot use spi with Arduino and  Edison,I tried using the following test code.
```
#include <stdio.h>
#include <unistd.h>
#include <errno.h>
#include <signal.h>
#include <stdlib.h>

#include "mraa.h"

#define LED_PIN 21
#define SPI_MISO_PIN 24
#define SPI_MOSI_PIN 11
#define SPI_SCK_PIN 10

/* SPI declaration */
#define SPI_BUS 5

/* SPI frequency in Hz */
#define SPI_FREQ 400000

uint16_t pat[] = { 0x01aa, 0x0255, 0x03aa, 0x0455, 0x05aa, 0x0655, 0x07aa, 0x0855 };
uint16_t pat_inv[] = { 0x0155, 0x02aa, 0x0355, 0x04aa, 0x0555, 0x06aa, 0x0755, 0x08aa };
uint16_t pat_clear[] = { 0x0100, 0x0200, 0x0300, 0x0400, 0x0500, 0x0600, 0x0700, 0x0800 };

int running = 0;

void sig_handler(int signo)
{
    if ( signo == SIGINT ) {
        printf("Closing GPIO\n", LED_PIN);
        running = -1;
    }
}

int main()
{
    /* Initialize mraa */
    mraa_result_t status = MRAA_SUCCESS;
    mraa_result_t r = MRAA_SUCCESS;
    mraa_spi_context spi;
    int i, j;
    mraa_init();

        //! [Interesting]
    /* initialize SPI bus */
    spi = mraa_spi_init(SPI_BUS);
    if (spi == NULL) {
        fprintf(stderr, "Failed to initialize SPI\n");
        mraa_deinit();
        return EXIT_FAILURE;
    }

    /* Create access to GPIO pin */
    mraa_gpio_context gpio;
    gpio = mraa_gpio_init(LED_PIN);
    if ( gpio == NULL ) {
        fprintf(stderr, "Error opening GPIO\n");
        exit(1);
    }

    /* set SPI frequency */
    status = mraa_spi_frequency(spi, SPI_FREQ);
    if (status != MRAA_SUCCESS)
        goto err_exit;

    /* set big endian mode */
    status = mraa_spi_lsbmode(spi, 0);
    if (status != MRAA_SUCCESS) {
        goto err_exit;
    }

    /* Set GPIO direction to out */
    r = mraa_gpio_dir(gpio, MRAA_GPIO_OUT);
    if ( r != MRAA_SUCCESS ) {
        mraa_result_print(r);
    }
    /* Create signal handler so we can exit gracefully */
    signal(SIGINT, sig_handler);

    /* Turn LED off and on forever until SIGINT (Ctrl+c) */
    while ( running == 0 ) {

        mraa_spi_write_buf_word(spi, pat, 16);

        r = mraa_gpio_write(gpio, 0);
        if ( r != MRAA_SUCCESS ) {
            mraa_result_print(r);
        }
        sleep(1);

        r = mraa_gpio_write(gpio, 1);
        if ( r != MRAA_SUCCESS ) {
            mraa_result_print(r);
        }
        sleep(1);
    }

    /* Clean up GPIO and exit */
    r = mraa_gpio_close(gpio);
    if ( r != MRAA_SUCCESS ) {
        mraa_result_print(r);
    }

err_exit:
    mraa_result_print(status);

    return r;
}

```
I can read GP183 output,but my SPI_SCK is NULL. I don't know why.
I have do install [spidev.asl](https://github.com/edison-fw/meta-acpi/blob/master/recipes-bsp/acpi-tables/samples/edison/spidev.asl)
```
root@bovia-deb:/bovia# ls /dev/spidev5.1
/dev/spidev5.1
root@bovia-deb:/bovia# ls -l /sys/kernel/config/acpi/table
total 0
drwxr-xr-x 2 root root 0 Jan  1  2000 arduino
drwxr-xr-x 2 root root 0 Jan  1  2000 i2c3028
drwxr-xr-x 2 root root 0 Jan  1  2000 leds
drwxr-xr-x 2 root root 0 Jan  1  2000 spidev

```
But my SPI is not work,do you know why?
Am i missing something?
Thank you.

htot, 2020-03-16:
I haven't tried this myself https://edison-fw.github.io/meta-intel-edison/4.4-hsu-i2c-spi.html#spi
Maybe you can find who contributed this piece of doc.

But likely it was not tested using mraa.

xlla, 2020-03-17:
> I cannot use spi with Arduino and Edison,I tried using the following test code.

Yes,  Arduino and Mraa can't work right now, until they both migrate to libgpiod.

xlla, 2020-03-17:
> But my SPI is not work,do you know why?

SPI will work if you use python with libgpiod.

shawnhsoia97, 2020-03-18:
Thank help for me.I give up using SPI directly.I still not work with SPI.
So I ues Mraa to control GPIO to mimic SPI function.

closed 2020-03-18 by shawnhsoia97