edison-fw/meta-intel-edison #97: blink-led.service on zeus

mwallnoefer, 2020-03-07:
The `blink-led` example fails on branch _zeus_ with the following exception:
```
Traceback (most recent call last):
  File "/usr/bin/blink-led", line 81, in <module>
    set_line("TRI_STATE_ALL", 0)
  File "/usr/bin/blink-led", line 66, in set_line
    line.request(consumer=line.owner().name(), type=gpiod.LINE_REQ_DIR_OUT)
AttributeError: 'NoneType' object has no attribute 'request'
```

mwallnoefer, 2020-03-07:
Solution found: https://github.com/htot/meta-intel-edison/commit/22fa91d79f55980b3941d61c9625476b7762f423

mwallnoefer, 2020-03-07:
Unfortunately the systemd service still stops in a warning:
```
Mar 07 10:37:43 edison systemd[1]: blink-led.service: Start request repeated too quickly.
Mar 07 10:37:43 edison systemd[1]: blink-led.service: Failed with result 'exit-code'.
Mar 07 10:37:43 edison systemd[1]: Failed to start Edison Arduino board LED Blinker.
```

htot, 2020-03-07:
See my comments https://github.com/htot/meta-intel-edison/commit/22fa91d79f55980b3941d61c9625476b7762f423

mwallnoefer, 2020-03-07:
Okay, I am fine with disabling it (at the `systemd` level?).

Just a question: this does not interfere with the Arduino functionality, correct?

htot, 2020-03-07:
I don't think so. The kernel does 'take' (hog) the LED ping, so it can not be used for anything else.

mwallnoefer, 2020-03-08:
I have tried to determine the location where the systemd link (runtime -> unit) gets added during build, but I cannot find it. It does not seem to happen in the receipt.

htot, 2020-03-08:
meta-intel-edison/meta-intel-edison-distro/recipes-core/post-install/files/post-install.sh?

But this should run only once. Ah, wait:
https://www.yoctoproject.org/docs/current/mega-manual/mega-manual.html#ref-classes-systemd

It's enabled by default, we need to turn it off.

mwallnoefer, 2020-03-09:
https://github.com/htot/meta-intel-edison/pull/9

htot, 2020-03-26:
> It's enabled by default, we need to turn it off.

I am reconsidering this. The problem is when the led.aml is loaded the kernel hogs the led pin, so blink-led should enable the led mux and terminate as the kernel control the led. 
But when led.aml is not loaded blink-led should run normally.

So this is a bug in blink-led that we need to fix.

htot, 2020-03-26:
Fixed I hope (5ad238581a4f85a65ad6585f24e20f380dac4b85) and force pushed to master and zeus.

mwallnoefer, 2021-02-03:
Just as a follow-up this patch would disable the _blink-led.service_ per default. I guess that a manual run for a one-time diagnosis would be better than an automatic launch at each start-up, since the LED could also be used by the _oobe_ wifi detection. Second, it is just wasted energy consumption...
```
diff --git a/meta-intel-edison-distro/recipes-support/blink-led/blink-led_0.1.bb b/meta-intel-edison-distro/recipes-support/blink-led/blink-led_0.1.bb
index 0e84904..7ae255f 100644
--- a/meta-intel-edison-distro/recipes-support/blink-led/blink-led_0.1.bb
+++ b/meta-intel-edison-distro/recipes-support/blink-led/blink-led_0.1.bb
@@ -7,6 +7,7 @@ FILESEXTRAPATHS_prepend := "${THISDIR}/files/:"
 
 inherit systemd
 SYSTEMD_SERVICE_${PN} = "blink-led.service"
+SYSTEMD_AUTO_ENABLE = "disable"
 
 RDEPENDS_${PN} = "libgpiod-python"
 

```

htot, 2021-02-03:
Ah thanks. Yes in the past blink-led needed to run continuously, but now we have a kernel module. I

htot, 2021-02-05:
I'll take this one and fix it in dunfell.

htot, 2021-03-02:
I hope it's working for you?

mwallnoefer, 2021-03-03:
This issue was ok I think 🤔, I still had problems with _mraa_: #123.

closed 2021-03-03 by mwallnoefer