edison-fw/meta-intel-edison #100: how to use PWM in warrior branch.

xlla, 2020-03-19:
please add some usage of pwm in acpi kernel, then we can driver dc motor and make more fun.

xlla, 2020-04-10:
When I try to switch GP182 soc mode to pwm, a kernel panic occur.
<pre>
root@edison:~# echo GP182 1 > /sys/kernel/debug/pinctrl_pinmux/pinmux
[  254.743653] BUG: unable to handle page fault for address: 38315000
[  254.749977] #PF: supervisor read access in kernel mode
[  254.755228] #PF: error_code(0x0000) - not-present page
[  254.760478] *pde = 00000000 
[  254.763430] Oops: 0000 [#1] SMP
[  254.766652] CPU: 0 PID: 652 Comm: sh Tainted: G        WC        5.5.0-edison-acpi-standard #1
[  254.775450] Hardware name: Intel Corporation Merrifield/BODEGA BAY, BIOS 542 2015.01.21:18.19.48
[  254.784442] EIP: __kmalloc+0x85/0x210
[  254.788194] Code: c0 0f 85 1e 01 00 00 8b 07 64 8b 50 04 64 03 05 cc b2 85 ca 8b 08 89 4d f0 85 c9 0f 84 54 01 00 00 8b 75 f0 8b 47 14 8d 4a 01 <8b> 1c 06 89 f0 8b 37 64 0f c7 0e 75 d0 8b 47 14 0f 18 04 03 3e 8d
[  254.807374] EAX: 00000000 EBX: 00000dc0 ECX: 000022cb EDX: 000022ca
[  254.813782] ESI: 38315000 EDI: f6001e40 EBP: f48dfe78 ESP: f48dfe5c
[  254.820188] DS: 007b ES: 007b FS: 00d8 GS: 00e0 SS: 0068 EFLAGS: 00010206
[  254.827128] CR0: 80050033 CR2: 38315000 CR3: 34672000 CR4: 001006d0
[  254.833529] Call Trace:
[  254.836043]  ? mrfld_pinctrl_pinmux_set+0x4f/0x1bc
[  254.840950]  mrfld_pinctrl_pinmux_set+0x4f/0x1bc
[  254.845677]  ? mrfld_pin_dbg_show+0xa0/0xa0
[  254.849961]  full_proxy_write+0x3e/0x70
[  254.853886]  ? full_proxy_poll+0x60/0x60
[  254.857900]  __vfs_write+0x2b/0x1d0
[  254.861473]  ? selinux_file_permission+0xf5/0x120
[  254.866288]  ? security_file_permission+0x2c/0x100
[  254.871190]  ? __switch_to_asm+0x28/0x50
[  254.875202]  ? __switch_to_asm+0x34/0x50
[  254.879214]  ? rw_verify_area+0x55/0x100
[  254.883228]  vfs_write+0x8f/0x1a0
[  254.886621]  ksys_write+0x4e/0xd0
[  254.890014]  sys_write+0x11/0x20
[  254.893318]  do_fast_syscall_32+0x7a/0x280
[  254.897512]  entry_SYSENTER_32+0xa5/0xf8
[  254.901525] EIP: 0xb7f378e5
[  254.904389] Code: e8 1c 00 00 00 89 d3 eb cf 8d 74 26 00 b8 40 42 0f 00 eb b5 8b 04 24 c3 8b 1c 24 c3 8b 3c 24 c3 90 51 52 55 89 e5 0f 34 cd 80 <5d> 5a 59 c3 90 90 90 90 8d 76 00 58 b8 77 00 00 00 cd 80 90 8d 76
[  254.923574] EAX: ffffffda EBX: 00000001 ECX: 01d40510 EDX: 00000008
[  254.929980] ESI: 01d40510 EDI: 00000008 EBP: b7efdd40 ESP: bfaaf1c0
[  254.936398] DS: 007b ES: 007b FS: 0000 GS: 0033 SS: 007b EFLAGS: 00000246
[  254.943334] Modules linked in: leds_gpio fb_ssd1331(C) fbtft(C) ti_ads7950 industrialio_triggered_buffer kfifo_buf rfcomm iptable_nat bnep spi_pxa2xx_platform dw_dmac usb_f_mass_storage usb_f_rndis u_ether usb_f_acm u_serial libcomposite pwm_lpss_pci pwm_lpss snd_sof_pci snd_sof_intel_byt intel_mrfld_adc snd_sof_intel_hda_common intel_mrfld_pwrbtn snd_sof_intel_ipc snd_sof snd_sof_xtensa_dsp snd_soc_acpi_intel_match snd_soc_acpi spi_pxa2xx_pci brcmfmac brcmutil hci_uart btbcm mmc_block extcon_intel_mrfld intel_soc_pmic_mrfld sdhci_pci cqhci sdhci led_class mmc_core
[  254.994552] CR2: 0000000038315000
[  254.998090] ---[ end trace 7f0f5a483eafb431 ]---
[  255.002879] EIP: __kmalloc+0x85/0x210
[  255.006669] Code: c0 0f 85 1e 01 00 00 8b 07 64 8b 50 04 64 03 05 cc b2 85 ca 8b 08 89 4d f0 85 c9 0f 84 54 01 00 00 8b 75 f0 8b 47 14 8d 4a 01 <8b> 1c 06 89 f0 8b 37 64 0f c7 0e 75 d0 8b 47 14 0f 18 04 03 3e 8d
[  255.025883] EAX: 00000000 EBX: 00000dc0 ECX: 000022cb EDX: 000022ca
[  255.032322] ESI: 38315000 EDI: f6001e40 EBP: f48dfe78 ESP: f48dfe5c
[  255.038762] DS: 007b ES: 007b FS: 00d8 GS: 00e0 SS: 0068 EFLAGS: 00010206
[  255.045733] CR0: 80050033 CR2: 38315000 CR3: 34672000 CR4: 001006d0
[  255.153016] BUG: unable to handle page fault for address: 38315000
[  255.159346] #PF: supervisor read access in kernel mode
[  255.164597] #PF: error_code(0x0000) - not-present page
[  255.169847] *pde = 00000000 
[  255.172800] Oops: 0000 [#2] SMP
[  255.176023] CPU: 0 PID: 635 Comm: login Tainted: G      D WC        5.5.0-edison-acpi-standard #1
[  255.185090] Hardware name: Intel Corporation Merrifield/BODEGA BAY, BIOS 542 2015.01.21:18.19.48
[  255.194080] EIP: __kmalloc_track_caller+0x82/0x210
[  255.198982] Code: c0 0f 85 21 01 00 00 8b 07 64 8b 50 04 64 03 05 cc b2 85 ca 8b 08 89 4d f0 85 c9 0f 84 57 01 00 00 8b 75 f0 8b 47 14 8d 4a 01 <8b> 1c 06 89 f0 8b 37 64 0f c7 0e 75 d0 8b 47 14 0f 18 04 03 3e 8d
[  255.218161] EAX: 00000000 EBX: 00000a20 ECX: 000022cb EDX: 000022ca
[  255.224567] ESI: 38315000 EDI: f6001e40 EBP: f4bd7d98 ESP: f4bd7d7c
[  255.230974] DS: 007b ES: 007b FS: 00d8 GS: 00e0 SS: 0068 EFLAGS: 00010206
[  255.237909] CR0: 80050033 CR2: 38315000 CR3: 3577a000 CR4: 001006d0
[  255.244312] Call Trace:
[  255.246827]  ? security_sid_to_context_core.isra.16+0x59/0x100
[  255.252799]  kmemdup+0x16/0x30
[  255.255931]  security_sid_to_context_core.isra.16+0x59/0x100
[  255.261722]  security_sid_to_context+0x1b/0x20
[  255.266272]  selinux_socket_getpeersec_stream+0x53/0xe0
[  255.271617]  ? selinux_sk_free_security+0x30/0x30
[  255.276435]  security_socket_getpeersec_stream+0x33/0x60
[  255.281876]  sock_getsockopt+0x960/0xdf0
[  255.285897]  ? unix_seqpacket_sendmsg+0x60/0x60
[... 53 lines cut ...]
[  255.637471]  sys_openat+0x16/0x20
[  255.640864]  do_fast_syscall_32+0x7a/0x280
[  255.645057]  entry_SYSENTER_32+0xa5/0xf8
[  255.649068] EIP: 0xb7fa68e5
[  255.651933] Code: e8 1c 00 00 00 89 d3 eb cf 8d 74 26 00 b8 40 42 0f 00 eb b5 8b 04 24 c3 8b 1c 24 c3 8b 3c 24 c3 90 51 52 55 89 e5 0f 34 cd 80 <5d> 5a 59 c3 90 90 90 90 8d 76 00 58 b8 77 00 00 00 cd 80 90 8d 76
[... cut ...]
[  255.786487] DS: 007b ES: 007b FS: 00d8 GS: 00e0 SS: 0068 EFLAGS: 00010206
[  255.793477] CR0: 80050033 CR2: 38315000 CR3: 34fee000 CR4: 001006d0
[  255.799975] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000009
[  255.807830] Kernel Offset: 0x8000000 from 0xc1000000 (relocation range: 0xc0000000-0xf77fdfff)
[  255.816638] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000009 ]---

FATAL: read from port failed: Device not configured
Skipping tty reset...

</pre>

xlla, 2020-04-13:
pwm is work.

closed 2020-04-13 by xlla

shawnhsoia97, 2021-05-14:
Hi @xlla,
I also want to set PWM mode, but I have not found `/sys/kernel/debug/pinctrl_pinmux/pinmux` in my kernel (branch zeus).
I tried used below code in my .asl files and use `/sys/class/pwm` class test , but which is not effect.
```
External (_SB_.PCI0.PWM0, DeviceObj)

Scope (\_SB.PCI0.PWM0)
{
    Name (RBUF, ResourceTemplate ()
    {
        PinFunction(Exclusive, PullUp, 0x0001,
            "\\_SB.FLIS", 0, ResourceConsumer, , ) { 12 }
    })
    Method (_CRS, 0, NotSerialized)
    {
        Return (RBUF)
    }
}
```

Please, Could you tell me how to use PWM? I need to switch GP12 to PWM mode1.
Thank you.

andy-shev, 2021-10-03:
> Hi @xlla, I also want to set PWM mode, but I have not found `/sys/kernel/debug/pinctrl_pinmux/pinmux` in my kernel (branch zeus). 
> Please, Could you tell me how to use PWM? I need to switch GP12 to PWM mode1. 

@xlla had been active almost couple of years ago and here is their branch with the experimental patches: https://github.com/xlla/linux/commits/eds-acpi-5.5.0-v1. We are free to accept the cleaned up version of the pinmux support (note there is some development regarding this feature in upstream as well in a similar way). So, feel free to take these and update, we will review and merge in the kernel in meta-intel-edison.