edison-fw/meta-intel-edison #101: add vim

mwallnoefer, 2020-03-23:
I think we should include the whole _vim_ in the Edison image to have at least one syntax highlighting-enabled editor pre-installed.

htot, 2020-03-23:
Might be a good idea to first try what dependencies get pulled in?

You can easily `bitbake vim` and use apt-get install (following https://edison-fw.github.io/meta-intel-edison/5.0-Creating-a-deb-repository.html)

mwallnoefer, 2020-03-25:
Okay, what regards to bitbake adding _vim_ has not caused any other dependencies to appear. In the end the build generated 11 DEB packages, but I don't think that all of them are necessary:
```
$ find -name *vim*
./corei7-64/vim_8.1.1518-r0_amd64.deb
./corei7-64/vim-doc_8.1.1518-r0_amd64.deb
./corei7-64/vim-tools_8.1.1518-r0_amd64.deb
./corei7-64/vim-dev_8.1.1518-r0_amd64.deb
./corei7-64/vim-syntax_8.1.1518-r0_amd64.deb
./corei7-64/vim-common_8.1.1518-r0_amd64.deb
./corei7-64/vim-tutor_8.1.1518-r0_amd64.deb
./corei7-64/vim-vimrc_8.1.1518-r0_amd64.deb
./corei7-64/vim-src_8.1.1518-r0_amd64.deb
./corei7-64/vim-dbg_8.1.1518-r0_amd64.deb
./corei7-64/vim-help_8.1.1518-r0_amd64.deb
```
We would need `vim`, `vim-tools`, `vim-syntax`, `vim-common`, `vim-vimrc` and maybe `vim-help`. Please find here their disk consumption:
```
$ du -hs vim_8.1.1518-r0_amd64.deb  vim-tools_8.1.1518-r0_amd64.deb  vim-syntax_8.1.1518-r0_amd64.deb vim-common_8.1.1518-r0_amd64.deb vim-vimrc_8.1.1518-r0_amd64.deb vim-help_8.1.1518-r0_amd64.deb 
1,2M	vim_8.1.1518-r0_amd64.deb
24K	vim-tools_8.1.1518-r0_amd64.deb
1,1M	vim-syntax_8.1.1518-r0_amd64.deb
2,3M	vim-common_8.1.1518-r0_amd64.deb
4,0K	vim-vimrc_8.1.1518-r0_amd64.deb
1,7M	vim-help_8.1.1518-r0_amd64.deb
```

htot, 2020-03-25:
The easiest way to check is use apt-get install[ from edison](https://edison-fw.github.io/meta-intel-edison/5.0-Creating-a-deb-repository.html) .

But it's fine by me to add if it's that limited.

mwallnoefer, 2020-03-26:
I suggest to do so, since we should provide at least one syntax highlighting-enabled editor. I first edited the `configure_edison` script directly on the board, but without the full vim it's very cumbersome. So I did it on the PC...

nano is very intuitive for the non-expert user, but provides no syntax highlighting. So in my eyes it works better for shorter text.

htot, 2020-03-26:
I did on the PC too. And then used Krusader as directory synchronizer.

htot, 2020-03-26:
```
root@edison:~# apt-get install vim
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  vim-common vim-help vim-syntax vim-tutor vim-vimrc
Suggested packages:
  diffutils
The following NEW packages will be installed:
  vim vim-common vim-help vim-syntax vim-tutor vim-vimrc
0 upgraded, 6 newly installed, 0 to remove and 0 not upgraded.
Need to get 6681 kB of archives.
After this operation, 0 B of additional disk space will be used.
```
This 40MB. Do we need vim-tools? It is not a dependency, but it's small.

mwallnoefer, 2020-03-27:
Did I get it right, these six packages amount to 40 MB after unpacking? And if we drop the help?

I am fine with dropping `vim-tools` I do not know about its exact purpose.

A valid alternative could be providing `mcedit` and the Midnight Commander instead, I guess it provides syntax highlighting as well.

htot, 2020-03-27:
Another way is: mount edison using sshfs from your host pc and use your hosts editor. Or if you are using Kubuntu, use kate which uses kde's kioslaves. And open the file using an url: fish://root@edison/usr/bin/file

mwallnoefer, 2020-03-27:
Ah moment: I see that on my system `nano` is providing syntax highlighting, but on the Edison it does not seem to do so. Maybe there is some extension missing?

htot, 2020-03-27:
That is possible. But we also had `less` without coloring, so when it was called from `journalctl` color codes were flying over the screen. This happened because `less` was provided by `busybox`. That problem was solved by providing `less` from it's own package.

mwallnoefer, 2020-03-28:
Okay after some investigation I found out that Yocto does not install an appropriate `nanorc` file (although the syntax highlighting definitions are copied though!): https://askubuntu.com/questions/90013/how-do-i-enable-syntax-highlighting-in-nano.

Moreover I figured out that also `busybox' ls` does not provide any coloring unless you request them explicitly over `ls --colors`.

htot, 2020-03-28:
Is all good for me. I don't mind adding `vim`.

BTW: I meant we replaced `busybox less` by `less`. `ls` is from coreutils.
According to [SE](https://superuser.com/questions/665274/how-to-make-ls-color-its-output-by-default-without-setting-up-an-alias#665285) we need to create aliases. Never did that but putting the suggested lines in `.profile` works for `sh` and in `.bashrc` for bash.

andy-shev, 2020-05-19:
@htot for best user experience I would even recommend to replace busybox with something more useful. Bysybox is good for developers or for products where user doesn't need to use shell.

htot, 2020-05-21:
True, when users want to develop on Edison. Generally compiled code is developed on the host. But @mwallnoefer finds as I have that developing python code is easier direct on Edison. Which can be worked around by sshfs or kioslaves of course.

Or by manually installing packages like vim or originals instead of scaled down busybox versions. 

Or we could create a developer layer that generates an image filled with developer tools?

andy-shev, 2020-05-22:
@htot I think it's a bit more complicated than that. So, first layer is a base with 2-3 variants: **minimum** (Busybox), **medium** and **luxury**. On top of each you can put other layer like **development**, **IoT**, **desktop**, etc...

htot, 2020-05-22:
We already have a minimal layer edison-image-minimal (could be minimized further, if anybody cares). And then the normal image (which includes the minimal). There is at least one user who has his own layer that also includes the minimal.

Afaiu we could put `packagegroup-core-full-cmdline` and this should pull all full versions of command supplied by `busybox`. I don't know if that will automatically drop `busybox` or even set alternatives to the full cmdline versions.

htot, 2020-06-01:
For now: I just added vim on edison-image.

closed 2020-06-01 by htot