edison-fw/meta-intel-edison #102: How to add or modify menuconfig?

shawnhsoia97, 2020-03-27:
Hi ,I want to modify driver or add new driver in menuconfig,so I modify `./out/linux64/build/tmp/work-shared/edison/kernel-source/drivers/usb/serial/Kconfig` to add new option.
But when `make edison-image`,I found that the modified file was re-created.
So do you know how to add new option or modify menuconfig?
Thank you so much!

htot, 2020-03-27:
Sure: https://edison-fw.github.io/meta-intel-edison/5.1-Bitbake-tricks#configuring-the-kernel-and-grab-the-kernel-fragment

closed 2020-03-27 by htot

shawnhsoia97, 2020-03-27:
Hi @htot ,you misunderstood me.
I mean I want Add New Kernel Configuration Options.
e.g.
`$vi ./out/linux64/build/tmp/work-shared/edison/kernel-source/drivers/net/usb/Kconfig`
Add below
```
config USB_NET_TEST
        tristate "Test"
        depends on USB_USBNET
        select USB_WDM
        help
          test

```
`$vi ./out/linux64/build/tmp/work-shared/edison/kernel-source/drivers/net/usb/Makefile` 
add` "obj-$(CONFIG_USB_NET_TEST)  += test.o"`

Menuconfig show:

[image]

But when make edison-image,I found that the modified file was removed and recreated,So my previous changes are restored to their original state.
So do you know how to solve it?

htot, 2020-03-27:
I see.

In that case you need to add a patch to the kernel. Bitbake will always download kernel again and unpack if you modify stuff. So there are 2 ways:
1) fork the kernel (on github or locally) and add you patches there and tell the recipe the new location of the fork and the new SHA (or HEAD) to checkout
2) create patch, save under files and tell the recipe to apply the patch after unpacking, if you look in the recipe your will see we are already applying a few patches

shawnhsoia97, 2020-03-30:
I get it! 
Really thank you very much!! ^^