edison-fw/meta-intel-edison #104: how to set vref for ADC chip

xlla, 2020-03-29:
I notice this in dmesg
<pre>
root@edison:~# dmesg |grep ads
[    0.089960] process: using mwait in idle threads
[ 1643.785328] ads7950 spi-ADS7951:00: Failed to retrieve ACPI GPIO chip
[ 1643.857991] ads7950 spi-ADS7951:00: spi-ADS7951:00 supply vref not found, using dummy regulator
</pre>

In Schema, REFP was controlled by J8,  how to tell ads7950 I was set AREF to 3.3v or 5v ?

htot, 2020-03-29:
I don't know what it means, but there are more dummy regulators:
```
root@edison:~# journalctl -b | grep regulator
regulator-dummy: no parameters
pca953x i2c-INT3491:00: i2c-INT3491:00 supply vcc not found, using dummy regulator
pca953x i2c-INT3491:01: i2c-INT3491:01 supply vcc not found, using dummy regulator
pca953x i2c-INT3491:02: i2c-INT3491:02 supply vcc not found, using dummy regulator
pca953x i2c-INT3491:03: i2c-INT3491:03 supply vcc not found, using dummy regulator
hci_uart_bcm serial0-0: serial0-0 supply vbat not found, using dummy regulator
hci_uart_bcm serial0-0: serial0-0 supply vddio not found, using dummy regulator
ads7950 spi-PRP0001:01: spi-PRP0001:01 supply vref not found, using dummy regulator
```

xlla, 2020-04-11:
I found it is use hardcode for acpi case, this is a warning message.

closed 2020-04-11 by xlla