edison-fw/meta-intel-edison #105: sof-audio-pci - failed to probe DSP hardware

xlla, 2020-03-30:
<pre>
[    9.612127] sof-audio-pci 0000:00:0d.0: error: no matching ASoC machine driver found - aborting probe
[    9.621634] sof-audio-pci 0000:00:0d.0: error: failed to get machine info -19
[    9.629027] sof-audio-pci 0000:00:0d.0: error: failed to probe DSP hardware!
</pre>

xlla, 2020-03-30:
<pre>
00:0d.0 Multimedia audio controller: Intel Corporation Device 119a (rev 01)

root@edison:~# lsmod |grep sof
snd_sof_pci            20480  0
snd_sof_intel_byt      20480  1 snd_sof_pci
snd_sof_intel_hda_common    49152  1 snd_sof_pci
snd_sof_intel_ipc      20480  1 snd_sof_intel_byt
snd_sof                81920  4 snd_sof_pci,snd_sof_intel_hda_common,snd_sof_intel_byt,snd_sof_intel_ipc
snd_sof_xtensa_dsp     16384  1 snd_sof_pci
snd_soc_acpi_intel_match    24576  1 snd_sof_pci
snd_soc_acpi           16384  2 snd_sof_pci,snd_soc_acpi_intel_match
</pre>

htot, 2020-03-30:
Thanks, this is a known issue resulting from enabling an experimental feature. It's harmless.

closed 2020-03-30 by htot

xlla, 2020-04-06:
in old version, soc seems work.
<pre>
[    1.597521] snd_soc_sst_platform: Enter:sst_soc_probe
[    1.893862] snd-soc-dummy snd-soc-dummy: ASoC: Failed to create platform debugfs directory
[    1.894121] merr_dpcm_dummy merr_dpcm_dummy.0:  snd-soc-dummy-dai <-> Headset-cpu-dai mapping ok
[    1.894268] merr_dpcm_dummy merr_dpcm_dummy.0:  snd-soc-dummy-dai <-> ssp2-codec mapping ok
[    1.894396] merr_dpcm_dummy merr_dpcm_dummy.0:  snd-soc-dummy-dai <-> snd-soc-dummy-dai mapping ok
</pre>

htot, 2020-04-06:
Yes this is a new driver (SOF is Sound Open Firmware).
@andy-shev is experimenting with it and turned on through [this commit](https://github.com/andy-shev/linux/commit/ecf088b615096b74b980fe5d21d3a7d6ab780d8c).

andy-shev, 2020-04-07:
Yes, sorry for that. Old one works with old firmware. I'm planning to give SOF a try when I will have more time for that (I have already few codecs on my table for couple of years...).

htot, 2020-04-07:
It doesn't really heart. And otherwise we could have reverted.

xlla, 2020-04-11:
I have prepared a pcm5102A breadboard for test.

andy-shev, 2020-05-29:
@xlla we need to talk to @plbossart about it.

htot, 2020-05-30:
> I've lost all my pointers to Edison setup though...If anyone has simple recipes to help me re-generate an image and add my own kernel + firmware files I am all ears.

This generates image: https://edison-fw.github.io/meta-intel-edison/1.2-Setting-up.html
Modifying kernel config: https://edison-fw.github.io/meta-intel-edison/5.1-Bitbake-tricks#configuring-the-kernel-and-grab-the-kernel-fragment

I'll be happy to help changing kernel or adding patches.

andy-shev, 2020-05-30:
@plbossart 

> I've lost all my pointers to Edison setup though...If anyone has _simple_ recipes to help me re-generate an image and add my own kernel + firmware files I am all ears.

If you familiar with Buildroot, I can help you: we have tree for it which we are using to create initrd with custom kernels. The infrastructure actually public: https://github.com/andy-shev/buildroot (`intel` branch). Only what you have to do as preparatory stuff is to update U-Boot (from now on you may simple take the latest official one): https://edison-fw.github.io/edison-wiki/u-boot-update (see `Option 2` which is likely your case).

Developer's wiki (a bit outdated): https://edison-fw.github.io/edison-wiki/

plbossart, 2020-06-02:
> > I've lost all my pointers to Edison setup though...If anyone has simple recipes to help me re-generate an image and add my own kernel + firmware files I am all ears.
> 
> This generates image: https://edison-fw.github.io/meta-intel-edison/1.2-Setting-up.html

This was almost good, but the last part is missing:

make image will in addition run the postbuild.sh script, which will create a complete image in out/current/build/toFlash/ ready for flashing using flashall or Flash Tool Lite.

-> I don't know where to run this, make image doesn't seem to work?

````
my_Edison_Workspace1/out/linux64/build$ ls -l
total 68
-rw-r--r--  1 pbossart pbossart 47708 Jun  1 21:22 bitbake-cookerdaemon.log
drwxr-xr-x  2 pbossart pbossart  4096 Jun  1 21:22 cache
drwxr-xr-x  2 pbossart pbossart  4096 Jun  1 18:37 conf
drwxr-xr-x  3 pbossart pbossart  4096 Jun  1 21:13 {IMAGE_ROOTFS}
drwxr-xr-x 14 pbossart pbossart  4096 Jun  1 19:02 tmp
````

htot, 2020-06-02:
> I don't know where to run this, make image doesn't seem to work?I can't figure out how to do this last flash...

If all is well, you created a link to the makefile in the 3rd step, in the directory above meta-intel-edison.

BTW: the build process created a {IMAGE_ROOTFS} directory. That's a bug, harmless though.

plbossart, 2020-06-02:
> > I don't know where to run this, make image doesn't seem to work? I can't figure out how to do this last flash...
> 
> If all is well, you created a link to the makefile in the 3rd step, in the directory above meta-intel-edison: my_Edison_Workspace1

oh, I missed this indeed. I am now having issues with flashall.sh

````
./flashall.sh
Using U-Boot target: edison-blankcdc
Now waiting for dfu device 8087:0a99
Please plug and reboot the board
dfu-util: Cannot open DFU device 8087:0a99
dfu-util: Cannot open DFU device 8087:0a99
dfu-util: Cannot open DFU device 8087:0a99
Timed out while waiting for dfu device 8087:0a99
DEBUG: lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 8087:0a2b Intel Corp. 
Bus 001 Device 005: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 004: ID 04d9:a0cd Holtek Semiconductor, Inc. USB Keyboard
Bus 001 Device 033: ID 1d6b:0104 Linux Foundation Multifunction Composite Gadget
Bus 001 Device 029: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
DEBUG: dfu-util -l
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Did you plug and reboot your board?
If yes, please try a recovery by calling this script with the --recovery option
````

I recompiled and install xfstk myself, but God knows if this is working on my Ubuntu 19.10 setup. I haven't bricked my boards yet, but they are still running the good old image.

htot, 2020-06-02:
Intel engineers maintain that it is not necessary to run flashall as root. My experience is: you must.
Maybe it's an Ubuntu thing.

BTW xfstk is only used when you run with the --recovery option (should not be needed now). Also, I found that xfstk is included with the Flash Tool Lite (I see the link here is broken: https://edison-fw.github.io/meta-intel-edison/2.0-Building-and-installing-the-image.html#flash-tool-lite. I'll try to fix that).

plbossart, 2020-06-02:
> Intel engineers maintain that it is not necessary to run flashall as root. My experience is: you must.

Yes, I forgot about that one. it's flashing nicely now.

> BTW xfstk is only used when you run with the --recovery option (should not be needed now). Also, I found that xfstk is included with the Flash Tool Lite (I see the link here is broken: [...]

I did have to recover my boards, not sure what happened last year when I updated uboot, but I chickened out and installed the Mac version of Phone Flash tool. It took several trials but I unbricked those boards.

if I see progress, I'll try to update @andy-shev defconfigs, there are some inconsistencies, the SOF driver should probe when fixed.

lybtongji, 2020-06-02:
> oh, I missed this indeed. I am now having issues with flashall.sh
>
> ```
> ./flashall.sh
> Using U-Boot target: edison-blankcdc
> Now waiting for dfu device 8087:0a99
> Please plug and reboot the board [...]

I also got this error after flashed the zeus image.

htot, 2020-06-02:
> if I see progress, I'll try to update @andy-shev defconfigs, there are some inconsistencies, the SOF driver should probe when fixed.

@plbossart that would be great. 

> I also got this error after flashed the zeus image.

@lybtongji Do you mean after trying to flash a 2nd time?

lybtongji, 2020-06-03:
> 
> 
> > if I see progress, I'll try to update @andy-shev defconfigs, there are some inconsistencies, the SOF driver should probe when fixed.
> 
> @plbossart that would be great.
> 
> > I also got this error after flashed the zeus image.
> 
> @lybtongji Do you mean after trying to flash a 2nd time?

I flashed the image again. It still not working.

run `flashall` :

```
>> flashall
Using U-boot target: edison-blankrndis
Now waiting for dfu device 8087:0a99
Please plug and reboot the board
Dfu device found
Flashing IFWI
Download        [=========================] 100%      4194304 bytes
Download done.
Download        [=========================] 100%      4194304 bytes
Download done.
Flashing U-Boot
Download        [=========================] 100%       532480 bytes
Download done.
Flashing U-Boot Environment
Download        [=========================] 100%        65536 bytes
Download done.
Flashing U-Boot Environment Backup
Download        [=========================] 100%        65536 bytes
Download done.
Rebooting to apply partiton changes
```

the serial output:

```
U-Boot 2020.04 (Jun 03 2020 - 01:52:20 +0000)

CPU:   Genuine Intel(R) CPU   4000  @  500MHz
DRAM:  980.6 MiB
WDT:   Started with servicing (60s timeout)
MMC:   mmc@ff3fc000: 0, mmc@ff3fa000: 1
Loading Environment from MMC... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Hit any key to stop autoboot:  0
Target:ifwi
Partitioning already done...
Partitioning already done...
Saving Environment to MMC... Writing to redundant MMC(0)... OK
#DOWNLOAD ... OK
Ctrl+C to exit ...
#DOWNLOAD ... OK
Ctrl+C to exit ...
#DOWNLOAD ... OK
Ctrl+C to exit ...
#DOWNLOAD ... OK
Ctrl+C to exit ...
#DOWNLOAD ... OK
Ctrl+C to exit ...
resetting ...

******************************
PSH KERNEL VERSION: b0182b2b
                WR: 20104000
******************************

SCU IPC: 0x800000d0  0xfffce92c

PSH miaHOB version: TNG.B0.VVBD.0000000c

microkernel built 11:24:08 Feb  5 2015

******* PSH loader *******
PCM page cache size = 192 KB
Cache Constraint = 0 Pages
Arming IPC driver ..
Adding page store pool ..
PagestoreAddr(IMR Start Address) = 0x04899000
pageStoreSize(IMR Size)          = 0x00080000

*** Ready to receive application ***

U-Boot 2020.04 (Jun 03 2020 - 01:52:20 +0000)

CPU:   Genuine Intel(R) CPU   4000  @  500MHz
DRAM:  980.6 MiB
WDT:   Started with servicing (60s timeout)
MMC:   mmc@ff3fc000: 0, mmc@ff3fa000: 1
Loading Environment from MMC... OK
In:    serial
Out:   serial
Err:   serial
Saving Environment to MMC... Writing to redundant MMC(0)... OK
Saving Environment to MMC... Writing to MMC(0)... OK
Net:   No ethernet found.
Hit any key to stop autoboot:  0
Target:blank
Partitioning using GPT
Writing GPT: success!
Saving Environment to MMC... Writing to redundant MMC(0)... OK
## Error: "boot_target_cmd" not defined
Unknown boot mode: boot
Saving Environment to MMC... Writing to MMC(0)... OK
Resetting to default boot mode and reboot...
resetting ...

******************************
PSH KERNEL VERSION: b0182b2b
                WR: 20104000
******************************

SCU IPC: 0x800000d0  0xfffce92c

PSH miaHOB version: TNG.B0.VVBD.0000000c

microkernel built 11:24:08 Feb  5 2015

******* PSH loader *******
PCM page cache size = 192 KB
Cache Constraint = 0 Pages
Arming IPC driver ..
Adding page store pool ..
PagestoreAddr(IMR Start Address) = 0x04899000
pageStoreSize(IMR Size)          = 0x00080000

*** Ready to receive application ***

U-Boot 2020.04 (Jun 03 2020 - 01:52:20 +0000)

CPU:   Genuine Intel(R) CPU   4000  @  500MHz
DRAM:  980.6 MiB
WDT:   Started with servicing (60s timeout)
MMC:   mmc@ff3fc000: 0, mmc@ff3fa000: 1
Loading Environment from MMC... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Hit any key to stop autoboot:  0
Target:blank
Partitioning already done...
## Error: "boot_target_cmd" not defined
Unknown boot mode: boot
Saving Environment to MMC... Writing to redundant MMC(0)... OK
Resetting to default boot mode and reboot...
resetting ...

......
```

The edison goes into a boot loop.

htot, 2020-06-03:
I see. Looks like `boot_target_cmd` is missing from env. I'll check tonight.

htot, 2020-06-04:
@lybtongji any luck?

lybtongji, 2020-06-05:
> 
> 
> @lybtongji any luck?

It's fine now.
took a long time to rebuild.

htot, 2020-06-05:
Ok, sorry for the bug.