edison-fw/meta-intel-edison #106: task do_kernel_configme does not exist for target linux-yocto

xlla, 2020-03-30:
<pre>
xlla@u18-1:~/develop/git/out/linux64/build$ bitbake linux-yocto -c kernel_configme -f
Loading cache: 100% |#######################################################################################################################| Time: 0:00:00
Loaded 5250 entries from dependency cache.
Parsing recipes: 100% |#####################################################################################################################| Time: 0:00:02
Parsing of 4102 .bb files complete (4096 cached, 6 parsed). 5256 targets, 269 skipped, 1 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "i686-poky-linux"
MACHINE              = "edison"
DISTRO               = "poky-edison"
DISTRO_VERSION       = "2.7.3"
TUNE_FEATURES        = "m32 corei7"
TARGET_FPU           = ""
DISTRO_NAME          = "Poky (Yocto Project Reference Distro)"
ROS_DISTRO           = "melodic"
ROS_VERSION          = "1"
ROS_PYTHON_VERSION   = "2"
meta                 
meta-poky            
meta-yocto-bsp       = "warrior:f475afc5df0837532dcd0f3a831ddc3aec8941f1"
meta-oe              
meta-python          
meta-networking      = "warrior:a24acf94d48d635eca668ea34598c6e5c857e3f8"
meta-intel           = "warrior:29ee4852a05931dcf856670d9d8a3c3077a40fe8"
meta-intel-edison-bsp 
meta-intel-edison-distro 
meta-intel-arduino   
meta-arduino         = "warrior-32-sketch-v2.1:5c7e5dfe24e00bf8f40e6f31d27059628a4ebd99"
meta-acpi            = "eds-5.0.0:525c0743a2b974acf126946331f70f7c270238eb"
meta-oic             = "master:f3e52b40178d966b260744b0ed91fb5b5e2062bd"
meta-qt5             = "warrior:20331bd6b696793a7aa23a3c035e5ebcf3075aa3"
meta-ros1-melodic    
meta-ros1            
meta-ros-common      = "warrior-draft:2cda90d23348766ff1bb1d6617c1f8e0f6181f3a"
workspace            = "<unknown>:<unknown>"

ERROR: Task do_kernel_configme does not exist for target linux-yocto (/home/xlla/develop/git/meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/linux-yocto_5.5.0.bb:do_kernel_configme). Close matches:
  do_menuconfig
ERROR: Command execution failed: 1

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

</pre>

htot, 2020-03-30:
you broke something (melodic? meta-arduino? meta-ros1?). I works for me:
```
ferry@kalamata:~/Develop/tmp/out/linux64/build$ bitbake linux-yocto -c kernel_configme -f
WARNING: Host distribution "ubuntu-19.10" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |##############################################################################################################################################################################| Time: 0:00:00
Loaded 3441 entries from dependency cache.
Parsing recipes: 100% |############################################################################################################################################################################| Time: 0:00:00
Parsing of 2311 .bb files complete (2310 cached, 1 parsed). 3442 targets, 305 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "i686-poky-linux"
MACHINE              = "edison"
DISTRO               = "poky-edison"
DISTRO_VERSION       = "3.0.2"
TUNE_FEATURES        = "m32 corei7"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "zeus:5e1f52edb7a9f790fb6cb5d96502f3690267c1b1"
meta-oe              
meta-python          
meta-networking      = "zeus:bb65c27a772723dfe2c15b5e1b27bcc1a1ed884c"
meta-intel           = "zeus:1e6dcb762fed9ab5d0265477e4a2ab56748750da"
meta-intel-edison-bsp 
meta-intel-edison-distro 
meta-intel-arduino   
meta-arduino         = "sketch_reset_v2:8a72d63252893b53e3b33ace7321077f26412065"
meta-acpi            = "eds-5.0.0:a4692530a015eb55df755a31bb738af3a056efbe"

NOTE: Tainting hash to force rebuild of task /home/ferry/Develop/tmp/meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/linux-yocto_5.5.0.bb, do_kernel_configme                         | ETA:  0:00:00
WARNING: /home/ferry/Develop/tmp/meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/linux-yocto_5.5.0.bb:do_kernel_configme is tainted from a forced run                                 | ETA:  0:00:00
Initialising tasks: 100% |#########################################################################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 0 Found 0 Missed 0 Current 30 (0% match, 100% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
NOTE: Tasks Summary: Attempted 242 tasks of which 241 didn't need to be rerun and all succeeded.

Summary: There were 2 WARNING messages shown.
```

xlla, 2020-04-07:
Not break anything. I just add those layers for ros , qt5 and devtool's workspace.
I found the reason, after finish kernel modify, it comes back.
<pre>
xlla@u18-1:~/develop/git/out/linux64/build$ devtool finish linux-yocto ~/develop/git/meta-intel-edison/meta-intel-edison-bsp
NOTE: Starting bitbake server...
Loading cache: 100% |#######################################################################################################################| Time: 0:00:00
Loaded 5250 entries from dependency cache.
Parsing recipes: 100% |#####################################################################################################################| Time: 0:00:03
Parsing of 4102 .bb files complete (4096 cached, 6 parsed). 5256 targets, 269 skipped, 1 masked, 0 errors.
INFO: Would remove config fragment /tmp/devtoolvqaeuvwk/tmprvonnts6/devtool-fragment.cfg
INFO: Updating file fbtft.cfg
INFO: Adding new patch 0001-add-mcu.patch
INFO: Adding new patch 0002-add-protected-pin-group-change.patch
INFO: Updating recipe linux-yocto_5.5.0.bb
INFO: Removing file /home/xlla/develop/git/meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/files/sof.cfg
INFO: Cleaning sysroot for recipe linux-yocto...
INFO: Leaving source tree /home/xlla/develop/git/out/linux64/build/workspace/sources/linux-yocto as-is; if you no longer need it then please delete it manually

</pre>

<pre>
xlla@u18-1:~/develop/git/out/linux64/build$ bitbake linux-yocto -c kernel_configme -f
Loading cache: 100% |#######################################################################################################################| Time: 0:00:00
Loaded 5251 entries from dependency cache.
Parsing recipes: 100% |#####################################################################################################################| Time: 0:00:01
Parsing of 4102 .bb files complete (4097 cached, 5 parsed). 5256 targets, 269 skipped, 1 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "i686-poky-linux"
MACHINE              = "edison"
DISTRO               = "poky-edison"
DISTRO_VERSION       = "2.7.3"
TUNE_FEATURES        = "m32 corei7"
TARGET_FPU           = ""
DISTRO_NAME          = "Poky (Yocto Project Reference Distro)"
ROS_DISTRO           = "melodic"
ROS_VERSION          = "1"
ROS_PYTHON_VERSION   = "2"
meta                 
meta-poky            
meta-yocto-bsp       = "warrior:f475afc5df0837532dcd0f3a831ddc3aec8941f1"
meta-oe              
meta-python          
meta-networking      = "warrior:a24acf94d48d635eca668ea34598c6e5c857e3f8"
meta-intel           = "warrior:29ee4852a05931dcf856670d9d8a3c3077a40fe8"
meta-intel-edison-bsp 
meta-intel-edison-distro 
meta-intel-arduino   
meta-arduino         = "warrior-32-sketch-v2.1:a0e7f76f495e8f6e37efd1bac779323236b61cb0"
meta-acpi            = "eds-5.0.0:525c0743a2b974acf126946331f70f7c270238eb"
meta-oic             = "master:f3e52b40178d966b260744b0ed91fb5b5e2062bd"
meta-qt5             = "warrior:20331bd6b696793a7aa23a3c035e5ebcf3075aa3"
meta-ros1-melodic    
meta-ros1            
meta-ros-common      = "warrior-draft:2cda90d23348766ff1bb1d6617c1f8e0f6181f3a"
workspace            = "<unknown>:<unknown>"

NOTE: Tainting hash to force rebuild of task /home/xlla/develop/git/meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/linux-yocto_5.5.0.bb, do_kernel_configme
WARNING: /home/xlla/develop/git/meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/linux/linux-yocto_5.5.0.bb.do_kernel_configme is tainted from a forced run
Initialising tasks: 100% |##################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 0 Found 0 Missed 0 Current 29 (0% match, 100% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
NOTE: Tasks Summary: Attempted 234 tasks of which 233 didn't need to be rerun and all succeeded.

Summary: There was 1 WARNING message shown.
</pre>

this is not what I want.
bitbake edison-image is so lag.

bitbake linux-yocto is lag too, it unpack source each time. when I found something is missing , edit source is difficulty for those patch files.

I want to edit kernel source easy and build/deploy quickly.
so I follow the manual use devtool modify linux-yocto, then  I can edit source visual. build is quickly too, it does not unpack source again and again. but the built kernel missing some modules, with those boot errors:
<pre>
[FAILED] Failed to start Load Kernel Modules.
See 'systemctl status systemd-modules-load.service' for details.

</pre>
It seems fragment cfg was not addon in devtool modify mode.
how to allow kernel fragment configuration work in devtool mode.

htot, 2020-04-07:
I never used `devtool` so I don't know. Instead of applying fragment, you can add a patch to your own (local) copy of the kernel and change the hash in the recipe, then you are sure it is applied.
Nevertheless, you might need to install kernel modules to /lib/modules manually.

Note: I am working on [using btrfs file system](https://github.com/htot/meta-intel-edison/blob/btrfs/docs/_docs/Edison/Moving%20to%20btrfs/6.0-Why-btrfs-is-great.md) that allows multiple kernels / images. There will be instructions on how to install 2nd kernel + modules.

If you modify kernel sources it will always clean and fetch the sources (from the bbcache, so relatively fast) and apply patches and rebuild the whole kernel. On my machine that takes 2 1/2 minutes. It's a pain, but not too bad (4C+HT + SSD).

But you could choose to build the kernel outside Yocto. @andy-shev explained me there are some caching tricks that can speed up that. But I forgot the details.

Also Yocto can build kernels using the externalsrc variable. Never understood what the advantage would be.

andy-shev, 2020-04-07:
> But you could choose to build the kernel outside Yocto. @andy-shev explained me there are some caching tricks that can speed up that. But I forgot the details.

It's a tool, called `ccache`, and it's in general for all C/C++ programs.

xlla, 2020-04-08:
I am a newbie for devtool, but love it quickly, strong recommend,   it's easy for development/test, fast deployment, no need bitbake xxx/ apt-get install xxx, create patches automatically.  

I am using an old hdd and inside a virtual machine, so build kernel is very lag.
devtool will remain my working source, so it build a little fast for iteration dev.
I found another method using  KBUILD_DEFCONFIG_KMACHINE to indicate a whole defconfig, I will try it.

@htot another question,
in past, to quick deploy kernel, bitbake linux-yocto, bitbake edison-image, then mount edison-image-edison.hddimg, copy all files to edison:/boot, it's very lag and complicated.

now devtool support quick deploy recipe to target, it works for normal recipe, but not suite kernel, I think it due to our specific wrap for kernel, here is the dry-run list, could you teach me how/where to put these kernel/modules correct, then I can create a script to do that.

<pre>
xlla@u18-1:~/develop/git/out/linux64/build$ devtool deploy-target -s -n linux-yocto edison
NOTE: Starting bitbake server...
Loading cache: 100% |#######################################################################################################################| Time: 0:00:00
Loaded 5251 entries from dependency cache.
Parsing recipes: 100% |#####################################################################################################################| Time: 0:00:01
Parsing of 4102 .bb files complete (4096 cached, 6 parsed). 5256 targets, 269 skipped, 1 masked, 0 errors.
Files to be deployed for linux-yocto on target edison:
  /lib/modules/5.5.0-edison-acpi-standard/modules.builtin.modinfo
  /lib/modules/5.5.0-edison-acpi-standard/modules.order
  /lib/modules/5.5.0-edison-acpi-standard/modules.builtin
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/leds/leds-gpio.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/leds/led-class.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/leds/trigger/ledtrig-activity.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/leds/trigger/ledtrig-timer.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/leds/trigger/ledtrig-default-on.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/leds/trigger/ledtrig-oneshot.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/leds/trigger/ledtrig-gpio.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/leds/trigger/ledtrig-heartbeat.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/thermal/intel/x86_pkg_temp_thermal.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/dma/dmatest.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/dma/dw/dw_dmac.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/platform/x86/intel_mrfld_pwrbtn.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/platform/x86/eeepc-laptop.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/iio/adc/intel_mrfld_adc.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/iio/adc/ti-ads7950.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/iio/buffer/kfifo_buf.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/iio/buffer/industrialio-triggered-buffer.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/net/wireless/broadcom/brcm80211/brcmutil/brcmutil.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/net/usb/smsc95xx.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/mmc/core/pwrseq_emmc.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/mmc/core/mmc_core.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/mmc/core/mmc_block.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/mmc/core/pwrseq_simple.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/mmc/host/sdhci-acpi.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/mmc/host/cqhci.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/mmc/host/sdhci-pci.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/mmc/host/sdhci.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/pwm/pwm-lpss-platform.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/pwm/pwm-lpss.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/pwm/pwm-lpss-pci.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/hid/hid-lg-g15.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/hid/hid-logitech.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/hid/hid-multitouch.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/hid/i2c-hid/i2c-hid.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/video/backlight/backlight.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/video/backlight/generic_bl.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/video/fbdev/ssd1307fb.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/input/sparse-keymap.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/input/input-leds.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/mfd/intel_soc_pmic_mrfld.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/spi/spidev.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/spi/spi-pxa2xx-pci.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/spi/spi-pxa2xx-platform.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/drivers/i2c/i2c-dev.ko
[... cut ...]
  /lib/modules/5.5.0-edison-acpi-standard/kernel/net/bluetooth/bnep/bnep.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/net/ipv6/netfilter/nf_log_ipv6.ko
  /lib/modules/5.5.0-edison-acpi-standard/kernel/fs/efivarfs/efivarfs.ko
  /boot/Module.symvers-5.5.0-edison-acpi-standard
  /boot/bzImage-5.5.0-edison-acpi-standard
  /boot/System.map-5.5.0-edison-acpi-standard
  /boot/bzImage
  /boot/config-5.5.0-edison-acpi-standard
  /boot/vmlinux-5.5.0-edison-acpi-standard
</pre>

@andy-shev  I will learn to use it, thanks!

htot, 2020-04-08:
The paths are correct (/boot, /lib/modules), but it depends what you want to do (on eMMC, on sdhc card, on usb stick), and if there is space to do that.

I am not using `devtool` so I don't know how to use.

xlla, 2020-04-09:
I want to deploy to eMMC.  space is enough, mmcblk0p8 - 2Gb, mmcblk0p7 - 64Mb.
the files to deploy to /boot seems differ from flashed /boot
<pre>
root@edison:~# ls -al /boot
total 12260
drwxr-xr-x  2 root root    16384 Jan  1  1970 .
drwxr-xr-x 24 root root     4096 Apr  7 01:35 ..
-rwxr-xr-x  1 root root 12105632 Apr  7 16:11 bzImage-initramfs-edison.bin
-rwxr-xr-x  1 root root   124964 Apr  7 16:11 ldlinux.c32
-rwxr-xr-x  1 root root    68608 Apr  7 16:11 ldlinux.sys
-rwxr-xr-x  1 root root   174144 Apr  7 16:11 libcom32.c32
-rwxr-xr-x  1 root root    22836 Apr  7 16:11 libutil.c32
-rwxr-xr-x  1 root root      676 Apr  7 16:11 syslinux.cfg
-rwxr-xr-x  1 root root    27052 Apr  7 16:11 vesamenu.c32
</pre>

I should create a script to translate to those bin/c32/sys/cfg files.

htot, 2020-04-09:
Yocto generates bzImage-initramfs-edison.bin,which is the only file you need to boot.
Dev-tool generates bzImage-5.5.0-edison-acpi-standard. This is without initramfs. If all modules you need to boot are built into kernel, this is all you need.

But AFAIK some drivers needed for boot need to be built as modules, not built-in. So to boot you need a initramfs. This can be built into kernel, or can be a separate cpio.

BTW after boot all other modules are needed, so /lib/modules need to be copied as well.

xlla, 2020-04-09:
after read task logs, I realize it's far complicated than my imagination.
cpio file was created in  task do_image_cpio
cpio file combined into initramfs in task do_bundle_initramfs
cfg files was created in task do_bootimg

so, deploy lib/modules/*.ko , bitbake edison-image and extract /boot contents to edison /boot is the only easiest method. :)

closed 2020-04-09 by xlla

htot, 2020-04-11:
I am moving away from built-in initramfs (see btrfs branch of htot). The reason is: there appears to be a limit of 16MB on the size. And I want to create an initramfs with sufficient tools to repair a non-booting fs. Currently the kernel + cpio is 20MB and has parted, btrfs and ext tools.
So maybe it is possible build kernel + cpio using devtool and save time.
But the most time will be saved by buying an ssd and making you PC dual bootable instead of using a VM.