edison-fw/meta-intel-edison #108: ADC: only A0 work.

xlla, 2020-04-06:
<pre>
root@edison:~# cat /sys/bus/iio/devices/iio\:device1/in_voltage0_raw
22
root@edison:~# cat /sys/bus/iio/devices/iio\:device1/in_voltage1_raw
cat: '/sys/bus/iio/devices/iio:device1/in_voltage1_raw': Input/output error
root@edison:~# cat /sys/bus/iio/devices/iio\:device1/in_voltage2_raw
cat: '/sys/bus/iio/devices/iio:device1/in_voltage2_raw': Input/output error
root@edison:~# cat /sys/bus/iio/devices/iio\:device1/in_voltage3_raw
cat: '/sys/bus/iio/devices/iio:device1/in_voltage3_raw': Input/output error
root@edison:~# cat /sys/bus/iio/devices/iio\:device1/in_voltage4_raw
cat: '/sys/bus/iio/devices/iio:device1/in_voltage4_raw': Input/output error
root@edison:~# cat /sys/bus/iio/devices/iio\:device1/in_voltage5_raw
cat: '/sys/bus/iio/devices/iio:device1/in_voltage5_raw': Input/output error
gpio-506 (TRI_STATE_ALL       |mraa                ) out hi 

</pre>

htot, 2020-04-06:
Again seems specific to your build. I have:
```
root@edison:~# cat /sys/bus/iio/devices/iio\:device1/in_voltage0_raw 
61
root@edison:~# cat /sys/bus/iio/devices/iio\:device1/in_voltage1_raw 
7
root@edison:~# cat /sys/bus/iio/devices/iio\:device1/in_voltage2_raw 
5
root@edison:~# cat /sys/bus/iio/devices/iio\:device1/in_voltage3_raw 
4
root@edison:~# cat /sys/bus/iio/devices/iio\:device1/in_voltage4_raw 
2816
root@edison:~# cat /sys/bus/iio/devices/iio\:device1/in_voltage5_raw 
2261
root@edison:~# cat /sys/bus/iio/devices/iio\:device1/in_voltage6_raw 
2063
root@edison:~# cat /sys/bus/iio/devices/iio\:device1/in_voltage7_raw 
58
```

xlla, 2020-04-08:
I don't know what cause that.
I found iio could not read channel except ch0 too.
<pre>
root@edison:~# iio_info 
Library version: 0.15 (git tag: v0.15)
Compiled with backends: local xml ip usb
IIO context created with local backend.
Backend version: 0.15 (git tag: v0.15)
Backend description string: Linux edison 5.5.0-edison-acpi-standard #1 SMP Tue Apr 7 08:01:25 UTC 2020 i686
IIO context has 1 attributes:
	local,kernel: 5.5.0-edison-acpi-standard
IIO context has 3 devices:
	iio:device1: ads7951 (buffer capable)
		9 channels found:
			voltage0:  (input, index: 0, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 21
				attr  1: scale value: 2.442002442
			voltage1:  (input, index: 1, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw ERROR: Input/output error (-5)
				attr  1: scale value: 2.442002442
			voltage2:  (input, index: 2, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw ERROR: Input/output error (-5)
				attr  1: scale value: 2.442002442
			voltage3:  (input, index: 3, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw ERROR: Input/output error (-5)
				attr  1: scale value: 2.442002442
			voltage4:  (input, index: 4, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw ERROR: Input/output error (-5)
				attr  1: scale value: 2.442002442
			voltage5:  (input, index: 5, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw ERROR: Input/output error (-5)
				attr  1: scale value: 2.442002442
			voltage6:  (input, index: 6, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw ERROR: Input/output error (-5)
				attr  1: scale value: 2.442002442
			voltage7:  (input, index: 7, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw ERROR: Input/output error (-5)
				attr  1: scale value: 2.442002442
			timestamp:  (input, index: 8, format: le:S64/64>>0)
		1 device-specific attributes found:
				attr  0: current_timestamp_clock value: realtime

		2 buffer-specific attributes found:
				attr  0: watermark value: 1
				attr  1: data_available value: 0
	iio:device0: mrfld_bcove_adc
		9 channels found:
			temp6:  (input)
			1 channel-specific attributes found:
				attr  0: raw value: 304
			voltage0:  (input)
			1 channel-specific attributes found:
				attr  0: raw value: 987
			temp3:  (input)
			1 channel-specific attributes found:
				attr  0: raw value: 478
			temp7:  (input)
			1 channel-specific attributes found:
				attr  0: raw value: 1
			temp4:  (input)
			1 channel-specific attributes found:
				attr  0: raw value: 1
			temp8:  (input)
			1 channel-specific attributes found:
				attr  0: raw value: 1
			resistance1:  (input)
			1 channel-specific attributes found:
				attr  0: raw value: 1020
			temp5:  (input)
			1 channel-specific attributes found:
				attr  0: raw value: 1
			current2:  (input)
			1 channel-specific attributes found:
				attr  0: raw value: 513
	iio_sysfs_trigger:
		0 channels found:
		2 device-specific attributes found:
				attr  0: add_trigger ERROR: Permission denied (-13)
				attr  1: remove_trigger ERROR: Permission denied (-13)

</pre>

and I found same error in your [post here](https://github.com/andy-shev/linux/issues/9),
only  raw value in voltage0 was read (8)

htot, 2020-04-08:
> and I found same error in your post here,

Good point, I forgot about that, I don't know how that got fixed.

htot, 2020-04-08:
So I have:
```
root@edison:~# iio_info  
Library version: 0.15 (git tag: v0.15)
Compiled with backends: local xml ip usb
IIO context created with local backend.
Backend version: 0.15 (git tag: v0.15)
Backend description string: Linux edison 5.5.0-edison-acpi-standard #1 SMP Thu Apr 2 20:06:36 UTC 2020 x86_64
IIO context has 1 attributes:
        local,kernel: 5.5.0-edison-acpi-standard
IIO context has 3 devices:
        iio_sysfs_trigger:
                0 channels found:
                2 device-specific attributes found:
                                attr  0: remove_trigger ERROR: Permission denied (-13)
                                attr  1: add_trigger ERROR: Permission denied (-13)
        iio:device1: ads7951 (buffer capable)
                9 channels found:
                        voltage0:  (input, index: 0, format: le:u12/16>>0)
                        2 channel-specific attributes found:
                                attr  0: raw value: 17
                                attr  1: scale value: 2.442002442
                        voltage1:  (input, index: 1, format: le:u12/16>>0)
                        2 channel-specific attributes found:
                                attr  0: raw value: 6
                                attr  1: scale value: 2.442002442
                        voltage2:  (input, index: 2, format: le:u12/16>>0)
                        2 channel-specific attributes found:
                                attr  0: raw value: 6
                                attr  1: scale value: 2.442002442
                        voltage3:  (input, index: 3, format: le:u12/16>>0)
                        2 channel-specific attributes found:
                                attr  0: raw value: 4
                                attr  1: scale value: 2.442002442
                        voltage4:  (input, index: 4, format: le:u12/16>>0)
                        2 channel-specific attributes found:
                                attr  0: raw value: 3034
                                attr  1: scale value: 2.442002442
                        voltage5:  (input, index: 5, format: le:u12/16>>0)
                        2 channel-specific attributes found:
                                attr  0: raw value: 2465
                                attr  1: scale value: 2.442002442
                        voltage6:  (input, index: 6, format: le:u12/16>>0)
                        2 channel-specific attributes found:
                                attr  0: raw value: 1179
                                attr  1: scale value: 2.442002442
                        voltage7:  (input, index: 7, format: le:u12/16>>0)
                        2 channel-specific attributes found:
                                attr  0: raw value: 192
                                attr  1: scale value: 2.442002442
                        timestamp:  (input, index: 8, format: le:S64/64>>0)
                1 device-specific attributes found:
                                attr  0: current_timestamp_clock value: realtime

                2 buffer-specific attributes found:
                                attr  0: watermark value: 1
                                attr  1: data_available value: 0
        iio:device0: mrfld_bcove_adc
                9 channels found:
                        temp3:  (input)
                        1 channel-specific attributes found:
                                attr  0: raw value: 0
                        temp7:  (input)
                        1 channel-specific attributes found:
                                attr  0: raw value: 1
                        current2:  (input)
                        1 channel-specific attributes found:
                                attr  0: raw value: 510
                        temp4:  (input)
                        1 channel-specific attributes found:
                                attr  0: raw value: 1
                        temp8:  (input)
                        1 channel-specific attributes found:
                                attr  0: raw value: 1
                        temp5:  (input)
                        1 channel-specific attributes found:
                                attr  0: raw value: 1
                        resistance1:  (input)
                        1 channel-specific attributes found:
                                attr  0: raw value: 0
                        temp6:  (input)
                        1 channel-specific attributes found:
                                attr  0: raw value: 334
                        voltage0:  (input)
                        1 channel-specific attributes found:
                                attr  0: raw value: 981
```
Note: the order of devices is not the same. I am using `x86_64`, you `i686`.
My meta-acpi is at [a4692530](https://github.com/edison-fw/meta-acpi/commit/a4692530a015eb55df755a31bb738af3a056efbe). And you?

xlla, 2020-04-10:
I have tested same firmware in another Edison, same result.
I guess it maybe caused by manual acpi-table-load, can you disable kernel and service acpi load,  do acpi-table-load in shell, then check the result?

xlla, 2020-04-10:
and I have combine all asl files into one.
<pre>
root@edison:~# ls -al /sys/kernel/config/acpi/table 
total 0
drwxr-xr-x 3 root root 0 Apr 10 15:37 .
drwxr-xr-x 3 root root 0 Jan  1  2000 ..
drwxr-xr-x 2 root root 0 Apr 10 15:37 oled-spi-leds
</pre>

when I do acpi-table-load, it will complain 
<pre>
gpioset: at least one GPIO line offset to value mapping must be specified
gpioset: at least one GPIO line offset to value mapping must be specified
</pre>

if I change sleep .2 to .5, it will find tristate line, but the characters on spi oled screen become big and  mistiness.

if I comment statement to skip pull down tristate line, and do acpi-table-load again, the spi oled screen works fine.

xlla, 2020-04-10:
I thought the logic of acpi-table-load is not obey the edison arduino hardware guide,
  
> Assert the tri_state_all signal to disconnect the shield pins
set mux/bias/soc mode
De-assert the tri_state_all signal to reconnect the shield pins

current flow is,
set mux/bias/soc mode
Assert the tri_state_all 
De-assert the tri_state_all

is it safety?  I found that quickly pull down/up tri_state_all after ssdt load will muss spi fb.

xlla, 2020-04-10:
I look into the ti-ads7950, 
ret = ti_ads7950_scan_direct(indio_dev, chan->address)

> Note: the order of devices is not the same. I am using `x86_64`, you `i686`

it seems the cause, I have add a lot of trace statements into ads7950,
found it will check data address when got a data in function ti_ads7950_read_raw,
only ch0 has a match address,  if I force it omit address check, then iio will read some wrong data but successful.
<pre>
	iio:device1: ads7951 (buffer capable)
		9 channels found:
			voltage0:  (input, index: 0, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 24
				attr  1: scale value: 2.442002442
			voltage1:  (input, index: 1, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 2054
				attr  1: scale value: 2.442002442
			voltage2:  (input, index: 2, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 5
				attr  1: scale value: 2.442002442
			voltage3:  (input, index: 3, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 2052
				attr  1: scale value: 2.442002442
			voltage4:  (input, index: 4, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 15
				attr  1: scale value: 2.442002442
			voltage5:  (input, index: 5, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 2062
				attr  1: scale value: 2.442002442
			voltage6:  (input, index: 6, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 0
				attr  1: scale value: 2.442002442
			voltage7:  (input, index: 7, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 2085
				attr  1: scale value: 2.442002442
			timestamp:  (input, index: 8, format: le:S64/64>>0)
		1 device-specific attributes found:
				attr  0: current_timestamp_clock value: realtime

		2 buffer-specific attributes found:
				attr  0: watermark value: 1
				attr  1: data_available value: 0
</pre> 

and here is the log,  for each channel, address[x] is actual address, expect addr[y] is expect address.
the pattern like, 
  ch0,ch1 want y 0
  ch2,ch3 want y 1
 ch4,ch5 want y 2
 ch6, ch7 want y 3

like a 64bit address be treated as 2 32 bit address in somewhere.

<pre>
[  107.762281] ads7950 read raw: ch[0].
[  107.762290] ads7950 read raw, scan direct, address [0].
[  107.762295] ads7950 scan dir, ch [0].
[  107.762301] ads7950 scan dir, cmd [1000]
[  107.762600] ads7950 scan dir, ret [0].
[  107.762607] ads7950 scan dir, rx [24].
[  107.762612] ads7950 read raw, expect addr[0].
[  107.762617] ads7950 read raw, val [24].
[  107.762839] ads7950 read raw: ch[0].
[  107.762846] ads7950 read raw, scale.
[  107.763236] ads7950 read raw: ch[1].
[  107.763244] ads7950 read raw, scan direct, address [1].
[  107.763248] ads7950 scan dir, ch [1].
[  107.763254] ads7950 scan dir, cmd [1080]
[  107.763522] ads7950 scan dir, ret [0].
[  107.763529] ads7950 scan dir, rx [2054].
[  107.763534] ads7950 read raw, expect addr[0].
[  107.763539] ads7950 read raw, val [2054].
[  107.763753] ads7950 read raw: ch[0].
[  107.763759] ads7950 read raw, scale.
[  107.764094] ads7950 read raw: ch[2].
[  107.764102] ads7950 read raw, scan direct, address [2].
[  107.764106] ads7950 scan dir, ch [2].
[  107.764112] ads7950 scan dir, cmd [1100]
[  107.764354] ads7950 scan dir, ret [0].
[  107.764361] ads7950 scan dir, rx [4101].
[  107.764366] ads7950 read raw, expect addr[1].
[  107.764370] ads7950 read raw, val [5].
[  107.764582] ads7950 read raw: ch[0].
[  107.764588] ads7950 read raw, scale.
[  107.764863] ads7950 read raw: ch[3].
[  107.764884] ads7950 read raw, scan direct, address [3].
[  107.764903] ads7950 scan dir, ch [3].
[  107.764960] ads7950 scan dir, cmd [1180]
[  107.765200] ads7950 scan dir, ret [0].
[  107.765207] ads7950 scan dir, rx [6148].
[  107.765212] ads7950 read raw, expect addr[1].
[  107.765217] ads7950 read raw, val [2052].
[  107.765429] ads7950 read raw: ch[0].
[  107.765435] ads7950 read raw, scale.
[  107.765708] ads7950 read raw: ch[4].
[  107.765715] ads7950 read raw, scan direct, address [4].
[  107.765720] ads7950 scan dir, ch [4].
[  107.765725] ads7950 scan dir, cmd [1200]
[  107.766030] ads7950 scan dir, ret [0].
[  107.766037] ads7950 scan dir, rx [8207].
[  107.766042] ads7950 read raw, expect addr[2].
[  107.766047] ads7950 read raw, val [15].
[  107.766266] ads7950 read raw: ch[0].
[  107.766285] ads7950 read raw, scale.
[  107.766569] ads7950 read raw: ch[5].
[  107.766576] ads7950 read raw, scan direct, address [5].
[  107.766581] ads7950 scan dir, ch [5].
[  107.766587] ads7950 scan dir, cmd [1280]
[  107.768279] ads7950 scan dir, ret [0].
[  107.768288] ads7950 scan dir, rx [10254].
[  107.768293] ads7950 read raw, expect addr[2].
[  107.768299] ads7950 read raw, val [2062].
[  107.768544] ads7950 read raw: ch[0].
[  107.768550] ads7950 read raw, scale.
[  107.768837] ads7950 read raw: ch[6].
[  107.768845] ads7950 read raw, scan direct, address [6].
[  107.768849] ads7950 scan dir, ch [6].
[  107.768855] ads7950 scan dir, cmd [1300]
[  107.770064] ads7950 scan dir, ret [0].
[  107.770073] ads7950 scan dir, rx [12288].
[  107.770078] ads7950 read raw, expect addr[3].
[  107.770083] ads7950 read raw, val [0].
[  107.770307] ads7950 read raw: ch[0].
[  107.770313] ads7950 read raw, scale.
[  107.770591] ads7950 read raw: ch[7].
[  107.770599] ads7950 read raw, scan direct, address [7].
[  107.770604] ads7950 scan dir, ch [7].
[  107.770620] ads7950 scan dir, cmd [1380]
[  107.770905] ads7950 scan dir, ret [0].
[  107.770962] ads7950 scan dir, rx [14373].
[  107.770968] ads7950 read raw, expect addr[3].
[  107.770973] ads7950 read raw, val [2085].
[  107.771187] ads7950 read raw: ch[0].
[  107.771193] ads7950 read raw, scale.
</pre>

xlla, 2020-04-11:
ads7951 use 20Mhz clock, my spi screen use 1Mhz clock, are they interfere each other?

xlla, 2020-04-11:
@andy-shev , please help me to check this.
I have read the code of ti-ads7950 and add some trace log to observe what happened to other channels, see above log.

it seems read from spi bus is ok, but address check failure, I printed the return value of ti_ads7950_scan_direct in "ads7950 scan dir, rx [xxx]" in dec.

when I convert them to hex, I found some bits looks very like actual address.
<pre>
24 | 18
2054 | 806
4101 | 1005
6148 | 1804
8207 | 200F
10254 | 280E
12288 | 3000
14373 | 3825
</pre>
if shift them right 11bit, then remain address seems matches channel.
<pre>
24 | 18 | 0 | 0
2054 | 806 | 1 | 1
4101 | 1005 | 10 | 2
6148 | 1804 | 11 | 3
8207 | 200F | 100 | 4
10254 | 280E | 101 | 5
12288 | 3000 | 110 | 6
14373 | 3825 | 111 | 7
</pre>

then I change hardcode 12 to 11 in macro TI_ADS7950_EXTRACT call, it is work.
each channel data look normal. 
<pre>
root@edison:~# iio_info
Library version: 0.15 (git tag: v0.15)
Compiled with backends: local xml ip usb
IIO context created with local backend.
Backend version: 0.15 (git tag: v0.15)
Backend description string: Linux edison 5.5.0-edison-acpi-standard #1 SMP Wed Apr 8 08:29:05 UTC 2020 i686
IIO context has 1 attributes:
	local,kernel: 5.5.0-edison-acpi-standard
IIO context has 3 devices:
	iio:device1: ads7951 (buffer capable)
		9 channels found:
			voltage0:  (input, index: 0, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 24
				attr  1: scale value: 2.442002442
			voltage1:  (input, index: 1, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 9
				attr  1: scale value: 2.442002442
			voltage2:  (input, index: 2, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 6
				attr  1: scale value: 2.442002442
			voltage3:  (input, index: 3, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 4
				attr  1: scale value: 2.442002442
			voltage4:  (input, index: 4, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 16
				attr  1: scale value: 2.442002442
			voltage5:  (input, index: 5, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 14
				attr  1: scale value: 2.442002442
			voltage6:  (input, index: 6, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 0
				attr  1: scale value: 2.442002442
			voltage7:  (input, index: 7, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 36
				attr  1: scale value: 2.442002442
			timestamp:  (input, index: 8, format: le:S64/64>>0)
		1 device-specific attributes found:
				attr  0: current_timestamp_clock value: realtime

		2 buffer-specific attributes found:
				attr  0: watermark value: 1
				attr  1: data_available value: 0
</pre>

xlla, 2020-04-11:
since @htot has report work fine on x86_64 for hardcode 12, is there any method to determine this hardcode dynamically?

the usage of ti-ads795x chip using kernel driver is merely in google search result,
I can't find more information about this.
In [kernel](https://github.com/torvalds/linux) repo, I can't create any issue, so I have to ask question here.

andy-shev, 2020-04-11:
@xlla Unfortunately I have no access to hardware till the Tuesday next week. Can you do the same dump in x86_64 environment (please use hex numbers where it makes sense)?

andy-shev, 2020-04-11:
@xlla and can you share all your patches to debug? (Perhaps as a branch in you kernel repository) 

Honestly, I don't see a problem in this driver, it might be somewhere else.

andy-shev, 2020-04-11:
This actually is interesting part...

	/*
	 * DMA (thus cache coherency maintenance) requires the
	 * transfer buffers to live in their own cache lines.
	 */
	u16 rx_buf[TI_ADS7950_MAX_CHAN + 2 + TI_ADS7950_TIMESTAMP_SIZE]
							____cacheline_aligned;
	u16 tx_buf[TI_ADS7950_MAX_CHAN + 2];
	u16 single_tx;
	u16 single_rx;

Recently driver gained single short support and these `single_*` had been added, though in case of Edison the small transfers (less than DMA burst, which is 8 bytes) should go via PIO mode. So, still no explanation for the bit shift.

andy-shev, 2020-04-11:
Another possible explanation is the CS signal phase (need oscilloscope to check) is deviating and thus the first (least significant) bit of measurement gone.

In any case 12 in the driver seems in accordance with data sheet. So, can't be changed.

xlla, 2020-04-12:
my branch is [here](https://github.com/xlla/linux/blob/eds-acpi-5.5.0-v1/drivers/iio/adc/ti-ads7950.c).

for @htot 's iio_info log,  if there is nothing connected to A4/A5, then the raw values of A4/A5 is abnormal.
for my fixed version, A0-A5 is all close to zero.

I have connected a ML8511 sensor to edison, measured voltage on A0/A1 is correct compare to multimeter.

I am afraid I couldn't switch to x86_64 now, there is no enough hd space to build another huge set of firmware/recipes, and I can image there must be a lot things to do to adapt x86_64 for my in progress kernel.

xlla, 2020-04-12:
> Another possible explanation is the CS signal phase (need oscilloscope to check) is deviating and thus the first (least significant) bit of measurement gone.

as I questioned, ads use 20Mhz clock, oled use 1Mhz clock, how do they coordinate in same clk line.

andy-shev, 2020-04-12:
> > Another possible explanation is the CS signal phase (need oscilloscope to check) is deviating and thus the first (least significant) bit of measurement gone.
> 
> as I questioned, ads use 20Mhz clock, oled use 1Mhz clock, how do they coordinate in same clk line.

Based on so called *SPI messages*. The clock is getting setup per each message.

andy-shev, 2020-04-12:
@xlla 
Can you update the debug log based on your latest patches?

xlla, 2020-04-12:
trigger here
<pre>
root@edison:/mnt/study/test# iio_info
Library version: 0.15 (git tag: v0.15)
Compiled with backends: local xml ip usb
IIO context created with local backend.
Backend version: 0.15 (git tag: v0.15)
Backend description string: Linux edison 5.5.0-edison-acpi-standard #1 SMP Wed Apr 8 08:29:05 UTC 2020 i686
IIO context has 1 attributes:
	local,kernel: 5.5.0-edison-acpi-standard
IIO context has 3 devices:
	iio:device1: ads7951 (buffer capable)
		9 channels found:
			voltage0:  (input, index: 0, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 606
				attr  1: scale value: 2.442002442
			voltage1:  (input, index: 1, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 1967
				attr  1: scale value: 2.442002442
			voltage2:  (input, index: 2, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 83
				attr  1: scale value: 2.442002442
			voltage3:  (input, index: 3, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 11
				attr  1: scale value: 2.442002442
			voltage4:  (input, index: 4, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 19
				attr  1: scale value: 2.442002442
			voltage5:  (input, index: 5, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 17
				attr  1: scale value: 2.442002442
			voltage6:  (input, index: 6, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 0
				attr  1: scale value: 2.442002442
			voltage7:  (input, index: 7, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 37
				attr  1: scale value: 2.442002442
			timestamp:  (input, index: 8, format: le:S64/64>>0)
		1 device-specific attributes found:
				attr  0: current_timestamp_clock value: realtime
</pre>

debug log here
<pre>
[16431.983482] ads7950 read raw: ch[0].
[16431.983491] ads7950 read raw, scale.
[16431.983699] ads7950 read raw: ch[0].
[16431.983705] ads7950 read raw, scale.
[16431.984526] ads7950 read raw: ch[0].
[16431.984533] ads7950 read raw, scale.
[16431.984723] ads7950 read raw: ch[0].
[16431.984729] ads7950 read raw, scale.
[16431.984981] ads7950 read raw: ch[0].
[16431.984987] ads7950 read raw, scale.
[16431.985162] ads7950 read raw: ch[0].
[16431.985168] ads7950 read raw, scale.
[16431.985341] ads7950 read raw: ch[0].
[16431.985347] ads7950 read raw, scale.
[16431.985520] ads7950 read raw: ch[0].
[16431.985526] ads7950 read raw, scale.
[16431.987509] ads7950 read raw: ch[0].
[16431.987518] ads7950 read raw, scan direct, address [0].
[... 64 lines cut ...]
[16432.004156] ads7950 read raw: ch[6].
[16432.004164] ads7950 read raw, scan direct, address [6].
[16432.004168] ads7950 scan dir, ch [6].
[16432.004174] ads7950 scan dir, cmd [1300]
[16432.004445] ads7950 scan dir, ret [0].
[16432.004452] ads7950 scan dir, rx [12288].
[16432.004459] ads7950 read raw, ret[12288] expect addr[3].
[16432.004464] ads7950 read raw, (8,4) addr[3000].
[16432.004470] ads7950 read raw, shift[0], realbits[12], val [0].
[16432.004692] ads7950 read raw: ch[0].
[16432.004698] ads7950 read raw, scale.
[16432.007214] ads7950 read raw: ch[7].
[16432.007223] ads7950 read raw, scan direct, address [7].
[16432.007228] ads7950 scan dir, ch [7].
[16432.007234] ads7950 scan dir, cmd [1380]
[16432.007494] ads7950 scan dir, ret [0].
[16432.007501] ads7950 scan dir, rx [14373].
[16432.007508] ads7950 read raw, ret[14373] expect addr[3].
[16432.007513] ads7950 read raw, (8,4) addr[3825].
[16432.007520] ads7950 read raw, shift[0], realbits[12], val [37].
[16432.007749] ads7950 read raw: ch[0].
[16432.007755] ads7950 read raw, scale.
</pre>

"ads7950 read raw, (8,4)" should be "ads7950 read raw, (11,4)", is a typo.

andy-shev, 2020-04-13:
Can you try to apply this patch (https://github.com/andy-shev/linux/commit/e436e0f9274ae0fe28faf0c9af720b8ffc0fc98b) against kernel and this one (https://github.com/westeri/meta-acpi/commit/2709b5021bb23502c95c7c44e606e598cc43bb87) against ASL code and test if it helps?

xlla, 2020-04-14:
Hi @andy-shev ,
I'v apply those patches and deploy to edison, here is the log.
<pre>
root@edison:~# iio_info 
Library version: 0.15 (git tag: v0.15)
Compiled with backends: local xml ip usb
IIO context created with local backend.
Backend version: 0.15 (git tag: v0.15)
Backend description string: Linux edison 5.5.0-edison-acpi-standard #1 SMP Wed Apr 8 08:29:05 UTC 2020 i686
IIO context has 1 attributes:
	local,kernel: 5.5.0-edison-acpi-standard
IIO context has 3 devices:
	iio:device1: ads7951 (buffer capable)
		9 channels found:
			voltage0:  (input, index: 0, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 605
				attr  1: scale value: 2.442002442
			voltage1:  (input, index: 1, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 1969
				attr  1: scale value: 2.442002442
			voltage2:  (input, index: 2, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 83
				attr  1: scale value: 2.442002442
			voltage3:  (input, index: 3, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 11
				attr  1: scale value: 2.442002442
			voltage4:  (input, index: 4, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 11
				attr  1: scale value: 2.442002442
			voltage5:  (input, index: 5, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 10
				attr  1: scale value: 2.442002442
			voltage6:  (input, index: 6, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 0
				attr  1: scale value: 2.442002442
			voltage7:  (input, index: 7, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 35
				attr  1: scale value: 2.442002442
			timestamp:  (input, index: 8, format: le:S64/64>>0)
		1 device-specific attributes found:
				attr  0: current_timestamp_clock value: realtime

		2 buffer-specific attributes found:
				attr  0: watermark value: 1
				attr  1: data_available value: 0
	iio:device0: mrfld_bcove_adc

</pre>

<pre>
root@edison:~# dmesg|grep ads
[    0.090550] process: using mwait in idle threads
[   46.892913] ads7950 init...
[   46.892967] ads7950 [ads7951], channels [9].
[   46.892974] ads7950 vref_mv [5000].
[   46.892999] ads7950 spi-PRP0001:00: spi-PRP0001:00 supply vref not found, using dummy regulator
[   46.893109] ads7950 vref ok.
[   46.893130] ads7950 triggered buffer ok.
[   46.897813] ads7950 adc chip ok.
[   46.902345] ads7950 add gpio chip[spi-PRP0001:00].
[   46.903858] ads7950 ready.
[   51.254389] ads7950 read raw: ch[0].
[... 80 lines cut ...]
[   51.274982] ads7950 read raw, scale.
[   51.275262] ads7950 read raw: ch[6].
[   51.275269] ads7950 read raw, scan direct, address [6].
[   51.275274] ads7950 scan dir, ch [6].
[   51.275279] ads7950 scan dir, cmd [1300]
[   51.277375] ads7950 scan dir, ret [0].
[   51.277384] ads7950 scan dir, rx [12288].
[   51.277391] ads7950 read raw, ret[12288] expect addr[3].
[   51.277396] ads7950 read raw, (8,4) addr[3000].
[   51.277413] ads7950 read raw, shift[0], realbits[12], val [0].
[   51.278633] ads7950 read raw: ch[0].
[   51.278640] ads7950 read raw, scale.
[   51.278929] ads7950 read raw: ch[7].
[   51.278936] ads7950 read raw, scan direct, address [7].
[   51.278941] ads7950 scan dir, ch [7].
[   51.278947] ads7950 scan dir, cmd [1380]
[   51.279223] ads7950 scan dir, ret [0].
[   51.279231] ads7950 scan dir, rx [14371].
[   51.279237] ads7950 read raw, ret[14371] expect addr[3].
[   51.279243] ads7950 read raw, (8,4) addr[3823].
[   51.279249] ads7950 read raw, shift[0], realbits[12], val [35].
[   51.281513] ads7950 read raw: ch[0].
[   51.281520] ads7950 read raw, scale.

</pre>

xlla, 2020-04-14:
> Ah, also you may drop the frequency to 10MHz and see if it helps. The chip is 20MHz compatible and SPI driver should put a divisor to guarantee that clock is not bigger than asked (in Edison case I think it's 16.667MHz), but still PCB can interfere with this frequency somehow.

I tried this, it seems work. look my log,
"all expect addr[x]", x is extracted by shift 12,  it is matches the ch, 
since my actual check statement is still using 11 bit, so all values of ch which above 0 is error in iio_info.

<pre>
root@edison:~# iio_info 
Library version: 0.15 (git tag: v0.15)
Compiled with backends: local xml ip usb
IIO context created with local backend.
Backend version: 0.15 (git tag: v0.15)
Backend description string: Linux edison 5.5.0-edison-acpi-standard #1 SMP Wed Apr 8 08:29:05 UTC 2020 i686
IIO context has 1 attributes:
	local,kernel: 5.5.0-edison-acpi-standard
IIO context has 3 devices:
	iio:device1: ads7951 (buffer capable)
		9 channels found:
			voltage0:  (input, index: 0, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw value: 1184
				attr  1: scale value: 2.442002442
			voltage1:  (input, index: 1, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw ERROR: Input/output error (-5)
				attr  1: scale value: 2.442002442
			voltage2:  (input, index: 2, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw ERROR: Input/output error (-5)
				attr  1: scale value: 2.442002442
			voltage3:  (input, index: 3, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw ERROR: Input/output error (-5)
				attr  1: scale value: 2.442002442
			voltage4:  (input, index: 4, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw ERROR: Input/output error (-5)
				attr  1: scale value: 2.442002442
			voltage5:  (input, index: 5, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw ERROR: Input/output error (-5)
				attr  1: scale value: 2.442002442
			voltage6:  (input, index: 6, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw ERROR: Input/output error (-5)
				attr  1: scale value: 2.442002442
			voltage7:  (input, index: 7, format: le:u12/16>>0)
			2 channel-specific attributes found:
				attr  0: raw ERROR: Input/output error (-5)
				attr  1: scale value: 2.442002442
			timestamp:  (input, index: 8, format: le:S64/64>>0)
		1 device-specific attributes found:
				attr  0: current_timestamp_clock value: realtime

		2 buffer-specific attributes found:
				attr  0: watermark value: 1

</pre>   

<pre>
[  811.212409] ads7950 spi-PRP0001:00: Failed to retrieve ACPI GPIO chip
[  811.314914] ads7950 init...
[  811.314963] ads7950 [ads7951], channels [9].
[  811.314970] ads7950 vref_mv [5000].
[  811.314994] ads7950 spi-PRP0001:00: spi-PRP0001:00 supply vref not found, using dummy regulator
[  811.315088] ads7950 vref ok.
[  811.315109] ads7950 triggered buffer ok.
[  811.320423] ads7950 adc chip ok.
[  811.321761] ads7950 add gpio chip[spi-PRP0001:00].
[  811.327584] ads7950 ready.
[... 73 lines cut ...]
[  824.157883] ads7950 read raw, ret[20501] expect addr[5].
[  824.157888] ads7950 read raw, (8,4) addr[5015].
[  824.158114] ads7950 read raw: ch[0].
[  824.158120] ads7950 read raw, scale.
[  824.158399] ads7950 read raw: ch[6].
[  824.158407] ads7950 read raw, scan direct, address [6].
[  824.158411] ads7950 scan dir, ch [6].
[  824.158417] ads7950 scan dir, cmd [1300]
[  824.158681] ads7950 scan dir, ret [0].
[  824.158688] ads7950 scan dir, rx [24576].
[  824.158694] ads7950 read raw, ret[24576] expect addr[6].
[  824.158699] ads7950 read raw, (8,4) addr[6000].
[  824.162128] ads7950 read raw: ch[0].
[  824.162135] ads7950 read raw, scale.
[  824.162416] ads7950 read raw: ch[7].
[  824.162424] ads7950 read raw, scan direct, address [7].
[  824.162429] ads7950 scan dir, ch [7].
[  824.162435] ads7950 scan dir, cmd [1380]
[  824.162704] ads7950 scan dir, ret [0].
[  824.162711] ads7950 scan dir, rx [28734].
[  824.162718] ads7950 read raw, ret[28734] expect addr[7].
[  824.162723] ads7950 read raw, (8,4) addr[703e].
[  824.163005] ads7950 read raw: ch[0].
[  824.163012] ads7950 read raw, scale.
</pre>

andy-shev, 2020-04-14:
@xlla Can you, please, test with original address extraction code?

xlla, 2020-04-14:
@andy-shev from my above log No.2 in backward order, 
the original address extraction code result is [xxx] in those 
"ads7950 read raw, ret[6155] expect addr[xxx]."

for channel 3, it is 1 by original address extract code. 
<pre>
[   51.268082] ads7950 read raw, scan direct, address [3].
[   51.268087] ads7950 scan dir, ch [3].
[   51.268093] ads7950 scan dir, cmd [1180]
[   51.268390] ads7950 scan dir, ret [0].
[   51.268397] ads7950 scan dir, rx [6155].
[   51.268403] ads7950 read raw, ret[6155] expect addr[1].
[   51.268408] ads7950 read raw, (8,4) addr[180b].
[   51.268478] ads7950 read raw, shift[0], realbits[12], val [11].
[   51.270200] ads7950 read raw: ch[0].
[   51.270207] ads7950 read raw, scale.
[   51.270980] ads7950 read raw: ch[4].
</pre>

[in this mraa manual](https://iotdk.intel.com/docs/master/mraa/edison.html) , it said Max spi speed is 6.25Mhz, is it right?
or for new linux kernel, it's performance improved?

andy-shev, 2020-04-14:
> @andy-shev from my above log No.2 in backward order,
> the original address extraction code result is [xxx] in those
> "ads7950 read raw, ret[6155] expect addr[xxx]."
> 
> for channel 3, it is 1 by original address extract code.
> 
> [   51.268082] ads7950 read raw, scan direct, address [3].
> [   51.268087] ads7950 scan dir, ch [3].
> [   51.268093] ads7950 scan dir, cmd [1180]
> [   51.268390] ads7950 scan dir, ret [0].
> [   51.268397] ads7950 scan dir, rx [6155].
> [   51.268403] ads7950 read raw, ret[6155] expect addr[1].
> [   51.268408] ads7950 read raw, (8,4) addr[180b].
> [   51.268478] ads7950 read raw, shift[0], realbits[12], val [11].
> [   51.270200] ads7950 read raw: ch[0].
> [   51.270207] ads7950 read raw, scale.
> [   51.270980] ads7950 read raw: ch[4].

So, I didn't get. Do patch fix the issue or not?

> [in this mraa manual](https://iotdk.intel.com/docs/master/mraa/edison.html) , it said Max spi speed is 6.25Mhz, is it right?
> or for new linux kernel, it's performance improved?

htot, 2020-04-14:
> It always was up to 25MHz in hardware, the old BSP simple mistaken by unknown reasons.

Interesting. I thought SPI and HSU had similar max speeds. But SPI appears significantly faster.

xlla, 2020-04-15:
> So, I didn't get. Do patch fix the issue or not?

It is not fix the issue.

slow bus speed to 10Mhz eliminate the issue.

andy-shev, 2020-04-15:
> > So, I didn't get. Do patch fix the issue or not?
> 
> It is not fix the issue.

In any case it's right thing to do. This ADC, as many others, is 16-bit word size.

> slow bus speed to 10Mhz eliminate the issue.

A-ha, it's something. So, we need to have oscilloscope charts to understand what's going on there. JFYI: 20MHz is actually 12.5MHz since base clock is 25MHz, but it may be a bug somewhere and OS thinks wrong about actual bus speed.

So, I can update ASL for now with 10MHz speed until we will understand the issue clearly.

xlla, 2020-04-16:
@andy-shev I am connect edison to a usb-hub due to lack port of my computer,
and the usb-hub was powered by a 5V adapter from AC socket, will it interfere with clock signal.

I only have a handheld oscilloscope, 40Mhz baudwidth, 250M samples/s.
I can observe oled's clock in 1Mhz, but can't watch ADC's.  
the oled's clock signal is very noise and skipped every 2 seconds.

xlla, 2020-04-16:
I realize that I must trigger ADC to do a read so it will transmit clock signal.
I write a simple app to continue read A0/A1, then I can do a measurement.
[image]

xlla, 2020-04-16:
some article said, edison's main clock,
If the SoC is in S0 state, 50Mhz
If the SoC is in S0i2 state, 19.2Mhz
If the SoC is in S0i3 state, 32Khz

for spi,
as master, 25Mhz,
as slave, 16.67Mhz

andy-shev, 2020-04-16:
> I realize that I must trigger ADC to do a read so it will transmit clock signal.
> I write a simple app to continue read A0/A1, then I can do a measurement.
> [image]

Is it two channel oscilloscope? Because what we need is phase between SCK (Clock) and MISO (RxD) signals.

> some article said, edison's main clock,
> If the SoC is in S0 state, 50Mhz
> If the SoC is in S0i2 state, 19.2Mhz
> If the SoC is in S0i3 state, 32Khz

It a system clock, I think it's actually 100MHz.

> for spi,
> as master, 25Mhz,

Yes, that's what we have, 25, 12.5, 8.333, 6.25, ...

> as slave, 16.67Mhz

That's I don't know

xlla, 2020-04-17:
sck vs miso
I don't know if it useful, so I just take one photo.
[image]

> Yes, that's what we have, 25, 12.5, 8.333, 6.25, ...

should we set spi peripheral's clock aligning those clock, or we can set spi clock freely, like 1234.5678khz?

andy-shev, 2020-04-17:
> sck vs miso
> I don't know if it useful, so I just take one photo.

Yeah, we need better one. If I have time , I would check in our lab.

> > Yes, that's what we have, 25, 12.5, 8.333, 6.25, ...
> 
> should we set spi peripheral's clock aligning those clock, or we can set spi clock freely, like 1234.5678khz?

andy-shev, 2020-05-19:
Colleague of mine has been investigating similar issue on another platform with another SPI controller. But he has, for his case, an idea that signal propagation thru level shifters and other discrete components might shift phase enough to get corrupted data. However, ADC7951 is connected directly to the SPI bus without any level discrete components. My colleague also found that even with simple external loopback another platform fails at higher frequencies. So, it means somebody has to look to the electrical specifications of the GPIO / pin control buffers and muxes used inside SoC. It might be phase delay because of them. In any case, I have updated the frequency in the ASL code to be 10MHz and it seems working realiably.

closed 2020-05-19 by andy-shev

htot, 2020-05-19:
That sounds reasonable, 25MHz is not a low frequency especially if SPI lines are loaded by capacitance from the external wiring.

In other projects I found it is easy to get the clock polarity wrong, and then communication still works until you get a capacitive load on the lines. But in those cases reducing clock frequency usually does not help.

xlla, 2020-05-29:
@andy-shev  I am glad to heard that so I can stick on 32bit kernel with confidence.