edison-fw/meta-intel-edison #114: Debian install linux-headers version problem

shawnhsoia97, 2020-06-22:
Hi guys,I want to install linux-headers,because I need to build WiFi adapter driver.
But I encountered some problems,I can't found the corresponding version.
I have tried to run $sudo apt-get install linux-headers-$(uname -r) to get headers of current kernel version
But Kernel version (uname -r) is 5.4.0-edison-acpi-standard,so can't found the corresponding version.
How to solve it?
Thank.

htot, 2020-06-22:
I sounds like I had this question before. Is the driver you need in the kernel tree? If so, just [enable it in the kernel](https://edison-fw.github.io/meta-intel-edison/5.1-Bitbake-tricks#configuring-the-kernel-and-grab-the-kernel-fragment) and let Yocto build it. 

Otherwise, I would expect if you build the SDK, the SDK should be able to build the driver (this I haven't tried).

shawnhsoia97, 2020-06-23:
kernel menuconfig is not support this driver.I need to driver of Realtek rtl8812au,so I need to build it manually by kernel headers source.

I think I should have built the SDK of  kernel headers,but this not build the driver,because not  found the corresponding kernel version.

htot, 2020-06-23:
I have no experience with this, but I find: https://stackoverflow.com/questions/49559438/yocto-how-to-add-out-of-tree-device-driver

And the Yocto Manual points to a template: https://www.yoctoproject.org/docs/current/kernel-dev/kernel-dev.html#incorporating-out-of-tree-modules

As I understand you don't build the module in debian for the host, but within the Yocto environment which has the build environment including kernel headers you need for the target (edison).

Maybe this will help?

shawnhsoia97, 2020-07-02:
Hi @htot ,I tried for several days,but still can't install.
I follow [https://www.yoctoproject.org/docs/current/kernel-dev/kernel-dev.html#incorporating-out-of-tree-modules](url)

This failed when I  following variables(MACHINE_EXTRA_RDEPENDS) in the Yocto Project Reference Manual and include the module in the images,module is not included in the image.

I don't know why,there is not even any error message.it looks like this variables(MACHINE_EXTRA_RDEPENDS) is not executed at all.

I looked for some answers.Maybe this image does not contain this variables?

This variable affects all images based on packagegroup-base,which does not include the core-image-minimal or core-image-full-cmdline images.[https://www.yoctoproject.org/docs/3.1/ref-manual/ref-manual.html#var-MACHINE_EXTRA_RDEPENDS](url)

do you know how include packagegroup-base to edison-image?

I tried IMAGE_INSTALL_append = "        packagegroup-base",but not effective

htot, 2020-07-02:
We have [edison-image-minimal.bb](https://github.com/edison-fw/meta-intel-edison/blob/4ed83663a9b3b63d312a43d98805414384815295/meta-intel-edison-distro/recipes-core/images/edison-image-minimal.bb#L4), maybe just edit that?

Otherwise, maybe one of the Yocto mailing lists can help?

shawnhsoia97, 2020-07-02:
> We have [edison-image-minimal.bb](https://github.com/edison-fw/meta-intel-edison/blob/4ed83663a9b3b63d312a43d98805414384815295/meta-intel-edison-distro/recipes-core/images/edison-image-minimal.bb#L4), maybe just edit that?

You are right!!!! Thank you for help!
out-of-tree modules is installed after Add packagegroup-base .

Also I tried build branch zeus,but perf recipe have problem.

```
 ModuleNotFoundError: No module named '_sysconfigdata'
| Makefile:186: recipe for target '/mnt/yocto/yocto_zeus/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/bpf_helper_defs.h' failed
| make[3]: *** [/mnt/yocto/yocto_zeus/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/bpf_helper_defs.h] Error 1
| make[3]: *** Deleting file '/mnt/yocto/yocto_zeus/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/bpf_helper_defs.h'
| Makefile.perf:778: recipe for target '/mnt/yocto/yocto_zeus/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/libbpf.a' failed
| make[2]: *** [/mnt/yocto/yocto_zeus/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/libbpf.a] Error 2
| make[2]: *** Waiting for unfinished jobs....
| make[3]: Nothing to be done for '/mnt/yocto/yocto_zeus/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/plugins/libtraceevent-dynamic-list'.
| Makefile:653: arch/x86/Makefile: No such file or directory
|   CC       /mnt/yocto/yocto_zeus/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/pmu-events/pmu-events.o
|   LD       /mnt/yocto/yocto_zeus/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/pmu-events/pmu-events-in.o
| make[3]: *** No rule to make target 'arch/x86/Makefile'.  Stop.
| Makefile.perf:224: recipe for target 'sub-make' failed
| make[1]: *** [sub-make] Error 2
| Makefile:69: recipe for target 'all' failed
| make: *** [all] Error 2
| make: Leaving directory '/mnt/yocto/yocto_zeus/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/tools/perf'
| WARNING: exit code 1 from a shell command.
|
ERROR: Task (/mnt/yocto/yocto_zeus/meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/perf/perf.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5849 tasks of which 5792 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /mnt/yocto/yocto_zeus/meta-intel-edison/meta-intel-edison-bsp/recipes-kernel/perf/perf.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
Makefile:53: recipe for target 'edison-image' failed
make: *** [edison-image] Error 1

```

htot, 2020-07-02:
I have zeus on 1c7d015f and you? Last time I updated yocto perf worked.

shawnhsoia97, 2020-07-03:
Okay ,I pull to [1c7d015]( https://github.com/edison-fw/meta-intel-edison/commit/1c7d015f320a09479e015c6ef81bf0a545eae55d),this problem is solved.

Thank you!

shawnhsoia97, 2020-07-07:
@htot  I still encounter same problem ,I thought it could be solved after git pull.
this error log:
```
NOTE: Setscene tasks completed
ERROR: perf-1.0-r9 do_compile: oe_runmake failed
ERROR: perf-1.0-r9 do_compile: Execution of '/mnt/yocto/yocto_zeus_original/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/temp/run.do_compile.14707' failed with exit code 1:
make: Entering directory '/mnt/yocto/yocto_zeus_original/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/tools/perf'
  BUILD:   Doing 'make -j40' parallel build

Auto-detecting system features:
...                         dwarf: [ on  ]
...            dwarf_getlocations: [ on  ]
...                         glibc: [ on  ]
...                          gtk2: [ OFF ]
...                      libaudit: [ OFF ]
...                        libbfd: [ on  ]
...                        libcap: [ OFF ]
...                        libelf: [ on  ]
...                       libnuma: [ OFF ]
...        numa_num_possible_cpus: [ OFF ]
...                       libperl: [ OFF ]
...                     libpython: [ on  ]
...                     libcrypto: [ on  ]
...                     libunwind: [ on  ]
...            libdw-dwarf-unwind: [ on  ]
...                          zlib: [ on  ]
...                          lzma: [ on  ]
...                     get_cpuid: [ on  ]
...                           bpf: [ on  ]
...                        libaio: [ on  ]
...                       libzstd: [ OFF ]
...        disassembler-four-args: [ on  ]

Makefile.config:506: DWARF support is off, BPF prologue is disabled
Makefile.config:714: Missing perl devel files. Disabling perl scripting support, please install perl-ExtUtils-Embed/libperl-dev
Makefile.config:843: No libzstd found, disables trace compression, please install libzstd-dev[el] and/or set LIBZSTD_DIR
Makefile.config:854: No libcap found, disables capability support, please install libcap-devel/libcap-dev
Makefile.config:922: No libbabeltrace found, disables 'perf data' CTF format support, please install libbabeltrace-dev[el]/libbabeltrace-ctf-dev
  DESCEND  plugins
  GEN      /mnt/yocto/yocto_zeus_original/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/bpf_helper_defs.h
Failed to import the site module
Traceback (most recent call last):
  File "/usr/lib/python3.6/site.py", line 570, in <module>
    main()
  File "/usr/lib/python3.6/site.py", line 556, in main
    known_paths = addusersitepackages(known_paths)
  File "/usr/lib/python3.6/site.py", line 288, in addusersitepackages
    user_site = getusersitepackages()
  File "/usr/lib/python3.6/site.py", line 264, in getusersitepackages
    user_base = getuserbase() # this will also set USER_BASE
  File "/usr/lib/python3.6/site.py", line 254, in getuserbase
    USER_BASE = get_config_var('userbase')
  File "/usr/lib/python3.6/sysconfig.py", line 607, in get_config_var
    return get_config_vars().get(name)
  File "/usr/lib/python3.6/sysconfig.py", line 550, in get_config_vars
    _init_posix(_CONFIG_VARS)
  File "/usr/lib/python3.6/sysconfig.py", line 421, in _init_posix
    _temp = __import__(name, globals(), locals(), ['build_time_vars'], 0)
ModuleNotFoundError: No module named '_sysconfigdata'
Makefile:186: recipe for target '/mnt/yocto/yocto_zeus_original/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/bpf_helper_defs.h' failed
make[3]: *** [/mnt/yocto/yocto_zeus_original/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/bpf_helper_defs.h] Error 1
make[3]: *** Deleting file '/mnt/yocto/yocto_zeus_original/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/bpf_helper_defs.h'
Makefile.perf:778: recipe for target '/mnt/yocto/yocto_zeus_original/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/libbpf.a' failed
make[2]: *** [/mnt/yocto/yocto_zeus_original/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/libbpf.a] Error 2
make[2]: *** Waiting for unfinished jobs....
make[3]: Nothing to be done for '/mnt/yocto/yocto_zeus_original/out/linux64/build/tmp/work/edison-poky-linux/perf/1.0-r9/perf-1.0/plugins/libtraceevent-dynamic-list'.
Makefile:653: arch/x86/Makefile: No such file or directory
make[3]: *** No rule to make target 'arch/x86/Makefile'.  Stop.
Makefile.perf:224: recipe for target 'sub-make' failed
make[1]: *** [sub-make] Error 2
[... 201 lines cut ...]
    To make this work a pulseaudio server needs to be started on edison.
    1) create a user
    2) systemctl --user start pulseaudio
    3) sudo setfacl -m u:<user name>:rw /dev/snd/*
    4) paplay "music.wav"

    On the host this will play to the Multifunction Composite Gadget input device.

    Signed-off-by: Ferry Toth <ftoth@exalondelft.nl>

commit 2ff5fd1455a93e3ba257ac5b4d9d7a93f92cfec9
Author: Ferry Toth <ftoth@exalondelft.nl>
Date:   Sat Jun 6 18:19:36 2020 +0200

    gadget: add usb audio

    Signed-off-by: Ferry Toth <ftoth@exalondelft.nl>

commit 6eed5de9997324f4423d805a2d2c8611578aea24
Author: Ferry Toth <ftoth@exalondelft.nl>
Date:   Sat Jun 6 17:56:42 2020 +0200

    DISTRO_FEATURES: add pulseaudio

```

shawnhsoia97, 2020-07-07:
Hi @htot ,
My `/dev/disk/by-partlabel/home` is not work
fstab:
```
/dev/mmcblk1    /          auto    nodev,noatime,discard,barrier=1,data=ordered,noauto_da_alloc    1  1
proc            /proc      proc    defaults              0  0
devpts          /dev/pts   devpts  mode=0620,gid=5       0  0
tmpfs           /run               tmpfs      mode=0755,nodev,nosuid,strictatime 0  0
tmpfs           /var/volatile      tmpfs      defaults              0  0

/dev/disk/by-partlabel/factory  /factory        auto    noauto,x-systemd.automount,nosuid,nodev,noatime,discard     1   1
/dev/disk/by-partlabel/boot     /boot       auto    noauto,x-systemd.automount,nosuid,nodev,noatime,discard     1   1
/dev/disk/by-partlabel/home     /boxxx      auto    noauto,x-systemd.automount,nosuid,nodev,noatime,discard,barrier=1,data=ordered,noauto_da_alloc 1   1

```
error log:
```
root@boxxx-xxx:~# cd /boxxx/
[  282.524058] audit: type=1006 audit(1592209075.662:2): pid=597 uid=0 subj=kernel old-auid=4294967295 auid=0 tty=(none) old-ses=4294967295 ses=1 res=1
[  287.920692] fuseblk: Unknown parameter 'discard'
[  288.016495] fuseblk: Unknown parameter 'discard'
-sh: cd: /bovia/: No such device
```
Have you any idea solve it? Thank

**Also**  **my rootsf is not enough**,I known I can remove the OTA partition and grow the rootfs partition from Linux,but I need to in yocto.

Because Edison IMAGE_SIZE_MB is too big (Edison IMAGE_SIZE_MB = 1531 and EDISON_ROOTFS_MB = 1583),so I need more EDISON_ROOTFS to install more package,as far as I know removing partition 9 can increase rootfs to 2GB.

Have you any idea for it?

htot, 2020-07-07:
1) after pulling I think you may need to do `make clean`
2) I don't know what you are doing, you are mounting home on /boxxx, using fuse? What format is is you home?
3) remove paramers that are not understood by fuse
4) I am working on btrfs as rootfs. There is a branch called btrfs that is WIP. One of the first things I do is make more space. Some docs here: https://github.com/htot/meta-intel-edison/tree/btrfs/docs/_docs/Edison/Moving%20to%20btrfs

shawnhsoia97, 2020-07-10:
1.OKay,I will try again
2.Because if it is mounted on the original /home,it will cover the /home of the rootfs,so I mount to /boxxx.Yes,I have install fuse,but i didn't make too many changes to fstab but only modified mount point.
3.OKay,I remove "discard,barrier=1,data=ordered,noauto_da_alloc" parameters that are not understood  by fuse,but still unable to enter the /boxxx.
```
root@boxxx-xxx:~# cd /boxxx/
-sh: cd: /boxxx/: No such device
```
4.okay ,I will try this solutions

Also I heard that yocto can install recipes to different partitions so that the available rootfs partition becomes larger.For example,install vim tools to the /dev/disk/by-partlabel/home.
Have you heard this way?Do you know how to do this?

htot, 2020-07-10:
2. There is no reason to mount ext4 with fuse, something is wrong. Maybe try to mount manually then when that succeeds modify fstab (-t ext4 or so?, replace auto in fstab by ext4?)

shawnhsoia97, 2020-07-13:
I  try to mount manually `mount -t  ext4 /dev/disk/by-partlabel/home /boxxx`
I get error `[   50.241506] EXT4-fs (mmcblk0p10): VFS: Can't find ext4 filesystem`

So I found this [https://serverfault.com/a/868014](https://serverfault.com/a/868014)

then solved 

```
root@boxxx-xxx:~# mkfs.ext4 /dev/disk/by-partlabel/home
mke2fs 1.44.5 (15-Dec-2018)
64-bit filesystem support is not enabled.  The larger fields afforded by this feature enable full-strength checksumming.  Pass -O 64bit to rectify.
Discarding device blocks: done
Creating filesystem with 336123 4k blocks and 84128 inodes
Filesystem UUID: 1bc1d55c-eb8b-4b73-91e4-cfcc08f44c6b
Superblock backups stored on blocks:
        32768, 98304, 163840, 229376, 294912

Allocating group tables: done
Writing inode tables: done
Creating journal (8192 blocks): done
Writing superblocks and filesystem accounting information: done

root@boxxx-xxx:~# mount -t ext4 /dev/disk/by-partlabel/home /boxxx
root@boxxx-xxx:~# cd /boxxx/
root@boxxx-xxx:/bovia# ls
lost+found
```
But I don't know why ? I remember it was normal before.
Maybe you know? and how to solve this in yocto? Thank !!!

shawnhsoia97, 2020-07-14:
1. I clone branch zeus again and rebuild, But still same error.(perf error)
4. Must I switch to btrfs only in the branch "brtfs" ?

htot, 2020-07-14:
I'm building zeus and btrfs without problems. So it must be that I have something cached that breaks on a fresh download. I'll try on a fresh clone.

shawnhsoia97, 2020-07-14:
Okay , Thank you help

also if I want to use btrsf,do I need to build that in branch zeus or branch btrfs?

htot, 2020-07-14:
I just checked out 1c7d015f (zeus), did `make clean` and further `rm -rf bbcache`. Then built `make image` without problems, except a few warnings:
```
WARNING: expat-native-2.2.8-r0 do_fetch: Failed to fetch URL https://downloads.sourceforge.net/expat/expat-2.2.8.tar.bz2, attempting MIRRORS if available
WARNING: gstreamer1.0-plugins-base-1.16.2-r0 do_fetch: Failed to fetch URL https://gstreamer.freedesktop.org/src/gst-plugins-base/gst-plugins-base-1.16.2.tar.xz, attempting MIRRORS if available
WARNING: gstreamer1.0-plugins-good-1.16.2-r0 do_fetch: Failed to fetch URL https://gstreamer.freedesktop.org/src/gst-plugins-good/gst-plugins-good-1.16.2.tar.xz, attempting MIRRORS if available
WARNING: sof-fw-1.5-r0 do_populate_lic: sof-fw: No generic license file exists for: BSD3 in any provider
WARNING: edison-image-1.0-r0 do_rootfs: The license listed BSD3 was not in the licenses collected for recipe sof-fw
NOTE: Tasks Summary: Attempted 4395 tasks of which 2 didn't need to be rerun and all succeeded.
```
The `btrfs` builds on top of zeus, and adds btrfs to U-Boot and kernel as well as btrfs-tools, so yes.

shawnhsoia97, 2020-07-15:
For me this problem still exists.
But I found this [Solution](http://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/?id=3a1a18ba9d28adb5562eabe9ec354f6d93154f5c),so I solved that error.
Maybe this is problem caused by Python3 ?

shawnhsoia97, 2020-07-15:
This cause of the problem is caused by python3.6
I upgrade to python3.8,that is solved.

htot, 2020-07-15:
Hmm. In that case, Yocto should be building it's own version of python, maybe you could report a bug?

shawnhsoia97, 2020-07-15:
Okay , So should I report that case to "pull requests" or "issues"?

Also I build branch btrfs now,but I found btrfsSnapshot.py need sudo,so maybe add sudo in Makefile?

Also run btrfsSnapshot.py have error 
```
Mount failed: ""
Already mounted?
```
I found "udisksctl loop-setup" not mount to Mountpoint.
```
shawn@boxxx-xxx:/mnt/yocto/yocto_btrfs$ sudo ./meta-intel-edison/utils/flash/btrfsSnapshot.py /mnt/yocto/yocto_btrfs/out/current/build
Deploy dir /mnt/yocto/yocto_btrfs/out/current/build/tmp/deploy/images/edison/
CompletedProcess(args='/usr/bin/udisksctl loop-setup --no-user-interaction -f /mnt/yocto/yocto_btrfs/out/current/build/tmp/deploy/images/edison/edison-image-edison.btrfs', returncode=0, stdout='Mapped file /mnt/yocto/yocto_btrfs/out/current/build/tmp/deploy/images/edison/edison-image-edison.btrfs as /dev/loop0.\n', stderr='')
CompletedProcess(args='lsblk -o MOUNTPOINT -n /dev/loop0', returncode=0, stdout='\n', stderr='')
Mount failed: ""
Already mounted?
shawn@boxxx-xxx:/mnt/yocto/yocto_btrfs$ lsblk
NAME   MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
loop0    7:0    0  1.9G  0 loop
sr0     11:0    1 1024M  0 rom
vda    252:0    0  500G  0 disk
└─vda1 252:1    0  500G  0 part /mnt/yocto
vdb    252:16   0   50G  0 disk
├─vdb1 252:17   0    1M  0 part
└─vdb2 252:18   0   50G  0 part /
```
So I think need to add  below after `udisksctl loop-setup`
 ```
cmd_mkimg = '/usr/bin/udisksctl mount -b ' + loop
    ret = subprocess.run(cmd_mkimg, shell=True, capture_output=True, text=True)
```
this error is soved after use that solution,but I'm not sure if that is correct,what do you think?

htot, 2020-07-15:
Bugs in Poky should go to https://bugzilla.yoctoproject.org/

btrfsSnapshot.py processes the btrfs image (post build) and creates a snapshot. You can later send multiple snapshots to edison, but is non-essential for now. 

It is a work in progress, I don't know why it failed for you. May because failed first time, is already mounted 2nd time and fails again.

shawnhsoia97, 2020-07-16:
Ok ,I reported.

I switch to btrfs now. but I encounter some problem.
resizepart 8 have error :
```
(parted) resizepart 8
Warning: Partition /dev/mmcblk0p8 is being used. Are you sure you want to
continue?
Yes/No? yes
End?  [3342335s]? 2
Error: Can't have the end before the start! (start sector=196608 length=-196605)

```
print:
```
Number  Start     End       Size      File system  Name         Flags
 1      2048s     6143s     4096s                  u-boot0      msftdata
 2      6144s     8191s     2048s                  u-boot-env0  msftdata
 3      8192s     12287s    4096s                  u-boot1      msftdata
 4      12288s    14335s    2048s                  u-boot-env1  msftdata
 5      14336s    16383s    2048s     ext2         factory      msftdata
 6      16384s    65535s    49152s                 panic        msftdata
 7      65536s    196607s   131072s   fat16        boot         msftdata
 8      196608s   3342335s  3145728s  ext4         rootfs       msftdata
10      4849664s  7634910s  2785247s  btrfs        home         msftdata

````
I think maybe is caused by  mmcblk0p8 type ? 
run `mkfs.btrfs -f /dev/mmcblk0p8` can solved that.but I don't know if that is correct.what do you think?
also `/dev/mmcblk0p8` is busy ,so I umount that .
log:

```
/ # mount -t btrfs /dev/mmcblk0p8 /media/rootfs
mount: mounting /dev/mmcblk0p8 on /media/rootfs failed: Device or resource busy
/ # umount /dev/mmcblk0p8
/ # mount -t btrfs /dev/mmcblk0p8 /media/rootfs
mount: mounting /dev/mmcblk0p8 on /media/rootfs failed: Invalid argument
/ # mount btrfs -t btrfs /dev/mmncblk0p8 /media/rootfs
BusyBox v1.31.0 (2020-07-15 03:20:09 UTC) multi-call binary.

Usage: mount [OPTIONS] [-o OPT] DEVICE NODE
/ # mkfs.btrfs -f /dev/mmcblk0p8
btrfs-progs v5.2.2
See http://btrfs.wiki.kernel.org for more information.

Detected a SSD, turning off metadata duplication.  Mkfs with -m dup if you want to force metadata duplication.
Label:              (null)
UUID:               530f00dd-35e9-4338-a165-3ee0cb03bfe2
Node size:          16384
Sector size:        4096
Filesystem size[ 2683.098350] BTRFS: device fsid 530f00dd-35e9-4338-a165-3ee0cb03bfe2 devid 1 transid 5 /dev/mmcblk0p8 scanned by mkfs.btrfs (494)
:    1.50GiB
Block group profiles:
  Data:             single            8.00MiB
  Metadata:         single            8.00MiB
  System:           single            4.00MiB
SSD detected:       yes
Incompat features:  extref, skinny-metadata
Number of devices:  1
Devices:
   ID        SIZE  PATH
    1     1.50GiB  /dev/mmcblk0p8

/ # parted
GNU Parted 3.2
Using /dev/mmcblk0
Welcome to GNU Parted! Type 'help' to view a list of commands.
(parted) resizepart 8
End?  [1711MB]? 2
Error: Can't have the end before the start! (start sector=196608 length=-192701)

(parted) resizepart 1
End?  [3145kB]? 1
Error: Can't have the end before the start! (start sector=2048 length=-94)
(parted) print
Model: MMC H4G1d (sd/mmc)
Disk /dev/mmcblk0: 3909MB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags:

Number  Start   End     Size    File system  Name         Flags
 1      1049kB  3146kB  2097kB               u-boot0      msftdata
 2      3146kB  4194kB  1049kB               u-boot-env0  msftdata
 3      4194kB  6291kB  2097kB               u-boot1      msftdata
 4      6291kB  7340kB  1049kB               u-boot-env1  msftdata
 5      7340kB  8389kB  1049kB  ext2         factory      msftdata
 6      8389kB  33.6MB  25.2MB               panic        msftdata
 7      33.6MB  101MB   67.1MB  fat16        boot         msftdata
 8      101MB   1711MB  1611MB  btrfs        rootfs       msftdata
10      2483MB  3909MB  1426MB  btrfs        home         msftdata

(parted) resizepart 8 2482
(parted) print
Model: MMC H4G1d (sd/mmc)
Disk /dev/mmcblk0: 3909MB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags:

Number  Start   End     Size    File system  Name         Flags
 1      1049kB  3146kB  2097kB               u-boot0      msftdata
 2      3146kB  4194kB  1049kB               u-boot-env0  msftdata
 3      4194kB  6291kB  2097kB               u-boot1      msftdata
 4      6291kB  7340kB  1049kB               u-boot-env1  msftdata
 5      7340kB  8389kB  1049kB  ext2         factory      msftdata
 6      8389kB  33.6MB  25.2MB               panic        msftdata
 7      33.6MB  101MB   67.1MB  fat16        boot         msftdata
 8      101MB   2482MB  2381MB  btrfs        rootfs       msftdata
10      2483MB  3909MB  1426MB  btrfs        home         msftdata

```

btrfs scan `argument not found,maybe should is `btrfs device scan`?
`mount btrfs -t btrfs /dev/mmncblk0p8 /media/rootfs ` should is mount -t btrfs /dev/**mmcblk0p8** /media/rootfs `?

htot, 2020-07-16:
I don't understand what you are trying to do. Which instructions are you following?

shawnhsoia97, 2020-07-17:
I follow [6.1-How-to-switch-to-btrfs](https://github.com/htot/meta-intel-edison/blob/btrfs/docs/_docs/Edison/Moving%20to%20btrfs/6.1-How-to-switch-to-btrfs.md) now.

I want to use btrifs.
Should I frist modify edison.env according to [6.1-How-to-switch-to-btrfs](https://github.com/htot/meta-intel-edison/blob/btrfs/docs/_docs/Edison/Moving%20to%20btrfs/6.1-How-to-switch-to-btrfs.md)  ?

shawnhsoia97, 2020-07-17:
Also I following **Boot into rescue image**,Interrupt U-Boot and type `run do_rescue`.
I got this error:
```
Hit any key to stop autoboot:  0
=> run do_rescue
9767712 bytes read in 425 ms (21.9 MiB/s)
9548339 bytes read in 416 ms (21.9 MiB/s)
Valid Boot Flag
Setup Size = 0x00003800
Magic signature found
Using boot protocol version 2.0f
Linux kernel version 5.6.0-edison-acpi-standard (oe-user@oe-host) #1 SMP Fri Jul 17 03:09:57 UTC 2020
Building boot_params at 0x00090000
Loading bzImage at address 100000 (9753376 bytes)
Magic signature found
Initial RAM disk at linear address 0x06000000, size 25165824 bytes
Kernel command line: "quiet debugshell=0 tty1 console=ttyS2,115200n8 root=/dev/mmcblk0p7 rootfstype=vfat systemd.unit=multi-user.target hardware_id=00"
Magic signature found

Starting kernel ...

[    2.397624] Initramfs unpacking failed: invalid magic at start of compressed archive
Scanning for Btrfs filesystems
Starting version 243.2+
Kernel with acpi enabled detected
Loading acpi tables
[    3.471826] pca953x i2c-INT3491:00: failed writing register
[    3.487856] pca953x i2c-INT3491:01: failed writing register
[    3.498878] pca953x i2c-INT3491:02: failed writing register
[    3.515174] pca953x i2c-INT3491:03: failed writing register
gpioset: at least one GPIO line offset to value mapping must be specified
gpioset: at least one GPIO line offset to value mapping must be specified
Waiting for root device /dev/mmcblk0p7
  0...
Mounted filesystems
Available block devices
major minor  #blocks  name

Cannot find /dev/mmcblk0p7 or no /sbin/init present , dropping to a shell

sh: can't access tty; job control turned off

```
I guess maybe caused by variables in U-Boot ?
```
boot_rescue=zboot ${loadaddr} 0x1800000 0x6000000 0x1800000
bootargs_rescue=debugshell=0 tty1 console=ttyS2,115200n8 root=/dev/mmcblk0p7 rootfstype=vfat systemd.unit=multi-user.target hardware_id=00
do_rescue=setenv bootargs ${bootargs_common} ${bootargs_rescue}; run load_rescue; run boot_rescue
load_rescue=load mmc 0:7 ${loadaddr} bzImage;load mmc 0:7 0x6000000 initrd
```
Because root should in the /dev/mmcblk0p8?

htot, 2020-07-17:
> Should I frist modify edison.env according to 6.1-How-to-switch-to-btrfs ?

Yes. And the rescue image is then on p7.

htot, 2020-07-18:
But you are right, I haven't provided U-Boot headers yet.  Sorry, I  need to clean op the mess first. In the meanwhile, this file
[btrfs.env.txt](https://github.com/edison-fw/meta-intel-edison/files/4942422/btrfs.env.txt) shows my env as I have now.

Note, I just force pushed my latest 0b63173f

htot, 2020-07-19:
+ 1 untested patch that might help you in the right direction 934bdd73

htot, 2020-07-20:
Now it's starting to work, I added another patch with fixes 56364588

First I would start to flash normally using flashall.sh or FlashToolLite, this will flash the ext4 image as always, but also allow you to do `run do_rescue` from U-Boot if needed so you can repair or investigate. This will break your `update` partition, your data there will be lost. If you want to use it you need to reformat it.

If it works normally then you could go and try (see details in the 6.1-How-to-switch-to-btrfs.md) from the `out/current/build/tmp/deploy/images/edison/u-boot-envs/`:
```
    sudo dfu-util -v -d 8087:0a99 --alt u-boot-env0 -D edison-btrfs.bin
    sudo dfu-util -v -d 8087:0a99 --alt u-boot-env1 -D edison-btrfs.bin -R
```
It should be rebooting and repartitioning now. If not force it with `run do_force_partition`.
This will erase your `update` and `home` partition.
Then (from `out/current/build/tmp/deploy/images/edison/`):
```
sudo dfu-util -v -d 8087:0a99 --alt rootfs -D edison-image-edison.btrfs
```
In the now very large partition p8 (called home) the btrfs image will be installed.

The documentation will be needing fixes, so if you find problems and care to send a patch, you are welcome.

Results:
```
root@edison:~# mount | grep btrfs
/dev/mmcblk0p8 on / type btrfs (rw,relatime,compress=lzo,ssd,space_cache,subvolid=256,subvol=/@)
/dev/mmcblk0p8 on /boot type btrfs (rw,noatime,compress=lzo,ssd,space_cache,subvolid=258,subvol=/@boot)
/dev/mmcblk0p8 on /lib/modules type btrfs (rw,noatime,compress=lzo,ssd,space_cache,subvolid=259,subvol=/@modules)
/dev/mmcblk0p8 on /home type btrfs (rw,noatime,compress=lzo,ssd,space_cache,subvolid=257,subvol=/@home,x-systemd.automount)

root@edison:~# df | grep mmc
/dev/mmcblk0p8   3719148 708520   2898600  20% /
/dev/mmcblk0p8   3719148 708520   2898600  20% /boot
/dev/mmcblk0p8   3719148 708520   2898600  20% /lib/modules
/dev/mmcblk0p8   3719148 708520   2898600  20% /home
/dev/mmcblk0p5      1003     21       911   3% /factory
```

shawnhsoia97, 2020-08-07:
Hi
Sorry i only have time try now.
I succeed rearrange partition (remove p9)
but failure to install btrfs image
run `sudo dfu-util -v -d 8087:0a99 --alt u-boot-env0 -D edison-btrfs.bin` show:
```
-alt u-boot-env0 -D edison-btrfs.bin
[sudo] password for bovia: 
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available

```

shawnhsoia97, 2020-08-07:
and I run` run do_force_flash_os`
show:
```
=> run do_force_flash_os
Saving Environment to MMC... Writing to MMC(0)... OK
Couldn't find part #9 on mmc device #0
DFU entities configuration failed!
(partition table does not match dfu_alt_info?)
dfu - Device Firmware Upgrade

Usage:
dfu <USB_controller> [<interface> <dev>] [<timeout>] [list]
  - device firmware upgrade via <USB_controller>
    on device <dev>, attached to interface
    <interface>
    [<timeout>] - specify inactivity timeout in seconds
    [list] - list available alt settings

```

htot, 2020-08-08:
> ```
> dfu-util: No DFU capable USB device available
> ```
That means flashing by DFU didn't work. Probably because `run do_force_flash_os` is not starting. And that happens because your partitions are not matching `dfu_alt_info`.

Could you try to `run do_force_partition` to see if that repartitions everything and then try again?

shawnhsoia97, 2020-08-10:
Okay salve it after run `do_force_partition`,but I got error by run `sudo dfu-util -v -d 8087:0a99 --alt rootfs -D edison-image-edison.btrfs`.

```
=> run do_force_flash_os
Saving Environment to MMC... Writing to MMC(0)... OK
################################################################################################################################################################################################Request would exceed designated area!
```

```
-D edison-image-edison.btrfs
[sudo] password for bovia: 
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 8087:0a99
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #7 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 4096
Copying data from PC to DFU device
Download	[===================      ]  78%   1615241216 bytes failed!
state(10) = dfuERROR, status(14) = Something went wrong, but the device does not know what it was

```
it's weird,my partition should increase to 2.1 GB,I don;t know why heppen that.

htot, 2020-08-10:
Can you do in U-Boot:
```
printenv do_dfu_alt_info_mmc
printenv partitions
```
It should be as [this](https://github.com/htot/meta-intel-edison/blob/56364588871eba9606c7422104c227db2ad0a68f/meta-intel-edison-bsp/recipes-bsp/u-boot/files/target_env/btrfs.env#L12), if not we need to fix `edison-btrfs.bin` first.

As you can see there is no `rootfs` partition anymore, it is now called `home`. So the dfu command should now be changed to:
```
sudo dfu-util -v -d 8087:0a99 --alt home -D edison-image-edison.btrfs
```

shawnhsoia97, 2020-08-11:
No, It's not.
```
=> printenv do_dfu_alt_info_mmc
do_dfu_alt_info_mmc=setenv dfu_alt_info "ifwi${hardware_id} raw 0 8192 mmcpart 1;ifwib${hardware_id} raw 0 8192 mmcpart 2;u-boot0 part 0 1;u-boot-env0 part 0 2;u-boot1 part 0 3;u-boot-env1 part 0 4;boot part 0 7;rootfs part 0 8;update part 0 9;home part 0 10;bzImage fat 0 7;initrd fat 0 7"
=> printenv partitions
partitions=uuid_disk=${uuid_disk};name=u-boot0,start=1MiB,size=2MiB,uuid=${uuid_uboot0};name=u-boot-env0,size=1MiB,uuid=${uuid_uboot_env0};name=u-boot1,size=2MiB,uuid=${uuid_uboot1};name=u-boot-env1,size=1MiB,uuid=${uuid_uboot_env1};name=factory,size=1MiB,uuid=${uuid_factory};name=panic,size=24MiB,uuid=${uuid_panic};name=boot,size=64MiB,uuid=${uuid_boot};name=rootfs,size=1536MiB,uuid=${uuid_rootfs};name=update,size=736MiB,uuid=${uuid_update};name=home,size=-,uuid=${uuid_home};

```
How to fix edison-btrfs.bin? because my btrfs.env same as [this](https://github.com/htot/meta-intel-edison/blob/56364588871eba9606c7422104c227db2ad0a68f/meta-intel-edison-bsp/recipes-bsp/u-boot/files/target_env/btrfs.env#L12)

htot, 2020-08-12:
`edison-btrfs.bin` is generated from `edison-btrfs.env`.

You can flash it with:
```
sudo dfu-util -v -d 8087:0a99 --alt u-boot-env0 -D edison-btrfs.bin
sudo dfu-util -v -d 8087:0a99 --alt u-boot-env1 -D edison-btrfs.bin -R
```
Or you can manually edit of course in U-Boot command line (if you want to make changes permanent use `saveenv`).

shawnhsoia97, 2020-08-14:
Thank you for help
I follow below step:
1. Build and `flashall`
2. run `/sbin/post-install.sh script,`
3. Rearrange partitions (delete update partition) and `/lib/systemd/systemd-growfs /`
4. `run do_force_partition`
5. `run do_force_flash_os `and 
```
    sudo dfu-util -v -d 8087:0a99 --alt u-boot-env0 -D edison-btrfs.bin
    sudo dfu-util -v -d 8087:0a99 --alt u-boot-env1 -D edison-btrfs.bin -R
```

6.   `run do_force_flash_os ` 
      and  `sudo dfu-util -v -d 8087:0a99 --alt home -D edison-image-edison.btrfs`

Now have the large rootfs

But also I have the some question.because did modify `edison-btrfs.env` , and run step 5 ~ 6,so do I need to perform step 3?
Question 2,I modify `edison-image.bb` with add tools needed (ex: minicom),then perform step 5 ~ 6,but **I found that the tools is not installed**.Do I need to follow step1 ~ 6?So when I install ext4 Zeus image,the space still be insufficient,right?

htot, 2020-08-14:
When you log into edison you can see the size of your partitions with `partx /dev/mmcblk0`. You can check the usage of the filesystem with `btrfs fi us /`. If your filesystem does not use the full partition you can grow it, otherwise not needed.

I don't know why your tools are not installed. But you can always `bitbake your_tool` and then install it, see [here](https://htot.github.io/meta-intel-edison/5.0-Creating-a-deb-repository.html).

htot, 2020-08-14:
BTW of your above steps, I was hoping the correct steps would be 1, 5, 6. That's what I did here (if I remember correctly)

If  so, we could add an option `--btrfs` to `flashall.sh` to do steps 5 and 6.

But I am too slow right now. My computer is in our attic and currently temperature there is +38C.

shawnhsoia97, 2020-08-17:
No,I not use bitbake your_tool and install it,because I need to install tools into IMAGE on Yocto.

I think tools of `edison-image.bb`  may not be install to IMAGE?because `IMAGE_SIZE_MB` incorrect.
```
./meta-intel-edison/utils/flash/postBuild.sh /mnt/yocto/yocto_btrfs/out/current/build
EDISON_ROOTFS_MB = , IMAGE_SIZE_MB = 1257
```
`IMAGE_SIZE_MB` should is 15XX, because branch warrior is that size.

is the `image_appand tools` install to either `edison-image-edison.ext4` or `edison-image-edison.btrfs`?

htot, 2020-08-17:
No, that is not true. There is more than sufficient space. And the contents of ext4 and btrfs are the same.

Maybe you have a typo? It should be:
```
IMAGE_INSTALL_append = " your_tool"
```

shawnhsoia97, 2020-08-18:
Sorry,this is my bad,what I mean is that the size of tools needed to installed on `branch warrior` is 15xxMB, but  the size of tools needed to installed on `branch warrior` is 1257MB.
Should the size be the same normally?

> Maybe you have a typo? It should be:
> 
> ```
> IMAGE_INSTALL_append = " your_tool"
> ```
Also that is my bad again.I am using `IMAGE_INSTALL_append`,but I made a typo when reply.

htot, 2020-08-18:
I don't understand, what tools? Normally the size of the rootfs is 700MB, as you can see from `df` above, used is 690060kB. Do you mean you want to install additional tools with size of 800MB?

shawnhsoia97, 2020-08-18:
Yes,my install tools with size of 800MB.
my install tools:

```
IMAGE_INSTALL_append = " v4l-utils"
IMAGE_INSTALL_append = " gnupg"
IMAGE_INSTALL_append = " minicom"
IMAGE_INSTALL_append = " hostapd"
IMAGE_INSTALL_append = " dnsmasq"
IMAGE_INSTALL_append = " usb-modeswitch"
IMAGE_INSTALL_append = " usb-modeswitch-data"
IMAGE_INSTALL_append = " wvdial"
IMAGE_INSTALL_append = " ppp"
IMAGE_INSTALL_append = " ppp-dialin"
IMAGE_INSTALL_append = " libqmi"
IMAGE_INSTALL_append = " cmake"
IMAGE_INSTALL_append = " opkg"
IMAGE_INSTALL_append = " util-linux"
IMAGE_INSTALL_append = " gawk"
IMAGE_INSTALL_append = " openssh"
IMAGE_INSTALL_append = " libnetfilter-conntrack"
IMAGE_INSTALL_append = " nginx"
IMAGE_INSTALL_append = " exfat-utils"
IMAGE_INSTALL_append = " fuse-exfat"
IMAGE_INSTALL_append = " gunicorn"
IMAGE_INSTALL_append = " python3-pip"
IMAGE_INSTALL_append = " python3-dbus"
IMAGE_INSTALL_append = " python3-flask"
IMAGE_INSTALL_append = " python3-pika"
IMAGE_INSTALL_append = " python3-flask-login"
IMAGE_INSTALL_append = " python3-flask-sqlalchemy"
IMAGE_INSTALL_append = " python3-flask-restful"
IMAGE_INSTALL_append = " python3-gevent"
IMAGE_INSTALL_append = " python3-psutil"
IMAGE_INSTALL_append = " rtc-3028"
IMAGE_INSTALL_append = " language"
IMAGE_INSTALL_append = " openvpn"
IMAGE_INSTALL_append = " openconnect"
IMAGE_INSTALL_append = " zlib"
IMAGE_INSTALL_append = " softether"
IMAGE_INSTALL_append = " libtext-iconv-perl"
IMAGE_INSTALL_append = " jq"
IMAGE_INSTALL_append = " networkmanager"
IMAGE_INSTALL_append = " alsa-lib"
IMAGE_INSTALL_append = " alsa-utils"
IMAGE_INSTALL_append = " pulseaudio"
IMAGE_INSTALL_append = " tzdata"
IMAGE_INSTALL_append = " gdb"
IMAGE_INSTALL_append = " vim"
IMAGE_INSTALL_append = " speedtest-cli"
IMAGE_INSTALL_append = " quectel"
IMAGE_INSTALL_append = " pciutils"

```

htot, 2020-08-18:
I would recommend not install hostapd, dnsmasq, networkmanager as conman already provides access point. It can also do vpn but I haven't tried. And opkg is not needed when you are using dpkg.
Also, I thinkg alsa, pulseaudio, txdata are already installed.

Which of these packages is so large?

shawnhsoia97, 2020-08-18:
Because my partner like use  hostapd, dnsmasq, networkmanager ,so I have to use these.

the large packages : vim cmake  networkmanager

But the question now,I can't install package to image,I found that even ext4 Zeus image  has not package(cmake,hostapd...).
Strange,rtner like use  hostapd, dnsmasq, networkmanager ,so I have to use these.

the large packages : vim cmake  networkmanager

But the question now,I can't install package to image,I found that even ext4 Zeus image  has not package(cmake,hostapd...).
Strange,this should installed
```
 bitbake -s | grep host
ghostscript                                          :9.27-r0
ghostscript-native                                   :9.27-r0
hostapd                                               :2.9-r0
nativesdk-packagegroup-sdk-host                      :1.0-r12
```
 ```
yocto_btrfs$ cat out/linux64/build/buildhistory/images/edison/glibc/edison-image/installed-packages.txt | grep hos
hostapd_2.9-r0_amd64.deb
libnss-myhostname2_1:243.2-r0_amd64.deb
perl-module-net-hostent_5.30.1-r0_amd64.deb
perl-module-sys-hostname_5.30.1-r0_amd64.deb
yocto_btrfs$ cat out/linux64/build/buildhistory/images/edison/glibc/edison-image/installed-packages.txt | grep cmake
cmake_3.15.3-r0_amd64.deb

```

shawnhsoia97, 2020-08-18:
No matter how the package is installed,the size so the image will nor change.

The size of installing and not installing above package the image size is the same.

htot, 2020-08-18:
Installed packages should be listed in `tmp/deploy/images/edison/edison-image-edison.manifest`

shawnhsoia97, 2020-08-19:
Looks to have installed.
```
yocto_btrfs$ cat out/linux64/build/tmp/deploy/images/edison/edison-image-edison.manifest | grep host
hostapd corei7-64 2.9-r0
libnss-myhostname2 corei7-64 1:243.2-r0
perl-module-net-hostent corei7-64 5.30.1-r0
perl-module-sys-hostname corei7-64 5.30.1-r0
```

But I am sure it is not install to edison-image-edison.ext4

```
Sstate summary: Wanted 25 Found 7 Missed 18 Current 2230 (28% match, 99% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
WARNING: edison-image-1.0-r0 do_rootfs: The license listed BSD3-Clause was not in the licenses collected for recipe sof-fw
NOTE: Tasks Summary: Attempted 5950 tasks of which 5885 didn't need to be rerun and all succeeded.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 2 seconds

Summary: There was 1 WARNING message shown.
./meta-intel-edison/utils/flash/postBuild.sh /mnt/yocto/yocto_btrfs/out/current/build
EDISON_ROOTFS_MB = , IMAGE_SIZE_MB = 1257
./meta-intel-edison/utils/flash/postBuild.sh: line 35: [: -lt: unary operator expected
1+0 records in
1+0 records out
4194304 bytes (4.2 MB, 4.0 MiB) copied, 0.0138364 s, 303 MB/s
1+0 records in
1+0 records out
4194304 bytes (4.2 MB, 4.0 MiB) copied, 0.0116446 s, 360 MB/s
1+0 records in
1+0 records out
4194304 bytes (4.2 MB, 4.0 MiB) copied, 0.0109405 s, 383 MB/s
1+0 records in
1+0 records out
4194304 bytes (4.2 MB, 4.0 MiB) copied, 0.0116314 s, 361 MB/s
1+0 records in
1+0 records out
4194304 bytes (4.2 MB, 4.0 MiB) copied, 0.0127391 s, 329 MB/s
1+0 records in
1+0 records out
4194304 bytes (4.2 MB, 4.0 MiB) copied, 0.0138895 s, 302 MB/s
1+0 records in
1+0 records out
4194304 bytes (4.2 MB, 4.0 MiB) copied, 0.0142471 s, 294 MB/s
Image Name:   Edison Updater script
Created:      Wed Aug 19 02:26:47 2020
Image Type:   PowerPC Linux Script (uncompressed)
Data Size:    22200 Bytes = 21.68 KiB = 0.02 MiB
Load Address: 00010000
Entry Point:  00010000
Contents:
   Image 0: 22192 Bytes = 21.67 KiB = 0.02 MiB
**** Done ***
Files ready to flash in /mnt/yocto/yocto_btrfs/out/current/build/toFlash/
Run the flashall script there to start flashing.

```

No matter how I  `IMAGE_INSTALL_append` the package,the` EDISON_ROOTFS_MB = , IMAGE_SIZE_MB = 1257 ` above will not change.

htot, 2020-08-19:
You can look under build/tmp...hostapd, in the tmp directory are a order file with the build step and the build log files of each step. Also the directory structure that is used to build the packages.

Finally there is build/tmp.../edison-image, here the packages are installed, normally with dpkg, check this log here too. From this the image is built. hostapd should be there no matter the size of the image.

shawnhsoia97, 2020-08-20:
I checked build/tmp.../edison-image,the package is here.
```
yocto_btrfs/out/linux64/build/tmp/work/edison-poky-linux/edison-image/1.0-r0$ cat image_initial_manifest | grep hostapd
mip,hostapd
```

but my edison there is not the package after flash image .
I tied build and flash branch zeus ,that is normally.

```
WARNING: edison-image-1.0-r0 do_rootfs: The license listed BSD3 was not in the licenses collected for recipe sof-fw
NOTE: Tasks Summary: Attempted 5702 tasks of which 5394 didn't need to be rerun and all succeeded.

Summary: There was 1 WARNING message shown.
./meta-intel-edison/utils/flash/postBuild.sh /mnt/yocto/yocto_zeus_original/out/current/build
EDISON_ROOTFS_MB = 1538, IMAGE_SIZE_MB = 1535
1+0 records in
1+0 records out

```
Strange,branch zeus and btrfs should be the same,right?
Can you try to` IMAGE_INSTALL_append` some package on branch btrfs?

htot, 2020-08-21:
In one case you have:
```
EDISON_ROOTFS_MB = , IMAGE_SIZE_MB = 1257
```

in the other:
```
EDISON_ROOTFS_MB = 1538, IMAGE_SIZE_MB = 1535
```
That is strange, maybe you are right and you need to increase the size of the image. I can only think of [here](https://github.com/htot/meta-intel-edison/blob/4ed83663a9b3b63d312a43d98805414384815295/meta-intel-edison-distro/recipes-core/images/edison-image-minimal.bb#L32).

shawnhsoia97, 2020-08-24:
Sorry, I how so foolish,I hope that you can excuse my foolish behavior.
No,I was not understand what your mean.I apologize.I understand what your mean now.

Now I change IMAGE_ROOTFS_SIZE to 10048576.(will I set too large?)
And then I get:
```
WARNING: edison-image-1.0-r0 do_rootfs: The license listed BSD3-Clause was not in the licenses collected for recipe sof-fw
NOTE: Tasks Summary: Attempted 5765 tasks of which 5701 didn't need to be rerun and all succeeded.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 2 seconds

Summary: There was 1 WARNING message shown.
./meta-intel-edison/utils/flash/postBuild.sh /mnt/yocto/yocto_btrfs/out/current/build
EDISON_ROOTFS_MB = , IMAGE_SIZE_MB = 9813
./meta-intel-edison/utils/flash/postBuild.sh: line 35: [: -lt: unary operator expected
1+0 records in
```
But I found keep increasing  IMAGE_ROOTFS_SIZE and also IMAGE_SIZE_MB will keep increasing,that without limit.

And also I mount edison-image-edison.ext4 and look for my need packages,but not found.

```
/yocto_btrfs$ sudo mount -t auto out/linux64/build/tmp/deploy/images/edison/edison-image-edison.ext4 /tmp/btrfs_ex4/
/yocto_btrfs$ cd /tmp/btrfs_ex4/
/tmp/btrfs_ex4$ ls etc/
alsa                    busybox.links.suid    dhcp          fstab.btrfs    intel_mcu    localtime       mtab            os-release  pulse                       security       slsh.rc       syslog.conf  usb_modeswitch.conf
apt                     ca-certificates       dnsmasq.conf  fw_env.config  iptables     login.defs      netconfig       pam.d       release                     sensord.conf   ssh           systemd      usb_modeswitch.d
avahi                   ca-certificates.conf  dnsmasq.d     group          issue        machine-id      network         passwd      resolv.conf                 sensors3.conf  ssl           terminfo     vconsole.conf
bash_completion         chatscripts           dpkg          gshadow        issue.net    mke2fs.conf     NetworkManager  pki         resolv-conf.NetworkManager  sensors.d      sudoers       timestamp    version
bash_completion.d       cron.daily            environment   host.conf      ld.so.cache  modprobe.d      nginx           ppp         resolv-conf.systemd         services       sudoers.d     timezone     wgetrc
bindresvport.blacklist  dbus-1                fancontrol    hostname       ld.so.conf   modules-load.d  nsswitch.conf   profile     rpc                         shadow         sudoers.dist  tmpfiles.d   wpa_supplicant.conf
bluetooth               default               fonts         hosts          libnl        mosquitto       ofono           profile.d   screenrc                    shells         sysconfig     udev         xdg
busybox.links.nosuid    depmod.d              fstab         inputrc        locale.conf  motd         

/tmp/btrfs_ex4$ find . -name hostapd*
find: ‘./var/cache/ldconfig’: Permission denied
find: ‘./var/lib/sudo’: Permission denied
find: ‘./var/lib/NetworkManager’: Permission denied
find: ‘./var/lib/openvpn’: Permission denied
find: ‘./lost+found’: Permission denied
find: ‘./etc/sudoers.d’: Permission denied
```
I should found hostapd but not.

htot, 2020-08-24:
Ok, no problem.
Can you check `tmp/out/linux64/build/tmp/work/edison-poky-linux/edison-image/1.0-r0/rootfs/` if you can find `hostapd*`?

shawnhsoia97, 2020-08-31:
Yes,I can found `hostapd*`,now I found hostapd in my Edison.
I  don't think it make sense to increase `IMAGE_ROOTFS_SIZE`,the reason should be that there was a problem with my file,but it is resolved now,Thank.

But I have the new problem,when installed  more package,appeared `edison-image-edison.btrfs` is too big.I don't understand what I should have more space ,right?

```
images/edison$ sudo dfu-util -v -d 8087:0a99 --alt home -D edison-image-edison.btrfs
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: File size is too big: Success

```

Also if I have flashed ext4 image and installed btrfs image once.
When I want update brifs image ,can I just do `sudo dfu-util -v -d 8087:0a99 --alt home -D edison-image-edison.btrfs` ? don't reflash ext4 image.

htot, 2020-08-31:
> But I have the new problem,when installed more package,appeared `edison-image-edison.btrfs` is too big.I don't understand what I should have more space ,right?
> 
> ```
> dfu-util: File size is too big: Success
> ```
The partition is too small probably. Try `partx /dev/mmcblk0` to see the partitions.
> Also if I have flashed ext4 image and installed btrfs image once.
> When I want update brifs image ,can I just do `sudo dfu-util -v -d 8087:0a99 --alt home -D edison-image-edison.btrfs` ? don't reflash ext4 image.

Yes, that should work.

shawnhsoia97, 2020-09-01:
This is  size of partition if `edison-image-edison.btrfs = 1981MB`
```
 partx /dev/mmcblk0
NR  START     END SECTORS SIZE NAME        UUID
 1   2048    6143    4096   2M u-boot0     0b484ad1-10d8-404f-ba7e-ccead78a7ac6
 2   6144    8191    2048   1M u-boot-env0 269de5cf-537d-de4d-9cea-9bec786dcca8
 3   8192   12287    4096   2M u-boot1     fa474346-c071-574e-b954-c47510693215
 4  12288   14335    2048   1M u-boot-env1 71f2d63f-5705-4d41-9d53-498988f37ec6
 5  14336   16383    2048   1M factory     5d62fae6-44d9-4549-a81f-28c14abf6a9a
 6  16384   65535   49152  24M panic       0d195330-516f-c847-b2a6-ee75a26a3685
 7  65536  196607  131072  64M boot        b55a5768-3ed3-0942-b74b-879f6390dbab
 8 196608 7634910 7438303 3.6G home        dc317641-c89d-ad4c-ba2d-21add117539a
```

`dfu-util: File size is too big: Success` ,if `edison-image-edison.btrfs = 2557MB`

```
images/edison$ sudo dfu-util -v -d 8087:0a99 --alt home -D edison-image-edison.btrfs
[sudo] password for bovia: 
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: File size is too big: Success

```
Maybe is my computer memory is too small?because I not connect the 2 USB cable to the USB port,that also get this error if  `edison-image-edison.btrfs = 2557MB`,but if  `edison-image-edison.btrfs = 1981MB`,that not get any error,that is just wait USB connect.

htot, 2020-09-01:
I see. Maybe there is a 2GB limit for dfu-util? I haven't tried that size.

Would it work for you to boot the rescue image, then copy the btrfs image from the sd card to the emmc using `dd`?

[14 more replies]

closed 2020-09-16 by shawnhsoia97