edison-fw/meta-intel-edison #117: Error during flashing Edison Image Warrior64 on Windows 10

Quarer, 2020-10-06:
I'm getting troubles with OS installation on Intel Edison Breakout board.
Using Windows 10 x64, Intel Platform Flash Tool Lite 5.8.9.0. Successfully installed older [Sumo32 Image](https://github.com/edison-fw/meta-intel-edison/releases/download/v1.0/edison-image-4b7e1298.tar.gz).
How I flash Image:
1. Launch Platform Flashing Tool,
2. Extracting downloaded image from archive
3. Browsing to select _FlashEdison.json_ file.
4. Usually start flashing, and connecting Edison.

With newest image Warrior64, Intel's flasher shows error:
```
10/06/20 11:43:35.638  INFO   : Platform Flash Tool Lite initialized successfully
10/06/20 11:50:41.354  ERROR  : D:\X\edison-debian-decb4c8\toFlash\edison_ifwi-dbg-00.bin not found
10/06/20 11:50:41.360  ERROR  : Failed to load D:\X\edison-debian-decb4c8\toFlash\FlashEdison.json
```
So I have moved all the files from directory _GNUSparseFile.0_ and leave them into main directory, so far without errors
```
10/06/20 11:58:04.569  DEBUG  : Json Flash file parsed (D:/X/edison-debian-decb4c8/toFlash)
10/06/20 11:58:04.570  INFO   : Ready to flash!
10/06/20 11:58:04.817  DEBUG  : FlashConfiguration:
name=Windows
startState=dnx_fw
brief=RNDIS USB driver (for Windows)
description=U-Boot RNDIS environment
documentation=
warning=
hiddenToEndUser=no
id=rndis_env
cmd[0, m=yes, t=no, s=no] Notify ${helperFile}
cmd[1, m=yes, t=42000, s=no] Wait for device with status dnx_fw
cmd[2, m=yes, t=no, s=no] Notify Edison board plugged
cmd[3, m=yes, t=180000, s=no] xFSTK Downloader:
        IFWI: D:\X\edison-debian-decb4c8\toFlash\edison_ifwi-dbg-00.bin
        IFWI DnX: D:\X\edison-debian-decb4c8\toFlash\edison_dnx_fwr.bin
        OS: D:\X\edison-debian-decb4c8\toFlash\u-boot-edison.img
        OS DnX: D:\X\edison-debian-decb4c8\toFlash\edison_dnx_osr.bin
        GP Flag: 0x80000007
cmd[4, m=yes, t=5000, s=no] Sleeping for 5 seconds
cmd[5, m=yes, t=120000, s=no] Wait for device with status pos
cmd[6, m=yes, t=60000, s=no] dfuDownloader "--verbose" "--alt" "u-boot0" "-D" "D:\X\edison-debian-decb4c8\toFlash\u-boot-edison.bin" "-t" "4096"
cmd[7, m=yes, t=60000, s=no] dfuDownloader "--verbose" "--alt" "u-boot-env0" "-D" "D:\X\edison-debian-decb4c8\toFlash\u-boot-envs\edison-blankrndis.bin" "-t" "4096"
cmd[8, m=yes, t=60000, s=no] dfuDownloader "--verbose" "--alt" "u-boot-env1" "-D" "D:\X\edison-debian-decb4c8\toFlash\u-boot-envs\edison-blankrndis.bin" "-t" "4096" "-R"
cmd[9, m=yes, t=5000, s=no] Sleeping for 5 seconds
cmd[10, m=yes, t=100000, s=no] Wait for device with status pos
cmd[11, m=yes, t=60000, s=no] dfuDownloader "--verbose" "--alt" "boot" "-D" "D:\X\edison-debian-decb4c8\toFlash\edison-image-edison.hddimg" "-t" "4096"
cmd[12, m=yes, t=1200000, s=no] dfuDownloader "--verbose" "--alt" "rootfs" "-D" "D:\X\edison-debian-decb4c8\toFlash\edison-image-edison.ext4" "-t" "4096" "-R"
cmd[13, m=yes, t=no, s=no] Notify Edison board needs to reboot<br />
                Please do not unplug it for 2 minutes<br /><br />
                On Arduino board, led will stop blinking at boot completed<br /><br />
                <img src='images/Edison-arduino-blink-led.png' width='390' />
10/06/20 11:58:04.818  DEBUG  : Monitoring D:\X\edison-debian-decb4c8\toFlash\FlashEdison.json
10/06/20 11:58:04.818  INFO   : Flash file D:\X\edison-debian-decb4c8\toFlash\FlashEdison.json [Windows] loaded
```

There is the detailed log:
```
10/06/20 11:58:04.818  DEBUG  : Monitoring D:\X\edison-debian-decb4c8\toFlash\FlashEdison.json
10/06/20 11:58:04.818  INFO   : Flash file D:\X\edison-debian-decb4c8\toFlash\FlashEdison.json [Windows] loaded
10/06/20 11:59:24.751  INFO   : [Port 1/1/3] New device detected on 1/1/3 (id=0100020000000002)
10/06/20 11:59:33.798  INFO   : [Port 1/1/3] Device lost on 1/1/3
10/06/20 11:59:58.389  INFO   : [Port 1/1/3] New device detected on 1/1/3 (id=0100020000000002)
10/06/20 11:59:59.084  INFO   : [Port 1/1/3] Start flashing the flash configuration with dnx_fw start state for B282A67AE3557604F2C232FF97E34BCC
10/06/20 11:59:59.877  DEBUG  : Device is already in the required state, no need to reboot it
10/06/20 11:59:59.878  INFO   : [Port 1/1/3] Starting flash ...
10/06/20 11:59:59.878  INFO   : [Port 1/1/3] Running `Notify ${helperFile}` command
10/06/20 11:59:59.965  INFO   : Displaying notification during 42000 ms
10/06/20 11:59:59.970  INFO   : [Port 1/1/3] Command `Notify ${helperFile}` succeed
10/06/20 12:00:00.020  INFO   : [Port 1/1/3] Running `Wait for device with status dnx_fw on port 1/1/3` command
10/06/20 12:00:00.020  INFO   : [Port 1/1/3] Command `Wait for device with status dnx_fw on port 1/1/3` succeed
10/06/20 12:00:00.020  INFO   : [Port 1/1/3] Running `Notify Edison board plugged` command
10/06/20 12:00:00.044  INFO   : Displaying notification during 5000 ms
[... 160 lines cut ...]
Please plug and reboot the board
Dfu device found
Flashing IFWI
Download        [=========================] 100%      4194304 bytes
Download done.
Download        [=========================] 100%      4194304 bytes
Download done.
Flashing U-Boot
Download        [=========================] 100%       475648 bytes
Download done.
Flashing U-Boot Environment
Download        [=========================] 100%        65536 bytes
Download done.
Flashing U-Boot Environment Backup
Download        [=========================] 100%        65536 bytes
Download done.
Rebooting to apply partiton changes
Dfu device not found Timeout
Did you plug and reboot your board?
If yes, please try a recovery by calling this script with the --recovery option
```

Thanks for any help.
Greetings,
Tom

htot, 2020-10-06:
I'm not sure where GNUSparseFile.0 dir.- comes from. The directory structure I see is:
[image]

Quarer, 2020-10-06:
So WinRar and 7-zip in Windows "sees" one more directory _GNUSparseFile.0_ 
[image]
[image]

I downloaded file with OS one more time on linux, and you are right. There is no folder like _GNUSparseFile.0_  under linux.

Log from linux:
```
root@u0 _delete_me_ :( # tar -xf edison-image-4b7e1298.tar.gz
root@u0 _delete_me_ # tree
.
├── edison-image-4b7e1298.tar.gz
└── toFlash
    ├── edison_dnx_fwr.bin
    ├── edison_dnx_osr.bin
    ├── edison_ifwi-dbg-00.bin
    ├── edison_ifwi-dbg-00-dfu.bin
    ├── edison_ifwi-dbg-01.bin
    ├── edison_ifwi-dbg-01-dfu.bin
    ├── edison_ifwi-dbg-02.bin
    ├── edison_ifwi-dbg-02-dfu.bin
    ├── edison_ifwi-dbg-03.bin
    ├── edison_ifwi-dbg-03-dfu.bin
    ├── edison_ifwi-dbg-04.bin
    ├── edison_ifwi-dbg-04-dfu.bin
    ├── edison_ifwi-dbg-05.bin
    ├── edison_ifwi-dbg-05-dfu.bin
    ├── edison_ifwi-dbg-06.bin
    ├── edison_ifwi-dbg-06-dfu.bin
    ├── edison-image-edison.ext4
    ├── edison-image-edison.hddimg
    ├── filter-dfu-out.js
    ├── flashall.bat
    ├── flashall.sh
    ├── FlashEdison.json
    ├── helper
    │   ├── helper.html
    │   └── images
    │       ├── Edison-arduino-blink-led.png
    │       ├── Edison-arduino.png
    │       └── Edison-breakout-board.png
    ├── ota_update.scr
    ├── package-list.txt
    ├── u-boot-edison.bin
    ├── u-boot-edison.img
    └── u-boot-envs
        ├── edison-blankcdc.bin
        ├── edison-blankrndis.bin
        ├── edison-ifwi.bin
        └── edison-prod.bin

4 directories, 35 files
```

But I still dont know how to flash it correctly 😢. After flashing IFWI and U-Boot device stuck, not showing anymore. I suppose it stuck in bootloop.

htot, 2020-10-06:
You lost me. Your first log shows:
```
cmd[12, m=yes, t=1200000, s=no] dfuDownloader "--verbose" "--alt" "rootfs" "-D" "D:\X\edison-debian-decb4c8\toFlash\edison-image-edison.ext4" "-t" "4096" "-R"
cmd[13, m=yes, t=no, s=no] Notify Edison board needs to reboot<br />
                Please do not unplug it for 2 minutes<br /><br />
```
So it looks like everything flashed. How are you trying to contact Edison?

Quarer, 2020-10-06:
> You lost me. Your first log shows:
>
> ```
> cmd[12, m=yes, t=1200000, s=no] dfuDownloader "--verbose" "--alt" "rootfs" "-D" "D:\X\edison-debian-decb4c8\toFlash\edison-image-edison.ext4" "-t" "4096" "-R" [...]

Ahh,  my bad, this part shouldn't be there... Its because i merged both folders (moved, replaced files from old _edison-image-4b7e1298_ to _edison-image-decb4c8_. My bad, sorry for that.

Here is the log from now, flashing (OS _edison-image-decb4c8_) on Windows 10 x64, using Intel Platform Flash Tool Lite:
```
10/06/20 21:05:04.389  DEBUG  : Json Flash file parsed (D:/X/edison-image-decb4c8/toFlash)
10/06/20 21:05:04.390  INFO   : Ready to flash!
10/06/20 21:05:04.662  DEBUG  : FlashConfiguration:
name=Windows
startState=dnx_fw
brief=RNDIS USB driver (for Windows)
description=U-Boot RNDIS environment
documentation=
warning=
hiddenToEndUser=no
id=rndis_env
cmd[0, m=yes, t=no, s=no] Notify ${helperFile}
cmd[1, m=yes, t=42000, s=no] Wait for device with status dnx_fw
cmd[2, m=yes, t=no, s=no] Notify Edison board plugged
cmd[3, m=yes, t=180000, s=no] xFSTK Downloader:
        IFWI: D:\X\edison-image-decb4c8\toFlash\edison_ifwi-dbg-00.bin
        IFWI DnX: D:\X\edison-image-decb4c8\toFlash\edison_dnx_fwr.bin
        OS: D:\X\edison-image-decb4c8\toFlash\u-boot-edison.img
        OS DnX: D:\X\edison-image-decb4c8\toFlash\edison_dnx_osr.bin
        GP Flag: 0x80000007
cmd[4, m=yes, t=5000, s=no] Sleeping for 5 seconds
cmd[5, m=yes, t=120000, s=no] Wait for device with status pos
cmd[6, m=yes, t=60000, s=no] dfuDownloader "--verbose" "--alt" "u-boot0" "-D" "D:\X\edison-image-decb4c8\toFlash\u-boot-edison.bin" "-t" "4096"
cmd[7, m=yes, t=60000, s=no] dfuDownloader "--verbose" "--alt" "u-boot-env0" "-D" "D:\X\edison-image-decb4c8\toFlash\u-boot-envs\edison-blankrndis.bin" "-t" "4096"
cmd[8, m=yes, t=60000, s=no] dfuDownloader "--verbose" "--alt" "u-boot-env1" "-D" "D:\X\edison-image-decb4c8\toFlash\u-boot-envs\edison-blankrndis.bin" "-t" "4096" "-R"
cmd[9, m=yes, t=5000, s=no] Sleeping for 5 seconds
cmd[10, m=yes, t=100000, s=no] Wait for device with status pos
cmd[11, m=yes, t=60000, s=no] dfuDownloader "--verbose" "--alt" "boot" "-D" "D:\X\edison-image-decb4c8\toFlash\edison-image-edison.hddimg" "-t" "4096"
cmd[12, m=yes, t=1200000, s=no] dfuDownloader "--verbose" "--alt" "rootfs" "-D" "D:\X\edison-image-decb4c8\toFlash\edison-image-edison.ext4" "-t" "4096" "-R"
cmd[13, m=yes, t=no, s=no] Notify Edison board needs to reboot<br />
                Please do not unplug it for 2 minutes<br /><br />
                On Arduino board, led will stop blinking at boot completed<br /><br />
                <img src='images/Edison-arduino-blink-led.png' width='390' />
10/06/20 21:05:04.662  DEBUG  : Monitoring D:\X\edison-image-decb4c8\toFlash\FlashEdison.json
10/06/20 21:05:04.662  INFO   : Flash file D:\X\edison-image-decb4c8\toFlash\FlashEdison.json [Windows] loaded
10/06/20 21:05:23.235  INFO   : [Port 1/1/2] New device detected on 1/1/2 (id=0100010000000002)
10/06/20 21:05:25.248  INFO   : [Port 1/1/2] Start flashing the flash configuration with dnx_fw start state for B282A67AE3557604F2C232FF97E34BCC
10/06/20 21:05:25.250  DEBUG  : Device is already in the required state, no need to reboot it
10/06/20 21:05:25.250  INFO   : [Port 1/1/2] Starting flash ...
10/06/20 21:05:25.250  INFO   : [Port 1/1/2] Running `Notify ${helperFile}` command
10/06/20 21:05:25.353  INFO   : Displaying notification during 42000 ms
10/06/20 21:05:25.366  INFO   : [Port 1/1/2] Command `Notify ${helperFile}` succeed
10/06/20 21:05:25.366  INFO   : [Port 1/1/2] Running `Wait for device with status dnx_fw on port 1/1/2` command
10/06/20 21:05:25.366  INFO   : [Port 1/1/2] Command `Wait for device with status dnx_fw on port 1/1/2` succeed
10/06/20 21:05:25.366  INFO   : [Port 1/1/2] Running `Notify Edison board plugged` command
10/06/20 21:05:25.403  INFO   : Displaying notification during 5000 ms
10/06/20 21:05:25.406  INFO   : [Port 1/1/2] Command `Notify Edison board plugged` succeed
10/06/20 21:05:25.406  INFO   : [Port 1/1/2] Running `"C:\Program Files (x86)\Intel\Platform Flash Tool Lite\xfstkFlashTool.exe" "-1" "D:\X\edison-image-decb4c8\toFlash\edison_dnx_fwr.bin" "D:\X\edison-image-decb4c8\toFlash\edison_ifwi-dbg-00.bin" "D:\X\edison-image-decb4c8\toFlash\edison_dnx_osr.bin" "D:\X\edison-image-decb4c8\toFlash\u-boot-edison.img" "0x80000007" "none" "none" "0"` command
10/06/20 21:05:25.748  INFO   : [Port 1/1/2] Disabling xFSTK Misc DnX
10/06/20 21:05:25.748  INFO   : [Port 1/1/2] Disabling xFSTK wipe out ifwi option
10/06/20 21:05:25.748  INFO   : [Port 1/1/2] Using xFSTK Solo interface (new API)
10/06/20 21:05:25.748  INFO   : [Port 1/1/2] xFSTK Soft Fuse not set
10/06/20 21:05:25.748  INFO   : [Port 1/1/2] XFSTK Downloader API 1.8.1
10/06/20 21:05:25.748  INFO   : [Port 1/1/2] Copyright (c) 2015 Intel Corporation
10/06/20 21:05:25.748  INFO   : [Port 1/1/2] fwdnx -- D:\X\edison-image-decb4c8\toFlash\edison_dnx_fwr.bin
10/06/20 21:05:25.748  INFO   : [Port 1/1/2] fwimage -- D:\X\edison-image-decb4c8\toFlash\edison_ifwi-dbg-00.bin
10/06/20 21:05:25.748  INFO   : [Port 1/1/2] osdnx -- D:\X\edison-image-decb4c8\toFlash\edison_dnx_osr.bin
10/06/20 21:05:25.748  INFO   : [Port 1/1/2] osimage -- D:\X\edison-image-decb4c8\toFlash\u-boot-edison.img
10/06/20 21:05:25.748  INFO   : [Port 1/1/2] gpflags -- 0x80000007
10/06/20 21:05:25.748  INFO   : [Port 1/1/2] softfuse --
[... cut ...] iled
10/06/20 21:06:21.283  ERROR  : [Port 1/1/2] Flash failed (Command type: Ifwi)
10/06/20 21:06:21.288  TRACE  : Sending system tray notification: "Flash failed"
10/06/20 21:06:21.398  INFO   : [Port 1/1/2] Device lost on 1/1/2
```
[image]

> How are you trying to contact Edison?

With older verison of your software (_edison-image-4b7e1298_ flashing correctly) I'm connecting second USB cable, windows detect new device as COM port. To comunication using Putty with COM[number] and speed 115200.

htot, 2020-10-06:
Hmm. As you can see the part that is failing is xfstk. `Flashtool Lite` flashes in 2 stages, `xfstk` then dfu.
That would be the same as with the flashall script doing flashall --recovery (uses xfstk) and then flashall (uses dfu).

Now the funny thing is: xfstk communicates with the built-in bootloader, this can not be destroyed. It downloads all kind of binary stuff (ifwi) and then installs a partition table, a partition with u-boot and the environment variables u-boot needs. Normally xfstk is  not needed, only if you manually break the environment variables so dfu can't work anymore.

So only `flashall` would normally be needed.

You are seeing:
```
10/06/20 21:05:31.541  DEBUG  : [Port 1/1/2] XFSTK-LOG--Windriver Error: 0xC0000011, HC status: The host controller has set the Transaction Error (XactErr) bit in the transfer descriptor's status field
```
Which means there was a communication error between `xfstk` tool and edison.

This could mean (and I hate it when people do this to me), your usb cable might be bad (I had this too). Or there may be a problem with windows 10 and the driver (I heard people complaining). So you might want to try:
- other, better (maybe even otg compatible) cable
- booting a ubuntu live usb, `agt-get install dfu-util` then try `./flashall.sh` (or dual boot into Ubuntu if you can). I don't have much faith in running Ubuntu in a Virtualbox for this.

I see you also flashed with `flashall.bat` with more success:
```
Flashing U-Boot Environment Backup
Download        [=========================] 100%        65536 bytes
Download done.
Rebooting to apply partiton changes
Dfu device not found Timeout
```
This means u-boot environment was flashed, Edison wants to reboot, maybe repartition then continue to flash boot partition then rootfs partition. This problem I also had due to same faulty usb cable, host pc failed to detect usb unplug/plug.
- so again suspicion that you have a bad usb cable.
- but now you can manualy solve it, through putty press `<esc>` during boot. Now you are in u-boot. Do `run do_force_flash_os`, Edison will start waiting for dfu. These lines from `flashall.bat` (I only did this from linux, not windows):
```
echo Flashing boot partition ^(kernel^)
call:flash-command --alt boot -D "%BASE_DIR%edison-image-edison.hddimg"
if %errorlevel% neq 0 ( exit /b %errorlevel%)

echo Flashing rootfs, ^(it can take up to 5 minutes... Please be patient^)
call:flash-command --alt rootfs -D "%BASE_DIR%edison-image-edison.ext4" -R
```
with call:flash-command is `dfu-util -d 8087:0a99`. If the cable is really bad, you will have a large chance that flashing the rootfs will still fail in the middle. Otherwise it should be able to boot now. BTW -R will cause edison to reboot. If you omit that:
```
    #  
    DFU complete CRC32: 0xf340088e  
    DOWNLOAD ... OK  
    Ctrl+C to exit ...  
    boot > reset  
    resetting ...  
```

Quarer, 2020-10-07:
At start I have to say, that I replaced low-quality cables. Flashed older firmwares without problems, but...

I have no idea, why device wont boot to U-Boot (after flashing newest os).

Console output on flashing **older** OS - _edison-image-4b7e1298_:

```
******************************
PSH KERNEL VERSION: b0182b2b
                WR: 20104000
******************************

SCU IPC: 0x800000d0  0xfffce92c

PSH miaHOB version: TNG.B0.VVBD.0000000c

microkernel built 11:24:08 Feb  5 2015

******* PSH loader *******
PCM page cache size = 192 KB
Cache Constraint = 0 Pages
Arming IPC driver ..
Adding page store pool ..
PagestoreAddr(IMR Start Address) = 0x04899000
pageStoreSize(IMR Size)          = 0x00080000

*** Ready to receive application ***

U-Boot 2018.09 (Jan 19 2019 - 14:58:11 +0000)

CPU: x86_64, vendor Intel, device 406a8h
DRAM:  980.6 MiB
MMC:   mmc@ff3fc000: 0, mmc@ff3fa000: 1
Loading Environment from MMC... OK
In:    serial
Out:   serial
Err:   serial
Saving Environment to MMC... Writing to redundant MMC(0)... OK
Saving Environment to MMC... Writing to MMC(0)... OK
Net:   No ethernet found.
Hit any key to stop autoboot:  0
Target:ifwi
Partitioning using GPT
Writing GPT: success!
Saving Environment to MMC... Writing to redundant MMC(0)... OK
Partitioning already done...
Saving Environment to MMC... Writing to MMC(0)... OK
#DOWNLOAD ... OK
Ctrl+C to exit ...
#DOWNLOAD ... OK
Ctrl+C to exit ...
#DOWNLOAD ... OK
Ctrl+C to exit ...
resetting ...

******************************
PSH KERNEL VERSION: b0182b2b
                WR: 20104000
******************************

SCU IPC: 0x800000d0  0xfffce92c

PSH miaHOB version: TNG.B0.VVBD.0000000c

microkernel built 11:24:08 Feb  5 2015

******* PSH loader *******
PCM page cache size = 192 KB
Cache Constraint = 0 Pages
Arming IPC driver ..
[... 135 lines cut ...]
Flashing U-Boot
Download	[=========================] 100%       475136 bytes
Flashing U-Boot Environment
Download	[=========================] 100%        65536 bytes
Flashing U-Boot Environment Backup
Download	[=========================] 100%        65536 bytes
Rebooting to apply partition changes
Now waiting for dfu device 8087:0a99
Timed out while waiting for dfu device 8087:0a99
DEBUG: lsusb
Bus 001 Device 009: ID 8086:e005 Intel Corp. 
Bus 001 Device 003: ID 090c:1000 Silicon Motion, Inc. - Taiwan (formerly Feiya Technology Corp.) Flash Drive
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
DEBUG: dfu-util -l
dfu-util 0.9
```

I would say even that my device is broken, but then it shouldn't be working (flashing firmwares) at all.

USB 2.0 cable got only 4 wires inside: +5V, GDN, D-, D+. I used that cables before to flash older firmwares, they worked.

Is there any chance to use older U-Boot and new image with `boot `and `rootfs`?

htot, 2020-10-08:
So, now the problem is unplug/plug event is not detected after U-Boot is updated. Might be a combination with vmware.
You can try to install the boot and rootfs image manually, that will likely work.

BTW I don't see how apt-get couldn't find the dfu-util package. That is not normal.

Quarer, 2020-10-08:
I would say, that after update of U-Boot, device cannot entry to U-Bot mode.
Whole booting stops on:
```

******************************
PSH KERNEL VERSION: b0182b2b
                WR: 20104000
******************************

SCU IPC: 0x800000d0  0xfffce92c

PSH miaHOB version: TNG.B0.VVBD.0000000c

microkernel built 11:24:08 Feb  5 2015

******* PSH loader *******
PCM page cache size = 192 KB
Cache Constraint = 0 Pages
Arming IPC driver ..
Adding page store pool ..
PagestoreAddr(IMR Start Address) = 0x04899000
pageStoreSize(IMR Size)          = 0x00080000

*** Ready to receive application ***
```

> You can try to install the boot and rootfs image manually, that will likely work.

I agree, but to install boot and rootfs I have to be in U-Boot mode, is that correct ?

htot, 2020-10-08:
I see, I missed that. You start with a functioning U-Boot 2018.09, then flash it with the U-Boot from the Warrior image (2020.01) and then U-Boot doesn't load. Never had that, but I will try to reproduce maybe tomorrow evening.

In the meanwhile you might want to copy the older U-Boot image over to your Warrior image and flash that, I would expect that to work.

htot, 2020-10-08:
OK I see. U-Boot [First run sets hardware_id environment variable which is read from
System Controller Unit (SCU)](First run sets hardware_id environment variable which is read from
System Controller Unit (SCU).). And the hardware_id is used to select edison_ifwi-dbg-xx.bin. So, if you have a different hardware_id that could explain why there is a different result. You make make this visible by adding -> into flashall.sh (or similar with flashall.bat):
```
function flash-dfu-ifwi {
	ifwi_hwid_found=`dfu-util -l -d ${USB_VID}:${USB_PID} | grep -c $1`
	if [ $ifwi_hwid_found -ne 0 ];
	then
->             dfu-util -l -d ${USB_VID}:${USB_PID} | grep $1
		flash-command ${@:2}
	fi
}
```

I just tried myself and got (hardware_id = 00):
```
Found DFU: [8087:0a99] ver=0223, devnum=32, cfg=1, intf=0, path="3-8", alt=0, name="ifwi00", serial="cbd210a7487174ddb6fdc06e815d8e87"
```
But I note another thing:
```
Flashing U-Boot
Download        [=========================] 100%       479232 bytes
```
In the above screenshots your have varying results: 475136, 475648. I think still something goes wrong unpacking the gz.
So here is my toFlash listing:
```
ferry@delfion:~/tmp/edison/edison-warrior/toFlash$ ls -l
totaal 1127388
-rw-rw-r-- 1 ferry ferry      98196 feb 14  2020 edison_dnx_fwr.bin
-rw-rw-r-- 1 ferry ferry     148996 feb 14  2020 edison_dnx_osr.bin
-rw-rw-r-- 1 ferry ferry    4194468 feb 14  2020 edison_ifwi-dbg-00.bin
-rw-rw-r-- 1 ferry ferry    4194304 feb 14  2020 edison_ifwi-dbg-00-dfu.bin
-rw-rw-r-- 1 ferry ferry    4194468 feb 14  2020 edison_ifwi-dbg-01.bin
-rw-rw-r-- 1 ferry ferry    4194304 feb 14  2020 edison_ifwi-dbg-01-dfu.bin
-rw-rw-r-- 1 ferry ferry    4194468 feb 14  2020 edison_ifwi-dbg-02.bin
-rw-rw-r-- 1 ferry ferry    4194304 feb 14  2020 edison_ifwi-dbg-02-dfu.bin
-rw-rw-r-- 1 ferry ferry    4194468 feb 14  2020 edison_ifwi-dbg-03.bin
-rw-rw-r-- 1 ferry ferry    4194304 feb 14  2020 edison_ifwi-dbg-03-dfu.bin
-rw-rw-r-- 1 ferry ferry    4194468 feb 14  2020 edison_ifwi-dbg-04.bin
-rw-rw-r-- 1 ferry ferry    4194304 feb 14  2020 edison_ifwi-dbg-04-dfu.bin
-rw-rw-r-- 1 ferry ferry    4194468 feb 14  2020 edison_ifwi-dbg-05.bin
-rw-rw-r-- 1 ferry ferry    4194304 feb 14  2020 edison_ifwi-dbg-05-dfu.bin
-rw-rw-r-- 1 ferry ferry    4194468 feb 14  2020 edison_ifwi-dbg-06.bin
-rw-rw-r-- 1 ferry ferry    4194304 feb 14  2020 edison_ifwi-dbg-06-dfu.bin
-rw-r--r-- 1 ferry ferry 1073741824 feb 14  2020 edison-image-edison.ext4
-rw-r--r-- 1 ferry ferry   14758912 feb 14  2020 edison-image-edison.hddimg
-rw-rw-r-- 1 ferry ferry        742 feb 14  2020 filter-dfu-out.js
-rw-rw-r-- 1 ferry ferry       7225 feb 14  2020 flashall.bat
-rwxrwxr-x 1 ferry ferry       5935 okt  8 22:52 flashall.sh
-rw-rw-r-- 1 ferry ferry       5939 feb 14  2020 FlashEdison.json
-rw-rw-r-- 1 ferry ferry      84517 okt  8 22:57 flash.log
drwxrwxr-x 1 ferry ferry         34 feb 14  2020 helper
-rw-rw-r-- 1 ferry ferry      22264 feb 14  2020 ota_update.scr
-rw-rw-r-- 1 ferry ferry      24839 feb 14  2020 package-list.txt
-rw-r--r-- 1 ferry ferry     479232 feb 14  2020 u-boot-edison.bin
-rwxr-xr-x 1 ferry ferry    6291968 feb 14  2020 u-boot-edison.img
drwxr-xr-x 1 ferry ferry        140 feb 14  2020 u-boot-envs
```
Please check the file sizes (and note that you also have different dates).

Quarer, 2020-10-10:
> In the meanwhile you might want to copy the older U-Boot image over to your Warrior image and flash that, I would expect that to work.

So I tried to boot os with old U-Boot. Like You said, I tried to flash warrior.
[image]

Log from CMD:
```
D:\X>"C:\Program Files (x86)\Intel\Platform Flash Tool Lite\dfu-util.exe" "--alt" "boot" "-D" "D:\X\edison-image-decb4c8\toFlash\edison-image-edison.hddimg" "-t" "4096" "-d" "8087:0a99" "--path" "2-1.2"
dfu-util 0.8.4-msvc

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 8087:0a99
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #6 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Copying data from PC to DFU device
Download        [=========================] 100%     14205440 bytes
Download done.
state(7) = dfuMANIFEST, status(0) = No error condition is present
state(2) = dfuIDLE, status(0) = No error condition is present
Done!

D:\X>"C:\Program Files (x86)\Intel\Platform Flash Tool Lite\dfu-util.exe" "--alt" "rootfs" "-D" "D:\X\edison-image-decb4c8\toFlash\edison-image-edison.ext4" "-t" "4096" "-R" "-d" "8087:0a99" "--path" "2-1.2"
dfu-util 0.8.4-msvc

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 8087:0a99
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #7 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Copying data from PC to DFU device
Download        [=========================] 100%    579686912 bytes
Download done.
state(7) = dfuMANIFEST, status(0) = No error condition is present
state(2) = dfuIDLE, status(0) = No error condition is present
Done!
Resetting USB to switch back to runtime mode

D:\X>
```

Log from Edison via Putty:

```

******************************
PSH KERNEL VERSION: b0182b2b
                WR: 20104000
******************************

[... 627 lines cut ...]

edison login: 
```

### Thank You Sir, a lot. Really great job! 

One more thing, I have tried to setup Wi-Fi on Edison. Command `configure_edison  --setup` and it found nothing, output:
```
Configure Edison: WiFi Connection

Failed to stop hostapd.service: Unit hostapd.service not loaded.
Failed to connect to non-global ctrl_ifname: (nil)  error: No such file or directory
Failed to connect to non-global ctrl_ifname: (nil)  error: No such file or directory

0 :     Rescan for networks
1 :     Exit WiFi Setup
2 :     Manually input a hidden SSID

Enter 0 to rescan for networks.
Enter 1 to exit.
Enter 2 to input a hidden network SSID:
```

Can you give me some advice how to setup Wi-Fi 2,4Ghz connection?
On [edison-fw/meta-intel-edison/wiki](https://github.com/edison-fw/meta-intel-edison/wiki) under section _Get Wifi to work_ there is no article.

htot, 2020-10-10:
### Leason learned for me
I won't be using gz as archive any more. I never new there are compatibility issues between linux and windows. @andy-shev is recommending xz, I'll investigate that.

### oobe
configure_edison is part of oobe, for which @mwallnoefer has a PR [here](https://github.com/edison-fw/edison-oobe/pull/6 ). Unfortunately I haven't been able to work on that much, but will soon.

The PR should be improving on the CLI, while breaking the web interface a bit that I was working on. After we give it a little more love both will be working.

Our help pages are here: https://edison-fw.github.io/meta-intel-edison/
Wifi setup (using connman): https://edison-fw.github.io/meta-intel-edison/4.2-networking.html#wifi

Actually connman is working these days very well, much better then when edison was started. It does at least: ethernet over usb (only [eem](https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-bsp/recipes-support/gadget/files/conf-gadget.sh), I did not implement rndis as windows hosts need) , wifi and bluetooth as well as tethering (no hostapd needed).

On first boot the wifi will be working as AP. If you want connman to stop that:
```
technologies
tether wifi off
```

htot, 2020-10-10:
BTW I just unpacked tar.gzip on 32bit Win7 using 7-zip 19.00 (x86) and I see no problems (no GNUsparesefile.0, file sizes correct).
I've tried with:
```
-rw-rw-r-- 1 ferry ferry 138006963 okt 10 13:53 toFlash.7z
-rw-rw-r-- 1 ferry ferry 198437406 okt 10 13:49 toFlash.tar.gz
-rw-rw-r-- 1 ferry ferry 137629692 okt 10 13:57 toFlash.tar.xz
-rw-rw-r-- 1 ferry ferry 198129452 okt 10 13:50 toFlash.zip
```
None had problems, but I think 7z is the most convenient on both platforms, because it skips the tar step (and knowing that zip looses file time precision).

Quarer, 2020-10-10:
> Our help pages are here: https://edison-fw.github.io/meta-intel-edison/
> Wifi setup (using connman): https://edison-fw.github.io/meta-intel-edison/4.2-networking.html#wifi

Thanks 😄, works:

```
root@Edison:/# ifconfig
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:902 errors:0 dropped:0 overruns:0 frame:0
          TX packets:902 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:60446 (59.0 KiB)  TX bytes:60446 (59.0 KiB)

wlan0     Link encap:Ethernet  HWaddr fc:c2:de:41:70:a3
          inet addr:10.9.0.200  Bcast:10.9.255.255  Mask:255.255.0.0
          inet6 addr: fe80::fec2:deff:fe41:70a3/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST DYNAMIC  MTU:1500  Metric:1
          RX packets:4616 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3966 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:366604 (358.0 KiB)  TX bytes:417107 (407.3 KiB)

root@Edisonek:/# ping google.com
PING google.com (172.217.18.110): 56 data bytes
64 bytes from 172.217.18.110: seq=0 ttl=118 time=59.909 ms
64 bytes from 172.217.18.110: seq=1 ttl=118 time=65.190 ms
64 bytes from 172.217.18.110: seq=2 ttl=118 time=56.589 ms
64 bytes from 172.217.18.110: seq=3 ttl=118 time=56.230 ms
^C
--- google.com ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 56.230/59.479/65.190 ms
root@Edison:/#
```

> BTW I just unpacked tar.gzip on 32bit Win7 using 7-zip 19.00 (x86) and I see no problems (no GNUsparesefile.0, file sizes correct).

Hard to believe, here you have test of archive-`edison-image-decb4c8.tar.gz` on:
- Winsows XP Pro x32 - [Windows_xp_x32.mp4](https://www.radario1-selling.cf/Windows_xp_x32.mp4)
- Windows 10 Pro x64 - [Windows_10_x64.mp4](https://www.radario1-selling.cf/Windows_10_x64.mp4)
On this both systems tested archives shows only ~640MB compressed files. Tested few hours ago.

Also I have one more question, where can I find packages for Edison? 
Directory `/etc/apt/sources.list.d` seems to be empty. 

```
root@Edison:/etc/apt/sources.list.d# ls
root@Edison:/etc/apt/sources.list.d#
```

Using:
```
root@Edison:/etc/apt/sources.list.d# cat /etc/*release
ID="poky-edison"
NAME="Poky (Yocto Project Reference Distro)"
VERSION="2.7.2 (warrior)"
VERSION_ID="2.7.2"
PRETTY_NAME="Poky (Yocto Project Reference Distro) 2.7.2 (warrior)"
EDISON-3.0
Copyright Intel 2015
```

On other linux-based devices I can entry source-list of packages.
Example 1:
```
root@OZ1:~# cat /etc/opkg/distfeeds.conf
src/gz openwrt_core http://downloads.openwrt.org/releases/18.06.8/targets/ar71xx/tiny/packages
src/gz openwrt_kmods http://downloads.openwrt.org/releases/18.06.8/targets/ar71xx/tiny/kmods/4.9.214-1-db2d5425c674d1648616e3c37760e80e
src/gz openwrt_base http://downloads.openwrt.org/releases/18.06.8/packages/mips_24kc/base
src/gz openwrt_luci http://downloads.openwrt.org/releases/18.06.8/packages/mips_24kc/luci
src/gz openwrt_packages http://downloads.openwrt.org/releases/18.06.8/packages/mips_24kc/packages
src/gz openwrt_routing http://downloads.openwrt.org/releases/18.06.8/packages/mips_24kc/routing
src/gz openwrt_telephony http://downloads.openwrt.org/releases/18.06.8/packages/mips_24kc/telephony
root@OZ1:~#
```
Example 2:
```
root@monkey ~ # cat /etc/apt/sources.list
#------------------------------------------------------------------------------#
#                   OFFICIAL DEBIAN REPOS
#------------------------------------------------------------------------------#

###### Debian Main Repos
deb http://deb.debian.org/debian/ stable main contrib non-free
deb-src http://deb.debian.org/debian/ stable main contrib non-free

deb http://deb.debian.org/debian/ stable-updates main contrib non-free
deb-src http://deb.debian.org/debian/ stable-updates main contrib non-free

deb http://deb.debian.org/debian-security stable/updates main
deb-src http://deb.debian.org/debian-security stable/updates main

deb http://ftp.debian.org/debian buster-backports main
deb-src http://ftp.debian.org/debian buster-backports main

#------------------------------------------------------------------------------#
#                      UNOFFICIAL  REPOS
#------------------------------------------------------------------------------#

###### 3rd Party Binary Repos
###Debian Multimedia
deb [arch=amd64,i386] https://www.deb-multimedia.org buster main non-free

###video-dl
#deb http://repo.daniil.it lenny main

root@monkey ~ #
```

Quarer, 2020-10-10:
> Our help pages are here: https://edison-fw.github.io/meta-intel-edison/
> Wifi setup (using connman): https://edison-fw.github.io/meta-intel-edison/4.2-networking.html#wifi

Thanks 😄, works:

```
root@Edison:/# ifconfig
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:902 errors:0 dropped:0 overruns:0 frame:0
          TX packets:902 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:60446 (59.0 KiB)  TX bytes:60446 (59.0 KiB)

wlan0     Link encap:Ethernet  HWaddr fc:c2:de:41:70:a3
          inet addr:10.9.0.200  Bcast:10.9.255.255  Mask:255.255.0.0
          inet6 addr: fe80::fec2:deff:fe41:70a3/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST DYNAMIC  MTU:1500  Metric:1
          RX packets:4616 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3966 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:366604 (358.0 KiB)  TX bytes:417107 (407.3 KiB)

root@Edison:/# ping google.com
PING google.com (172.217.18.110): 56 data bytes
64 bytes from 172.217.18.110: seq=0 ttl=118 time=59.909 ms
64 bytes from 172.217.18.110: seq=1 ttl=118 time=65.190 ms
64 bytes from 172.217.18.110: seq=2 ttl=118 time=56.589 ms
64 bytes from 172.217.18.110: seq=3 ttl=118 time=56.230 ms
^C
--- google.com ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 56.230/59.479/65.190 ms
root@Edison:/#
```

> BTW I just unpacked tar.gzip on 32bit Win7 using 7-zip 19.00 (x86) and I see no problems (no GNUsparesefile.0, file sizes correct).

Hard to believe, here you have test of archive-`edison-image-decb4c8.tar.gz` on:
- Winsows XP Pro x32 - [Windows_xp_x32.mp4](https://www.radario1-selling.cf/Windows_xp_x32.mp4)
- Windows 10 Pro x64 - [Windows_10_x64.mp4](https://www.radario1-selling.cf/Windows_10_x64.mp4)
On this both systems tested archives shows only ~640MB compressed files. Tested few hours ago.

Also I have one more question, where can I find packages for Edison? 
Directory `/etc/apt/sources.list.d` seems to be empty. 

```
root@Edison:/etc/apt/sources.list.d# ls
root@Edison:/etc/apt/sources.list.d#
```

Using:
```
root@Edison:/etc/apt/sources.list.d# cat /etc/*release
ID="poky-edison"
NAME="Poky (Yocto Project Reference Distro)"
VERSION="2.7.2 (warrior)"
VERSION_ID="2.7.2"
PRETTY_NAME="Poky (Yocto Project Reference Distro) 2.7.2 (warrior)"
EDISON-3.0
Copyright Intel 2015
```

On other linux-based devices I can entry source-list of packages.
Example 1:
```
root@OZ1:~# cat /etc/opkg/distfeeds.conf
src/gz openwrt_core http://downloads.openwrt.org/releases/18.06.8/targets/ar71xx/tiny/packages
src/gz openwrt_kmods http://downloads.openwrt.org/releases/18.06.8/targets/ar71xx/tiny/kmods/4.9.214-1-db2d5425c674d1648616e3c37760e80e
src/gz openwrt_base http://downloads.openwrt.org/releases/18.06.8/packages/mips_24kc/base
src/gz openwrt_luci http://downloads.openwrt.org/releases/18.06.8/packages/mips_24kc/luci
src/gz openwrt_packages http://downloads.openwrt.org/releases/18.06.8/packages/mips_24kc/packages
src/gz openwrt_routing http://downloads.openwrt.org/releases/18.06.8/packages/mips_24kc/routing
src/gz openwrt_telephony http://downloads.openwrt.org/releases/18.06.8/packages/mips_24kc/telephony
root@OZ1:~#
```
Example 2:
```
root@monkey ~ # cat /etc/apt/sources.list
#------------------------------------------------------------------------------#
#                   OFFICIAL DEBIAN REPOS
#------------------------------------------------------------------------------#

###### Debian Main Repos
deb http://deb.debian.org/debian/ stable main contrib non-free
deb-src http://deb.debian.org/debian/ stable main contrib non-free

deb http://deb.debian.org/debian/ stable-updates main contrib non-free
deb-src http://deb.debian.org/debian/ stable-updates main contrib non-free

deb http://deb.debian.org/debian-security stable/updates main
deb-src http://deb.debian.org/debian-security stable/updates main

deb http://ftp.debian.org/debian buster-backports main
deb-src http://ftp.debian.org/debian buster-backports main

#------------------------------------------------------------------------------#
#                      UNOFFICIAL  REPOS
#------------------------------------------------------------------------------#

###### 3rd Party Binary Repos
###Debian Multimedia
deb [arch=amd64,i386] https://www.deb-multimedia.org buster main non-free

###video-dl
#deb http://repo.daniil.it lenny main

root@monkey ~ #
```

htot, 2020-10-10:
OK I replayed here and here is what happened: I downloaded the file, ungzipped it with no problem, deleted the original. Then later gezipped, xz, zip and 7z it. The re-gzipped one unpacks without errors on windows. So apparently it was gzipped with a buggy gzip (from Ubuntu 19.10). I've 7z then files now and uploaded, I hope they are fixed now.

Yeah, I know how other distros work, I have Kubuntu on all my machines. In fact I have Win10 only in a virtual box.

There are distributions that distribute binaries, others distribute source packages. But Yocto is in fact a distribution builder.
It downloads sources, compiles them, packs into debs (or rpm or ipk) and creates your own repository.  The work is done using a "recipe" that is executed through a tool called `bitbake`. There are many recipes, see https://layers.openembedded.org/layerindex/branch/zeus/recipes/.

If a recipe exists in a layer, the layer needs to be added to setup.sh (if not already) and then let bitbake build it (`bitbake vim` or so), then let `bitbake` build the repo. See https://edison-fw.github.io/meta-intel-edison/5.0-Creating-a-deb-repository.html

TLDR:
To install a package, you need to create the deb first. Building the complete image as you downloaded takes about 3 hours on my machine (Core i7 / 100kiops SSD / btrfs).

If you are looking for an image that "just boots" and installs binary deb from an existing repo, that is what the debian image (Buster) does.

Quarer, 2020-10-10:
Okey, got it. Thanks for everything.
[image]

closed 2020-10-10 by Quarer