edison-fw/meta-intel-edison #118: How to change IFWI of Edison?

shawnhsoia97, 2020-11-09:
Hello , I have a question. I need to change IFWI of Edison, because the some GPIO default is turn on or blink when plug power(exp: GP12 is blink and GP13 is turn on), I think caused by firmware, and I want to clean this default config.

As far as I know that maybe can replace intel_mcu.bin in  /lib/firmware to change firmware, so first I think need to create new Binary Disk Image File.

I found some a way that is create one .bin file by Eclipse, but I don't understand how to use this and not sure this way is correct .
I try to create new project by Eclipse, but which crashed in target connection.

I don't know this idea is correct. Does anyone have any idea? Thank !

htot, 2020-11-09:
I think you are confusing stuff:
- LED  DS2 should be blinking, it's on DIG13 and controlled by GP40.
- IFWI is installed by flashall. Afaik it's needed to bring up the CPU (before U-Boot). intel_mcu.bin is needed afaik to load stuff to the MCU (by linux). They are not the same.
- the LED blinks because GP40 is described in acpi as a LED with trigger heartbeat by [leds.asl](https://github.com/htot/meta-acpi/blob/master/recipes-bsp/acpi-tables/samples/edison/leds.asli). If you don't want that remove [here](https://github.com/edison-fw/meta-intel-edison/blob/ee0cb3be855a06494e7159ddb050671d5df408c3/meta-intel-edison-bsp/conf/machine/edison.conf#L30). 
- the other LED is afaik not programmable (see schematic)

shawnhsoia97, 2020-11-10:
Thank for your reply.
But I am not referring to LED DS2 ans GP40, I mean to GP12 and GP13 that is connect to LED by the my self designed board.
BTW I have removed leds.asl and arduino.asl, because I am not use Edison on Arduino board.

As I mentioned earlier about that the GP13 will turn on and CP12 will blink twice at  bring up the CPU (before U-Boot).I don't want this.

I am not sure which one correctly which either IFWI or intel_mcu.bin. The both is compiled to MCU(quark) in Edison ,right? AFAIK this issue is MCU caused. 

Also maybe you know how to solve by change  either IFWI or intel_mcu.bin?

htot, 2020-11-10:
For your hardware, I can't say.

IFWI files are here: https://github.com/edison-fw/meta-intel-edison/tree/master/utils/flash/ifwi/edison
MCU file here: https://github.com/edison-fw/meta-intel-edison/tree/master/meta-intel-edison-bsp/recipes-support/edison-mcu/files
In both cases we don't have the sources, so you would need to reverse engineer.

shawnhsoia97, 2020-11-10:
Okay thank your help.

BTW there is really little information about these. I will keep find the way. Thank!

closed 2020-11-10 by shawnhsoia97