edison-fw/meta-intel-edison #119: Failed to fetch URL http://hisham.hm/htop/releases/2.2.0/htop-2.2.0.tar.gz

Kudesnick, 2020-11-11:
htop-2.2.0-r0 do_fetch: Failed to fetch URL http://hisham.hm/htop/releases/2.2.0/htop-2.2.0.tar.gz, attempting MIRRORS if available
default settings, master branch

this utilite located to https://github.com/hishamhm/htop

htot, 2020-11-11:
I see. This needs to be fixed in openembedded, and I see it is in gatesgarth and master. [Myself](https://github.com/htot/meta-intel-edison) I am on zeus and working to get btrfs support. And then we need to move up to dunfell.

So, we could patch the recipe in poky (through setup), overload it with our own fixed version or copy in gatesgarth version. I'd prefer the 2nd option. Do you have a patch available?

Kudesnick, 2020-11-15:
I'm still not very good at yocto system, maybe I need to edit the file ./out/linux64/poky/meta-openembedded/meta-oe/recipes-support/htop/htop_2.2.0.bb, but at the moment I can not check it. I will definitely attach a patch if I find a competent solution.

htot, 2020-11-15:
Yes, and that modification could be turned into a patch to be applied at setup time. But I would just copy recipes-support/htop into meta-intel-edison and edit that.

htot, 2020-12-05:
Looks like we created similar patches. I applied yours against warrior as I had already included mine to zeus. Thanks!

closed 2020-12-05 by htot