edison-fw/meta-intel-edison #124: BTRFS rebooted crash

shawnhsoia97, 2021-02-25:
I update to latest zeus version.
I have happened to the strange status
When I did input command reboot, the system have crashed that is stop, this like below:
[image]

But I replug power of the Edison, this system boot normally
[image]

htot, 2021-02-25:
It seems not a crash, but waiting (or looping) inside U-Boot. I'm a little confused, your booting a btrfs image, but U-boot says Target: blank and I believe now I have Target: btrfs. Who did you install?

BTW: I am currently on the dunfell branch as this has the diff with zeus here: https://github.com/edison-fw/meta-intel-edison/compare/zeus...htot:dunfell

closed 2021-02-25 by shawnhsoia97

shawnhsoia97, 2021-02-25:
Sorry this is my HW issue.
I am sorry if this has caused you any misunderstand.

Thank for your quick reply.