edison-fw/meta-intel-edison #125: boot failure

humanoid2050, 2021-02-28:
I have an Edison on a stack of sparkfun carrier boards, including the base block, the SD card block, and the UART block. I have tried multiple things, but in all cases the board fails to boot to the root fs. I think the root of the issue may be related to the following line:
`Initramfs unpacking failed: invalid magic at start of compressed archive`

The full boot log is here:
```
Target:blank
Partitioning already done...
Flashing already done...
9767712 bytes read in 422 ms (22.1 MiB/s)
9558813 bytes read in 414 ms (22 MiB/s)
Valid Boot Flag
Setup Size = 0x00003800
Magic signature found
Using boot protocol version 2.0f
Linux kernel version 5.6.0-edison-acpi-standard (oe-user@oe-host) #1 SMP Sun Feb 21 20:56:44 UTC 2021
Building boot_params at 0x00090000
Loading bzImage at address 100000 (9753376 bytes)
Magic signature found
Initial RAM disk at linear address 0x06000000, size 25165824 bytes
Kernel command line: "quiet root=/dev/mmcblk0p8 rootfstype=ext4 console=ttyS2,115200n8 earlyprintk=ttyS2,115200n8,keep loglevel=4 systemd.unit=multi-user.target hardware_id=00"
Magic signature found

Starting kernel ...

[    2.400612] Initramfs unpacking failed: invalid magic at start of compressed archive
Scanning for Btrfs filesystems
Starting version 243.2+
Kernel with acpi enabled detected
Loading acpi tables
[    3.495505] pca953x i2c-INT3491:00: failed writing register
[    3.519364] pca953x i2c-INT3491:01: failed writing register
[    3.542308] pca953x i2c-INT3491:02: failed writing register
[    3.564625] pca953x i2c-INT3491:03: failed writing register
gpioset: at least one GPIO line offset to value mapping must be specified
gpioset: at least one GPIO line offset to value mapping must be specified
Waiting for root device /dev/mmcblk0p8
Found device '/run/media/mmcblk0p8'
Init found, booting...
mount: can't find /boot in /etc/fstab
mount: mounting /boot on /realroot/boot failed: Invalid argument
mount: can't find /lib/modules in /etc/fstab
mount: mounting /lib/modules on /realroot/lib/modules failed: Invalid argument
[   10.050126] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43340-sdio for chip BCM43340/2
[   10.445670] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43340-sdio for chip BCM43340/2
[   10.512171] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   10.582658] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43340/2 wl0: Oct 23 2017 08:41:23 version 6.10.190.70 (r674464) FWID 01-98d71006
[FAILED] Failed to mount /home.
See 'systemctl status home.mount' for details.
[DEPEND] Dependency failed for Post install service.
[  OK  ] Stopped Set the bluetooth hardware address.
[  OK  ] Stopped target Bluetooth.
[  OK  ] Stopped target Host and Network Name Lookups.
[  OK  ] Stopped target Sound Card.
[  OK  ] Stopped target Timers.
[  OK  ] Stopped Daily apt download activities.
[  OK  ] Stopped Daily Cleanup of Temporary Directories.
[  OK  ] Stopped target System Time Synchronized.
[  OK  ] Stopped target System Time Set.
         Stopping Save/Restore Sound Card State...
         Stopping Edison Arduino board LED Blinker...
         Stopping Bluetooth service...
[  OK  ] Stopped File System Check on /dev/disk/by-partlabel/home.
         Stopping Load/Save Random Seed...
         Stopping Load/Save RF Kill Switch Status...
         Stopping Kernel Logging Service...
         Stopping System Logging Service...
         Stopping Daemon to handle arduino sketches...
         Stopping The Edison status and configuration service...
[  OK  ] Removed slice system-getty.slice.
         Stopping Telephony service...
[... 592 lines cut ...]
 179        2       1024 mmcblk0p2
 179        3       2048 mmcblk0p3
 179        4       1024 mmcblk0p4
 179        5       1024 mmcblk0p5
 179        6      24576 mmcblk0p6
 179        7      65536 mmcblk0p7
 259        0    1572864 mmcblk0p8
 259        1     753664 mmcblk0p9
 259        2    1392623 mmcblk0p10
 179       24   31276032 mmcblk1
Cannot find /dev/mmcblk0p7 or no /sbin/init present , dropping to a shell

sh: can't access tty; job control turned off
/ # [    7.788171] EXT4-fs (mmcblk0p5): mounted filesystem without journal. Opts: (null)
[    7.789258] EXT4-fs (mmcblk0p8): recovery complete
[    7.805125] EXT4-fs (mmcblk0p8): mounted filesystem with ordered data mode. Opts: (null)
[    7.815595] mount (382) used greatest stack depth: 13768 bytes left
[    7.825591] FAT-fs (mmcblk0p7): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[    7.863921] EXT4-fs (mmcblk1p1): mounted filesystem with ordered data mode. Opts: (null)

/ #
```
I have tried building directly on my host system running Ubuntu 20.04, as well as in dockerized environments based in Ubuntu18.04 and even 16.04. Results are always the same. I have tried changing the load address of the initramfs. Results are always the same. I have tried checking out different branches of this repo. Results are always the same, except for old ones like sumo which fail `make setup` due to a patch that won't apply and warrior that fails `make image` due to `undefined reference to stime`.

Can someone point me in the right direction?

htot, 2021-02-28:
Unfortunately this
> Initramfs unpacking failed: invalid magic at start of compressed archive

is not related to your problems. It means the kernel is expecting a built-in initramfs, but it's not, we load it separately. I have the same same message, it's harmless.

Afaict you are having at least 2 problems.

1. You don't have an Edison Arduino board, but you are loading the  arduino.asl [here](https://github.com/edison-fw/meta-intel-edison/blob/d06bc2177e3b5bb2fa6c8d4805e7d904f30d6e8e/meta-intel-edison-bsp/conf/machine/edison.conf#L30). This causes the complaints on `pca953x`, as you probably don't have them. Question is, what asl would you need instead to enable your hardware. But let's try to boot first.
2. Something is wrong with mmcblk0p8 or with /etc/fstab on it. So, because mounting fails systemd starts stopping services. Try again to go into rescue and then manually mount (if not already mounted) mmcblk0p8 and then check /etc/fstab. I'm not sure which branch you are using (zeus?). Maybe I made a mistake in fstab, if so, you should be able to manually fix it.

humanoid2050, 2021-02-28:
@htot Thanks for taking the time to reply. 

From uboot, I can do `ls mmc 0:8` and get a nice dump of the partition contents. Side note, uboot can't detect the presence of an SD card, but I suspect that can be filed along with the .asl issues.

If I boot into rescue, I do get a prompt and I can poke around. The partitions seem to be fine, as `/dev/mmcblk*` lists all expected partitions. Additionally, `/run/media/` contains automatically mounted entries for `mmcblk0p5  mmcblk0p7  mmcblk0p8  mmcblk1`.

Contents of fstab:
```
/run/media/mmcblk0p8/etc # cat fstab
tmpfs           /var/volatile      tmpfs      defaults              0  0

/dev/disk/by-partlabel/factory  /factory        auto    noauto,x-systemd.automount,nosuid,nodev,noatime,discard     1   1
/dev/disk/by-partlabel/home     /home       auto    nofail,x-systemd.automount,noatime,discard  1   1
```
Checking for those partitions (at least in the intramfs env):
```
/dev/disk/by-partlabel # ls
boot         home         rootfs       u-boot-env1  u-boot1
factory      panic        u-boot-env0  u-boot0      update
```

I am using zeus. 

I also see in the boot log 
```
mount: can't find /boot in /etc/fstab
mount: mounting /boot on /realroot/boot failed: Invalid argument
mount: can't find /lib/modules in /etc/fstab
mount: mounting /lib/modules on /realroot/lib/modules failed: Invalid argument
```
But what I can't find is what/where the requests for mounting `/boot/` and `/lib/modules` are coming from. Is that relevant?

humanoid2050, 2021-02-28:
Did a little more digging. By using the `./flashall --btrfs` option instead of `make flash` from the project root, it works fine now. Just a mismatch somewhere in the install process?

htot, 2021-02-28:
> From uboot, I can do ls mmc 0:8 and get a nice dump of the partition contents. Side note, uboot can't detect the presence of an SD card, but I suspect that can be filed along with the .asl issues.

That's not normal. U-Boot can these days actually boot directly from the SD card.

I don't see why mounting home failed, still. This happens by the init script which is inside the initrd. After executing, loading needed modules and mounting it does a switchroot to systemd, same environment as do_rescue. You can find it here: https://github.com/edison-fw/meta-intel-edison/blob/zeus/meta-intel-edison-distro/recipes-core/initrdscripts/initramfs-framework/init

Installing the btrfs image, you get a slightly modified fstab. This is what I use so it is better tested. Still the ext image worked last time I tried.

Good that btrfs image is working though.

humanoid2050, 2021-03-01:
I have discovered that in the non-btrfs path of `flashall.sh`, the rootfs is flashed but nothing is done for the home partition. As a result, mounting home fails due to lack of a proper filesystem. If I boot into recovery and run `mkfs.ext4 /dev/mmcblk0p10` and then reboot, everything is fine. Perhaps I missed a step?

htot, 2021-03-01:
I had the intention not to break the home partition :-( But good that the recovery partition works.

humanoid2050, 2021-03-01:
I see your intent. Sometimes there are scripts that fire on first boot. If there is such a thing in this environment, could it be used to check for the existence of an ext4 fs on the home partition, and make one if it is missing? Otherwise at least a note in the online documentation might be helpful. Either way, I'm happy to close this issue.

htot, 2021-03-02:
I have seen reports with the original firmware that if home is damaged you couldn't boot.

There is a script that runs on first boot (post-install.sh). But I think it runs after systemd. Best thing would be to have systemd fsck the partition if it's broken or create one if missing. But I thought it already did that.

closed 2021-03-06 by htot

mwallnoefer, 2021-04-11:
@htot The following warnings apply for me as well. Could we quiet them somehow?

> I also see in the boot log
>
> ```
> mount: can't find /boot in /etc/fstab
> mount: mounting /boot on /realroot/boot failed: Invalid argument
> mount: can't find /lib/modules in /etc/fstab
> mount: mounting [...]

htot, 2021-04-11:
Hi @mwallnoefer it is done by this script (in the initrd) https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-distro/recipes-core/initrdscripts/initramfs-framework/init

Question: are you using Zeus or Dunfell? And are you booting ext4 or btrfs image?

mwallnoefer, 2021-04-11:
Dunfell with ext4

htot, 2021-04-11:
Ok so it seems `mount -n /boot || true` and `mount -n --move /boot /realroot/boot || true` don't just silently fail as I hoped :-(

Other then that it's harmless.