edison-fw/meta-intel-edison #127: Preempt-RT support

mwallnoefer, 2021-03-25:
Interesting to see how the _PREEMPT_RT_ effort continues its way to be integrated into the mainline kernel: https://wiki.linuxfoundation.org/realtime/start

Could it make sense to build the Edison images with a _preempt_rt_-based kernel per default? Also to provide an alternative to the internal MCU which currently isn't really usable afaik?

htot, 2021-03-25:
I like it, but in case of troubles might be difficult to get support from kernel developers? 
If found the bugs hurting us (eem ethernet over gadget) are the same for -rt and normal kernel. So the only real downside it reduced performance that you can notice as a reduced ethernet speed (iperf3).

BTW we have a Telegram group called Intel Edison https://t.me/IntelEdison

mwallnoefer, 2021-03-26:
> If found the bugs hurting us (eem ethernet over gadget) are the same for -rt and normal kernel. So the only real downside it reduced performance that you can notice as a reduced ethernet speed (iperf3).

Okay, maybe things get better in the future, when the whole _rt_ patchset lands in the vanilla kernel.

htot, 2021-03-26:
I would say reduced performance in some area's is to be expected, in particular like eem ethernet (gadget) as it involves dwc3 interrupts. The -rt patch converts interrupts to tasklets so they can be made interruptible by higher priority tasklets. This is the essence of the -rt patch, to reduced latencies at the expense of throughput.

With the -rt patch is see latencies 50 - 100 us, without -rt patch sometimes spikes to 100ms.

htot, 2021-09-07:
With dunfell there is a -rt kernel provided.

closed 2021-09-07 by htot